zigpy/zha-device-handlers #1407: Tuya Smart Air box - TS0601 _TZE200_yvx5lh6k

kan84, 2022-03-03:
Hi,

I have been using Tuya Smart Air box - TS0601 _TZE200_yvx5lh6k for couple of months with Homeassistant without any issue.
Yesterday when i update homeassistant to Home Assistant 2022.3.0 all of the four entities became unavailable.

After device was reset, it was visible in Homeassistant to be added as a new device. As soon as device was added and interview was complete device entities shows as unavailable. 
I have attached the device signature below for the device let me know is there are logs which i can pull  

Device Signature

{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0051",
      "in_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0xef00"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZE200_yvx5lh6k",
  "model": "TS0601",
  "class": "zigpy.device.Device"
}

MattWestb, 2022-03-03:
`"class": "zigpy.device.Device"` = the system have not loading the quirk for the device.
Then the device signature have not being changed it cant being the device but one user have deleting your device in the quirk that is making it not working longer https://github.com/zigpy/zha-device-handlers/pull/1346.

Configure local quirks in ZHA https://github.com/zigpy/zha-device-handlers/discussions/693#discussioncomment-857274 and putting the renamed quirk in the  local quirk folder after renaming it to `ts0601_air_quality.py` (deleting the ` .txt `) and restarting HA and the quirk shall loading OK for your devices.

kan84, 2022-03-03:
Thanks, i ll try the solution but i guess every time i upgrade i ll have to put quirk back. 
That explains why it suddenly stopped working after the homeassistant update.

MattWestb, 2022-03-03:
No then updating HA is the local quirks not deleted then its on your local system and not in the HA container.

But if the quirk is being updated in HA and working your shall deleting the local one for not getting problem in the future.

Pleas reporting back then have testing !!

I heaving one fight for getting the change rolled back and not braking devices that  have working for 9 mounts but need feedback for getting it thru.

kan84, 2022-03-03:
I just tested the way you told me by putting the quirk in /config/custom_zha_quirks/ts0601_air_quality.py and I can confirm it is working for me.
I can see all four entities I was seeing before. So whatever update came with Home assistant 2020.03 was responsible for breaking the device which was working fine before.

kan84, 2022-03-03:
Thanks for the insights. I am tracking your other issue as well.

MattWestb, 2022-03-05:
Pleas leave comment in reversing PR for getting it merged #1410.

closed 2022-09-08 by github-actions[bot]