zigpy/zha-device-handlers #1413: [BUG] tuya LIDL color lights have losing all color mode

MattWestb, 2022-03-06:
**Describe the bug**
I have one LIDL E14 RGBWW TS0505A _TZ3000_odygigth and one LIDL RGBWW LED Strip TS0505A_TZ3000_riwp3k79 and both have working OK with RBG and CT untill update to HA 2022.3.
Now i have both the CT slider and can changing the color temperature and one color wheel and its not working then the device is still on TC mode and dont listening on color commands.

So the missing command is the selecting color mode not the color change command.
**To Reproduce**
Steps to reproduce the behavior:
Have deleting the LED stripe from the "old" test system and putting in one new test HA with RCP and using bellows so its one clean install.
**Expected behavior**
Changing color mode working OK.
**Screenshots**
If applicable, add screenshots to help explain your problem.

**Additional context**
All IKEA CWS (RGBW / RGBWW) lights can switching color mode OK (CWS1, CWS3 E14, E27 and GU10).

If need more information pleas ask and i trying getting it.

MattWestb, 2022-03-06:
If like having it in Zigpy i can reposing it there but i think its not needed.
I think its little patching in Zigpy that have making the braking and not somthing in device handling. 

The LED strips have working OK around 15 mounts (from the first week it was released in EU).

MattWestb, 2022-03-06:
Adding the E14 in https://github.com/zigpy/zha-device-handlers/blob/dev/zhaquirks/lidl/rgbcct.py and ZHA is loading the quirk but the function is not changes = no color mode change made in the device then clicking new colors.

Plus one ZHA bug found: If setting light level to minimum (1%) the light is still on but ZHA is flagging it off and taking the color wheel and TC slider away but not the light level slider.
If setting 2% light level the light is on both on the device and in ZHA GUI.

MattWestb, 2022-03-06:
Fast sniffing ZHA is sending one `on` command and then new color command Mired for CT and X, Y for color and all locks like IKEA CWS that using Mired for CT and X,Y for color.
So very likely is LIDL / tuya not liking getting one ON command before the color change command (like IKEA was not liking transition time with on and dim commands).

I dont have so much experience with color modes so pleas say is need more information.

MattWestb, 2022-03-06:
The 1% bug is **not** related to ZHA is one "undocumented future" of tuya lights that is reporting status off then its still on in very low light level.
IKEA CWS 1 and 3 gen is working OK

mdeweerd, 2022-03-09:
I also have color issues, but I am still on HA 2022.2.X .  When asking for warm color, I get white on several lights.  IMHO something broke when I updated mid january from an early 2021.12 version .

I notice a setting "applyRedFix" in Z2M:
https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/devices/lidl.js#L535-L545

And 'turnsOffAtBrightness' for other devices: https://github.com/Koenkk/zigbee-herdsman-converters/blob/f89b828b1ec9a793c90bde2edf7931a56d1bd752/devices/innr.js#L24

This search may interest you: https://github.com/Koenkk/zigbee-herdsman-converters/search?q=applyRedFix  (Also checkout the second page!)
 
The color correction is applied here:
https://github.com/Koenkk/zigbee-herdsman-converters/blob/50f047f0f333e05f068bdbaef76e4790304bb9b4/converters/toZigbee.js#L987-L990

And also appears in the [README](https://github.com/Koenkk/zigbee-herdsman-converters/blob/67d503a3e140308e61a7163bf0f8424b32184ac4/README.md#documentation-of-definition-meta-property) but refers to the code.

Maybe this helps fix our colors in ZHA.

MattWestb, 2022-03-09:
It can being the bug is implanted 2022.2 then i have not  testing color mode on my LIDL light after Christmas and my IKEA CWS i have using after and was working OK so more then likely.
The red problem i have not testing then i normally only using "soft colors" in my production system but its likely needing the same for ZHA for getting the lights working well.

The 1 = off is out off Zigbee spes and many other things Z2M have doing like 255 = max and 0 is also used and its all reserved if reading the Light and occupancy (shall being red as one supplement to ZB3 papers) https://zigbeealliance.org/wp-content/uploads/2019/11/docs-15-0014-05-0plo-Lighting-OccupancyDevice-Specification-V1.0.pdf
**The minimum light level SHALL be 0x01 and the maximum light level SHALL be 0xfe.**

And also the ZCL R8 is interesting that you finding in the reference docks discussion.

mdeweerd, 2022-03-09:
I find that the ZCL specification is generally a sufficient ressource and I have the latest "published" one.

It's when I go to the soft colors, that I get "white" in the middle of the scale it's soft and on the other end (white side) I get white.

MattWestb, 2022-03-09:
I have "finding" ZCL R8 and have linked it in the dev docks but we dont have getting the spec for Zigbee R23 that Silabs have implanted so we dont knowing all new things in it.

I was testing my E14 LIDL RGBW with one tuya TS004F Knob on one tuya ZBGW and it was doing all color and dimer commands very bad from and was moving the knob to my test ZHA and putting it the same group (not binding adding the remote as one light!!) and its was controlling one IKEA CWS very nicely (compared with tuya light) but its not part of this issued.

mdeweerd, 2022-03-09:
Specifications can be requested at https://csa-iot.org/developer-resource/specifications-download-request/ .  Only the R22 is available there.  The R23 is said to be "upcoming" in Silicon Labs' document so they may still be working on it, possibly because of Matter?

MattWestb, 2022-03-21:
I was moving my 1 tuya / LIDL RGBWW lights to my new test setup that **was** running the RCP plugin but was being broken then i was updating Armbian and i have flashing one EZSP 6.10.3.0 and repairing all device plus adding the LIDL lighs and they is working OK !!
The old test setup is running Home Assistant 2022.3.0 containerized and the new Home Assistant 2022.3.1 supervised.

I think i shall testing updating both to the same version but its looks very strange then its have not being any changing in ZHA that i have seen that can making this kind of trouble if not being somthing in the shared libs.

mdeweerd, 2022-03-21:
I just updated my entire system (HAOS) to the latest versions, HA is 2022.3.5 .
I still have the color issue.

MattWestb, 2022-03-24:
I was updating my new test system to Home Assistant 2022.3.7 and the 2 LIDL bulbs is working OK. 
Deleting the E14 RGBWW and adding it back to my "old" test system (2022.3.0) and and its working :-((((
Perhaps i was having one old junk saved in the system that was making it not working OK then its looks it being somthing not working OK in ZHA GUI and perhaps not wrong in the Zigbee layer.

@mdeweerd can you trying deleting one of your lights and restarting HA so its have flushing all old data and paring it and see how its working ?

mdeweerd, 2022-03-24:
With or without my quirk ?

mdeweerd, 2022-03-24:
With quirk, it's still the same as before.  I removed the device, did 'ha core restart' and rejoined the device.
The entity name changed.

closed 2022-09-27 by github-actions[bot]