zigpy/zha-device-handlers #1417: [Device Support Request] SML003 sensitivity setting

Ulrar, 2022-03-07:
**Is your feature request related to a problem? Please describe.**
I bought an SML003 (hue motion sensor) to replace my Ikea ones because they support setting the sensitivity.
However once paired in ZHA, that's not an exposed setting and unfortunately the default is way too sensitive.

**Describe the solution you'd like**
Add a way to set the sensitivity, Z2M apparently supports it

**Device signature - this can be acquired by removing the device from ZHA and pairing it again from the add devices screen. Be sure to add the entire content of the log panel after pairing the device to a code block below this line.**

```json
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4107, maximum_buffer_size=82, maximum_incoming_transfer_size=128, server_mask=11264, maximum_outgoing_transfer_size=128, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "2": {
      "profile_id": 260,
      "device_type": "0x0107",
      "in_clusters": [
        "0x0000",
        "0x0001",
        "0x0003",
        "0x0400",
        "0x0402",
        "0x0406"
      ],
      "out_clusters": [
        "0x0000",
        "0x0003",
        "0x0006",
        "0x0019"
      ]
    }
  },
  "manufacturer": "Signify Netherlands B.V.",
  "model": "SML003",
  "class": "zigpy.device.Device"
}
```

**Additional context**
[This thread](https://community.home-assistant.io/t/zha-phillips-hue-motion/191855/13) seems to suggest the way to do it is by setting ultrasonic_u_to_o_threshold and pir_u_to_o_threshold clusters but that doesn't seem to do much currently. Might be better to look into how Z2M does it.

Thanks

MattWestb, 2022-03-07:
The sensitivity is implanted on the cluster 
https://github.com/zigpy/zha-device-handlers/blob/e4e62504738ced0c54c983ee5094a3a6af0a741d/zhaquirks/philips/__init__.py#L78-L80
But its looks your device is heaving one other Endpoint / cluster setting and the quirk is not being loaded for it.
Can you reading the software version on basic cluster and posting it ?

Also trying deleting it and reading it and look if you is getting one other cluster setting.

Ulrar, 2022-03-07:
Hey,

Sure, not sure which one exactly you want so here goes : 
sw_build_id (0x4000) : 2.53.6
zcl_version (0x0000) : 2
app_version (0x0001) : 2
stack_version (0x0002) : 1
hw_version (0x0003) : 1

Let me know if that's not what you were looking for.
It's a brand new SML003 bought on amazon and received this morning.
Thanks

MattWestb, 2022-03-07:
I only have 2 SML001 and cant comparing the version but perhaps some other user can and can saying if its updated or not.

I think the quirk need being updated with your device signature and the new manufacture name and also need some testing so knowing if its working OK.

Ulrar, 2022-03-07:
Sure. The entities that show up do seem to be working fine as far as I can tell : 

[image]

I've bound it directly to a group of lights and it's been turning them on and off fine, and also reporting occupancy and 'on/off' to HA at the same time without a hitch. Luminance and temperature also look about right.

And if you tell me what files I can edit them in my container to test it out

MattWestb, 2022-03-07:
The quirk is in https://github.com/zigpy/zha-device-handlers/blob/dev/zhaquirks/philips/motion.py.
You need downloading and configuring local quirks in HA and adding the quirk in the new folder  https://github.com/zigpy/zha-device-handlers/discussions/693#discussioncomment-857274.
Then you need edeting or bets making one new device class for your new version (for testing editing the old is OK but need one new class for adding it in ZHA).

The old device class is here:
https://github.com/zigpy/zha-device-handlers/blob/e4e62504738ced0c54c983ee5094a3a6af0a741d/zhaquirks/philips/motion.py#L32
The signature must looks 100% as your device for the system shall loading it also the manufacture name and its can being imported from the INIT https://github.com/zigpy/zha-device-handlers/blob/e4e62504738ced0c54c983ee5094a3a6af0a741d/zhaquirks/philips/__init__.py#L36.
The replacement shall looking the same but this line must being other then in the signature for getting the extra attributes on the cluster.

After every change you need restarting HA for the system using the changes.

MattWestb, 2022-03-07:
Add the `SIGNIFY` at the end of this line and its being imported in the quirk:
https://github.com/zigpy/zha-device-handlers/blob/e4e62504738ced0c54c983ee5094a3a6af0a741d/zhaquirks/philips/motion.py#L29 like this:  `from zhaquirks.philips import PHILIPS, OccupancyCluster, SIGNIFY`

MattWestb, 2022-03-07:
Like is being done here:
https://github.com/zigpy/zha-device-handlers/blob/e4e62504738ced0c54c983ee5094a3a6af0a741d/zhaquirks/philips/rwl022.py#L42 (Its one other quirk and wrong device ID you need to fixing that in the motion quirk).

Then all this:
https://github.com/zigpy/zha-device-handlers/blob/e4e62504738ced0c54c983ee5094a3a6af0a741d/zhaquirks/philips/motion.py#L42-L79
Need being as your device i looking.

If that is OK is the system loading the quirk for your device.

Next step is fixing the 
https://github.com/zigpy/zha-device-handlers/blob/e4e62504738ced0c54c983ee5094a3a6af0a741d/zhaquirks/philips/motion.py#L81
section so its using all cluster your device is having.

MattWestb, 2022-03-07:
You dont need repairing the device only restarting HA.
If you is getting on the device card or device signature   `"class": "zigpy.device.Device"` = the system was not  loading the quirk for the device = Signature is not OK.

If you is making some _bad things_ in the quirk it can resulting ZHA is not starting.

Ulrar, 2022-03-07:
Just noticed that my coordinator firmware was outdated (slightly newer zstack available) so I did the upgrade to 20211217 just now to be sure but no luck, that didn't change anything.

MattWestb, 2022-03-07:
I think (= not knowing) using the cluster number shall being OK in the signature but try using the real names if some its not OK defined like this shall being all your cluster with names:
```
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    PowerConfiguration.cluster_id,
                    Identify.cluster_id,
                    IlluminanceMeasurement.cluster_id,
                    TemperatureMeasurement.cluster_id,
                    OccupancySensing.cluster_id,
                ],
                OUTPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Identify.cluster_id,
                    OnOff.cluster_id,
                    Ota.cluster_id,
                ],
```
If you is setting up loging for quirks in HA you can see if the system is loading your new quirk (its loading the local after the internal ones in the log) in the logfiles then HA is starting by setting this in HA log setting:
```
    zhaquirks: debug
    custom_zha_quirks: debug
```

MattWestb, 2022-03-07:
Changing the `DEVICE_TYPE: zha.DeviceType.OCCUPANCY_SENSOR,` in the signature.

MattWestb, 2022-03-07:
And the `PROFILE_ID: zll.PROFILE_ID,`

MattWestb, 2022-03-07:
Sorry the last was wrong shall being `PROFILE_ID: zha.PROFILE_ID,`

Ulrar, 2022-03-08:
Hey,

That code works, it did load the new quirk.

[image]

I'm not seeing anything different though, should I ?
Thanks !

Ulrar, 2022-03-08:
Ah I see, I can now find the occupancy sensitivity in the manager clusters menu. I just set it to 1 and that seems to have worked.
Thanks for the help !

Can we get that added to the official quirk somehow ?

MattWestb, 2022-03-08:
So only the new import and the new device class was making all :-))

If you like you can making one PR for adding the changes of i can doing it for you.
I suggesting changing the name of the device class to `SignifyMotion` then its not coming any new with the name Philips and its making it "cleaner".

If Philips have putting in more functions we can adding them then we finding out how its working and you is getting them with HA updates.

And for info ZHA is looking implanting more switches and GUI elements for known extra functions like we have doing with all tuya TRVs (but not so god done for the user but looks nice and working OK) but **if** its coming we dont knowing but the sensitivity is one of the good function that being in the GUI.

Ulrar, 2022-03-08:
Sure, I created #1420 just now with a new class named `SignifyMotion` as suggested, hopefully that looks good.

So if I understand the difference correctly it looks like my device has a few extra output clusters, or did I miss something ?
Does that mean it supports extra binding options that the previous motion sensors might not have had ?

MattWestb, 2022-03-08:
The large thing is that Philips have taking the Zigbee Light Link function away = not possible "paring" it direct with one or more lights like IKEA is doing and i think Philips was able doing before (without coordinator). 
I think also its new Zigbee module (very likely one Silabs) and the Zigbee stack is working little different but shall being OK and is more up to date then the old ones.

I have 2 "old" indoor one that i using mostly for light measuring but they is always "jumping around" and using the worse routers the can finding (its the old Zigbee module that is bad).
Z2M have one very long running issue with connection problem with the "old " ones and its not solved.
I hope your is working  better in real life then the old ones (Silabs module is normally having  very good radio  on the chip) !!

Then i finding one good working light measuring device i putting them in the Black Box for bad Zigbee devices. 

What i knowing is ZHA binding all "good" cluster for being bond but you need testing how its working for knowing. Also try do one "reconfigure this device" from the device card and you can see how ZHA is doing it.

Ulrar, 2022-03-08:
Oh I see, they removed ZLL and left only ZHA ?
That's probably for the best, that was causing issues on other devices for me.

I do know what you mean, I have end devices connecting to the farthest possible router for some reason (especially philips hue dimmer switches !) but I just checked and that motion sensor is connected to one of the spots it's bound to in the same hallway, seems logical.
It's been working well so far but it's only been a couple of days, we'll see how it goes in the next few months ! If it keeps working well I'm planning to replace all of my ikea motion sensors with these.

Thanks for approving the PR !

MattWestb, 2022-03-08:
Yes and no. Philips and IKEA first hub / systems was ZLL and IKEA was very earls updating the devices to Zigbee 3 (now only the old motion sensor is ZLL plus the HUB) and Zigbee 3 is having updated  HA 1.X and ZLL 1.X implanted in the standard and is 95% compatible (some ZLL cluster commands is not 100%). IKEA is still having ZLL / Touch Link in all device and can being used in ZLL and Zigbee 3 system and as stand alone with one or more lights.

The IKEA motion sensor is not one real motion sensor its one _On/Off controller_ (sending on with off time to lights) by Zigbee 3 standard and its OK for doing that but not for doing "other things" then real HA motion sensor is better for automatons but cant (normally) being bonded (sending commands to lights) light groups.

Dont forgetting deleting the local quirk then the PR have being merged and released in next HA so your system is using updated quirk then its being updated.

closed 2022-03-11 by dmulcahey

Ulrar, 2022-03-12:
Hi,

I see this has been merged now (thanks !), how do I know which version this will be in ?
Does this get pulled on every minor HA release or only major ones ?

Thanks

dmulcahey, 2022-03-12:
There is no set frequency

PR #1420 (2022-03-11): Adding SignifyMotion class supporting new signature of SML003
Hi,

This resolves #1417.
Let me know if I need to change anything.

Thank you