zigpy/zha-device-handlers #1429: [Device Support Request] Tuya bulb _TZ3210_bfwvfyx1 TS0505B

sergey-suloev, 2022-03-12:
Tuya bulb _TZ3210_bfwvfyx1 TS0505B isn't fully supported by ZHA, only ON/OFF switch is supported.
Need full support for the device.

**Device signature **
```
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x010d",
      "in_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0008",
        "0x0300",
        "0x1000",
        "0xef00"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "242": {
      "profile_id": 41440,
      "device_type": "0x0061",
      "in_clusters": [],
      "out_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "_TZ3210_bfwvfyx1",
  "model": "TS0505B",
  "class": "zigpy.device.Device"
}
```

**Additional context**

MattWestb, 2022-03-12:
Your device is saying its one  `EXTENDED_COLOR_LIGHT = 0x010D` but is one tuya as you knowing its oft little not in sync with the Zigbee standard.
tuya quirk have not so many added and its looks its only dim ones.
You can looking on the LIDL that have adding patching for TC and RGBCT lights   https://github.com/zigpy/zha-device-handlers/tree/dev/zhaquirks/lidl and trying adding your device in the quirk and see if you can getting it working OK.

I have one LIDL (tuya re branded)  LED stripe controller with RGBCT and one E14 RGBCT blub that have the RGB and the CT but for the 2 last HA releases is the CT working OK but the RGB color wheel is not working and very likely you is getting the same problem then the quirk is working OK but the problem is in ZHA with the color wheel.

sergey-suloev, 2022-03-12:
@MattWestb 

This seems to be the one I need https://github.com/zigpy/zha-device-handlers/blob/dev/zhaquirks/lidl/rgbcct.py

MattWestb, 2022-03-12:
If the endpoints and cluster is the same only adding your device ID and it shall loading OK and you is getting the color fixed (but still one bug in HA).
If different cluster you need doing one new device class for it.

sergey-suloev, 2022-03-12:
No, this is a totally different device.

MattWestb, 2022-03-12:
That is making it much more tricky getting all things working.
Start  trying copy one device class that looks good and giving it one new name and fixing the signature so the system is loading the quirk.
Then you can start testing different tricks in the replace part.
I thin we have 2 other "hybrid devices" but i dont knowing if they have getting all thing working OK.

MattWestb, 2022-03-24:
If looking in Z2M is they adding your device as one "normal" tuya RGBWW light without using the extra cluster. https://github.com/Koenkk/zigbee-herdsman-converters/commit/cd05e8ac993efe9e48da09638ca3789032ef7d26
So try doing one new device class and the other but with the endpoints and cluster as your device and its shall working OK.
If "finding" more good functions later we can adding  them in the quirk later.

sergey-suloev, 2022-03-24:
@MattWestb  could you explain please ? I am not sure I understand what you suggest

MattWestb, 2022-03-24:
The original LIDL quirk have one device class:
https://github.com/zigpy/zha-device-handlers/blob/df9bdca7ff597fe23aff8fe946630927f6e93bd7/zhaquirks/lidl/rgbcct.py#L36
Make one new device class in the quirk (copy all of it) and changing its name and the cluster in the signature is its looks like your device.
If you like you can using the same replacement then we dont  have any new functions adding in it.

You can importing the the 0xef00 cluster from tuya INIT like being done here:
https://github.com/zigpy/zha-device-handlers/blob/df9bdca7ff597fe23aff8fe946630927f6e93bd7/zhaquirks/tuya/ts0501bs.py#L26
And using it in the signature like here:
https://github.com/zigpy/zha-device-handlers/blob/df9bdca7ff597fe23aff8fe946630927f6e93bd7/zhaquirks/tuya/ts0501bs.py#L50

Then you have getting the signature working OK ZHA is loading the quirk for the device and you shall getting  the right  function of it.

MattWestb, 2022-03-25:
Can you posting the Zigbee device signature from the device card with your quirk loaded ?

sergey-suloev, 2022-03-25:
https://github.com/sergey-suloev/zha-device-handlers/tree/dev/zhaquirks/tuya/bulb

MattWestb, 2022-03-25:
>Can you posting the **Zigbee device signature** from the **device card** with your quirk loaded ?

I like to see how the quirk is allocating the clusters then the quirks is loaded.

sergey-suloev, 2022-03-25:
@MattWestb I just updated to a new HomeAssistant version and ZHA is failing now.

2022-03-25 23:16:12 DEBUG (MainThread) [zhaquirks] Loading quirks module zhaquirks.tuya.air
2022-03-25 23:16:12 ERROR (MainThread) [homeassistant.config_entries] Error setting up entry FT232R USB UART - FT232R USB UART, s/n: A50285BI - FTDI for zha
File "/usr/src/homeassistant/homeassistant/components/zha/__init__.py", line 99, in async_setup_entry
File "/usr/local/lib/python3.9/site-packages/zhaquirks/__init__.py", line 394, in setup
File "/usr/local/lib/python3.9/site-packages/zhaquirks/tuya/air/__init__.py", line 13, in <module>
from zhaquirks.tuya import DPToAttributeMapping, TuyaLocalCluster, TuyaNewManufCluster
ImportError: cannot import name 'DPToAttributeMapping' from 'zhaquirks.tuya' (/usr/local/lib/python3.9/site-packages/zhaquirks/tuya/__init__.py)
2022-03-25 23:16:12 DEBUG (MainThread) [homeassistant.core] Bus:Handling <Event component_loaded[L]: component=zha>
2022-03-25 23:17:56 ERROR (MainThread) [homeassistant.components.websocket_api.http.connection] [548229863888] Received invalid command: zha/device
2022-03-25 23:17:56 ERROR (MainThread) [homeassistant.components.websocket_api.http.connection] [548229863888] Received invalid command: zha/device
2022-03-25 23:18:01 ERROR (MainThread) [homeassistant.components.websocket_api.http.connection] [548229863888] Received invalid command: zha/device
2022-03-25 23:18:01 ERROR (MainThread) [homeassistant.components.websocket_api.http.connection] [548229863888] Received invalid command: zha/device

sergey-suloev, 2022-03-25:
```

{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x010d",
      "in_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0008",
        "0x0300",
        "0x1000"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "242": {
      "profile_id": 41440,
      "device_type": "0x0061",
      "in_clusters": [],
      "out_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "_TZ3210_bfwvfyx1",
  "model": "TS0505B",
  "class": "zhaquirks.tuya.bulb.rgbcct.TuyaRGBCCTLight"
}

```

sergey-suloev, 2022-03-25:
[image]

MattWestb, 2022-03-25:
I was comparing it with one LIDL E14 RGBWW abd my Zigbee signature looks the same (my dont need the quirk but was having problem and was doing one).
```
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x010d",
      "in_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0008",
        "0x0300",
        "0x1000"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "242": {
      "profile_id": 41440,
      "device_type": "0x0061",
      "in_clusters": [],
      "out_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "_TZ3000_odygigth",
  "model": "TS0505A",
  "class": "rgbcct.RGBCCTLight"
}
```
And you device card is writing level, light_color, on_off in the auto naming so the system is detecting it as so.

If you is putting the light on and then clicking on the name you is getting the  settings.
Then clicking in the upper right corner on the setting symbol you shall getting the advance setting for color and brightness like this:
[image]
 for color and brightness like this:

sergey-suloev, 2022-03-26:
@MattWestb yes, I made it work 

thanks

closed 2022-03-26 by sergey-suloev

sergey-suloev, 2022-03-30:
@MattWestb it works well without my customization

MattWestb, 2022-03-30:
I was thinking that but i and one dev was / is having problem with some LIDL (tuya) lights that cant changing colors only CT but its one other strange things we have not founding way.

Great that you have getting it working well !!!

daviswieck, 2022-07-07:
Hi.. I am having a similar problem.... 
The code is identical to the original post, just a different manufacturer. 

This is my first time diving into ZHA... How do I set up the custom quirk? Can I do it locally? Or do I need to pull from github? 

```
{
      "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
      "endpoints": {
        "1": {
          "profile_id": 260,
          "device_type": "0x010d",
          "in_clusters": [
            "0x0000",
            "0x0003",
            "0x0004",
            "0x0005",
            "0x0006",
            "0x0008",
            "0x0300",
            "0x1000",
            "0xef00"
          ],
          "out_clusters": [
            "0x000a",
            "0x0019"
          ]
        },
        "242": {
          "profile_id": 41440,
          "device_type": "0x0061",
          "in_clusters": [],
          "out_clusters": [
            "0x0021"
          ]
        }
      }
    },
  "manufacturer": "_TZ3210_vaiyrvd1",
  "model": "TS0505B",
  "class": "zigpy.device.Device"
}
```

MattWestb, 2022-07-07:
Some information is in the discussion https://github.com/zigpy/zha-device-handlers/discussions/693#discussioncomment-857274 and is working OK.

Then one device have getting one integrated quirk in ZHA dont forgetting deleting the local files so you is using the latest quirk in your system and changes in ZHA can braking your local if doing some changes in the future.

daviswieck, 2022-07-07:
So I've made the folder (/config/custom_zha_quirks/) and uploaded these (rgbcct.py, __init__.py) - https://github.com/zigpy/zha-device-handlers/issues/1429#issuecomment-1079392154 - from above... in the rgbcct.py I changed the `44        MODELS_INFO: [("_TZ3210_vaiyrvd1", "TS0505B")]` to my device id. Once I restart HA, and reconnect the light it should work?

MattWestb, 2022-07-07:
If the system is loading the quirk you shall getting the quirk name in the class instead the standard without quirk `"class": "zigpy.device.Device"`.

Then the quirk you is linked is not verified working and was not merged i cant grantee its working OK. 
You can also trying using the LIDL quirk that shall working and only putting in your device id and it shall working if all cluster on the endpoints is the same and restarting HA after doing the changes so the system is using the updated code.

VCTGomes, 2024-11-29:
Unfortunately it's still missing Do Not Disturb feature.

flybrys, 2024-11-29:
I just purchased a light that has this signature. Seems to work ok with manual control but the color temp range is wrong, and also when the light is triggered by automations or scenes, it does not work. It only works with toggling the switch in the UI.

Also getting an error in HA logs since this joined:

> [Min mireds is 0, setting to 153] Please open an issue on the quirks repo to have this device corrected

Any suggestions on how to correct this? Or should I open a new issue?