zigpy/zha-device-handlers #1438: [BUG] TS004X ignoring duplicate frame

sanasol, 2022-03-21:
**Describe the bug**
Some times my tuya buttons doesn't trigger automation since handler thinks its duplicate frame.
Looks like it blocks 2 frames for duplicate, so zha_event not triggered.
```
2022-03-21 16:46:28 DEBUG (MainThread) [zigpy_znp.api] Received command: ZDO.SrcRtgInd.Callback(DstAddr=0x360A, Relays=[0xDC21])
2022-03-21 16:46:28 DEBUG (Thread-3) [aiosqlite] executing functools.partial(<built-in method execute of sqlite3.Connection object at 0x7f7a24ab70>, 'INSERT INTO relays_v7 VALUES (?, ?)\n                        ON CONFLICT (ieee)\n                        DO UPDATE SET relays=excluded.relays', (0c:43:14:ff:fe:86:af:b0, b'\x01!\xdc'))
2022-03-21 16:46:28 DEBUG (Thread-3) [aiosqlite] operation functools.partial(<built-in method execute of sqlite3.Connection object at 0x7f7a24ab70>, 'INSERT INTO relays_v7 VALUES (?, ?)\n                        ON CONFLICT (ieee)\n                        DO UPDATE SET relays=excluded.relays', (0c:43:14:ff:fe:86:af:b0, b'\x01!\xdc')) completed
2022-03-21 16:46:28 DEBUG (Thread-3) [aiosqlite] executing functools.partial(<built-in method commit of sqlite3.Connection object at 0x7f7a24ab70>)
2022-03-21 16:46:28 DEBUG (Thread-3) [aiosqlite] operation functools.partial(<built-in method commit of sqlite3.Connection object at 0x7f7a24ab70>) completed
2022-03-21 16:46:28 DEBUG (MainThread) [zigpy_znp.api] Received command: AF.IncomingMsg.Callback(GroupId=0x0000, ClusterId=6, SrcAddr=0x360A, SrcEndpoint=1, DstEndpoint=1, WasBroadcast=<Bool.false: 0>, LQI=105, SecurityUse=<Bool.false: 0>, TimeStamp=2757151, TSN=0, Data=b'\x01\x01\xFD\x00', MacSrcAddr=0xDC21, MsgResultRadius=28)
2022-03-21 16:46:28 DEBUG (MainThread) [zigpy.zcl] [0x360a:1:0x0006] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=False disable_default_response=False> manufacturer=None tsn=1 command_id=253>
2022-03-21 16:46:28 DEBUG (MainThread) [zigpy.zcl] [0x360a:1:0x0006] ZCL request 0x00fd: [0]
2022-03-21 16:46:28 DEBUG (MainThread) [tuya] TS004X: ignoring duplicate frame
2022-03-21 16:46:28 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x360A:1:0x0006]: received 'press_type' command with [0] args on cluster_id '6' tsn '1'
2022-03-21 16:46:29 DEBUG (MainThread) [zigpy_znp.api] Received command: ZDO.SrcRtgInd.Callback(DstAddr=0x360A, Relays=[0xDC21])
2022-03-21 16:46:29 DEBUG (Thread-3) [aiosqlite] executing functools.partial(<built-in method execute of sqlite3.Connection object at 0x7f7a24ab70>, 'INSERT INTO relays_v7 VALUES (?, ?)\n                        ON CONFLICT (ieee)\n                        DO UPDATE SET relays=excluded.relays', (0c:43:14:ff:fe:86:af:b0, b'\x01!\xdc'))
2022-03-21 16:46:29 DEBUG (Thread-3) [aiosqlite] operation functools.partial(<built-in method execute of sqlite3.Connection object at 0x7f7a24ab70>, 'INSERT INTO relays_v7 VALUES (?, ?)\n                        ON CONFLICT (ieee)\n                        DO UPDATE SET relays=excluded.relays', (0c:43:14:ff:fe:86:af:b0, b'\x01!\xdc')) completed
2022-03-21 16:46:29 DEBUG (Thread-3) [aiosqlite] executing functools.partial(<built-in method commit of sqlite3.Connection object at 0x7f7a24ab70>)
2022-03-21 16:46:29 DEBUG (Thread-3) [aiosqlite] operation functools.partial(<built-in method commit of sqlite3.Connection object at 0x7f7a24ab70>) completed
2022-03-21 16:46:29 DEBUG (MainThread) [zigpy_znp.api] Received command: AF.IncomingMsg.Callback(GroupId=0x0000, ClusterId=6, SrcAddr=0x360A, SrcEndpoint=1, DstEndpoint=1, WasBroadcast=<Bool.false: 0>, LQI=102, SecurityUse=<Bool.false: 0>, TimeStamp=2819239, TSN=0, Data=b'\x01\x01\xFD\x00', MacSrcAddr=0xDC21, MsgResultRadius=28)
2022-03-21 16:46:29 DEBUG (MainThread) [zigpy.zcl] [0x360a:1:0x0006] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=False disable_default_response=False> manufacturer=None tsn=1 command_id=253>
2022-03-21 16:46:29 DEBUG (MainThread) [zigpy.zcl] [0x360a:1:0x0006] ZCL request 0x00fd: [0]
2022-03-21 16:46:29 DEBUG (MainThread) [tuya] TS004X: ignoring duplicate frame
2022-03-21 16:46:29 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x360A:1:0x0006]: received 'press_type' command with [0] args on cluster_id '6' tsn '1'
```

```
2022-03-21 17:06:05 DEBUG (MainThread) [zigpy_znp.api] Received command: AF.DataRequestExt.Rsp(Status=<Status.SUCCESS: 0>)
2022-03-21 17:06:05 DEBUG (MainThread) [zigpy_znp.api] Received command: AF.DataConfirm.Callback(Status=<Status.SUCCESS: 0>, Endpoint=1, TSN=65)
2022-03-21 17:06:07 DEBUG (MainThread) [zigpy_znp.api] Received command: ZDO.SrcRtgInd.Callback(DstAddr=0x1669, Relays=[0xDC21])
2022-03-21 17:06:07 DEBUG (Thread-3) [aiosqlite] executing functools.partial(<built-in method execute of sqlite3.Connection object at 0x7f7a24ab70>, 'INSERT INTO relays_v7 VALUES (?, ?)\n                        ON CONFLICT (ieee)\n                        DO UPDATE SET relays=excluded.relays', (5c:02:72:ff:fe:06:72:8e, b'\x01!\xdc'))
2022-03-21 17:06:07 DEBUG (Thread-3) [aiosqlite] operation functools.partial(<built-in method execute of sqlite3.Connection object at 0x7f7a24ab70>, 'INSERT INTO relays_v7 VALUES (?, ?)\n                        ON CONFLICT (ieee)\n                        DO UPDATE SET relays=excluded.relays', (5c:02:72:ff:fe:06:72:8e, b'\x01!\xdc')) completed
2022-03-21 17:06:07 DEBUG (Thread-3) [aiosqlite] executing functools.partial(<built-in method commit of sqlite3.Connection object at 0x7f7a24ab70>)
2022-03-21 17:06:07 DEBUG (Thread-3) [aiosqlite] operation functools.partial(<built-in method commit of sqlite3.Connection object at 0x7f7a24ab70>) completed
2022-03-21 17:06:07 DEBUG (MainThread) [zigpy_znp.api] Received command: AF.IncomingMsg.Callback(GroupId=0x0000, ClusterId=6, SrcAddr=0x1669, SrcEndpoint=1, DstEndpoint=1, WasBroadcast=<Bool.false: 0>, LQI=102, SecurityUse=<Bool.false: 0>, TimeStamp=13512143, TSN=0, Data=b'\x01\x01\xFD\x00', MacSrcAddr=0xDC21, MsgResultRadius=28)
2022-03-21 17:06:07 DEBUG (MainThread) [zigpy.zcl] [0x1669:1:0x0006] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=False disable_default_response=False> manufacturer=None tsn=1 command_id=253>
2022-03-21 17:06:07 DEBUG (MainThread) [zigpy.zcl] [0x1669:1:0x0006] ZCL request 0x00fd: [0]
2022-03-21 17:06:07 DEBUG (MainThread) [tuya] TS004X: ignoring duplicate frame
2022-03-21 17:06:07 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x1669:1:0x0006]: received 'press_type' command with [0] args on cluster_id '6' tsn '1'
2022-03-21 17:06:08 DEBUG (MainThread) [zigpy_znp.api] Received command: ZDO.SrcRtgInd.Callback(DstAddr=0x1669, Relays=[0xDC21])
2022-03-21 17:06:08 DEBUG (Thread-3) [aiosqlite] executing functools.partial(<built-in method execute of sqlite3.Connection object at 0x7f7a24ab70>, 'INSERT INTO relays_v7 VALUES (?, ?)\n                        ON CONFLICT (ieee)\n                        DO UPDATE SET relays=excluded.relays', (5c:02:72:ff:fe:06:72:8e, b'\x01!\xdc'))
2022-03-21 17:06:08 DEBUG (Thread-3) [aiosqlite] operation functools.partial(<built-in method execute of sqlite3.Connection object at 0x7f7a24ab70>, 'INSERT INTO relays_v7 VALUES (?, ?)\n                        ON CONFLICT (ieee)\n                        DO UPDATE SET relays=excluded.relays', (5c:02:72:ff:fe:06:72:8e, b'\x01!\xdc')) completed
2022-03-21 17:06:08 DEBUG (Thread-3) [aiosqlite] executing functools.partial(<built-in method commit of sqlite3.Connection object at 0x7f7a24ab70>)
2022-03-21 17:06:08 DEBUG (Thread-3) [aiosqlite] operation functools.partial(<built-in method commit of sqlite3.Connection object at 0x7f7a24ab70>) completed
2022-03-21 17:06:08 DEBUG (MainThread) [zigpy_znp.api] Received command: AF.IncomingMsg.Callback(GroupId=0x0000, ClusterId=6, SrcAddr=0x1669, SrcEndpoint=1, DstEndpoint=1, WasBroadcast=<Bool.false: 0>, LQI=99, SecurityUse=<Bool.false: 0>, TimeStamp=13574411, TSN=0, Data=b'\x01\x01\xFD\x00', MacSrcAddr=0xDC21, MsgResultRadius=28)
2022-03-21 17:06:08 DEBUG (MainThread) [zigpy.zcl] [0x1669:1:0x0006] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=False disable_default_response=False> manufacturer=None tsn=1 command_id=253>
2022-03-21 17:06:08 DEBUG (MainThread) [zigpy.zcl] [0x1669:1:0x0006] ZCL request 0x00fd: [0]
[... cut ...]
2022-03-21 13:26:33 DEBUG (MainThread) [zigpy.zcl] [0xc670:1:0x0006] TS004X: send default response [custom]
2022-03-21 13:27:32 DEBUG (MainThread) [zigpy.zcl] [0xc670:1:0x0006] TS004X: send default response [custom]
2022-03-21 13:27:35 DEBUG (MainThread) [zigpy.zcl] [0x1669:1:0x0006] TS004X: send default response [custom]
2022-03-21 13:45:45 DEBUG (MainThread) [tuya] TS004X: ignoring duplicate frame (custom)
2022-03-21 13:45:45 DEBUG (MainThread) [zigpy.zcl] [0x360a:1:0x0006] TS004X: send default response [custom]
```

**Expected behavior**
Turn on light by button

sanasol, 2022-03-21:
My buttons is TS0041-TS0043, not TS004f

> is both frames sent from the device the same command (= have the same tsn 1 and 1) and its one bug in the device firmware that is making it sending cone command more times also then we is sending default response to it from the quirk so it shall not doing that if being normal Zigbee device.

I cannot replace button firmware, so probably only way is fix it by quirk.

Also not sure that its device bug. Maybe frames somehow doubled while transferring through relay.

> Look is you is getting pressed, double and long in development tools > Evenets and putting in zha_event in listening to and clicking start listening and you can see what events ZHA is getting from the device.

I see events when everything goes fine.
But few times per day every button(different models 41-42-43) not triggering zha_event.

Log file contains two duplicate frame error or single duplicate frame and default response.
Like this
```
2022-03-21 13:45:45 DEBUG (MainThread) [tuya] TS004X: ignoring duplicate frame
2022-03-21 13:45:45 DEBUG (MainThread) [zigpy.zcl] [0x360a:1:0x0006] TS004X: send default response
```
In this case zha_event not triggered for some reason.

MattWestb, 2022-03-21:
In the last log you was posting  is the default response to late for the device getting it before reseeding the command (its think its was not  reached the receiver device and is re sending it).

The device is sending unicst and is using sequence number (tsn) for having control if the if the frame is being lost or duplicated but uncast is never being duplicated in the network if its working OK. If the sender is not getting one default response of one unicast it  shall being repeated then its one lost frame but the TS004X is all not doing it OK.

The ZHA is sending default response and ignoring duplicate message is OK and its generated of the device quirk and its no error only info then the device is doing it wrong and they is indicating that the system have receiving the command and you shall have getting one `zha_events`.

Still look if you is getting events from the device and the rest is only info in the log from the quirk and no errors in the system (only bad behaving device).

The TS0041-TS0043, and TS004f is using the same commands and also have the same bugs in the events ("scene") mode commands that is not Zigbee standard but is working OK with the quirk we is using.

closed 2022-09-24 by github-actions[bot]