zigpy/zha-device-handlers #1439: [Device Support Request] TS0601 Thermostat  _TZE200_cpmgn2cf

RamiRihawi, 2022-03-22:
**Is your feature request related to a problem? Please describe.**
This version of the device is not compatible with ZHA yet.

**Describe the solution you'd like**
The device is added without any entities.
I want to be able to control at least the close and open functionality of the device

**Device signature - this can be acquired by removing the device from ZHA and pairing it again from the add devices screen.

Please find below device signature in code block and Logs while adding the device

`{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0051",
      "in_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0xef00"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZE200_cpmgn2cf",
  "model": "TS0601",
  "class": "zigpy.device.Device"
}`

LOGS:
> [0x0000:zdo] ZDO request ZDOCmd.Mgmt_Permit_Joining_rsp: [<Status.SUCCESS: 0>]
[0x0000:zdo] No handler for ZDO request:ZDOCmd.Mgmt_Permit_Joining_rsp([<Status.SUCCESS: 0>])
New device 0x7728 (90:35:ea:ff:fe:2a:87:81) joined the network
[0x7728] Scheduling initialization
Received frame on uninitialized device <Device model=None manuf=None nwk=0x7728 ieee=90:35:ea:ff:fe:2a:87:81 is_initialized=False> from ep 0 to ep 0, cluster 19: b'\x85(w\x81\x87*\xfe\xff\xea5\x90\x80'
[0x7728:zdo] ZDO request ZDOCmd.Device_annce: [0x7728, 90:35:ea:ff:fe:2a:87:81, 128]
Tries remaining: 3
[0x7728] Requesting 'Node Descriptor'
Tries remaining: 2
[0x7728] Extending timeout for 0x31 request
Received frame on uninitialized device <Device model=None manuf=None nwk=0x7728 ieee=90:35:ea:ff:fe:2a:87:81 is_initialized=False> from ep 0 to ep 0, cluster 32770: b'1\x00(w\x02@\x80\x02\x10RR\x00\x00,R\x00\x00'
[0x7728] Got Node Descriptor: NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)
[0x7728] Discovering endpoints
Tries remaining: 3
[0x7728] Extending timeout for 0x32 request
Received frame on uninitialized device <Device model=None manuf=None nwk=0x7728 ieee=90:35:ea:ff:fe:2a:87:81 is_initialized=False> from ep 0 to ep 0, cluster 32773: b'2\x00(w\x01\x01'
[0x7728] Discovered endpoints: [1]
[0x7728] Initializing endpoints [<Endpoint id=1 in=[] out=[] status=<Status.NEW: 0>>]
[0x7728:1] Discovering endpoint information
Tries remaining: 3
[0x7728] Extending timeout for 0x33 request
Received frame on uninitialized device <Device model=None manuf=None nwk=0x7728 ieee=90:35:ea:ff:fe:2a:87:81 is_initialized=False> from ep 0 to ep 0, cluster 32772: b'3\x00(w\x14\x01\x04\x01Q\x00\x01\x04\x00\x00\x04\x00\x05\x00\x00\xef\x02\x19\x00\n\x00'
[0x7728:1] Discovered endpoint information: SizePrefixedSimpleDescriptor(endpoint=1, profile=260, device_type=81, device_version=1, input_clusters=[0, 4, 5, 61184], output_clusters=[25, 10])
Unknown cluster 61184
[0x7728] Extending timeout for 0x34 request
[0x7728:1:0x0000] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=52 command_id=Command.Read_Attributes_rsp>
[0x7728] Read model 'TS0601' and manufacturer '_TZE200_cpmgn2cf' from <Endpoint id=1 in=[basic:0x0000, groups:0x0004, scenes:0x0005, None:0xEF00] out=[ota:0x0019, time:0x000A] status=<Status.ZDO_INIT: 1>>
[0x7728] Discovered basic device information for <Device model='TS0601' manuf='_TZE200_cpmgn2cf' nwk=0x7728 ieee=90:35:ea:ff:fe:2a:87:81 is_initialized=True>
Device is initialized <Device model='TS0601' manuf='_TZE200_cpmgn2cf' nwk=0x7728 ieee=90:35:ea:ff:fe:2a:87:81 is_initialized=True>
Checking quirks for _TZE200_cpmgn2cf TS0601 (90:35:ea:ff:fe:2a:87:81)
Considering <class 'zhaquirks.xbee.xbee_io.XBeeSensor'>
Fail because endpoint list mismatch: {232, 230} {1}
Considering <class 'zhaquirks.xbee.xbee3_io.XBee3Sensor'>
Fail because endpoint list mismatch: {232, 230} {1}
Considering <class 'zhaquirks.smartthings.tag_v4.SmartThingsTagV4'>
[... 139 lines cut ...]
[0x7728:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=68 command_id=1>
[0x7728:1:0xef00] Unknown cluster-specific command 1
[0x7728:1:0xef00] ZCL request 0x0001: b'\x00?k\x02\x00\x04\x00\x00\x00\x14'
[0x7728:1:0xef00] No handler for cluster command 1
[0x7728:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=69 command_id=1>
[0x7728:1:0xef00] Unknown cluster-specific command 1
[0x7728:1:0xef00] ZCL request 0x0001: b'\x00@l\x02\x00\x04\x00\x00\x00\x0f'
[0x7728:1:0xef00] No handler for cluster command 1
[0x7728:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=70 command_id=1>
[0x7728:1:0xef00] Unknown cluster-specific command 1
[0x7728:1:0xef00] ZCL request 0x0001: b'\x00Am\x02\x00\x04\x00\x00\x00\x00'
[0x7728:1:0xef00] No handler for cluster command 1
[0x7728:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=71 command_id=1>
[0x7728:1:0xef00] Unknown cluster-specific command 1
[0x7728:1:0xef00] ZCL request 0x0001: b'\x00Bn\x01\x00\x01\x00'
[0x7728:1:0xef00] No handler for cluster command 1
[0x82F5:1:0x0b04]: failed to get attributes '['active_power', 'rms_current', 'rms_voltage']' on 'electrical_measurement' cluster: 
[0x53B7:1:0x0500]: Updated alarm state: 0
[0x7728:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=72 command_id=1>
[0x7728:1:0xef00] Unknown cluster-specific command 1
[0x7728:1:0xef00] ZCL request 0x0001: b'\x00Co\x04\x00\x01\x00'
[0x7728:1:0xef00] No handler for cluster command 1

**Additional context**
[https://www.aliexpress.com/item/4001065119510.html?spm=a2g0o.order_list.0.0.21ef1802WETkFk](url)

MattWestb, 2022-03-22:
Its one "clasic" quirk for this device but your new device ID is not added but can being done in the quirk and also in ZHA for getting all functions if it.
Add your ID on one new line here:
https://github.com/zigpy/zha-device-handlers/blob/039a31f1ca15821e800bdb347eb47fdb02455f31/zhaquirks/tuya/ts0601_trv.py#L1422-L1429
And also in ZHA here
https://github.com/home-assistant/core/blob/78a41022ade7f070db6ed5c7678a0ba1171b72e6/homeassistant/components/zha/climate.py#L595-L608
Its also one working local quirk with more function but its not merged by you can installing it as local quirk: https://github.com/zigpy/zha-device-handlers/pull/1192 (its also need the adding in ZHA climate.py).

Using local quirk is described here https://github.com/zigpy/zha-device-handlers/discussions/693#discussioncomment-857274.
The ZHA climate.py need being edited in the HA container https://github.com/zigpy/zha-device-handlers/discussions/693#discussion-1237451.

RamiRihawi, 2022-03-22:
I would appreciate your help on how to add the device ID to ts0601_trv.py
All i have right now is HA installed on RPi 4 with ZHA added as an integration.

MattWestb, 2022-03-22:
Patched `climate.py` for ZHA container and `ts0601_trv.py` (classic) and and (neo) `ts0601_trv_moes.py`. Dont use both at the same time then it can making problems.
[ts0601_trv.zip](https://github.com/zigpy/zha-device-handlers/files/8327524/ts0601_trv.zip)

Copy the `climate.py` to `/usr/src/homeassistant/homeassistant/components/zha/climate.py` in the HA container with command `cp /usr/src/homeassistant/homeassistant/components/zha/climate.py /config/climate.py` inside the HA container after have copy the climate.py to the HA confing directory.
The quirk you can configure local quirk and copy one of them in the new folder and restarting HA and it shall being loaded is all is OK.

RamiRihawi, 2022-03-22:
Thank you very much for your help,
One thing though, i do not have HA installed as a container, but as a full system on RPi4.
That means i cannot use cp commands.

Do you have another suggestion?

MattWestb, 2022-03-22:
If you is running on one real OS is the path different but i dont remember it but the end is the same then you is finding the local ha files in your system.

RamiRihawi, 2022-03-22:
All is working now, Thank you a lot.

closed 2022-03-22 by RamiRihawi

MattWestb, 2022-03-23:
Witch version did you trying ?
I like making one PR for adding it in the system permanent so not need doing all one more time then updating HA.

RamiRihawi, 2022-03-23:
From the Zip folder you sent me i used first "ts0601_trv.py" which reviled only three entities. 
and i added the climate.py to the config folder of HA.

Then i tried the second one "ts0601_trv_moes.py" after removing the first one.
now i have all entities listed.

Reproduktor, 2022-11-04:
Hi all, I get this problem when launching HA with this quirk (HA 2022.11):

``TypeError: `manufacturer_attributes` is deprecated. Copy the parent class's `attributes` dictionary and update it with your manufacturer-specific `attributes`. Make sure to specify that it is manufacturer-specific through the  appropriate constructor or tuple!``

Any ideas?

Frogowitsch, 2023-01-08:
Same problem here as Reproduktor. Is the quirk not compatible with newer HA versions or what can be the problem ?

Frogowitsch, 2023-01-09:
I get a lot of functions from this tvr, but what i need is force valve open and close. This one does not show up. Any idea what the problem can be ? All other functions are there.

bennipi1287, 2023-02-27:
> I get a lot of functions from this tvr, but what i need is force valve open and close. This one does not show up. Any idea what the problem can be ? All other functions are there.

How did you fix the problem?

Theo16, 2023-06-19:
I have HA with ZHA lasted version and my valve _TZE200_cpmgn2cf TS0601 does not show entities like valve position, window open etc. like in zigbee2mqtt. Is it possible to have this entities in ZHA?

pepelillo, 2023-10-05:
Same here, same device same behaviour.

PR #1443 (2022-03-24): Adding 2 new tuya TRVs
Adding `_TZE200_cpmgn2cf` and `_TZE200_9sfg7gm0` that is new / updated version of the normal TH TRVs.

Fix #1439