zigpy/zha-device-handlers #1440: Quirky tripper door sensor battery not reporting

cbarth3, 2022-03-22:
I am creating this issue as this was the last post I could find about this. I can see voltage info but the entity shows as unkown.

I can confirm, the latest version of HA, just added the Quirky tripper door sensor and seeing this:
[image]

_Originally posted by @42ism in https://github.com/zigpy/zha-device-handlers/issues/33#issuecomment-792041390_

MattWestb, 2022-03-22:
You need making one quirk and adding this in the replacement:
https://github.com/zigpy/zha-device-handlers/blob/039a31f1ca15821e800bdb347eb47fdb02455f31/zhaquirks/__init__.py#L188.
of the `PowerConfiguration` cluster then the system shall calculating the battery from the voltage,

MattWestb, 2022-03-22:
Its looks some device from the same manufacture have implanting on fix in one other quirk:
https://github.com/zigpy/zha-device-handlers/blob/039a31f1ca15821e800bdb347eb47fdb02455f31/zhaquirks/sercomm/szwtd02n.py#L21-L25

cbarth3, 2022-03-23:
Thanks for info. Not sure how I can implement that into my home assistant.

javicalle, 2022-03-23:
Probably just adding the MODEL to the existing quirk would do the job:
```python
class SZWTD02N(CustomDevice):
    """Custom device representing Sercomm SZ-WTD02N flood sensor."""

    signature = {
        #  <SimpleDescriptor endpoint=1 profile=260 device_type=1026
        #  device_version=0
        #  input_clusters=[0, 1, 3, 1280, 32, 2821]
        #  output_clusters=[3, 25]>
        MODELS_INFO: [
            (SERCOMM, "SZ-WTD02N_SF"),
            (SERCOMM, "Tripper"),
        ],
```

You need to edit the `szwtd02n.py` file in your installation or copy in a local quirks path:

There are some guides here:
* https://github.com/zigpy/zha-device-handlers/discussions/693

closed 2022-05-08 by cbarth3