zigpy/zha-device-handlers #1444: [BUG] New Ikea Firmware breaks Ikea-Switches

idunnololz, 2022-03-24:
**Describe the bug**
Ikea light & blind switches cannot control zigbee groups after firmware update.

**To Reproduce**
Steps to reproduce the behavior:
1. Update switch to latest firmware.
2. Switch can be bound to groups but cannot control them.

**Expected behavior**
Switches should work as before the firmware update. They should be able to control zigbee groups.

**Additional context**
I noticed this exact bug reported on the deconz github which was later resolved, perhaps the fix is similar?
Bug report on Deconz: https://github.com/dresden-elektronik/deconz-rest-plugin/issues/5449
Fix on Deconz: https://github.com/dresden-elektronik/deconz-rest-plugin/pull/5538/commits/5ea825f7e9d7c2df80797d266d045070148b3f88

idunnololz, 2022-03-24:
Device signature:
```
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4476, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0203",
      "in_clusters": [
        "0x0000",
        "0x0001",
        "0x0003",
        "0x0009",
        "0x0020",
        "0x1000",
        "0xfc7c"
      ],
      "out_clusters": [
        "0x0003",
        "0x0004",
        "0x0006",
        "0x0008",
        "0x0019",
        "0x0102",
        "0x1000"
      ]
    }
  },
  "manufacturer": "IKEA of Sweden",
  "model": "TRADFRI open/close remote",
  "class": "zhaquirks.ikea.opencloseremote.IkeaTradfriOpenCloseRemote"
}
```

MattWestb, 2022-03-24:
ZHA is doing all it can with all updated IKEA and we have updating some devices that have changeling behavior for getting them working OK
But IKEA have taking away the group binding and implanting device binding and we cant implanting it in IKEAs firmware then we dont have the code they is using and also we doing have the certificate they is signing there firmware for doing OTA updates.

Pleas contact IKEA and doing on reclamation of your controlling devices.

MattWestb, 2022-03-24:
If you is having the skills and one SWD-probe you can flashing one old firmware in all you IKEA controllers that we have extracted from old OTA files.

More info (from ZHA and Z2M users) https://github.com/zigpy/zigpy/discussions/660#discussioncomment-1565541.

idunnololz, 2022-03-24:
I see, thank you. I will try to downgrade the firmware later today.

closed 2022-09-27 by github-actions[bot]