zigpy/zha-device-handlers #1457: 3A Smart Switch shows as Light instead of switch

austwhite, 2022-04-03:
Not sure if this is in the right place and I am not overly technical.

I have a 3A Smart Switch.  It shows in Deconz as a Switch, model LXN56-LC27LX1.3
I wanted to move over to ZHA, but this device is showing as a light instead of a switch in ZHA.
Can this be updated at all, or if there is a way I can update it can someone tell me, in relatively simple terms, how?

Also, if this is posted in the wrong place, please guide me to the right place.

Thanks

Hedda, 2022-04-07:
FYI, while not solving the root issue for this, know that helpers were introduced in Home Assistant 2022.4 release as workarounds:

https://www.home-assistant.io/blog/2022/04/06/release-20224/#switch-as-x

"Switch as X" can be found under "Configuration > Automations & Scripts > Helpers" and when editing the switch entity.

https://www.home-assistant.io/integrations/switch_as_x

The "Switch as X” helper lets you convert a switch to a light, but in your case, I guess you would need a new "Light as X" helper?

https://www.home-assistant.io/blog/2022/04/06/release-20224/#more-helpers-now-available-via-the-ui

Again such a workaround would not solve the root cause of a switch device presenting itself as a light device in its firmware.

Tropaion, 2022-04-10:
@austwhite Do you have a link to the device and the device signature?

Hedda, 2022-04-11:
@austwhite  Yes you should have filled in the form for "Device support request" which asked for device signature

**Device signature - this can be acquired by removing the device from ZHA and pairing it again from the add devices screen. Be sure to add the entire content of the log panel after pairing the device to a code block below this line.**

https://www.home-assistant.io/integrations/zha#reporting-issues

If issue is related to a specific Zigbee device, provide device Zigbee signature. Signature is available at Configuration -> Devices & Services -> Zigbee Home Automation (click Configure) -> Devices (pick your device) -> Zigbee Device Signature

TheJulianJES, 2022-04-17:
Also see https://www.home-assistant.io/integrations/zha/#modifying-the-device-type on how to modify the device type reported by ZHA to Home Assistant.

austwhite, 2022-04-18:
Sorry for the delay responding to this.  Switch as X won't help as it detects as a light and should be a switch.
I'll get the additional information Device a signature and link to the device.

austwhite, 2022-04-18:
@Tropaion 
I have stayed with Deconz for now, even though I would prefer to go to ZHA.
Not sure how to find the device signature in Deconz, but I have this information
Manufacturer: 3A Smart Home DE
Model Identifier: LXN56-LC27LX1.3
Type: Router

Also found this in Zigbee to MQTT https://www.zigbee2mqtt.io/devices/HGZB-01A.html
This is the same device.   
Sorry,  I am not so good with Zigbee stuff.
If there is not enough to work with, or this needs to go elsewhere I will close it off relog it when I work out how to get the information

closed 2022-04-18 by austwhite

Hedda, 2022-04-19:
@austwhite  Please just leave open for now.