zigpy/zha-device-handlers #1458: [Device Support Request] Tuya Smart Plug TS011F by _TZ3000_v1pdxuqq

arantes555, 2022-04-03:
**Is your feature request related to a problem? Please describe.**
Power metering does not work on HA. Only on/off works.

**Describe the solution you'd like**
Making power metering work.

**Device signature - this can be acquired by removing the device from ZHA and pairing it again from the add devices screen. Be sure to add the entire content of the log panel after pairing the device to a code block below this line.**

```
[0x0000:zdo] ZDO request ZDOCmd.Mgmt_Permit_Joining_req: [60, <Bool.false: 0>]
New device 0x774f (a4:c1:38:fb:b6:16:43:7c) joined the network
[0x774f] Scheduling initialization
Tries remaining: 3
[0x774f] Requesting 'Node Descriptor'
Tries remaining: 2
[0x774f] Extending timeout for 0x17 request
Device 0x774f (a4:c1:38:fb:b6:16:43:7c) joined the network
[0x774f] Scheduling initialization
[0x774f] Canceling old initialize call
Received frame on uninitialized device <Device model=None manuf=None nwk=0x774F ieee=a4:c1:38:fb:b6:16:43:7c is_initialized=False> from ep 0 to ep 0, cluster 19: b'\x04Ow|C\x16\xb6\xfb8\xc1\xa4\x8e'
[0x774f:zdo] ZDO request ZDOCmd.Device_annce: [0x774F, a4:c1:38:fb:b6:16:43:7c, 142]
Tries remaining: 3
[0x774f] Requesting 'Node Descriptor'
Tries remaining: 2
[0x774f] Extending timeout for 0x19 request
Received frame on uninitialized device <Device model=None manuf=None nwk=0x774F ieee=a4:c1:38:fb:b6:16:43:7c is_initialized=False> from ep 0 to ep 0, cluster 32770: b'\x17\x00Ow\x01@\x8eA\x11BB\x00\x00*B\x00\x00'
[0x774f] Got Node Descriptor: NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)
[0x774f] Discovering endpoints
Tries remaining: 3
Received frame on uninitialized device <Device model=None manuf=None nwk=0x774F ieee=a4:c1:38:fb:b6:16:43:7c is_initialized=False> from ep 0 to ep 0, cluster 32770: b'\x19\x00Ow\x01@\x8eA\x11BB\x00\x00*B\x00\x00'
[0x774f] Got Node Descriptor: NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)
[0x774f] Discovering endpoints
Tries remaining: 3
Received frame on uninitialized device <Device model=None manuf=None nwk=0x774F ieee=a4:c1:38:fb:b6:16:43:7c is_initialized=False> from ep 0 to ep 0, cluster 32773: b'\x1b\x00Ow\x02\x01\xf2'
Received frame on uninitialized device <Device model=None manuf=None nwk=0x774F ieee=a4:c1:38:fb:b6:16:43:7c is_initialized=False> from ep 0 to ep 0, cluster 32773: b'\x1d\x00Ow\x02\x01\xf2'
[0x774f] Discovered endpoints: [1, 242]
[0x774f] Initializing endpoints [<Endpoint id=1 in=[] out=[] status=<Status.NEW: 0>>, <Endpoint id=242 in=[] out=[] status=<Status.NEW: 0>>]
[0x774f:1] Discovering endpoint information
Tries remaining: 3
[0x774f] Discovered endpoints: [1, 242]
[0x774f] Initializing endpoints [<Endpoint id=1 in=[] out=[] status=<Status.NEW: 0>>, <Endpoint id=242 in=[] out=[] status=<Status.NEW: 0>>]
[0x774f:1] Discovering endpoint information
Tries remaining: 3
Received frame on uninitialized device <Device model=None manuf=None nwk=0x774F ieee=a4:c1:38:fb:b6:16:43:7c is_initialized=False> from ep 0 to ep 0, cluster 32772: b'\x1f\x00Ow\x1a\x01\x04\x01\n\x01\x01\x07\x03\x00\x04\x00\x05\x00\x06\x00\x00\xe0\x01\xe0\x00\x00\x02\x19\x00\n\x00'
[0x774f:1] Discovered endpoint information: SizePrefixedSimpleDescriptor(endpoint=1, profile=260, device_type=266, device_version=1, input_clusters=[3, 4, 5, 6, 57344, 57345, 0], output_clusters=[25, 10])
Unknown cluster 57344
Unknown cluster 57345
[0x774f:242] Discovering endpoint information
Tries remaining: 3
Received frame on uninitialized device <Device model=None manuf=None nwk=0x774F ieee=a4:c1:38:fb:b6:16:43:7c is_initialized=False> from ep 0 to ep 0, cluster 32772: b'!\x00Ow\x1a\x01\x04\x01\n\x01\x01\x07\x03\x00\x04\x00\x05\x00\x06\x00\x00\xe0\x01\xe0\x00\x00\x02\x19\x00\n\x00'
[0x774f:1] Discovered endpoint information: SizePrefixedSimpleDescriptor(endpoint=1, profile=260, device_type=266, device_version=1, input_clusters=[3, 4, 5, 6, 57344, 57345, 0], output_clusters=[25, 10])
Unknown cluster 57344
Unknown cluster 57345
[0x774f:242] Discovering endpoint information
Tries remaining: 3
Received frame on uninitialized device <Device model=None manuf=None nwk=0x774F ieee=a4:c1:38:fb:b6:16:43:7c is_initialized=False> from ep 0 to ep 0, cluster 32772: b'#\x00Ow\n\xf2\xe0\xa1a\x00\x00\x00\x01!\x00'
[0x774f:242] Discovered endpoint information: SizePrefixedSimpleDescriptor(endpoint=242, profile=41440, device_type=97, device_version=0, input_clusters=[], output_clusters=[33])
Received frame on uninitialized device <Device model=None manuf=None nwk=0x774F ieee=a4:c1:38:fb:b6:16:43:7c is_initialized=False> from ep 0 to ep 0, cluster 32772: b'%\x00Ow\n\xf2\xe0\xa1a\x00\x00\x00\x01!\x00'
[0x774f:242] Discovered endpoint information: SizePrefixedSimpleDescriptor(endpoint=242, profile=41440, device_type=97, device_version=0, input_clusters=[], output_clusters=[33])
[0x774f:1:0x0000] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=39 command_id=Command.Read_Attributes_rsp>
[0x774f] Read model 'TS011F' and manufacturer '_TZ3000_v1pdxuqq' from <Endpoint id=1 in=[identify:0x0003, groups:0x0004, scenes:0x0005, on_off:0x0006, None:0xE000, None:0xE001, basic:0x0000] out=[ota:0x0019, time:0x000A] status=<Status.ZDO_INIT: 1>>
[0x774f] Discovered basic device information for <Device model='TS011F' manuf='_TZ3000_v1pdxuqq' nwk=0x774F ieee=a4:c1:38:fb:b6:16:43:7c is_initialized=True>
Device is initialized <Device model='TS011F' manuf='_TZ3000_v1pdxuqq' nwk=0x774F ieee=a4:c1:38:fb:b6:16:43:7c is_initialized=True>
Checking quirks for _TZ3000_v1pdxuqq TS011F (a4:c1:38:fb:b6:16:43:7c)
Considering <class 'zhaquirks.tuya.ts011f_plug.Plug_4AC_2USB'>
Fail because endpoint list mismatch: {1, 2, 3, 4, 5, 242} {1, 242}
Considering <class 'zhaquirks.tuya.ts011f_plug.Plug_3AC_4USB'>
Fail because endpoint list mismatch: {242, 1, 2, 3} {1, 242}
Considering <class 'zhaquirks.tuya.ts011f_plug.Plug_1AC'>
Fail because endpoint list mismatch: {242, 11} {1, 242}
[... 203 lines cut ...]
Duplicate 148 TSN
[0x774f:1:0x000a] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=False disable_default_response=False> manufacturer=None tsn=148 command_id=Command.Read_Attributes>
[0x774f:1:0x000a] ZCL request 0x0000: [[7]]
Duplicate 148 TSN
[0x774f:1:0x000a] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=False disable_default_response=False> manufacturer=None tsn=148 command_id=Command.Read_Attributes>
[0x774f:1:0x000a] ZCL request 0x0000: [[7]]
Duplicate 148 TSN
[0x774f:1:0x0006] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=149 command_id=Command.Report_Attributes>
[0x774f:1:0x0006] ZCL request 0x000a: [[Attribute(attrid=0, value=<TypeValue type=Bool, value=Bool.true>)]]
[0x774f:1:0x0006] Attribute report received: on_off=1
[... cut ...]
[0xa136:1:0x0001] Attribute report received: battery_percentage_remaining=137
[0xa136:1:0x0405] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=55 command_id=Command.Report_Attributes>
[0xa136:1:0x0405] ZCL request 0x000a: [[Attribute(attrid=0, value=<TypeValue type=uint16_t, value=6161>)]]
[0xa136:1:0x0405] Attribute report received: measured_value=6161
```

**Additional context**

MattWestb, 2022-04-03:
Can you posting one link to the place the device was bayed then its dont have any cluster for power meriting.
Also is it having other functions that can being added in one new quirk class ?
If you like testing adding this device class to the normal https://github.com/zigpy/zha-device-handlers/blob/dev/zhaquirks/tuya/ts011f_plug.py and configuring local quirk in HA config and putting the new quirk in the directory you have doing for local quirk and restarting HA:
```
class Plug_1AC_GPP(CustomDevice):
    """Tuya 1 outlet with restore power state support."""

    signature = {
        MODEL: "TS011F",
        ENDPOINTS: {
            # <SimpleDescriptor endpoint=1 profile=260 device_type=266
            # device_version=1
            # input_clusters=[0, 3, 4, 5, 6, 57344, 57345]
            # output_clusters=[10, 25]>
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_PLUG_IN_UNIT,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Identify.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    OnOff.cluster_id,
                    TuyaZBE000Cluster.cluster_id,
                    TuyaZBExternalSwitchTypeCluster.cluster_id,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            },
            # <SimpleDescriptor endpoint=242 profile=41440 device_type=97
            # device_version=1
            # input_clusters=[]
            # output_clusters=[33]>
            242: {
                PROFILE_ID: 41440,
                DEVICE_TYPE: 97,
                INPUT_CLUSTERS: [],
                OUTPUT_CLUSTERS: [GreenPowerProxy.cluster_id],
            },
        },
    }
    replacement = {
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_PLUG_IN_UNIT,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Identify.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaZBOnOffAttributeCluster,
                    TuyaZBE000Cluster,
                    TuyaZBExternalSwitchTypeCluster,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            },
            242: {
                PROFILE_ID: 41440,
                DEVICE_TYPE: 97,
                INPUT_CLUSTERS: [],
                OUTPUT_CLUSTERS: [GreenPowerProxy.cluster_id],
            },
        },
    }
```
If it working you is getting power on state, light mode and switch mode and in the near future children lock for the button.

arantes555, 2022-04-03:
> Can you posting one link to the place the device was bayed then its dont have any cluster for power meriting.

I am not sure I understand your question, but if you are asking for where I bought this plug, I bought it from AliExpress. Here's the link : https://www.aliexpress.com/item/1005002248492657.html . It is indeed advertised as having power monitoring.

As for testing with the provided quirk, I will try setting up a local quirks directory and trying this one, thank you very much ! I'll report back here when it is done.

arantes555, 2022-04-06:
@MattWestb I just tried to custom quirk. From what I see in the "Device info" section, the quirk seems to be applied correctly.

However, I am not sure what it is supposed to do, but I do not see any difference before / after I added the quirk : the plug still has exactly the same entities an On/off control, a diagnostics identify button which does not seem to do anything, and disabled entities for LQI and RSSI. There are no new entities for power measurement, or anything related to child safety or anything else.

I also tried adding another fresh plug of the same model to ZHA. Same behaviour (except that it somehow keeps flashing the internal LED). Here's the resulting log : 
```
[0x0000:zdo] ZDO request ZDOCmd.Mgmt_Permit_Joining_req: [60, <Bool.false: 0>]
New device 0x592b (a4:c1:38:b3:53:5a:3f:3e) joined the network
[0x592b] Scheduling initialization
Tries remaining: 3
[0x592b] Requesting 'Node Descriptor'
Tries remaining: 2
[0x592b] Extending timeout for 0xe7 request
Device 0x592b (a4:c1:38:b3:53:5a:3f:3e) joined the network
[0x592b] Scheduling initialization
[0x592b] Canceling old initialize call
Received frame on uninitialized device <Device model=None manuf=None nwk=0x592B ieee=a4:c1:38:b3:53:5a:3f:3e is_initialized=False> from ep 0 to ep 0, cluster 19: b'\x00+Y>?ZS\xb38\xc1\xa4\x8e'
[0x592b:zdo] ZDO request ZDOCmd.Device_annce: [0x592B, a4:c1:38:b3:53:5a:3f:3e, 142]
Tries remaining: 3
[0x592b] Requesting 'Node Descriptor'
Tries remaining: 2
[0x592b] Extending timeout for 0xe9 request
Received frame on uninitialized device <Device model=None manuf=None nwk=0x592B ieee=a4:c1:38:b3:53:5a:3f:3e is_initialized=False> from ep 0 to ep 0, cluster 32770: b'\xe7\x00+Y\x01@\x8eA\x11BB\x00\x00*B\x00\x00'
[0x592b] Got Node Descriptor: NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)
[0x592b] Discovering endpoints
Tries remaining: 3
Received frame on uninitialized device <Device model=None manuf=None nwk=0x592B ieee=a4:c1:38:b3:53:5a:3f:3e is_initialized=False> from ep 0 to ep 0, cluster 32773: b'\xeb\x00+Y\x02\x01\xf2'
[0x592b] Discovered endpoints: [1, 242]
[0x592b] Initializing endpoints [<Endpoint id=1 in=[] out=[] status=<Status.NEW: 0>>, <Endpoint id=242 in=[] out=[] status=<Status.NEW: 0>>]
[0x592b:1] Discovering endpoint information
Tries remaining: 3
Received frame on uninitialized device <Device model=None manuf=None nwk=0x592B ieee=a4:c1:38:b3:53:5a:3f:3e is_initialized=False> from ep 0 to ep 0, cluster 32772: b'\xed\x00+Y\x1a\x01\x04\x01\n\x01\x01\x07\x03\x00\x04\x00\x05\x00\x06\x00\x00\xe0\x01\xe0\x00\x00\x02\x19\x00\n\x00'
Received frame on uninitialized device <Device model=None manuf=None nwk=0x592B ieee=a4:c1:38:b3:53:5a:3f:3e is_initialized=False> from ep 0 to ep 0, cluster 32772: b'\xed\x00+Y\x1a\x01\x04\x01\n\x01\x01\x07\x03\x00\x04\x00\x05\x00\x06\x00\x00\xe0\x01\xe0\x00\x00\x02\x19\x00\n\x00'
[0x592b] Invalid state on future for 0xed seq -- probably duplicate response
[0x592b:1] Discovered endpoint information: SizePrefixedSimpleDescriptor(endpoint=1, profile=260, device_type=266, device_version=1, input_clusters=[3, 4, 5, 6, 57344, 57345, 0], output_clusters=[25, 10])
Unknown cluster 57344
Unknown cluster 57345
[0x592b:242] Discovering endpoint information
Tries remaining: 3
Received frame on uninitialized device <Device model=None manuf=None nwk=0x592B ieee=a4:c1:38:b3:53:5a:3f:3e is_initialized=False> from ep 0 to ep 0, cluster 32772: b'\xef\x00+Y\n\xf2\xe0\xa1a\x00\x00\x00\x01!\x00'
[0x592b:242] Discovered endpoint information: SizePrefixedSimpleDescriptor(endpoint=242, profile=41440, device_type=97, device_version=0, input_clusters=[], output_clusters=[33])
[0x592b:1:0x0000] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=241 command_id=Command.Read_Attributes_rsp>
[0x592b:zdo] ZDO request ZDOCmd.Node_Desc_req: [0x0000]
[0x592b:zdo] No handler for ZDO request:ZDOCmd.Node_Desc_req([0x0000])
[0x592b] Read model 'TS011F' and manufacturer '_TZ3000_v1pdxuqq' from <Endpoint id=1 in=[identify:0x0003, groups:0x0004, scenes:0x0005, on_off:0x0006, None:0xE000, None:0xE001, basic:0x0000] out=[ota:0x0019, time:0x000A] status=<Status.ZDO_INIT: 1>>
[0x592b] Discovered basic device information for <Device model='TS011F' manuf='_TZ3000_v1pdxuqq' nwk=0x592B ieee=a4:c1:38:b3:53:5a:3f:3e is_initialized=True>
Device is initialized <Device model='TS011F' manuf='_TZ3000_v1pdxuqq' nwk=0x592B ieee=a4:c1:38:b3:53:5a:3f:3e is_initialized=True>
Checking quirks for _TZ3000_v1pdxuqq TS011F (a4:c1:38:b3:53:5a:3f:3e)
Considering <class 'custom_ts011f.Plug_1AC_GPP'>
Found custom device replacement for a4:c1:38:b3:53:5a:3f:3e: <class 'custom_ts011f.Plug_1AC_GPP'>
[... cut ...] Address|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)])
```

MattWestb, 2022-04-06:
The device dont have any power measurement, cluster implanted so you is not getting it in the GUI.
The quirk is adding power on restore and switch mode and also light mode for the LED.
Was reading the revues from the device but its 2 or more times translated so not easy knowing the real but its looks some is complaining its not having power measurement implanted.
And Z2M have moving it to tuya plugs without power measuring https://github.com/Koenkk/zigbee-herdsman-converters/commit/767e4149a1ea526270a39d69acea5b9610216603.

Only way knowing 100% is its have its implanted is connecting it to one tuya ZBGW and look if tuya is using with or without power measurement.

closed 2022-10-10 by github-actions[bot]