zigpy/zha-device-handlers #1463: [Device Support Request] Mercator Ikuu 1 button Dimmer (SSWD01)

ToastySefac, 2022-04-07:
**Is your feature request related to a problem? Please describe.**
I have a Mercator Ikuu Light Dimmer (SSWD01) which identifies as a Tuya TS0601 Dimmer (_TZE200_swaamsoy). It interviews ok in ZHA, but once done, HA only gives a on/off option which doesn’t do anything. 

**Describe the solution you'd like**
I'd like to have it identify correctly and have standard options, such an on/off, dimmer brightness up/down, etc

**Device signature - this can be acquired by removing the device from ZHA and pairing it again from the add devices screen. Be sure to add the entire content of the log panel after pairing the device to a code block below this line.**
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0104",
      "in_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0008",
        "0x0300",
        "0xef00"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZE200_swaamsoy",
  "model": "TS0601",
  "class": "zigpy.device.Device"
}

**Additional context**

The model is in the Tuya "ts0601_dimmer.py" quirk file already. I've added the quirk to my custom folder and linked in configuration.yaml, but it doesn't seem to be helping.

Appreciate any help you could give. Thanks

MattWestb, 2022-04-09:
Your device need being added in one quirk for working in ZHA and we have getting its working with most tuya dimmers but its little more work getting all in the right place.
Look in this issue https://github.com/zigpy/zha-device-handlers/issues/1302 for the progress and if you can helping testing the quirk for getting it working OK and being added in zha as soon is possible then its many user like having it so there devices is working in ZHA.

javicalle, 2022-04-11:
I believe that you can simple change the `DEVICE_TYPE` to make it work:
```python

class TuyaSingleSwitchDimmerPlus(TuyaDimmerSwitch):
    """Tuya touch switch device."""

    signature = {
        # "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, 
        # reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, 
        # manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, 
        # descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, 
        # *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
        MODELS_INFO: [
            ("_TZE200_swaamsoy", "TS0601"),
        ],
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.DIMMER_SWITCH,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Identify.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    OnOff.cluster_id,
                    LevelControl.cluster_id,
                    Color.cluster_id,
                    TuyaManufCluster.cluster_id,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            }
        },
    }

    replacement = {
        ENDPOINTS: {
            1: {
                DEVICE_TYPE: zha.DeviceType.ON_OFF_LIGHT,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Identify.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaOnOffMCU,
                    TuyaInWallLevelControl,
                    Color.cluster_id,
                    TuyaLevelControlManufCluster,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            }
        }
    }
```

The device also exposes 'normal' `OnOff` and `LevelControl` clusters. This quirk will ignore the 'normal' clusters and make use of MCU commands.

ToastySefac, 2022-04-11:
Updated and now getting this error:
```
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 335, in async_setup
    result = await component.async_setup_entry(hass, self)
  File "/usr/src/homeassistant/homeassistant/components/zha/__init__.py", line 99, in async_setup_entry
    setup_quirks(config)
  File "/usr/local/lib/python3.9/site-packages/zhaquirks/__init__.py", line 409, in setup
    importer.find_module(modname).load_module(modname)
  File "<frozen importlib._bootstrap_external>", line 529, in _check_name_wrapper
  File "<frozen importlib._bootstrap_external>", line 1029, in load_module
  File "<frozen importlib._bootstrap_external>", line 854, in load_module
  File "<frozen importlib._bootstrap>", line 274, in _load_module_shim
  File "<frozen importlib._bootstrap>", line 711, in _load
  File "<frozen importlib._bootstrap>", line 680, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 850, in exec_module
  File "<frozen importlib._bootstrap>", line 228, in _call_with_frames_removed
  File "/config/custom_zha_quirks/ts0601_dimmer.py", line 38, in <module>
    class TuyaSingleSwitchDimmerPlus(TuyaDimmerSwitch):
  File "/config/custom_zha_quirks/ts0601_dimmer.py", line 56, in TuyaSingleSwitchDimmerPlus
    Identify.cluster_id,
NameError: name 'Identify' is not defined
```

ToastySefac, 2022-04-11:
I've updated the file with your code above and get this error:
```
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 335, in async_setup
    result = await component.async_setup_entry(hass, self)
  File "/usr/src/homeassistant/homeassistant/components/zha/__init__.py", line 99, in async_setup_entry
    setup_quirks(config)
  File "/usr/local/lib/python3.9/site-packages/zhaquirks/__init__.py", line 409, in setup
    importer.find_module(modname).load_module(modname)
  File "<frozen importlib._bootstrap_external>", line 529, in _check_name_wrapper
  File "<frozen importlib._bootstrap_external>", line 1029, in load_module
  File "<frozen importlib._bootstrap_external>", line 854, in load_module
  File "<frozen importlib._bootstrap>", line 274, in _load_module_shim
  File "<frozen importlib._bootstrap>", line 711, in _load
  File "<frozen importlib._bootstrap>", line 680, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 850, in exec_module
  File "<frozen importlib._bootstrap>", line 228, in _call_with_frames_removed
  File "/config/custom_zha_quirks/ts0601_dimmer.py", line 27, in <module>
    class TuyaSingleSwitchDimmerPlus(TuyaDimmerSwitch):
  File "/config/custom_zha_quirks/ts0601_dimmer.py", line 45, in TuyaSingleSwitchDimmerPlus
    Identify.cluster_id,
NameError: name 'Identify' is not defined
```

javicalle, 2022-04-11:
Get all the `import` from the previous version:
```python
from typing import Optional, Union

from zigpy.profiles import zha
from zigpy.quirks import CustomCluster
import zigpy.types as t
from zigpy.zcl import foundation
from zigpy.zcl.clusters.general import Basic, GreenPowerProxy, Groups, Identify, LevelControl, OnOff, Ota, Scenes, Time
from zigpy.zcl.clusters.lighting import Color

from zhaquirks.const import (
    DEVICE_TYPE,
    .../....
```

ToastySefac, 2022-04-11:
Thanks. That's getting closer. I have the light turning on and off, but the dimming functionality is missing. It's also introduced color and temperature controls, which this physical device doesn't control (but that doesn't matter much).
```
2022-04-11 20:20:53 WARNING (MainThread) [zigpy.zcl] Unknown cluster 0xEF00
2022-04-11 20:20:53 WARNING (MainThread) [zigpy.zcl] Unknown cluster 0xEF00
2022-04-11 20:20:55 WARNING (MainThread) [zigpy.zcl] [0x6BCF:1:0xef00] No 'handle_set_data_response' tuya handler found for set_data_response(data=TuyaCommand(status=89, tsn=233, dp=6, data=TuyaData(dp_type=<TuyaDPType.VALUE: 2>, function=0, raw=b'\x00\x00\x00\x00', *payload=0)))
2022-04-11 20:20:59 WARNING (MainThread) [homeassistant.components.google_assistant.trait] Entity light.tze200_swaamsoy_ts0601_e365f4fe_level_light_color_on_off has incorrect color temperature 0
2022-04-11 20:21:01 WARNING (MainThread) [zigpy.zcl] [0x6BCF:1:0xef00] No 'handle_set_data_response' tuya handler found for set_data_response(data=TuyaCommand(status=89, tsn=234, dp=6, data=TuyaData(dp_type=<TuyaDPType.VALUE: 2>, function=0, raw=b'\x00\x00\x00\x00', *payload=0)))
2022-04-11 20:22:01 WARNING (MainThread) [zigpy.zcl] [0x6BCF:1:0xef00] No 'handle_set_data_response' tuya handler found for set_data_response(data=TuyaCommand(status=89, tsn=235, dp=6, data=TuyaData(dp_type=<TuyaDPType.VALUE: 2>, function=0, raw=b'\x00\x00\x00\x00', *payload=0)))
2022-04-11 20:23:01 WARNING (MainThread) [zigpy.zcl] [0x6BCF:1:0xef00] No 'handle_set_data_response' tuya handler found for set_data_response(data=TuyaCommand(status=89, tsn=236, dp=6, data=TuyaData(dp_type=<TuyaDPType.VALUE: 2>, function=0, raw=b'\x00\x00\x00\x00', *payload=0)))
2022-04-11 20:23:42 WARNING (MainThread) [homeassistant.components.google_assistant.trait] Entity light.tze200_swaamsoy_ts0601_e365f4fe_level_light_color_on_off has incorrect color temperature 0
2022-04-11 20:23:47 WARNING (MainThread) [homeassistant.components.google_assistant.trait] Entity light.tze200_swaamsoy_ts0601_e365f4fe_level_light_color_on_off has incorrect color temperature 0
2022-04-11 20:23:47 WARNING (MainThread) [homeassistant.components.google_assistant.trait] Entity light.tze200_swaamsoy_ts0601_e365f4fe_level_light_color_on_off has incorrect color temperature 0
2022-04-11 20:23:48 WARNING (MainThread) [homeassistant.components.google_assistant.trait] Entity light.tze200_swaamsoy_ts0601_e365f4fe_level_light_color_on_off has incorrect color temperature 0
2022-04-11 20:23:48 WARNING (MainThread) [homeassistant.components.google_assistant.trait] Entity light.tze200_swaamsoy_ts0601_e365f4fe_level_light_color_on_off has incorrect color temperature 0
2022-04-11 20:24:01 WARNING (MainThread) [zigpy.zcl] [0x6BCF:1:0xef00] No 'handle_set_data_response' tuya handler found for set_data_response(data=TuyaCommand(status=89, tsn=242, dp=6, data=TuyaData(dp_type=<TuyaDPType.VALUE: 2>, function=0, raw=b'\x00\x00\x00\x00', *payload=0)))
2022-04-11 20:24:24 WARNING (MainThread) [homeassistant.components.google_assistant.trait] Entity light.tze200_swaamsoy_ts0601_e365f4fe_level_light_color_on_off has incorrect color temperature 0
2022-04-11 20:24:28 WARNING (MainThread) [homeassistant.components.google_assistant.trait] Entity light.tze200_swaamsoy_ts0601_e365f4fe_level_light_color_on_off has incorrect color temperature 0
2022-04-11 20:24:31 WARNING (MainThread) [homeassistant.components.google_assistant.trait] Entity light.tze200_swaamsoy_ts0601_e365f4fe_level_light_color_on_off has incorrect color temperature 0
2022-04-11 20:24:33 WARNING (MainThread) [homeassistant.components.google_assistant.trait] Entity light.tze200_swaamsoy_ts0601_e365f4fe_level_light_color_on_off has incorrect color temperature 0
2022-04-11 20:24:33 WARNING (MainThread) [homeassistant.components.google_assistant.trait] Entity light.tze200_swaamsoy_ts0601_e365f4fe_level_light_color_on_off has incorrect color temperature 0
2022-04-11 20:24:38 WARNING (MainThread) [homeassistant.components.google_assistant.trait] Entity light.tze200_swaamsoy_ts0601_e365f4fe_level_light_color_on_off has incorrect color temperature 0
2022-04-11 20:24:40 WARNING (MainThread) [homeassistant.components.google_assistant.trait] Entity light.tze200_swaamsoy_ts0601_e365f4fe_level_light_color_on_off has incorrect color temperature 0
```

javicalle, 2022-04-11:
It seems that the quirk does not fit the device.  Please, without turning HA off, reboot the switch and attach any traces it generates.

javicalle, 2022-04-11:
> It's also introduced color and temperature controls, which this physical device doesn't control (but that doesn't matter much).

If not useful, the `Color` cluster can be removed from the replacement part.

According to https://github.com/Koenkk/zigbee-herdsman-converters/pull/3004 most of the DPs match with our definition, so is not clear why didn't work.

In addition to the traces of the device restart (for example unplugging it and turning it back on), traces of when the physical device is operated would be very useful. Look at the logger config to get all the relevant traces:
* https://www.home-assistant.io/integrations/zha/#debug-logging

ToastySefac, 2022-04-11:
> It seems that the quirk does not fit the device. Please, without turning HA off, reboot the switch and attach any traces it generates.

I restarted the device by cutting the power at the mains power / fuse box. It has now restarted and is working perfectly. I will keep testing it over the next few days, **but thank you for your help @javicalle . I really appreciate it!**

> If not useful, the `Color` cluster can be removed from the replacement part.

This worked as well.

For anyone else with this device, here is the final ts0601_dimmer.py contents:
```
"""Tuya based touch switch."""
from typing import Optional, Union

from zigpy.profiles import zha
from zigpy.quirks import CustomCluster
import zigpy.types as t
from zigpy.zcl import foundation
from zigpy.zcl.clusters.general import Basic, GreenPowerProxy, Groups, Identify, LevelControl, OnOff, Ota, Scenes, Time
from zigpy.zcl.clusters.lighting import Color

from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODELS_INFO,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
)
from zhaquirks.tuya import (
    TuyaDimmerSwitch,
    TuyaLevelControl,
    TuyaManufacturerClusterOnOff,
    TuyaManufacturerLevelControl,
    TuyaManufCluster,
    TuyaOnOff,
)
from zhaquirks.tuya.mcu import (
    TuyaInWallLevelControl,
    TuyaLevelControlManufCluster,
    TuyaOnOff as TuyaOnOffMCU,
)

class TuyaSingleSwitchDimmerPlus(TuyaDimmerSwitch):
    """Tuya touch switch device."""

    signature = {
        # "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, 
        # reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, 
        # manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, 
        # descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, 
        # *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
        MODELS_INFO: [
            ("_TZE200_swaamsoy", "TS0601"),
        ],
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.DIMMER_SWITCH,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Identify.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    OnOff.cluster_id,
                    LevelControl.cluster_id,
                    Color.cluster_id,
                    TuyaManufCluster.cluster_id,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            }
        },
    }

    replacement = {
        ENDPOINTS: {
            1: {
                DEVICE_TYPE: zha.DeviceType.ON_OFF_LIGHT,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Identify.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaOnOffMCU,
                    TuyaInWallLevelControl,
                    TuyaLevelControlManufCluster,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            }
        }
    }
```

ToastySefac, 2022-04-13:
I've tested this for a couple of days. Other than a warning in the logs, it is working perfectly, both through HA and physically. Are the warnings anything to worry about?

```
Logger: zigpy.zcl
Source: /usr/local/lib/python3.9/site-packages/zhaquirks/tuya/__init__.py:1388
First occurred: 4:18:15 PM (251 occurrences)
Last logged: 8:27:50 PM

[0x6BCF:1:0xef00] No 'handle_set_data_response' tuya handler found for set_data_response(data=TuyaCommand(status=9, tsn=54, dp=6, data=TuyaData(dp_type=<TuyaDPType.VALUE: 2>, function=0, raw=b'\x00\x00\x00\x00', *payload=0)))
[0x6BCF:1:0xef00] No 'handle_set_data_response' tuya handler found for set_data_response(data=TuyaCommand(status=9, tsn=55, dp=6, data=TuyaData(dp_type=<TuyaDPType.VALUE: 2>, function=0, raw=b'\x00\x00\x00\x00', *payload=0)))
[0x6BCF:1:0xef00] No 'handle_set_data_response' tuya handler found for set_data_response(data=TuyaCommand(status=9, tsn=56, dp=6, data=TuyaData(dp_type=<TuyaDPType.VALUE: 2>, function=0, raw=b'\x00\x00\x00\x00', *payload=0)))
[0x6BCF:1:0xef00] No 'handle_set_data_response' tuya handler found for set_data_response(data=TuyaCommand(status=9, tsn=57, dp=6, data=TuyaData(dp_type=<TuyaDPType.VALUE: 2>, function=0, raw=b'\x00\x00\x00\x00', *payload=0)))
[0x6BCF:1:0xef00] No 'handle_set_data_response' tuya handler found for set_data_response(data=TuyaCommand(status=9, tsn=80, dp=6, data=TuyaData(dp_type=<TuyaDPType.VALUE: 2>, function=0, raw=b'\x00\x00\x00\x00', *payload=0)))
```

javicalle, 2022-04-13:
The warnings are only informative that there is a config of the device (datapoint 6) that the implementation is not considering. It seems that device reports from datapoints 1, 2, 3 and 6.

DP 1 and 2 are the `on_off` and `current_level` and seems to be working OK.
DP 3 is configured by default to the dimming `minimum_level` value. Not sure if is correct.
DP 6 is not configured (and ZHA complains about it). For the value it has, this could be the `minimum_level` but I have not found conclusive information.

But it is also, according to the vendor device have a 'backlight status' function, other tuya devices have it at datapoint 15 but yours don't report at this DP, so no sure how can be configured (maybe one of the already known DP?).

Regards.

ToastySefac, 2022-06-12:
Any ideas as to what might be happening with this device? When I select a brightness, it jumps around wildly. I can't see any pattern to it.

https://user-images.githubusercontent.com/40307878/173210314-f6552ef4-8121-4ded-b0bb-d9a406e1a41e.mp4

I don't think it's a faulty device. I believe it's something happening within my Home Assitant/ConbeeII/ZHA setup because:
- Using it's brand name hub it works perfectly
- I have 2 identical dimmer devices which have the same symptoms
- I've had it working ok in zigee2mqtt. Have issues with other devices with z2m, so prefer to stay with ZHA
- I've tried with a fresh install of HA on a new virtual machine and had the same issue (to rule out an automation interferring with it)

ToastySefac, 2022-06-12:
Any ideas as to what might be happening with this device? When I select a brightness, it jumps around wildly. I can't see any pattern to it.

https://user-images.githubusercontent.com/40307878/173210357-7f34e80e-6bc9-45a1-8431-e8b7f851f30f.mp4

I don't think it's a faulty device. I believe it's something happening within my Home Assitant/ConbeeII/ZHA setup because:

- Using it's brand name hub it works perfectly
- I have 2 identical dimmer devices which have the same symptoms
- I've had it working ok in zigee2mqtt. Have issues with other devices with z2m, so prefer to stay with ZHA
- I've tried with a fresh install of HA on a new virtual machine and had the same issue (to rule out an automation interferring with it)

javicalle, 2022-06-12:
I can't say that would be the same behaviour, but some devices have a 'transition time' when you change the brightness.
My guess would be that the behaviour is:
* the brightness is changed in HA moving the bar
* when command received, the device reports it's status (not changed yet because of transition time)
* HA updates the status with the reported one (it comes back to initial value)
* if you wait enough, status is updated in HA (or maybe not, it would depends if device reports the intermediate status until the final status)

Debug logs would give more information of what is happening here.
Also, is it just an UI issue or is the device behavior also erratic?

ToastySefac, 2022-06-13:
I think your guess might be right.

In instances where I make a large change (example: from 20% to 80%) and when it reverts back to the original value in the UI, the device's physical brightness does not change. So, it appears the UI updates itself to show the true current state reported by the device.

I can force it to get to the desired brightness if I tap/click the brightness level twice, it jumps a bit, but seems to retain the second one.

The device's behaviour is not erratic, just the UI.

<details><summary>Debug log</summary>

```
2022-06-13 10:19:42 DEBUG (MainThread) [homeassistant.core] Bus:Handling <Event call_service[L]: domain=light, service=turn_on, service_data=entity_id=light.familyroom_dimmer, brightness_pct=55>
2022-06-13 10:19:42 DEBUG (MainThread) [zigpy.zcl] [0x1ABB:1:0x0008] Sending Tuya Cluster Command. Cluster Command is 4, Arguments are (140, 1)
2022-06-13 10:19:42 DEBUG (MainThread) [zigpy.zcl] [0x1ABB:1:0xef00] tuya_mcu_command: cluster_data=TuyaClusterData(endpoint_id=1, cluster_attr='on_off', attr_value=1, expect_reply=True)
2022-06-13 10:19:42 DEBUG (MainThread) [zigpy.zcl] [0x1ABB:1:0xef00] get_dp_mapping --> found DP: 1
2022-06-13 10:19:42 DEBUG (MainThread) [zigpy.zcl] [0x1ABB:1:0xef00] from_cluster_data: 1, DPToAttributeMapping(ep_attribute='on_off', attribute_name='on_off', dp_type=<TuyaDPType.BOOL: 1>, converter=None, dp_converter=None, endpoint_id=None)
2022-06-13 10:19:42 DEBUG (MainThread) [zigpy.zcl] [0x1ABB:1:0xef00] ztype: Bool.true
2022-06-13 10:19:42 DEBUG (MainThread) [zigpy.zcl] [0x1ABB:1:0xef00] from_value: [1, 1]
2022-06-13 10:19:42 DEBUG (MainThread) [zigpy.zcl] [0x1ABB:1:0xef00] raw: b'\x01'
2022-06-13 10:19:42 DEBUG (MainThread) [zigpy.zcl] [0x1ABB:1:0xef00] tuya_command: TuyaCommand(status=0, tsn=166, dp=1, data=TuyaData(dp_type=<TuyaDPType.BOOL: 1>, function=0, raw=b'\x01', *payload=<Bool.true: 1>))
2022-06-13 10:19:42 DEBUG (MainThread) [zigpy.zcl] [0x1ABB:1:0xef00] Sending request header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.CLUSTER_COMMAND: 1>, is_manufacturer_specific=True, is_reply=0, disable_default_response=0, reserved=0, *is_cluster=True, *is_general=False), manufacturer=4098, tsn=167, command_id=0, *is_reply=False)
2022-06-13 10:19:42 DEBUG (MainThread) [zigpy.zcl] [0x1ABB:1:0xef00] Sending request: set_data(data=TuyaCommand(status=0, tsn=166, dp=1, data=TuyaData(dp_type=<TuyaDPType.BOOL: 1>, function=0, raw=b'\x01', *payload=<Bool.true: 1>)))
2022-06-13 10:19:42 DEBUG (MainThread) [zigpy.zcl] [0x1ABB:1:0xef00] tuya_mcu_command: cluster_data=TuyaClusterData(endpoint_id=1, cluster_attr='current_level', attr_value=140, expect_reply=True)
2022-06-13 10:19:42 DEBUG (MainThread) [zigpy.zcl] [0x1ABB:1:0xef00] get_dp_mapping --> found DP: 2
2022-06-13 10:19:42 DEBUG (MainThread) [zigpy.zcl] [0x1ABB:1:0xef00] from_cluster_data: 2, DPToAttributeMapping(ep_attribute='level', attribute_name='current_level', dp_type=<TuyaDPType.VALUE: 2>, converter=<function TuyaLevelControlManufCluster.<lambda> at 0x7fc620f2a9d0>, dp_converter=<function TuyaLevelControlManufCluster.<lambda> at 0x7fc620f2aa60>, endpoint_id=None)
2022-06-13 10:19:42 DEBUG (MainThread) [zigpy.zcl] [0x1ABB:1:0xef00] converted: 549
2022-06-13 10:19:42 DEBUG (MainThread) [zigpy.zcl] [0x1ABB:1:0xef00] ztype: 549
2022-06-13 10:19:42 DEBUG (MainThread) [zigpy.zcl] [0x1ABB:1:0xef00] from_value: [4, 0, 0, 2, 37]
2022-06-13 10:19:42 DEBUG (MainThread) [zigpy.zcl] [0x1ABB:1:0xef00] raw: b'\x00\x00\x02%'
2022-06-13 10:19:42 DEBUG (MainThread) [zigpy.zcl] [0x1ABB:1:0xef00] tuya_command: TuyaCommand(status=0, tsn=168, dp=2, data=TuyaData(dp_type=<TuyaDPType.VALUE: 2>, function=0, raw=b'\x00\x00\x02%', *payload=620888064))
2022-06-13 10:19:42 DEBUG (MainThread) [zigpy.zcl] [0x1ABB:1:0xef00] Sending request header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.CLUSTER_COMMAND: 1>, is_manufacturer_specific=True, is_reply=0, disable_default_response=0, reserved=0, *is_cluster=True, *is_general=False), manufacturer=4098, tsn=169, command_id=0, *is_reply=False)
2022-06-13 10:19:42 DEBUG (MainThread) [zigpy.zcl] [0x1ABB:1:0xef00] Sending request: set_data(data=TuyaCommand(status=0, tsn=168, dp=2, data=TuyaData(dp_type=<TuyaDPType.VALUE: 2>, function=0, raw=b'\x00\x00\x02%', *payload=620888064)))
2022-06-13 10:19:42 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x1ABB:1:0x0008]: executed 'move_to_level_with_on_off' command with args: '(140, 1)' kwargs: '{}' result: Default_Response(command_id=4, status=<Status.SUCCESS: 0>)
2022-06-13 10:19:42 DEBUG (MainThread) [homeassistant.components.zha.entity] light.familyroom_dimmer: turned on: {'move_to_level_with_on_off': Default_Response(command_id=4, status=<Status.SUCCESS: 0>)}
2022-06-13 10:19:42 DEBUG (MainThread) [homeassistant.core] Bus:Handling <Event state_changed[L]: entity_id=light.familyroom_dimmer, old_state=<state light.familyroom_dimmer=on; supported_color_modes=[<ColorMode.BRIGHTNESS: 'brightness'>], color_mode=brightness, brightness=0, off_brightness=None, friendly_name=Family Room Dimmer, supported_features=41 @ 2022-06-13T10:09:08.036149+10:00>, new_state=<state light.familyroom_dimmer=on; supported_color_modes=[<ColorMode.BRIGHTNESS: 'brightness'>], color_mode=brightness, brightness=140, off_brightness=None, friendly_name=Family Room Dimmer, supported_features=41 @ 2022-06-13T10:09:08.036149+10:00>>
[... cut ...]
2022-06-13 10:20:18 DEBUG (MainThread) [zigpy_deconz.api] Device state changed response: [<DeviceState.128|APSDE_DATA_REQUEST_SLOTS_AVAILABLE|APSDE_DATA_CONFIRM|2: 166>, 0]
2022-06-13 10:20:18 DEBUG (MainThread) [zigpy_deconz.api] Command Command.aps_data_confirm (0,)
2022-06-13 10:20:19 DEBUG (MainThread) [zigpy_deconz.api] APS data confirm response for request with id 185: 00
2022-06-13 10:20:19 DEBUG (MainThread) [zigpy_deconz.api] Request id: 0xb9 'aps_data_confirm' for <DeconzAddressEndpoint address_mode=ADDRESS_MODE.NWK address=0x1abb endpoint=1>, status: 0x00
```

intra-aud, 2022-06-13:
Sorry to contaminate the thread with a question to @ToastySefac , you mention that you have the hub that also integrates with this mech. I don't suppose you can check to see what the firmware version is reported as ?

I seem to have an issue with coil whine and i'm wondering if there's a newer firmware that might solve this issue (since i don't have a hub to use with this).

ToastySefac, 2022-06-13:
I bought the Mercator hub just to check if any of the devices I have had any updates, they don't 😁

It says the following for the dimmer:
ZigBee Module: V1.0.0
MCU Module V2.0.0

I've got a whining too, but the switch is far enough away that I don't hear it. Let me know if you need anything else checked with the hub

intra-aud, 2022-06-13:
@ToastySefac  , Thank you for checking!

ToastySefac, 2022-06-13:
> @ToastySefac  , Thank you for checking!
> 
>  

Do you have the same issue as above with your dimmer?

intra-aud, 2022-06-13:
@ToastySefac  , I do not have the same behaviour. It appears mine works as i would expect, however if my dimmer is in the "off" state and i adjust the brightness it does turn on (which is annoying).

closed 2022-12-17 by github-actions[bot]

SamBRZ, 2023-05-12:
Hi Guys , a complete noob here ,so appreciate your help . i got the exact same light switch from Mercator Ikuu , and usinghome assitant OS ,  ZHA with Skyconnect , the switch interview comletes and i get a switch toggle which does nothing , also dont have any dimming options. i read through the whole discussion , but really dont know where to start and how to edit the device .py or even where to find it ! @ToastySefac appreciate your help since it seems like you got this working . thanks mate!

ToastySefac, 2023-05-12:
Here's my memory of what I did...

1. Add this into your configuration.yaml file:
```
zha:
  custom_quirks_path: /config/custom_zha_quirks/
```
2. Create a folder called "custom_zha_quirks" in /config
3. Create a new blank file called "ts0601_dimmer.py" and save it in /config/custom_zha_quirks/
4. Copy and paste the quirk into the new "ts0601_dimmer.py" file.
```
"""Tuya based touch switch."""
from typing import Optional, Union

from zigpy.profiles import zha
from zigpy.quirks import CustomCluster
import zigpy.types as t
from zigpy.zcl import foundation
from zigpy.zcl.clusters.general import Basic, GreenPowerProxy, Groups, Identify, LevelControl, OnOff, Ota, Scenes, Time
from zigpy.zcl.clusters.lighting import Color

from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODELS_INFO,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
)
from zhaquirks.tuya import (
    TuyaDimmerSwitch,
    TuyaLevelControl,
    TuyaManufacturerClusterOnOff,
    TuyaManufacturerLevelControl,
    TuyaManufCluster,
    TuyaOnOff,
)
from zhaquirks.tuya.mcu import (
    TuyaInWallLevelControl,
    TuyaLevelControlManufCluster,
    TuyaOnOff as TuyaOnOffMCU,
)

class TuyaSingleSwitchDimmerPlus(TuyaDimmerSwitch):
    """Tuya touch switch device."""

    signature = {
        # "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, 
        # reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, 
        # manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, 
        # descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, 
        # *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
        MODELS_INFO: [
            ("_TZE200_swaamsoy", "TS0601"),
        ],
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.DIMMER_SWITCH,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Identify.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    OnOff.cluster_id,
                    LevelControl.cluster_id,
                    Color.cluster_id,
                    TuyaManufCluster.cluster_id,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            }
        },
    }

    replacement = {
        ENDPOINTS: {
            1: {
                DEVICE_TYPE: zha.DeviceType.ON_OFF_LIGHT,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Identify.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaOnOffMCU,
                    TuyaInWallLevelControl,
                    TuyaLevelControlManufCluster,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            }
        }
    }
```
5. Delete anything in /config/custom_zha_quirks/__pycache__/
6. Restart Home Assistant

**To see if the quirk is being applied:**

1. Go into Settings>Devices>Devices and open the dimmer device
2. Under Device Info > Zigbee info, underneath _Power Source:Mains_ you should see an entry that says "_Quirk: ts0601_dimmer.TuyaSingleSwitchDimmerPlus_"

SamBRZ, 2023-05-13:
Hey mate , thanks a lot for your reply and step by step instructions.
i have followed all the steps and i am getting below error at step #6 when trying to restart HA.

Logger: homeassistant.components.websocket_api.http.connection
Source: components/hassio/__init__.py:563
Integration: Home Assistant WebSocket API (documentation, issues)
First occurred: 4:56:39 PM (1 occurrences)
Last logged: 4:56:39 PM

[546903718064] The system cannot restart because the configuration is not valid: Invalid config for [zha]: not a directory for dictionary value @ data['zha']['custom_quirks_path']. Got '/config/custom_zha_quirks/'. (See /config/configuration.yaml, line 17).
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/websocket_api/commands.py", line 199, in handle_call_service
    await hass.services.async_call(
  File "/usr/src/homeassistant/homeassistant/core.py", line 1849, in async_call
    task.result()
  File "/usr/src/homeassistant/homeassistant/core.py", line 1889, in _execute_service
    await cast(Callable[[ServiceCall], Awaitable[None]], handler.job.target)(
  File "/usr/src/homeassistant/homeassistant/components/hassio/__init__.py", line 563, in async_handle_core_service
    raise HomeAssistantError(
homeassistant.exceptions.HomeAssistantError: The system cannot restart because the configuration is not valid: Invalid config for [zha]: not a directory for dictionary value @ data['zha']['custom_quirks_path']. Got '/config/custom_zha_quirks/'. (See /config/configuration.yaml, line 17).

ToastySefac, 2023-05-13:
I'm still a novice but I think it's an issue with your custom_zha_quirks folder.
Check that you've created it in the right location. It should be: /config/custom_zha_quirks/

intra-aud, 2023-05-16:
I think toasty is right, it needs to be in the right place.

For my installation, the quirk file lives in
"/usr/share/hassio/homeassistant/custom_zha_quirks"

exceladdict, 2025-08-11:
Feel like this is a little bit of necroposting but wanted to share incase someone else decides to try the [Mercator IKUU SISWD01](https://www.ikuu.com.au/product/inline-switch-with-dimmer/) which shows up as a TS0601 _TZE200_ojzhk75b. As seems to be the case above with the SSWD01, It's identified automatically as a switch but does not function by default. 

Following the steps of @ToastySefac above, including creating a folder for /config/custom_ZHA_quirks (realised that "config" is referring to the root folder where configuration.YAML lives), and then inserting _TZE200_ojzhk75b in place of _TZE200_swaamsoy in the code below did the trick.

I thought I'd give it a shot because on googling the product code, I noticed that over in [mqtt land](https://github.com/Koenkk/zigbee2mqtt/issues/16375) the devices seem to be grouped together. 

If anyone else is trying the inline dimmer, suggest giving this a shot!

```
"""Tuya based touch switch."""
from typing import Optional, Union

from zigpy.profiles import zha
from zigpy.quirks import CustomCluster
import zigpy.types as t
from zigpy.zcl import foundation
from zigpy.zcl.clusters.general import Basic, GreenPowerProxy, Groups, Identify, LevelControl, OnOff, Ota, Scenes, Time
from zigpy.zcl.clusters.lighting import Color

from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODELS_INFO,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
)
from zhaquirks.tuya import (
    TuyaDimmerSwitch,
    TuyaLevelControl,
    TuyaManufacturerClusterOnOff,
    TuyaManufacturerLevelControl,
    TuyaManufCluster,
    TuyaOnOff,
)
from zhaquirks.tuya.mcu import (
    TuyaInWallLevelControl,
    TuyaLevelControlManufCluster,
    TuyaOnOff as TuyaOnOffMCU,
)

class TuyaSingleSwitchDimmerPlus(TuyaDimmerSwitch):
    """Tuya touch switch device."""

    signature = {
        # "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, 
        # reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, 
        # manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, 
        # descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, 
        # *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
        MODELS_INFO: [
            ("_TZE200_ojzhk75b", "TS0601"),
        ],
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.DIMMER_SWITCH,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Identify.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    OnOff.cluster_id,
                    LevelControl.cluster_id,
                    Color.cluster_id,
                    TuyaManufCluster.cluster_id,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            }
        },
    }

    replacement = {
        ENDPOINTS: {
            1: {
                DEVICE_TYPE: zha.DeviceType.ON_OFF_LIGHT,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Identify.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaOnOffMCU,
                    TuyaInWallLevelControl,
                    TuyaLevelControlManufCluster,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            }
        }
    }
```