zigpy/zha-device-handlers #1470: Reduce sensor update interval

JoryHogeveen, 2022-04-08:
Not sure if this is the best location for this question, if not, please just let me know and close this topic!

I'm having an issue with one of my non-battery powered sensors. Since it's a main powered device it registers as a router and updates almost every second or so.
It's about this device: https://github.com/zigpy/zha-device-handlers/issues/1406

I get the feeling it overflows the Zigbee network with messages.
Are there any available settings/clusters to be able to reduce the update interval?
The Tuya documentation isn't really helpful and I couldn't find other resources that answer this question.

Any help is greatly appreciated!

MattWestb, 2022-04-08:
I think Z2M users have complaining that this device is spamming the network.
Then its one tuya MCU device we can setting reporting on the Zigbee module but its not helping then the MCU is living its own life and sending its reports as often is like and we have not finding one command for changing it (if its implanted in the device firmware).

Some user is saying its one "expensive device" then having it connected with tiya cloud then must paying for every transaction the device is doing so i think its coming one update pretty soon if the manufacture is not going bankruptcy from the tuya cloud bills  ;-))

If you or some other user is having the device and one tzuya TVGW its interesting sniffing the its settingup the device to see is tuya is doing some more magic.

Leave the issue open so more user can reporting and perhaps we can finding somthing interesting that can helping or one work around that can reducing the logs in HA.

MattWestb, 2022-04-08:
Not in my head (or more true its in but i cant finding it for the moment then have too much problems with  the latest HA update and  the broken IKEA controllers).

Search Z2M git for your device ID or `Smart Air House keeper` and i think you is finding it.

JoryHogeveen, 2022-04-08:
Ok I will, thank you!

closed 2022-10-12 by github-actions[bot]

foice, 2024-08-06:
The issue is closed, but I do not find any solution. My device is updating every 5 seconds, which is too much. 
Is there a way to tweak the quirk to have to update less frequently?