zigpy/zha-device-handlers #1473: [Device Support Request] Aqara Light Sensor T1 shows up as mains power instead of battery

Ming-A, 2022-04-08:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

I have paired the device about 3 times and have also updated my coordinator (Tubes ZB CC2652P2) to the latest firmware, but the device still shows up as mains power instead of battery (the light sensor is working fine though)

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

Shows battery with the battery level instead of showing as mains power

**Device signature - this can be acquired by removing the device from ZHA and pairing it again from the add devices screen. Be sure to add the entire content of the log panel after pairing the device to a code block below this line.**

`{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|MainsPowered: 132>, manufacturer_code=4718, maximum_buffer_size=127, maximum_incoming_transfer_size=100, server_mask=11264, maximum_outgoing_transfer_size=100, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=True, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0106",
      "in_clusters": [
        "0x0000",
        "0x0001",
        "0x0003",
        "0x0400"
      ],
      "out_clusters": [
        "0x0003"
      ]
    }
  },
  "manufacturer": "LUMI",
  "model": "lumi.sen_ill.agl01",
  "class": "zigpy.device.Device"
}`

**Additional context**
Add any other context or screenshots about the feature request here.

Tropaion, 2022-04-09:
Normally it should work without a quirk since the device uses standard zigbee clusters but try if it works with the quirk I created for it: [custom_quirks.zip](https://github.com/zigpy/zha-device-handlers/files/8457643/custom_quirks.zip)

Ming-A, 2022-04-13:
> @Ming-A Normally it should work without a quirk since the device uses standard zigbee clusters but try if it works with the quirk I created for it: [custom_quirks.zip](https://github.com/zigpy/zha-device-handlers/files/8457643/custom_quirks.zip)

sorry for the late reply, thanks for helping with creating a quirk, currently I'm just looking around on how to add custom quirks and i see that I have to add 
````zha:
  enable_quirks: true
  custom_quirks_path: /config/zha_quirks```
into my config and put the custom quick into the folder path, will give it a try and see how it goes

Tropaion, 2022-04-14:
Maybe this helps: https://github.com/zigpy/zha-device-handlers/discussions/693
After a reboot the quirk should automatically apply.

Ming-A, 2022-04-15:
> Maybe this helps: #693 After a reboot the quirk should automatically apply.

thanks for sharing, i think i was able to pair the quick since it says quick here but the power is still under mains instead of battery
[image]

Tropaion, 2022-04-15:
Ok, thats weird, I'm pretty sure there isn't a problem with the code.
Can you manually read the cluster? If it returns Mains, it's definitely a problem with the device. If you have a Tuya Gateway, then you can look if there is a update for it.
[image]
But maybe @javicalle can tell more about it.

MattWestb, 2022-04-15:
Need patching the node description then the device is saying its main powered.
One example https://github.com/zigpy/zha-device-handlers/pull/989 making it battery powered in ZHA GUI.

Ming-A, 2022-04-15:
> Ok, thats weird, I'm pretty sure there isn't a problem with the code. Can you manually read the cluster? If it returns Mains, it's definitely a problem with the device. If you have a Tuya Gateway, then you can look if there is a update for it. [image] But maybe @javicalle can tell more about it.

[image]
mine is blank for some reason, tried clicking get zigbee attribute and its seems to not do anything (might be lag? not sure)

MattWestb, 2022-04-15:
Yes that is the false node description.
Can you reading the:
```
Clusters
PowerConfiguration1CRCluster (Endpoint id: 1, Id: 0x0001, Type: in)

Cluster Attribute
battery_percentage_remaining (id: 0x0021)
```
I dont knowing if the new Aqara is reporting battery on Zigbee standard cluster or in "Xiaomi" way.

javicalle, 2022-04-15:
HA is not showing battery icon because device declares to be `MainsPowered:
* `mac_capability_flags=<MACCapabilityFlags.AllocateAddress|MainsPowered: 132>`

Try modifyng in your quirk just the node descriptor like:
```python
    replacement = {
        NODE_DESCRIPTOR: zigpy.zdo.types.NodeDescriptor(
            logical_type=2,
            complex_descriptor_available=0,
            user_descriptor_available=0,
            reserved=0,
            aps_flags=0,
            frequency_band=8,
            mac_capability_flags=128,
            manufacturer_code=4718,
            maximum_buffer_size=127,
            maximum_incoming_transfer_size=100,
            server_mask=11264,
            maximum_outgoing_transfer_size=100,
            descriptor_capability_field=0,
        ),
    }
```

Tropaion, 2022-04-15:
@javicalle Didn't know this, learned something new today 😄 
BTW can this implicate the first device ```_TZE200_aycxwiau``` of the quirk?
Is it better in such a case to make a new class?

Tropaion, 2022-04-15:
@javicalle Didn't know that, learned something new today 😄 
Do you mean like this? [agl01.zip](https://github.com/zigpy/zha-device-handlers/files/8498163/agl01.zip)
Is there no need to import some external library?

javicalle, 2022-04-15:
I believe that the `ENDPOINTS` part isn't needed (don't have any difference).

> Is there no need to import some external library?
I don't think, but not tested 😓

Tropaion, 2022-04-15:
@javicalle Ok^^. Didn't know it was allowed to leave ```ENDPOINTS``` empty.
For @Ming-A, you can try this quirk: [agl01.zip](https://github.com/zigpy/zha-device-handlers/files/8498320/agl01.zip)

javicalle, 2022-04-15:
> Didn't know it was allowed to leave `ENDPOINTS` empty.

Yes it's. usually you need to chenge it, but not seem this case. Maybe if some cluster must be 'quirked' we will put in again.

dmulcahey, 2022-04-15:
If you don’t want to change it make sure to omit the key entirely.

javicalle, 2022-04-15:
@Tropaion I dont see the replacement part now...

I dont have looked all the problem/resolution but I means something like:
```python
"""Aquara Tuya Illumination Sensor."""

from zigpy.profiles import zha
from zigpy.quirks import CustomDevice
from zigpy.zcl.clusters.general import (
    Basic,
    Identify,
    PowerConfiguration,
)

from zigpy.zcl.clusters.measurement import IlluminanceMeasurement
from zigpy.zdo.types import NodeDescriptor

from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODELS_INFO,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
)

class IlluminanceSensor(CustomDevice):
    """Aquara Tuya Illumination Sensor."""
    """Devies uses standard clusters, quirk only clears Mains Powered to enable battery percentage"""

    def __init__(self, application, ieee, nwk, replaces):
        super().__init__(application, ieee, nwk, replaces)
        self.fix_mains_flag()

    async def get_node_descriptor(self) -> NodeDescriptor:
        super().get_node_descriptor()
        self.fix_mains_flag()
        return self.node_desc

    def fix_mains_flag(self):
        """Clear Mains Powered bit"""
        self.node_desc.mac_capability_flags &= (
            ~NodeDescriptor.MACCapabilityFlags.MainsPowered
        )

    signature = {
        #  <SimpleDescriptor endpoint=1, profile=260, device_type=262
        #  device_version=1
        #  input_clusters=[0, 1, 3, 1024]
        #  output_clusters=[3]>
        MODELS_INFO: [("LUMI", "lumi.sen_ill.agl01")],
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.LIGHT_SENSOR,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    PowerConfiguration.cluster_id,
                    Identify.cluster_id,
                    IlluminanceMeasurement.cluster_id,
                ],
                OUTPUT_CLUSTERS: [Identify.cluster_id],
            },
        },
    }
    replacement = {
        NODE_DESCRIPTOR: zigpy.zdo.types.NodeDescriptor(
            logical_type=2,
            complex_descriptor_available=0,
            user_descriptor_available=0,
            reserved=0,
            aps_flags=0,
            frequency_band=8,
            mac_capability_flags=128,
            manufacturer_code=4718,
            maximum_buffer_size=127,
            maximum_incoming_transfer_size=100,
            server_mask=11264,
            maximum_outgoing_transfer_size=100,
            descriptor_capability_field=0,
        ),
    }
```

Tropaion, 2022-04-15:
@javicalle I followed this quirk which @MattWestb posted which had the same problem: https://github.com/marciogranzotto/zha-device-handlers/blob/0daea0320571c185e423e203c9da7c73d4cbb90c/zhaquirks/tuya/ts0201_zemismart.py
It uses ```__init__``` to clear the Mains bit, I thought this is a bit better since it only clear one bit from the provided ```NODE_DESCRIPTOR``` and doesn't "replace" it entirely.

Tropaion, 2022-04-15:
Here again, like you said, also added two missing imports:
[agl01.zip](https://github.com/zigpy/zha-device-handlers/files/8498418/agl01.zip)

javicalle, 2022-04-16:
> It uses `__init__` to clear the Mains bit, I thought this is a bit better since it only clears one bit from the provided `NODE_DESCRIPTOR` and doesn't "replace" it entirely.

Ah! Ok now I see what you say.
You're right, it's another way to achieve the same thing.

Tropaion, 2022-04-16:
Which way would you use/recommend?

javicalle, 2022-04-16:
> Which way would you use/recommend?

There is no best or recommended way, it's a matter of programmer preference.
`replacement` is a more 'declarative' way, while the other is more 'programmatic'.
Each has its advantages and disadvantages. The important thing is that it is clear to understand (and to maintain). The way you've proposed is much clearer, but requires setting 'more places' (`__init__` & `get_node_descriptor`).

The more things we learn, the more ability to fix broken things 🤷🏻‍♂️

Ming-A, 2022-04-19:
> Here again, like you said, also added two missing imports: [agl01.zip](https://github.com/zigpy/zha-device-handlers/files/8498418/agl01.zip)

[image]

I have tried it again and this time Illuminance shows up as unavailable without any battery percentage as well

Tropaion, 2022-04-19:
Did you repair?

Ming-A, 2022-04-20:
> Did you repair?

[image]
so far i have only tried to reconfigure it but ill try to re-pair it again and see if it works :)

javicalle, 2022-04-29:
Any error logs in your traces?

Ming-A, 2022-04-29:
> error logs in your traces

how do you check for the traces error logs? i checked the normal HA logs and so far i don't see any that's related to ZHA

javicalle, 2022-04-29:
Usually there are in HA logs.
You can try to enable the debug level and repair again:
* https://www.home-assistant.io/integrations/zha/#debug-logging

Ming-A, 2022-04-30:
> Usually there are in HA logs. You can try to enable the debug level and repair again:
> 
> * https://www.home-assistant.io/integrations/zha/#debug-logging

I have enabled debug and it doesn't seem to show anything related to zha, here's the logs tho i got while repairing it (sorry for the long screenshot, i paired it using my phone HA app)

[image]
)

anderl80, 2022-05-03:
Are you using the Sonoff stick, too? For me connecting to Xiaomi Illumination Sensor (same as yours? https://www.amazon.de/gp/product/B08NZBJLP1/ref=ppx_yo_dt_b_asin_title_o03_s00?ie=UTF8&psc=1) doesn't even show LQI.

https://community.home-assistant.io/t/cannot-connect-mi-light-detection-sensor-to-zha/413702

Ming-A, 2022-05-05:
> Are you using the Sonoff stick, too? For me connecting to Xiaomi Illumination Sensor (same as yours? https://www.amazon.de/gp/product/B08NZBJLP1/ref=ppx_yo_dt_b_asin_title_o03_s00?ie=UTF8&psc=1) [...]

I'm using the Tubes ZB CC2652P2 cordinator.
for me it works fine on default, the only problem is it shows up as mains instead of battery power without any battery % indication.

javicalle, 2022-05-05:
Your traces seems normal to me and there is no device info.

Can you remove the battery (if possible) and attach the logs when battery is replaced?

According to Z2M, device reports `voltage` and `battery`:
* https://www.zigbee2mqtt.io/devices/GZCGQ11LM.html

That is the same device as yours?

Ming-A, 2022-05-08:
> 

yes, it should be this one, using ZHA instead of Z2M though. I have removed and placed the battery back in again, along with removing it and repairing it didn't work either.

the logs:
`[0x0000:zdo] ZDO request ZDOCmd.Mgmt_Permit_Joining_rsp: [<Status.SUCCESS: 0>]
[0x0000:zdo] No handler for ZDO request:ZDOCmd.Mgmt_Permit_Joining_rsp([<Status.SUCCESS: 0>])
[0x44FF:1:0x0006] Received ZCL frame: b'\x18\x30\x0A\x00\x00\x10\x01'
[0x44FF:1:0x0006] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=0, is_reply=1, disable_default_response=1, reserved=0, *is_cluster=False, *is_general=True), tsn=48, command_id=10, *is_reply=True)
[0x44FF:1:0x0006] Decoded ZCL frame: OnOff:Report_Attributes(attribute_reports=[Attribute(attrid=0x0000, value=TypeValue(type=Bool, value=<Bool.true: 1>))])
[0x44FF:1:0x0006] Received command 0x0A (TSN 48): Report_Attributes(attribute_reports=[Attribute(attrid=0x0000, value=TypeValue(type=Bool, value=<Bool.true: 1>))])
[0x44FF:1:0x0006] Attribute report received: on_off=<Bool.true: 1>
New device 0x236e (04:cf:8c:df:3c:8d:d0:f1) joined the network
[0x236e] Scheduling initialization
Received frame on uninitialized device <Device model=None manuf=None nwk=0x236E ieee=04:cf:8c:df:3c:8d:d0:f1 is_initialized=False> from ep 0 to ep 0, cluster 19: b'\xd3n#\xf1\xd0\x8d<\xdf\x8c\xcf\x04\x84'
[0x236e:zdo] ZDO request ZDOCmd.Device_annce: [0x236E, 04:cf:8c:df:3c:8d:d0:f1, 132]
Tries remaining: 3
[0x236e] Requesting 'Node Descriptor'
Tries remaining: 2
[0x236e] Extending timeout for 0x54 request
Received frame on uninitialized device <Device model=None manuf=None nwk=0x236E ieee=04:cf:8c:df:3c:8d:d0:f1 is_initialized=False> from ep 0 to ep 0, cluster 32770: b'T\x00n#\x02@\x84n\x12\x7fd\x00\x00,d\x00\x00'
[0x236e] Got Node Descriptor: NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|MainsPowered: 132>, manufacturer_code=4718, maximum_buffer_size=127, maximum_incoming_transfer_size=100, server_mask=11264, maximum_outgoing_transfer_size=100, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=True, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)
[0x236e] Discovering endpoints
Tries remaining: 3
[0x236e] Extending timeout for 0x55 request
Received frame on uninitialized device <Device model=None manuf=None nwk=0x236E ieee=04:cf:8c:df:3c:8d:d0:f1 is_initialized=False> from ep 1 to ep 1, cluster 0: b'\x18\x00\x0A\x05\x00\x42\x12\x6C\x75\x6D\x69\x2E\x73\x65\x6E\x5F\x69\x6C\x6C\x2E\x61\x67\x6C\x30\x31\x01\x00\x20\x1B'
[0x236e] Received ZCL while uninitialized on endpoint id 1, cluster 0x0000 id, hdr: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=0, is_reply=1, disable_default_response=1, reserved=0, *is_cluster=False, *is_general=True), tsn=0, command_id=10, *is_reply=True), payload: b'\x05\x00B\x12lumi.sen_ill.agl01\x01\x00 \x1b'
[0x236e] Uninitialized device command '10' params: Report_Attributes(attribute_reports=[Attribute(attrid=0x0005, value=TypeValue(type=CharacterString, value='lumi.sen_ill.agl01')), Attribute(attrid=0x0001, value=TypeValue(type=uint8_t, value=27))])
Received frame on uninitialized device <Device model=None manuf=None nwk=0x236E ieee=04:cf:8c:df:3c:8d:d0:f1 is_initialized=False> from ep 1 to ep 1, cluster 1024: b'\x18\x01\x0A\x00\x00\x21\x8D\x4A'
[0x236e] Received ZCL while uninitialized on endpoint id 1, cluster 0x0400 id, hdr: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=0, is_reply=1, disable_default_response=1, reserved=0, *is_cluster=False, *is_general=True), tsn=1, command_id=10, *is_reply=True), payload: b'\x00\x00!\x8dJ'
[0x236e] Uninitialized device command '10' params: Report_Attributes(attribute_reports=[Attribute(attrid=0x0000, value=TypeValue(type=uint16_t, value=19085))])
Received frame on uninitialized device <Device model=None manuf=None nwk=0x236E ieee=04:cf:8c:df:3c:8d:d0:f1 is_initialized=False> from ep 1 to ep 1, cluster 64704: b'\x1C\x5F\x11\x02\x0A\xF7\x00\x41\x27\x01\x21\x2D\x0C\x03\x28\x19\x04\x21\xA8\x01\x05\x21\x1E\x00\x06\x24\x03\x00\x00\x00\x00\x08\x21\x1B\x01\x0A\x21\x00\x00\x0C\x20\x01\x64\x23\x51\x00\x00\x00'
[0x236e] Received ZCL while uninitialized on endpoint id 1, cluster 0xfcc0 id, hdr: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=True, is_reply=1, disable_default_response=1, reserved=0, *is_cluster=False, *is_general=True), manufacturer=4447, tsn=2, command_id=10, *is_reply=True), payload: b"\xf7\x00A'\x01!-\x0c\x03(\x19\x04!\xa8\x01\x05!\x1e\x00\x06$\x03\x00\x00\x00\x00\x08!\x1b\x01\n!\x00\x00\x0c \x01d#Q\x00\x00\x00"
[0x236e] Uninitialized device command '10' params: Report_Attributes(attribute_reports=[Attribute(attrid=0x00F7, value=TypeValue(type=LVBytes, value=b'\x01!-\x0c\x03(\x19\x04!\xa8\x01\x05!\x1e\x00\x06$\x03\x00\x00\x00\x00\x08!\x1b\x01\n!\x00\x00\x0c \x01d#Q\x00\x00\x00'))])
[0xF8FE:1:0x000a] Received ZCL frame: b'\x00\x57\x00\x07\x00'
[... cut ...]
[0x236E](lumi.sen_ill.agl01): started initialization
[0x236E:ZDO](lumi.sen_ill.agl01): 'async_initialize' stage succeeded
[0x236E](lumi.sen_ill.agl01): power source: Battery or Unknown
[0x236E](lumi.sen_ill.agl01): completed initialization
Error handling '_save_attribute' event with (04:cf:8c:df:3c:8d:d0:f1, 1, 0, 4, 'LUMI') params: FOREIGN KEY constraint failed
Error handling '_save_attribute' event with (04:cf:8c:df:3c:8d:d0:f1, 1, 0, 5, 'lumi.sen_ill.agl01') params: FOREIGN KEY constraint failed`

Ming-A, 2022-05-08:
whats weird now is its not even showing the entity of the device anymore 
[image]

closed 2022-11-11 by github-actions[bot]