zigpy/zha-device-handlers #1476: Support for dimmer switch TS0601

yftj, 2022-04-09:
Hi hope you are well, I'd like to request support for [these dimmer switches](https://www.ozsmartthings.com.au/products/zigbee-dimmer-switch) (TS0601). They have a middle switch and a brightness increase on-top and decrease in the bottom. Please let me know if there's anything else you need. I'm new to GitHub so forgive my formatting. The zigbee signature is:

'''
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0051",
      "in_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0xef00"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZE200_1agwnems",
  "model": "TS0601",
  "class": "zigpy.device.Device"
}

'''

MattWestb, 2022-04-09:
Your device need being added in one quirk for working in ZHA and we have getting its working with most tuya dimmers but its little more work getting all in the right place.
Look in this issue https://github.com/zigpy/zha-device-handlers/issues/1302 for the progress and if you can helping testing the quirk for getting it working OK and being added in zha as soon is possible then its many user like having it so there devices is working in ZHA.

yftj, 2022-04-09:
> Your device need being added in one quirk for working in ZHA and we have getting its working with most tuya dimmers but its little more work getting all in the right place. Look in this issue #1302 [...]

yftj, 2022-04-09:
Sorry, How do I go about testing the quirk? is there a guide I can follow somewhere? Thank you

Tropaion, 2022-04-10:
@yftj There isn't, just read the latest commends here: https://github.com/zigpy/zha-device-handlers/issues/1302
This could help a bit: https://github.com/zigpy/zha-device-handlers/issues/1302#issuecomment-1094241762
Please close issue an continue in the mentioned issue.

javicalle, 2022-04-13:
Not sure if your device has been validate to work with the quirk. Have you tested it?

PR #1489 (2022-04-19): Implementations for `NO_MANUFACTURER_ID` dimmers
This PR adds device definitions for dimmers that require send commands with the `NO_MANUFACTURER_ID` (or what is the same `manufacturer_specific=False`).

A few comments:
* `tuya/mcu/TuyaLevelControl` now handles `move_to_level_with_on_off` command, sending an `on_off` command before the `current_level` command.
* the `TuyaLevelControlManufCluster` handles DPs up to 3 gangs.
* the new `NoManufacturerCluster` will force the `NO_MANUFACTURER_ID` in commands. This is needed for recent dimmers
* at some point I wanted to migrate the remaining classe [... cut]
closed 2022-04-19 by dmulcahey