zigpy/zha-device-handlers #1478: [BUG?] Tuya devices with cluster E000, E001, EF00 throwing warnung

Tropaion, 2022-04-09:
All Tuya devices which use these clusters: E000, E001, EF00 are throwing this warnung even though they work:
```
Logger: zigpy.zcl
Source: /usr/local/lib/python3.9/site-packages/zigpy/zcl/__init__.py:210
First occurred: 14:50:58 (18 occurrences)
Last logged: 14:50:58

Unknown cluster 0xE000
Unknown cluster 0xE001
Unknown cluster 0xEF00
```

MattWestb, 2022-04-09:
Some devices was starting doing it on the 0xEF00 and 0xEF01 before last update then i was adding TS0004 https://github.com/zigpy/zha-device-handlers/pull/1436.
Dont looks nice but looks working OK

dmulcahey, 2022-04-09:
https://github.com/zigpy/zigpy/issues/955

needs to be addressed in Zigpy

closed 2022-04-09 by dmulcahey