zigpy/zha-device-handlers #1481: [Device Support Request] TS0601_dimmer ZigBee groups support

Tropaion, 2022-04-09:
New issue related to: https://github.com/zigpy/zha-device-handlers/issues/1302#issuecomment-1094118490
Now that we have a working quirk, it would be nice if we could support ZigBee groups (@javicalle).

OlafZwe, 2022-04-10:
Is Zigbee groups support also required for scenes functionality in HA?

My TS0601 TuyaDoubleSwitchDimmer (_TZE200_e3oitdyu, model ms-105bz) has a bug (in HA 2022.4.1) when switching scenes specifically, where if the connected light(s) are off before the scene is activated and the scene is configured to turn the light(s) on, they do not turn on when the scene is activated.
If the light(s) are on before the scene and the scene changes their brightness or turns them off then the light(s) do correctly respond to the scene activating. Controlling the light(s) in HA without using scenes does work as expected, although with significant delay.

If this issue with scenes and TS0601_dimmer devices is not related to Zigbee groups support, please let me know and I will move this into a separate issue.

MattWestb, 2022-04-10:
HA have 2 types of groups and scenes "HA Groups and scenes" and Zigbee Groups and Zigbee scenes (the last is not implanted in ZHA) and its working in different ways.
HA is grouping lights and other device in one group and can also doing the same with scenes then the device have different setting in different  scenes.
Then sending commands to HA groups or changing scenes HA is sending one command for changing its state to every device as it was one device = many command is being presessed.

Zigbee groups is one or more braodcast address in the mesh network that is being used for sending and revising commands so adding one Zigbee light in one Zigbee group is the light listening to the braodcast  address and one controller (remote) is being bound to the group / broadcst address and is sending commands to that.
That is making the function is working then the host system is not online and also if the coordinator is not working.
Zigbee scenes is tagged with Zigbee groups and one light can being member of more groups and also having scenes from more groups (but the device must being member of all groups for the scenes added in it).
ZHA have not implanted all commands for setting up Zigbee scenes so its not working if not defining them with one external utility.

Normal Zigbee groups and scenes is working with normal Zigbeee light devices but it must have support for both for working OK. HA switches is not sending light commands and cant being bounded so they is not working with real Zigbee groups and must being "routes" theu the hist system (HA automatons) and is not working if host system is down.

Now tuya devices:
Most tuya Zigbee devices is working OK with Zigbee groups and scenes with some bugs / futures like the TS004F Rotating Kno in command mode can controlling groups light but not bonded to one Zugbee group but added as one light and its working OK.

Other tuya devices that  is not real light devices is not working with Zigbee groups and scenes like all TS0601  / MCU / DP device. The Zigbee module is having group and scene cluster but its looks like its not  working at  all and all MCU / DP commands is not being sent with normal Zigbee commands its being tunneled on the tuya cluster in unicat to the coordinator and not on with ZVL commands to Zigbee groups.

So its not possible binding one MCU device with one Zigbee group and sending command to it or using Zigbee scenes directly.

Possible is routeing the commands for Zigbee groups and scenes thru the host system (HA automatons) but then its not working the the host  system is down.

tuya is using direct liking (Zigbee groups) for Zigbee devices and for MCU / DP devices they is using the cloud  for routing commands and i think some they can using local in the ZBGW for routing  to devices is have connected but i have not seen that  in real.

So in the end all TS0601 / MCU device cant using Zgbee groups and scenes native (direct in the network layer)  and must being routed thru the host system > not autonomous function like rel Zigbee device can can doing (if configurated OK).

Also tuya definition of  groups and scenes is not the same as Zigbee is using like one tuya Zigbee scene switch is not sending Zigbee scene commands to one bounded Zigbee group, its sending no standard commands on On/Off cluster with unicaast to the coordinator that cant being used as real scene commands in the Zigbee network layer like real Zigbee light scene switches is doing.

If like using Zigbee light autonomous they must supporting Zigbee groups and scenes or its always must using the host system for doing the work that  shall being made in the network layer of the  Zigbee network.

If finding one way getting tuya MCU device sending ZCL commands on On/Off, Level; Color, Group and Scene cluster it word being great then tuya have making very nice MCU (Zigbee) devices but i think its not possible with our knowledge (or more true its not possible at all then its not implanted in the Zigbee module firmware).

MattWestb, 2022-04-10:
Implanting some kind of binding (not real Zigbee ones) that is routing the light commands from the device in ZHA to one Zigbee group shall being possible but i dont knowing how to s´do that in the code.

Adding one attribute in the quirk that is being used for getting the saved group and routing commands that  is coming in from the device to the group.
The attribute can being read and stetted with normal cluster commands and is being saved in the Zigbee.DB as normal attributes.

More out of the box thinking ???

Tropaion, 2022-04-10:
@MattWestb So what I understand, the uncommon tuya devices don't work with groups.
But the bseed zigbee light switches work pretty nice with groups, so I think it would make sense that the dimmer, since it is also a light device, should also work with groups.

MattWestb, 2022-04-10:
@Tropaion Zigbee dimmers shall working OK if the firmware is made OK.
tuya MCU devices is not using Zigbee group commands and if like sending group commands it must being thru the host system.

@javicalle If the device is having On/Off, Light level and group in cluster ZHA is displaying it in the Zigbee group panel and can being added (the TS004F dimmer switch is doing like that) and can being added in the group/s. But our problem if we is doing that is we must getting the group storing the group and using it for sending commands to it and receiving command from it.

ZHA is / can sending and receiving group commands if making groups they is have the same functions like one device  On/Off, light level and color.

The Principe is light is listening to one or more broadcast addresses and also filtering the groups to its table with groups is member of and if its one of its groups its executing the command received.

Light controller is sending commands to bonded groups like was one unicast but  to the boradcast address (normally its sleepers so its sending it to its parent router that is doing the broadcast).

Coordinators and controller bridges (Philips HUE and IKEA is Zigbee controller bridges) can doing bot sending and receiving commands to groups.

Then ZHA can sending commands to groups we only need finding out how to doing it from the quirk and also configuring the group sending it to for the device.
I think one attribute on tuya cluster for setting the group and saving it in zigbee.db and then the quirk can using it for sending commands to the group like it was being bond to the group but made in the host system.

I think @puddly can having some more good thinking around it and also @dmulcahey for getting it working it ZHA GUI.

In ZHA the status of one group is made from the status reported from the light in the group and then all light if off the group is off ala Zigbee standard. The problem is that one light can being in more groups and can making it little tricky. Controllers is not having any status / state they is only sending commands and the status / state is being reported from the light.

Its not one great way getting it working but i think its better then doing automatons in HA for every command sent in HA and its one 100% system online function so cant comparing with Zignee group binding that is made in the network layer without host system.

MattWestb, 2022-04-11:
If making one light device in ZHA with OnOff, Light level, color scene and group cluster that can being used in the ZHA group management as one normal light and its saving the group(s) in the zigbee.db then the quirk can using it for sending commands to the group thru the ZHA device.

Its 150% no Zigbee standard but calling it one tuya MCU light that tuya MCU quirks can using for sending commands to grouped device.

I think its much better then making HA automation for all functions on one device and more times then having more device.

In the end we is coming back to the point making one tuya MCU part in HA for Zigbee and other MCU devices (WiFI, BT and thred / Matter) and only tunneling MCU commands to it for going around the limits we have with advanced tuya MCU device like TRVs.

closed 2022-10-15 by github-actions[bot]