zigpy/zha-device-handlers #1482: [Device Support Request]  TS130F _TZ3000_wptayaqr BSEED Shutter Switch works but has additional switch i cannot explain to myself

itsolon, 2022-04-09:
**Is your feature request related to a problem? Please describe.**
Basically this device works out of the box, it seems no quirk is being loaded.
but it has some to me inexplainable switch - perhaps an issue but at least weird to me. Explanation or help appreciated.
a link to that device for better understanding which device i mean
https://www.bseed.com/collections/zigbee-series/products/bseed-zigbee-shutter-switch-smart-wall-touch-switch
TS130F _TZ3000_wptayaqr

**Describe the solution you'd like**
I dont know if possible but it would be nice to have a slider where the position of the shutter can be set. just like dimmer from shut to open.. 

**Device signature - this can be acquired by removing the device from ZHA and pairing it again from the add devices screen. Be sure to add the entire content of the log panel after pairing the device to a code block below this line.**

{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0202",
      "in_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0102",
        "0xe001"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "242": {
      "profile_id": 41440,
      "device_type": "0x0061",
      "in_clusters": [],
      "out_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "_TZ3000_wptayaqr",
  "model": "TS130F",
  "class": "zigpy.device.Device"
}]

**Additional context**
[image]

Tropaion, 2022-04-09:
@itsolon Try this quirk, should work if I didn't make a mistake: [ts130f.zip](https://github.com/zigpy/zha-device-handlers/files/8457760/ts130f.zip)

itsolon, 2022-04-09:
no obviously it will not be loaded
    }
  },
  "manufacturer": "_TZ3000_wptayaqr",
  "model": "TS130F",
  "class": "zigpy.device.Device"
}

Tropaion, 2022-04-09:
@itsolon Do you get some error?

Tropaion, 2022-04-09:
@itsolon Try again, not it should work: [ts130f.zip](https://github.com/zigpy/zha-device-handlers/files/8457802/ts130f.zip)

itsolon, 2022-04-09:
no quirk is not being loaded 
1
2
3
4
5
6
7
8
9
10
11
12
13
14
15
16
17
18
19
20
21
22
23
24
25
26
27
28
29
30
31
32
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0202",
      "in_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0102",
        "0xe001"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "242": {
      "profile_id": 41440,
      "device_type": "0x0061",
      "in_clusters": [],
      "out_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "_TZ3000_wptayaqr",
  "model": "TS130F",
  "class": "zigpy.device.Device"
}

itsolon, 2022-04-09:
sorry was sleeping

itsolon, 2022-04-09:
"device_type": "0x0061",
      "in_clusters": [],
      "out_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "_TZ3000_wptayaqr",
  "model": "TS130F",
  "class": "zigpy.device.Device"
}
#

itsolon, 2022-04-09:
please explain to me how to delete the pycache

Tropaion, 2022-04-09:
```__pychache__``` is a hidden folder so you can only delete it via terminal, I use the terminal Add-on for it. Go to the folder where the quirk is you changed and delete folder with ```rm -r __pychache__```

itsolon, 2022-04-09:
nothing is stupid... i am glad that you help.. 
i did delete the folder via terminal, reboot always neccessary? i reboot 
      "out_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "_TZ3000_wptayaqr",
  "model": "TS130F",
  "class": "zigpy.device.Device"
}

it does not load

took the latest ts130f from you

Tropaion, 2022-04-09:
Yes, to test custom work always in this order:

- Add/Change Quirk
- Delete ```__pychache__```
- Reboot
After that, the quirk should apply.

Did you enable custom quirks?

itsolon, 2022-04-09:
i already have one active from my 
 and.. which logs can i show you .. appreciate explanation

[image]

[image]

Tropaion, 2022-04-09:
@itsolon I don't need logs, found another mistake in the quirk: [ts130f.zip](https://github.com/zigpy/zha-device-handlers/files/8457827/ts130f.zip)
Happens when I'm tired 😄

itsolon, 2022-04-09:
}
  },
  "manufacturer": "_TZ3000_wptayaqr",
  "model": "TS130F",
  "class": "ts130f.TuyaTS130ESTC"
}

now it loads but,,

itsolon, 2022-04-09:
[image]

this is greyed out now.. which is good .. or do i have to repair now?

second good news is 

[image]

i dont really know if i had it before

itsolon, 2022-04-09:
UPDATE it was an orphaned entity , i only deletet and now its perfect
thank you very much

Tropaion, 2022-04-09:
@itsolon Cool, so did you confirm that everything works?
If yes, you can close the issue and I will make an PR tomorrow, the quirks should the be in the next release.

itsolon, 2022-04-09:
give me 5 minutes i will repair see if it works out of the box

itsolon, 2022-04-09:
i successfully repaired it fresh, and everything works
i can up down stop
and additionally i can use the slider for position. 
perfect work.!!!!

closed 2022-04-09 by itsolon