zigpy/zha-device-handlers #1491: [Device Support Request] TZE200_ntcy3xu1 Smoke Detector (Tuya)

mwitlox, 2022-04-13:
Unable to get this device add to ZHA or Z2MQTT (TZE200_ntcy3xu1 TS0601)

- No entities available
- At least Smoke Alarm and Battery support would be great
- Got 3 maybe possible to make a group to if 1 goes on all others also go on :)

Link to the product: https://nl.aliexpress.com/item/1005004028271868.html?spm=a2g0o.detail.1000014.6.60ea58e9JKBVqU&gps-id=pcDetailBottomMoreOtherSeller&scm=1007.40050.274735.0&scm_id=1007.40050.274735.0&scm-url=1007.40050.274735.0&pvid=88b4e6bd-d89d-40be-8c8c-daf880cd73b7&_t=gps-id:pcDetailBottomMoreOtherSeller,scm-url:1007.40050.274735.0,pvid:88b4e6bd-d89d-40be-8c8c-daf880cd73b7,tpp_buckets:668%232846%238114%231999&pdp_ext_f=%257B%2522sku_id%2522%253A%252212000027793000075%2522%252C%2522sceneId%2522%253A%252230050%2522%257D&pdp_pi=-1%253B5.55%253B-1%253B-1%2540salePrice%253BEUR%253Brecommend-recommend

[0x0000:zdo] ZDO request ZDOCmd.Mgmt_Permit_Joining_rsp: [<Status.SUCCESS: 0>]
[0x0000:zdo] No handler for ZDO request:ZDOCmd.Mgmt_Permit_Joining_rsp([<Status.SUCCESS: 0>])
New device 0xaaf3 (a4:c1:38:0e:9d:6f:69:67) joined the network
[0xaaf3] Scheduling initialization
Received frame on uninitialized device <Device model=None manuf=None nwk=0xAAF3 ieee=a4:c1:38:0e:9d:6f:69:67 is_initialized=False> from ep 0 to ep 0, cluster 19: b'\x07\xf3\xaagio\x9d\x0e8\xc1\xa4\x80'
[0xaaf3:zdo] ZDO request ZDOCmd.Device_annce: [0xAAF3, a4:c1:38:0e:9d:6f:69:67, 128]
Tries remaining: 3
[0xaaf3] Requesting 'Node Descriptor'
Tries remaining: 2
[0xaaf3] Extending timeout for 0xd2 request
Received frame on uninitialized device <Device model=None manuf=None nwk=0xAAF3 ieee=a4:c1:38:0e:9d:6f:69:67 is_initialized=False> from ep 0 to ep 0, cluster 32770: b'\xd2\x00\xf3\xaa\x02@\x80A\x11BB\x00\x00*B\x00\x00'
[0xaaf3] Got Node Descriptor: NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)
[0xaaf3] Discovering endpoints
Tries remaining: 3
[0xaaf3] Extending timeout for 0xd3 request
Received frame on uninitialized device <Device model=None manuf=None nwk=0xAAF3 ieee=a4:c1:38:0e:9d:6f:69:67 is_initialized=False> from ep 0 to ep 0, cluster 32773: b'\xd3\x00\xf3\xaa\x01\x01'
[0xaaf3] Discovered endpoints: [1]
[0xaaf3] Initializing endpoints [<Endpoint id=1 in=[] out=[] status=<Status.NEW: 0>>]
[0xaaf3:1] Discovering endpoint information
Tries remaining: 3
[0xaaf3] Extending timeout for 0xd4 request
Received frame on uninitialized device <Device model=None manuf=None nwk=0xAAF3 ieee=a4:c1:38:0e:9d:6f:69:67 is_initialized=False> from ep 0 to ep 0, cluster 32772: b'\xd4\x00\xf3\xaa\x14\x01\x04\x01Q\x00\x01\x04\x04\x00\x05\x00\x00\xef\x00\x00\x02\x19\x00\n\x00'
[0xaaf3:1] Discovered endpoint information: SizePrefixedSimpleDescriptor(endpoint=1, profile=260, device_type=81, device_version=1, input_clusters=[4, 5, 61184, 0], output_clusters=[25, 10])
Unknown cluster 0xEF00
[0xAAF3:1:0x0000] Sending request header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=False, is_reply=0, disable_default_response=0, reserved=0, *is_cluster=False, *is_general=True), tsn=213, command_id=<GeneralCommand.Read_Attributes: 0>, *is_reply=False)
[0xAAF3:1:0x0000] Sending request: Read_Attributes(attribute_ids=[4, 5])
[0xaaf3] Extending timeout for 0xd5 request
[0xAAF3:1:0x0000] Received ZCL frame: b'\x18\xD5\x01\x04\x00\x00\x42\x10\x5F\x54\x5A\x45\x32\x30\x30\x5F\x6E\x74\x63\x79\x33\x78\x75\x31\x05\x00\x00\x42\x06\x54\x53\x30\x36\x30\x31'
[0xAAF3:1:0x0000] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=0, is_reply=1, disable_default_response=1, reserved=0, *is_cluster=False, *is_general=True), tsn=213, command_id=1, *is_reply=True)
[0xAAF3:1:0x0000] Decoded ZCL frame: Basic:Read_Attributes_rsp(status_records=[ReadAttributeRecord(attrid=0x0004, status=<Status.SUCCESS: 0>, value=TypeValue(type=CharacterString, value='_TZE200_ntcy3xu1')), ReadAttributeRecord(attrid=0x0005, status=<Status.SUCCESS: 0>, value=TypeValue(type=CharacterString, value='TS0601'))])
[0xaaf3] Read model 'TS0601' and manufacturer '_TZE200_ntcy3xu1' from <Endpoint id=1 in=[groups:0x0004, scenes:0x0005, None:0xEF00, basic:0x0000] out=[ota:0x0019, time:0x000A] status=<Status.ZDO_INIT: 1>>
[0xaaf3] Discovered basic device information for <Device model='TS0601' manuf='_TZE200_ntcy3xu1' nwk=0xAAF3 ieee=a4:c1:38:0e:9d:6f:69:67 is_initialized=True>
Device is initialized <Device model='TS0601' manuf='_TZE200_ntcy3xu1' nwk=0xAAF3 ieee=a4:c1:38:0e:9d:6f:69:67 is_initialized=True>
Checking quirks for _TZE200_ntcy3xu1 TS0601 (a4:c1:38:0e:9d:6f:69:67)
Considering <class 'zhaquirks.xbee.xbee_io.XBeeSensor'>
Fail because endpoint list mismatch: {232, 230} {1}
Considering <class 'zhaquirks.xbee.xbee3_io.XBee3Sensor'>
Fail because endpoint list mismatch: {232, 230} {1}
Considering <class 'zhaquirks.smartthings.tag_v4.SmartThingsTagV4'>
Fail because device_type mismatch on at least one endpoint
Considering <class 'zhaquirks.smartthings.multi.SmartthingsMultiPurposeSensor'>
Fail because device_type mismatch on at least one endpoint
Considering <class 'zhaquirks.netvox.z308e3ed.Z308E3ED'>
Fail because device_type mismatch on at least one endpoint
Considering <class 'zhaquirks.gledopto.soposhgu10.SoposhGU10'>
Fail because endpoint list mismatch: {11, 13} {1}
Considering <class 'bellows.zigbee.application.EZSPCoordinator'>
Fail because device_type mismatch on at least one endpoint
'sensor' component -> 'RSSISensor' using ['basic']
'sensor' component -> 'LQISensor' using ['basic']
device - 0xAAF3:a4:c1:38:0e:9d:6f:69:67 entering async_device_initialized - is_new_join: True
device - 0xAAF3:a4:c1:38:0e:9d:6f:69:67 has joined the ZHA zigbee network
[0xAAF3](TS0601): started configuration
[0xAAF3:ZDO](TS0601): 'async_configure' stage succeeded
[0xAAF3:1:0x0000]: finished channel configuration
[0xAAF3:1:0x0019]: finished channel configuration
Error handling '_save_attribute' event with (a4:c1:38:0e:9d:6f:69:67, 1, 0, 4, '_TZE200_ntcy3xu1') params: FOREIGN KEY constraint failed
[0xAAF3:1:0x0000]: 'async_configure' stage succeeded
[0xAAF3:1:0x0019]: 'async_configure' stage succeeded
Error handling '_save_attribute' event with (a4:c1:38:0e:9d:6f:69:67, 1, 0, 5, 'TS0601') params: FOREIGN KEY constraint failed
[0xAAF3](TS0601): completed configuration
[0xAAF3](TS0601): stored in registry: ZhaDeviceEntry(name='_TZE200_ntcy3xu1 TS0601', ieee='a4:c1:38:0e:9d:6f:69:67', last_seen=1649867790.7504046)
[... 116 lines cut ...]
[0xAAF3:1:0x0000] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=0, is_reply=1, disable_default_response=0, reserved=0, *is_cluster=False, *is_general=True), tsn=190, command_id=10, *is_reply=True)
[0xAAF3:1:0x0000] Decoded ZCL frame: Basic:Report_Attributes(attribute_reports=[Attribute(attrid=0xFFDF, value=TypeValue(type=CharacterString, value='y  )\x13   )e   )\x12   )i   )i   )i   )i'))])
[0xAAF3:1:0x0000] Received command 0x0A (TSN 190): Report_Attributes(attribute_reports=[Attribute(attrid=0xFFDF, value=TypeValue(type=CharacterString, value='y  )\x13   )e   )\x12   )i   )i   )i   )i'))])
[0xAAF3:1:0x0000] Attribute report received: 0xFFDF='y  )\x13   )e   )\x12   )i   )i   )i   )i'
[0xAAF3:1:0x0000] Sending reply header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=False, is_reply=1, disable_default_response=1, reserved=0, *is_cluster=False, *is_general=True), tsn=190, command_id=<GeneralCommand.Default_Response: 11>, *is_reply=True)
[0xAAF3:1:0x0000] Sending reply: Default_Response(command_id=10, status=<Status.SUCCESS: 0>)
[... cut ...]
[0xB3A6:1:0x0b04] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=0, is_reply=1, disable_default_response=1, reserved=0, *is_cluster=False, *is_general=True), tsn=217, command_id=1, *is_reply=True)
[0xB3A6:1:0x0b04] Decoded ZCL frame: TuyaZBElectricalMeasurement:Read_Attributes_rsp(status_records=[ReadAttributeRecord(attrid=0x050B, status=<Status.SUCCESS: 0>, value=TypeValue(type=int16s, value=0)), ReadAttributeRecord(attrid=0x0508, status=<Status.SUCCESS: 0>, value=TypeValue(type=uint16_t, value=0)), ReadAttributeRecord(attrid=0x0505, status=<Status.SUCCESS: 0>, value=TypeValue(type=uint16_t, value=232))])

Tropaion, 2022-04-15:
Can you post the device signature?

mwitlox, 2022-04-15:
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0051",
      "in_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0xef00"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZE200_ntcy3xu1",
  "model": "TS0601",
  "class": "zigpy.device.Device"
}

Tropaion, 2022-04-15:
Did you try applying the quirk I posted? According to the device signature it should work.

mwitlox, 2022-04-15:
> e quirk I posted? According to the device signature it should work.

Figuring out right now howto to that haha. Just made a folder and uploaded the .py file. ZHA doesn't work anymore so still looking how to fix that so i can test again ^^

Tropaion, 2022-04-15:
@mwitlox Here is an explanation how to do it: https://github.com/zigpy/zha-device-handlers/discussions/693#discussioncomment-857274
ZHA can also stop working when the quirk has an error -> look in the logs about it.

mwitlox, 2022-04-15:
@Tropaion 

Logger: homeassistant.config_entries
Source: custom_zha_quirks/ts0601_smoke.py:22
First occurred: 21:06:55 (1 occurrences)
Last logged: 21:06:55

Error setting up entry Sonoff Zigbee 3.0 USB Dongle Plus, s/n: 2001c85f6f3bec1196259e957a0af07f - ITead for zha
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 335, in async_setup
    result = await component.async_setup_entry(hass, self)
  File "/usr/src/homeassistant/homeassistant/components/zha/__init__.py", line 99, in async_setup_entry
    setup_quirks(config)
  File "/usr/local/lib/python3.9/site-packages/zhaquirks/__init__.py", line 409, in setup
    importer.find_module(modname).load_module(modname)
  File "<frozen importlib._bootstrap_external>", line 529, in _check_name_wrapper
  File "<frozen importlib._bootstrap_external>", line 1029, in load_module
  File "<frozen importlib._bootstrap_external>", line 854, in load_module
  File "<frozen importlib._bootstrap>", line 274, in _load_module_shim
  File "<frozen importlib._bootstrap>", line 711, in _load
  File "<frozen importlib._bootstrap>", line 680, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 850, in exec_module
  File "<frozen importlib._bootstrap>", line 228, in _call_with_frames_removed
  File "/config/custom_zha_quirks/ts0601_smoke.py", line 22, in <module>
    from . import TuyaManufCluster, TuyaManufClusterAttributes
ImportError: attempted relative import with no known parent package

mwitlox, 2022-04-15:
Ok now i got this

Logger: homeassistant.config_entries
Source: zha_quirks/ts0601_smoke.py:83
First occurred: 21:25:57 (1 occurrences)
Last logged: 21:25:57

Error setting up entry Sonoff Zigbee 3.0 USB Dongle Plus, s/n: 2001c85f6f3bec1196259e957a0af07f - ITead for zha
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 335, in async_setup
    result = await component.async_setup_entry(hass, self)
  File "/usr/src/homeassistant/homeassistant/components/zha/__init__.py", line 99, in async_setup_entry
    setup_quirks(config)
  File "/usr/local/lib/python3.9/site-packages/zhaquirks/__init__.py", line 409, in setup
    importer.find_module(modname).load_module(modname)
  File "<frozen importlib._bootstrap_external>", line 529, in _check_name_wrapper
  File "<frozen importlib._bootstrap_external>", line 1029, in load_module
  File "<frozen importlib._bootstrap_external>", line 854, in load_module
  File "<frozen importlib._bootstrap>", line 274, in _load_module_shim
  File "<frozen importlib._bootstrap>", line 711, in _load
  File "<frozen importlib._bootstrap>", line 680, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 850, in exec_module
  File "<frozen importlib._bootstrap>", line 228, in _call_with_frames_removed
  File "/config/zha_quirks/ts0601_smoke.py", line 73, in <module>
    class TuyaSmokeDetector0601(CustomDevice):
  File "/config/zha_quirks/ts0601_smoke.py", line 83, in TuyaSmokeDetector0601
    ("_TZE200_aycxwiau", "TS0601")
TypeError: 'tuple' object is not callable

Tropaion, 2022-04-15:
Ok, I made another mistake, let's try again 😄 : [ts0601_smoke.zip](https://github.com/zigpy/zha-device-handlers/files/8497935/ts0601_smoke.zip)

mwitlox, 2022-04-15:
@Tropaion Wauw this is working! Now at least i can see if there is a smoke alarm! Now i can test to make a group of 3 when 1 goes off all give alarm

Still need to learn more about this quirk stuff

Do you know if they also provide a battery status or something?

Tropaion, 2022-04-15:
@mwitlox It should show battery percentage like this:
[image]
It would be nice if you could test and confirm that everything works, then I can add the work to ZHA

mwitlox, 2022-04-15:
Hmm only got this

sensor.rookmelder_beneden_0406f16f_basic_lqi
sensor.rookmelder_beneden_0406f16f_basic_rssi

No battery

Tropaion, 2022-04-15:
Weird, maybe try repairing the device.

mwitlox, 2022-04-15:
Yeah just did that, got 3 of them. Maybe the battery type is not supported? Got 2 AAA batteries in these things

Tropaion, 2022-04-15:
Normally it should show battery status. What does it say for the energy source? Mains or Battery/Unknown?
Maybe @MattWestb knows more about this device/case.

mwitlox, 2022-04-15:
It says: Battery or Unknown

Tropaion, 2022-04-15:
@mwitlox I'm not sure if this changes anything, but if you have time, try this: [motion_ac02.zip](https://github.com/zigpy/zha-device-handlers/files/8498337/motion_ac02.zip)

mwitlox, 2022-04-15:
@Tropaion I have time, just drinking some beer playing around with HA and stuff. Thanks for the support so far!

When i add the file and reboot i get this error again:

Error setting up entry Sonoff Zigbee 3.0 USB Dongle Plus, s/n: 2001c85f6f3bec1196259e957a0af07f - ITead for zha
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 335, in async_setup
    result = await component.async_setup_entry(hass, self)
  File "/usr/src/homeassistant/homeassistant/components/zha/__init__.py", line 99, in async_setup_entry
    setup_quirks(config)
  File "/usr/local/lib/python3.9/site-packages/zhaquirks/__init__.py", line 409, in setup
    importer.find_module(modname).load_module(modname)
  File "<frozen importlib._bootstrap_external>", line 529, in _check_name_wrapper
  File "<frozen importlib._bootstrap_external>", line 1029, in load_module
  File "<frozen importlib._bootstrap_external>", line 854, in load_module
  File "<frozen importlib._bootstrap>", line 274, in _load_module_shim
  File "<frozen importlib._bootstrap>", line 711, in _load
  File "<frozen importlib._bootstrap>", line 680, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 850, in exec_module
  File "<frozen importlib._bootstrap>", line 228, in _call_with_frames_removed
  File "/config/zha_quirks/motion_ac02.py", line 40, in <module>
    class MotionT1(XiaomiCustomDevice):
  File "/config/zha_quirks/motion_ac02.py", line 71, in MotionT1
    NODE_DESCRIPTOR: zigpy.zdo.types.NodeDescriptor(
NameError: name 'NODE_DESCRIPTOR' is not defined

Tropaion, 2022-04-15:
Sorry, posted the wrong file, wanted to post this one: [ts0601_smoke.zip](https://github.com/zigpy/zha-device-handlers/files/8498375/ts0601_smoke.zip) 😥

mwitlox, 2022-04-15:
Got this:

Error setting up entry Sonoff Zigbee 3.0 USB Dongle Plus, s/n: 2001c85f6f3bec1196259e957a0af07f - ITead for zha
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 335, in async_setup
    result = await component.async_setup_entry(hass, self)
  File "/usr/src/homeassistant/homeassistant/components/zha/__init__.py", line 99, in async_setup_entry
    setup_quirks(config)
  File "/usr/local/lib/python3.9/site-packages/zhaquirks/__init__.py", line 409, in setup
    importer.find_module(modname).load_module(modname)
  File "<frozen importlib._bootstrap_external>", line 529, in _check_name_wrapper
  File "<frozen importlib._bootstrap_external>", line 1029, in load_module
  File "<frozen importlib._bootstrap_external>", line 854, in load_module
  File "<frozen importlib._bootstrap>", line 274, in _load_module_shim
  File "<frozen importlib._bootstrap>", line 711, in _load
  File "<frozen importlib._bootstrap>", line 680, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 850, in exec_module
  File "<frozen importlib._bootstrap>", line 228, in _call_with_frames_removed
  File "/config/zha_quirks/ts0601_smoke.py", line 74, in <module>
    class TuyaSmokeDetector0601(CustomDevice):
  File "/config/zha_quirks/ts0601_smoke.py", line 106, in TuyaSmokeDetector0601
    NODE_DESCRIPTOR: zigpy.zdo.types.NodeDescriptor(
NameError: name 'NODE_DESCRIPTOR' is not defined

Tropaion, 2022-04-15:
There was a missing import:
[ts0601_smoke.zip](https://github.com/zigpy/zha-device-handlers/files/8498398/ts0601_smoke.zip)

mwitlox, 2022-04-15:
[image]

javicalle, 2022-04-16:
The current `NODE_DESCRIPTOR` seems good to me (`mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>`, `*is_mains_powered=False`), so I think that there is no need to include it on the replacement part.

Since device don't have a `PowerConfiguration` cluster I'm not sure how you should report it for HA to expose the value. I'll see if I can figure it out.

Please go back to the working version of the quirk and check if there are any errors in the traces that might give us a clue as to what might be going on.

Regards.

javicalle, 2022-04-16:
@Tropaion can you tell me which cluster/attribute reports the battery status?

Tropaion, 2022-04-16:
@javicalle I don't know, I don't have the device 😕

javicalle, 2022-04-16:
Ahhh, I supposed from a previous comment.
Then it is not sure that this device will ever reported its battery status... 🤔

Tropaion, 2022-04-16:
Yes, I think the device simply doesn't do it, since it doesn't have a ```PowerConfiguration``` cluster
I will just PR the quirk like this.

javicalle, 2022-04-16:
It could be that device reports low_battery alt DP: 14
Other posible DPs for this device are:
```
    wooxSwitch: 102,
    wooxBattery: 14,
    wooxSmokeTest: 8,
```
But not sure because they aren't in the current version...

@mwitlox can you try to enable debug logs, remove and put battery to the device and attache the logs?

mwitlox, 2022-04-19:
@javicalle 

I tried enable debug.. do you have a example howto debug only one device the right way?

logger:
  default: info
  logs:
    zigpy.application: debug
    zigpy_znp.zigbee.application: debug

Only Info is logged. Think i am doing something wrong :(

2022-04-19 19:54:20 DEBUG (MainThread) [zigpy_znp.zigbee.application] Request failed (Unsuccessful request status code: <Status.MAC_NO_ACK: 233>), retry attempt 1 of 5
2022-04-19 19:54:22 DEBUG (MainThread) [zigpy_znp.zigbee.application] Request failed (Unsuccessful request status code: <Status.MAC_NO_ACK: 233>), retry attempt 2 of 5
2022-04-19 19:54:24 DEBUG (MainThread) [zigpy_znp.zigbee.application] Request failed (Unsuccessful request status code: <Status.MAC_NO_ACK: 233>), retry attempt 3 of 5
2022-04-19 19:54:24 DEBUG (MainThread) [zigpy_znp.zigbee.application] Request failed (Unsuccessful request status code: <Status.MAC_NO_ACK: 233>), retry attempt 4 of 5
2022-04-19 19:54:25 DEBUG (MainThread) [zigpy_znp.zigbee.application] Request failed (Unsuccessful request status code: <Status.MAC_NO_ACK: 233>), retry attempt 5 of 5

2022-04-19 19:54:59 INFO (MainThread) [zigpy_znp.zigbee.application] ZDO device announce: nwk=0xB667, ieee=a4:c1:38:fa:6f:f1:06:04, capabilities=128
2022-04-19 19:54:59 INFO (MainThread) [zigpy.application] Device 0xb667 (a4:c1:38:fa:6f:f1:06:04) joined the network
2022-04-19 19:54:59 INFO (MainThread) [zigpy_znp.zigbee.application] TC device join: ZDO.TCDevInd.Callback(SrcNwk=0xB667, SrcIEEE=a4:c1:38:fa:6f:f1:06:04, ParentNwk=0x0000)
2022-04-19 19:54:59 INFO (MainThread) [zigpy.application] Device 0xb667 (a4:c1:38:fa:6f:f1:06:04) joined the network

There is also a button on the device. When it's mounted it clicks and does something. Some sort of check

Tropaion, 2022-04-19:
@mwitlox Enable debugging like this:
```
logger:
  default: info
  logs:
    zigpy: debug
    zhaquirks: debug
```

javicalle, 2022-04-19:
> do you have a example howto debug only one device the right way?

It's not possible to debug just one device. Enable the logs perform some actions and disable.
Logs can be setted (in mode YAML) from the `logger.set_level` service in developer tools section.

mwitlox, 2022-04-19:
[HA Debug Zig 19-14-2022.txt](https://github.com/zigpy/zha-device-handlers/files/8516369/HA.Debug.Zig.19-14-2022.txt)

javicalle, 2022-04-19:
@mwitlox have you tried to remove and put batteries from device while having debug traces? I can't see any report. If you already done, we can suppose that device is not reporting battery status. Maybe only low battery status (when battery will be low).

javicalle, 2022-04-20:
The device is reporting 3 DP:
* DP1: `command_id=1025` used in `TuyaIasZone` cluster ✅ 
* DP4: `command_id=260` settled in attribute `0x0104` reports values 1/0
* DP14: `command_id=1038` settled in attribute `0x040e` always with value `2`

DP4 can be a tamper sensor (`IasZone.ZoneStatus.Tamper`). Not implemented in any quirk.
Maybe DP14 is some discrete battery level (good, med, low), but it's just a guess. Some implementations are giving fake battery levels in function of this value (ex: good=100%, med=20%, low =1%). Not sure if there is a better way to make it accesible in HA.

goldfoxo, 2022-04-21:
I've just received one of the same sensors, and the quirk here works, thanks!

Playing around a little bit I think I those can confirm those values a little.

The `0x0104` attribute is the tamper sensor. It sets to `1` when the backplate is off, and `0` when put back on and the tamper button gets pressed in. Funnily though mine wasn't being reliable and I discovered it's stuck, I need to give it a jolt to realise the cover is off and send `1`.

`0x040e` does seem like it's a battery quality type thing. With fresh batteries it reports `2` on every update of anything else. Then I swapped one almost dead battery in, and it reported `0` once when it started up, but then didn't report again ever. Then I charged two dead batteries up just a tiny bit and tried those. It reported `2` just like brand new batteries for a few minutes, until it reported `0` just once and stopped again.

I suppose `0` does mean very low battery, then it stops sending updates (maybe to save battery). I haven't seen `1` yet.

mwitlox, 2022-04-21:
DP4: command_id=260 settled in attribute 0x0104 reports values 1/0 could be something like: 0 Mounted 1 Unmounted
DP14: command_id=1038 settled in attribute 0x040e always with value 2 Battery 100% 1 Battery 50% 0 Replace Battery 0-10%

Something strange happend today. I have a few sockets with power monitoring. Seems like it somehow uses the same quirk now as the smoke detectors. But i didn't change the socket or readd the socket. Sorry if my knowledge from zigbee devices is limited. Just a year ago i only used Tasmota Wifi devices. Maybe i need to change something? Can someone help with this?

[image]

[image]

{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x010a",
      "in_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0702",
        "0x0b04",
        "0xe000",
        "0xe001"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZ3000_gjnozsaz",
  "model": "TS011F",
  "class": "zhaquirks.tuya.ts011f_plug.Plug"
}

mwitlox, 2022-04-21:
Just turned quirks off then it works again but stil saying zhaquirks.tuya.ts011f_plug.Plug

[image]

Tropaion, 2022-04-21:
I don't understand the problem, the smoke detector quirk is ```ts0601_smoke``` and the device you should is ```TS011F``` and uses the quirk ```ts011f_plug```, which is the right one and not even a custom quirk.

mwitlox, 2022-04-21:
@Tropaion 

When i enable this:
[image]
All the TS011F sockets don't work properly anymore

mwitlox, 2022-04-21:
Ok sorry forget it. I readd the device, reboot HA serval times now all works again. Maybe something to do with todays update HA Core 2022.4.6

Back to the Smoke sensors.

javicalle, 2022-05-23:
I'm not familiar with the Zigbee IasZone cluster.
Also not familiar with the cluster behaviour in HA.
But... what do you think about something like?:
```python
TUYA_SMOKE_DETECTED_ATTR = 0x0401  # [0]/[1] [Detected]/[Clear]!
TUYA_SMOKE_TAMPED_ATTR = 0x0104  # [0]/[1] [Clear]/[Tamped]!
TUYA_SMOKE_BATTERY_ATTR = 0x040e  # [0]/[1]/[2] [Low]/[Med]]/[Full]!

class TuyaSmokeDetectorCluster(TuyaManufClusterAttributes):
    """Manufacturer Specific Cluster of the TS0601 smoke detector."""

    attributes = {
        TUYA_SMOKE_DETECTED_ATTR: ("smoke_detected", t.uint8_t, True),
        TUYA_SMOKE_TAMPED_ATTR: ("tamped_device", t.uint8_t, True),
        TUYA_SMOKE_BATTERY_ATTR: ("battery_status", t.uint8_t, True),
    }

    def _update_attribute(self, attrid, value):
        super()._update_attribute(attrid, value)
        if attrid == TUYA_SMOKE_DETECTED_ATTR:
            if value == 0:
                self.endpoint.device.ias_bus.listener_event(
                    "update_zone_status", IasZone.ZoneStatus.Alarm_1
                )
            else:
                self.endpoint.device.ias_bus.listener_event("update_zone_status", 0)
        elif attrid == TUYA_SMOKE_TAMPED_ATTR:
            if value == 1:
                self.endpoint.device.ias_bus.listener_event(
                    "update_zone_status", IasZone.ZoneStatus.Tamper
                )
            else:
                self.endpoint.device.ias_bus.listener_event("update_zone_status", 0)
        elif attrid == TUYA_SMOKE_BATTERY_ATTR:
            if value == 0:
                self.endpoint.device.ias_bus.listener_event(
                    "update_zone_status", IasZone.ZoneStatus.Battery
                )
            else:
                self.endpoint.device.ias_bus.listener_event("update_zone_status", 0)
        else:
            _LOGGER.warning(
                "[0x%04x:%s:0x%04x] unhandled attribute: 0x%04x",
                self.endpoint.device.nwk,
                self.endpoint.endpoint_id,
                self.cluster_id,
                attrid,
            )
```

Could do the job?

I don't know if this would be the 'standar' way to manage all the others device alerts (tamper, battery) but I don't think that a cluster for every alert would be the way to go.

madspoulsen, 2022-05-30:
I'm in on this. I can't contribute with much more than a virtual beer, but this looks nice so far. 
The tamper and battery integration would be so awesome! I too see those when i manually go in the cluster section. I'm just too noob to know how to make use of it from there sorry.
But i'm definitely looking forward to see a complete quirk with alle 3 sensors. :)

mwitlox, 2022-06-01:
> this would be the 'standar' way to manage all the others device alerts (tamper, battery) but I don't think that a cluster for every alert would be the way to go.

TUYA_SMOKE_DETECTED_ATTR = 0x0401  # [0]/[1] [Detected]/[Clear]!
TUYA_SMOKE_TAMPED_ATTR = 0x0104  # [0]/[1] [Clear]/[Tamped]!
TUYA_SMOKE_BATTERY_ATTR = 0x040e  # [0]/[1]/[2] [Low]/[Med]]/[Full]!

If this is possible with these smoke sensors this will be a great thing and the maximum that can be achieve i think. Let me know if i can test another quirk

closed 2022-12-05 by github-actions[bot]

alexd321, 2023-01-23:
Did anyone make any progress on this one? I tried @mwitlox 's suggested change to the quirk, but no success

javicalle, 2023-01-23:
Which part are you asking about?

The smoke detector must be already supported.

The extra detector status will require some implementation in HA and had not progressed.

alexd321, 2023-01-24:
Sorry for not being clear, yes the extra sensors like battery etc