zigpy/zha-device-handlers #1492: [Device Support Request] Mercator Ikuü Dimmer Switch Mechanism (SSWM-DIMZ)

intra-aud, 2022-04-14:
I have a Mercator Ikuu Dimmer Mechanism (SSWM-DIMZ) which identifies as a Tuya TS0601 Dimmer.
I have tried to use an existing quirk however it does not seem to fit the device and does not function correctly, the quirk that i attempted to use is https://github.com/zigpy/zha-device-handlers/issues/1463. 

I am still fairly new at zigpy so if i have missed something, please let me know and i'll be happy to attempt to assist in solving this issue and hopefully get it support in HA as well!

**Device signature:
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0051",
      "in_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0xef00"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZE200_9cxuhakf",
  "model": "TS0601",
  "class": "zigpy.device.Device"
}

javicalle, 2022-04-14:
It is the same device with the same functionallity?
Your device signature is quite diferent nad probably will match the `TuyaSingleSwitchDimmer` device:
* https://github.com/zigpy/zha-device-handlers/blob/5272b4c3e180f1c0c43c2a412c8428c12549e772/zhaquirks/tuya/ts0601_dimmer.py#L28

I suggest you to configure a local quirk and add your device to the list:
```python
        MODELS_INFO: [
            ("_TZE200_dfxkcots", "TS0601"),
            ("_TZE200_whpb9yts", "TS0601"),
            ("_TZE200_ebwgzdqq", "TS0601"),
            ("_TZE200_9i9dt8is", "TS0601"),
            ("_TZE200_swaamsoy", "TS0601"),
            ("_TZE200_0nauxa0p", "TS0601"),
            ("_TZE200_la2c2uo9", "TS0601"),
            ("_TZE200_9cxuhakf", "TS0601"),
        ],
```

Refs:
* https://github.com/zigpy/zha-device-handlers/discussions/693

intra-aud, 2022-04-17:
Hi @javicalle 

Thanks for your suggestion, i did apply the model to the quirk that you suggested and i can now see some controls to the device.

Now it appears to use that particular quirk

```
  "manufacturer": "_TZE200_9cxuhakf",
  "model": "TS0601",
  "class": "ts0601_dimmer.TuyaSingleSwitchDimmer"
```

The on/off control does activate the light and the intensity does appear to work too, however  there is an error that pops up when i try to use the entity provided which is:
```
"Failed to call service light/turn_off. 'Status' object is not subscriptable" or  
"Failed to call service light/turn_on. 'Status' object is not subscriptable"
```

So i guess were close but not exactly matching.

Here's what i found in the HA core log:

```
    await hass.services.async_call(
  File "/usr/src/homeassistant/homeassistant/core.py", line 1634, in async_call
    task.result()
  File "/usr/src/homeassistant/homeassistant/core.py", line 1671, in _execute_service
    await cast(Callable[[ServiceCall], Awaitable[None]], handler.job.target)(
  File "/usr/src/homeassistant/homeassistant/helpers/entity_component.py", line 204, in handle_service
    await service.entity_service_call(
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 677, in entity_service_call
    future.result()  # pop exception if have
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 955, in async_request_call
    await coro
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 714, in _handle_entity_call
    await result
  File "/usr/src/homeassistant/homeassistant/components/light/__init__.py", line 513, in async_handle_light_off_service
    await light.async_turn_off(**filter_turn_off_params(light, params))
  File "/usr/src/homeassistant/homeassistant/components/zha/light.py", line 343, in async_turn_off
    if isinstance(result, Exception) or result[1] is not Status.SUCCESS:
TypeError: 'Status' object is not subscriptable
```

javicalle, 2022-04-17:
Which HA version are you running?
The suggested code targets v2022.4.1 or .2 I don't remember exactly.

intra-aud, 2022-04-17:
Hi @javicalle 

I am running HA 2022.4.4

javicalle, 2022-04-18:
This error has been fixed in zigpy v0.0.70 included in HA from v2022.4.1:
* https://github.com/zigpy/zha-device-handlers/pull/1464

Have you any other local quirk in your installation (other than the `tuya/ts0601_dimmer.py` file)?
Could you enable the ZHA logs and attach the relevant info?
* https://www.home-assistant.io/integrations/zha/#debug-logging

intra-aud, 2022-04-19:
Ok, i thought i would try and upgrade today to the latest version of HA (2022.4.5) to see if this would help, unfortunately i still get the same error when attempting to use the entity to fire off the dimmer.

This time i did enable debug logging and i captured (what i think) is some relevant info as i can see an error in the log.

For you information i also have Lumi Curtin motors (E1) which also has a quirk file in the custom_zha_quirks folder.

`
2022-04-19 11:16:00 DEBUG (MainThread) [homeassistant.core] Bus:Handling <Event call_service[L]: domain=light, service=turn_on, service_data=entity_id=light.tze200_9cxuhakf_ts0601_71119afe_level_on_off>
2022-04-19 11:16:00 DEBUG (MainThread) [zigpy.zcl] [0xA9AE:1:0xef00] Sending request header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.CLUSTER_COMMAND: 1>, is_manufacturer_specific=True, is_reply=0, disable_default_response=0, reserved=0, *is_cluster=True, *is_general=False), manufacturer=4098, tsn=99, command_id=0, *is_reply=False)
2022-04-19 11:16:00 DEBUG (MainThread) [zigpy.zcl] [0xA9AE:1:0xef00] Sending request: set_data(param=Command(status=0, tsn=0, command_id=257, function=0, data=[1, 1]))
2022-04-19 11:16:00 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0xA9AE:1:0x0006]: executed 'on' command with args: '()' kwargs: '{}' result: Status.SUCCESS
2022-04-19 11:16:00 DEBUG (MainThread) [bellows.ezsp.protocol] Send command sendUnicast: (<EmberOutgoingMessageType.OUTGOING_DIRECT: 0>, 0xA9AE, EmberApsFrame(profileId=260, clusterId=61184, sourceEndpoint=1, destinationEndpoint=1, options=<EmberApsOption.APS_OPTION_ENABLE_ROUTE_DISCOVERY|APS_OPTION_RETRY: 320>, groupId=0, sequence=99), 100, b'\x05\x02\x10c\x00\x00\x00\x01\x01\x00\x01\x01')
2022-04-19 11:16:00 DEBUG (MainThread) [bellows.ezsp.protocol] Application frame 52 (sendUnicast) received: b'00b5'
2022-04-19 11:16:00 ERROR (MainThread) [homeassistant.components.websocket_api.http.connection] [139750229786480] 'Status' object is not subscriptable
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/websocket_api/commands.py", line 193, in handle_call_service
    await hass.services.async_call(
  File "/usr/src/homeassistant/homeassistant/core.py", line 1634, in async_call
    task.result()
  File "/usr/src/homeassistant/homeassistant/core.py", line 1671, in _execute_service
    await cast(Callable[[ServiceCall], Awaitable[None]], handler.job.target)(
  File "/usr/src/homeassistant/homeassistant/helpers/entity_component.py", line 204, in handle_service
    await service.entity_service_call(
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 677, in entity_service_call
    future.result()  # pop exception if have
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 955, in async_request_call
    await coro
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 714, in _handle_entity_call
    await result
  File "/usr/src/homeassistant/homeassistant/components/light/__init__.py", line 504, in async_handle_light_on_service
    await light.async_turn_on(**filter_turn_on_params(light, params))
  File "/usr/src/homeassistant/homeassistant/components/zha/light.py", line 258, in async_turn_on
    if isinstance(result, Exception) or result[1] is not Status.SUCCESS:
TypeError: 'Status' object is not subscriptable
2022-04-19 11:16:00 DEBUG (MainThread) [bellows.ezsp.protocol] Application frame 69 (incomingMessageHandler) received: b'00040100ef01014001000041a0c4aea9ffff070c0210630b0000'
2022-04-19 11:16:00 DEBUG (MainThread) [bellows.zigbee.application] Received incomingMessageHandler frame with [<EmberIncomingMessageType.INCOMING_UNICAST: 0>, EmberApsFrame(profileId=260, clusterId=61184, sourceEndpoint=1, destinationEndpoint=1, options=<EmberApsOption.APS_OPTION_ENABLE_ROUTE_DISCOVERY|APS_OPTION_RETRY: 320>, groupId=0, sequence=65), 160, -60, 0xa9ae, 255, 255, b'\x0c\x02\x10c\x0b\x00\x00']
2022-04-19 11:16:00 DEBUG (MainThread) [zigpy.zcl] [0xA9AE:1:0xef00] Received ZCL frame: b'\x0c\x02\x10c\x0b\x00\x00'
2022-04-19 11:16:00 DEBUG (MainThread) [zigpy.zcl] [0xA9AE:1:0xef00] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=True, is_reply=1, disable_default_response=0, reserved=0, *is_cluster=False, *is_general=True), manufacturer=4098, tsn=99, command_id=11, *is_reply=True)
2022-04-19 11:16:00 DEBUG (MainThread) [zigpy.zcl] [0xA9AE:1:0xef00] Decoded ZCL frame: TuyaManufacturerLevelControl:Default_Response(command_id=0, status=<Status.SUCCESS: 0>)
2022-04-19 11:16:00 DEBUG (MainThread) [bellows.ezsp.protocol] Application frame 63 (messageSentHandler) received: b'00aea9040100ef010140010000b5640000'
2022-04-19 11:16:00 DEBUG (MainThread) [bellows.zigbee.application] Received messageSentHandler frame with [<EmberOutgoingMessageType.OUTGOING_DIRECT: 0>, 43438, EmberApsFrame(profileId=260, clusterId=61184, sourceEndpoint=1, destinationEndpoint=1, options=<EmberApsOption.APS_OPTION_ENABLE_ROUTE_DISCOVERY|APS_OPTION_RETRY: 320>, groupId=0, sequence=181), 100, <EmberStatus.SUCCESS: 0>, b'']
2022-04-19 11:16:00 DEBUG (MainThread) [bellows.ezsp.protocol] Application frame 69 (incomingMessageHandler) received: b'00040100ef0101000100004298c2aea9ffff0a090a0200000101000101'
[... cut ...]
2022-04-19 11:16:00 DEBUG (MainThread) [zigpy.zcl] [0xA9AE:1:0xef00] Received command 0x02 (TSN 10): set_data_response(param=Command(status=0, tsn=0, command_id=257, function=0, data=[1, 1]))
2022-04-19 11:16:00 DEBUG (MainThread) [zhaquirks.tuya] cc:86:ec:ff:fe:9a:11:71 Received Attribute Report. Command is 2, Tuya Paylod values[Status : 0, TSN: 0, Command: 257, Function: 0, Data: [1, 1]]
2022-04-19 11:16:00 DEBUG (MainThread) [zhaquirks.tuya] cc:86:ec:ff:fe:9a:11:71 - Received switch event message, channel: 1, state: 1
`

BTW, thank you for all your help and patience to date!

javicalle, 2022-04-19:
Please, put the logs inside 3 apostrophes `\``.

Can you show us the content of the other quirk or point me to where you copied them from?
I think there may be something that is preventing some of the changes from the new version from being applied.

intra-aud, 2022-04-19:
> Please, put the logs inside 3 apostrophes ```.
> 
> Can you show us the content of the other quirk or point me to where you copied them from? I think there may be something that is preventing some of the changes from the new version from being applied.

Sure , the other quirk was taken from this ZHA-Device-Handler and is under Xiaomi -> Aquara -> [roller_curtain_e1.py]

(https://github.com/zigpy/zha-device-handlers/blob/dev/zhaquirks/xiaomi/aqara/roller_curtain_e1.py)

javicalle, 2022-04-19:
@intra-aud could you try to use this version for your quirk:
* https://github.com/zigpy/zha-device-handlers/blob/25b4a8478f0a54e1b7103a7eb1a2585da65e192a/zhaquirks/tuya/ts0601_dimmer.py

Replace all file content with this version and then add your device to the quirk:
```python
        MODELS_INFO: [
            ("_TZE200_dfxkcots", "TS0601"),
            ("_TZE200_whpb9yts", "TS0601"),
            ("_TZE200_ebwgzdqq", "TS0601"),
            ("_TZE200_9i9dt8is", "TS0601"),
            ("_TZE200_swaamsoy", "TS0601"),
            ("_TZE200_0nauxa0p", "TS0601"),
            ("_TZE200_la2c2uo9", "TS0601"),
            ("_TZE200_9cxuhakf", "TS0601"),
        ],
```

intra-aud, 2022-04-20:
Hi @javicalle 

I have replaced the quirk with your suggestion and added the needed model line to the file.

Glad to report that the error regarding light_on and light_off TypeError: 'Status' object is not subscriptable has been resolved.
(Yippee!)

I do notice that the "brightness" control appears to change state/level when the device is on or off

Example:
Off Status = Brightness 0 (slider to the far left)
On Status = Brightness 50 (slider to the middle of the line)
Slide the brightness to 75 (when dimmer is off) , light turns on to that selected brightness.

Is the intended behaviour?

Is there a way to set the brightness in an independent fashion of the on/off status?

My use case would be 80% brightness throughout the day but after 11:00pm i'd like the light to be 25% brightness  (so that i'm not blasted with light should i need to visit bathroom in the night)  :)

javicalle, 2022-05-03:
Sorry the delay. I have lost the track of this issue.

> Is the intended behaviour?

It's not a quirk thing. This is how HA behaves and has now been 'fixed' on Tuya devices 🤷🏻‍♂️ 

> Is there a way to set the brightness in an independent fashion of the on/off status?

I don't know of any way to do it using automations.
There is https://github.com/mdeweerd/zha-toolkit which allows invoking services. It probably works by writing the `current_level` attribute or calling the `move_to_level` command but not tested myself.

javicalle, 2022-05-03:
If device is working (despite the HA behaviour) you can create the PR to add your device to the mainstream?

Just edit from the web the file content and follow the instructions:
* https://github.com/zigpy/zha-device-handlers/blob/dev/zhaquirks/tuya/ts0601_dimmer.py

This way other users will be able to use the device in ZHA thanks to you!

javicalle, 2022-05-03:
It seems that there is a method that can be used:
```yaml
service: zha.set_zigbee_cluster_attribute
data:
  ieee: 00:0d:6f:00:05:7d:2d:34
  attribute: 0
  endpoint_id: 1
  cluster_id: 8
  value: '255'
```

Just replace the `ieee` value with the one from your device.

intra-aud, 2022-05-04:
Hi @javicalle 

I've added a PR so that support for the device can be evaluated and merged into main branch (thanks for guiding me through how git works!).

The method that you mention seems a little strange, took me a little while to understand how the clusters work but i think i got it in the end. Strangely the issue with behaviour is still there, when the switch is in the "off" state i can write the value of "255" to the device but then the light automatically turns "on" at full brightness by itself.

Its like the cluster automatically turns on based off a value change.

Should i raise this behaviour issue separately as the basic on and off seems to to work.

javicalle, 2022-05-04:
> I've added a PR so that support for the device can be evaluated and merged into main branch (thanks for guiding me through how git works!).

Just one more thing, could you add in the first comment of the PR the reference to this issue? Like this:
```
Fixes: #1492
```
This way when the PR is merged, this issue is automatically closed.

> Should i raise this behaviour issue separately as the basic on and off seems to to work.

If you have the opportunity to perform some more tests, of course.
There are a couple of things that have changed lately, but I'm not sure they justify what you're reporting.
But we better review it in a specific issue.

intra-aud, 2022-05-04:
@javicalle 

Done, gee that's an awfully handy thing to link the issue via a comment!

javicalle, 2022-05-04:
Ummm, almost done, but edit the first comment and put the reference in that one.

intra-aud, 2022-05-04:
> Ummm, almost done, but edit the first comment and put the reference in that one.

@javicalle  You know, its been on of those days where nothing seems right and i haven't done the correct thing. Regardless hopefully that fixes it.

javicalle, 2022-05-04:
Perfect now!
Thanks for your work.

intra-aud, 2022-10-11:
No problems, all done :)