zigpy/zha-device-handlers #1495: AduroSmart ERIA Smart Wireless Dimming Switch Remote

jasondostal, 2022-04-14:
**Is your feature request related to a problem? Please describe.**
I have an AduroSmart ERIA Smart Wireless Dimming Switch Remote paired with a Conbee II stick.  I have Home Assistant running using Docker on Ubuntu.  When I first setup the remote, the on, off, dim up and dim down buttons all worked.  Recently, the dim up and down buttons ceased functioning but the on/off buttons still do.

I've replaced batteries, repaired the device, reconfigured, the device, removed it from HA, etc.  Debugging the automation, I see the triggers for on and off firing but it is not triggering on up/down.

**Describe the solution you'd like**
Up/down dim buttons work 

**Device signature - this can be acquired by removing the device from ZHA and pairing it again from the add devices screen. Be sure to add the entire content of the log panel after pairing the device to a code block below this line.**
`ew device 0x544d (00:15:8d:00:01:98:d3:94) joined the network
[0x544d] Scheduling initialization
Received frame on uninitialized device <Device model=None manuf=None nwk=0x544D ieee=00:15:8d:00:01:98:d3:94 is_initialized=False> from ep 0 to ep 0, cluster 19: b'\xe8MT\x94\xd3\x98\x01\x00\x8d\x15\x00\x80'
[0x544d:zdo] ZDO request ZDOCmd.Device_annce: [0x544D, 00:15:8d:00:01:98:d3:94, 128]
Tries remaining: 3
[0x544d] Requesting 'Node Descriptor'
Tries remaining: 2
[0x544d] Extending timeout for 0x23 request
Received frame on uninitialized device <Device model=None manuf=None nwk=0x544D ieee=00:15:8d:00:01:98:d3:94 is_initialized=False> from ep 0 to ep 0, cluster 32770: b'#\x00MT\x02@\x807\x10\x7fd\x00\x00*d\x00\x00'
[0x544d] Got Node Descriptor: NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4151, maximum_buffer_size=127, maximum_incoming_transfer_size=100, server_mask=10752, maximum_outgoing_transfer_size=100, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)
[0x544d] Discovering endpoints
Tries remaining: 3
[0x544d] Extending timeout for 0x25 request
Received frame on uninitialized device <Device model=None manuf=None nwk=0x544D ieee=00:15:8d:00:01:98:d3:94 is_initialized=False> from ep 0 to ep 0, cluster 32773: b'%\x00MT\x01\x01'
[0x544d] Discovered endpoints: [1]
[0x544d] Initializing endpoints [<Endpoint id=1 in=[] out=[] status=<Status.NEW: 0>>]
[0x544d:1] Discovering endpoint information
Tries remaining: 3
[0x544d] Extending timeout for 0x27 request
Received frame on uninitialized device <Device model=None manuf=None nwk=0x544D ieee=00:15:8d:00:01:98:d3:94 is_initialized=False> from ep 0 to ep 0, cluster 32772: b"'\x00MT\x1e\x01\x04\x01 \x08\x02\x05\x00\x00\x03\x00\x08\x00\x00\x10\xcc\xfc\x06\x03\x00\x04\x00\x06\x00\x08\x00\x00\x10\xcc\xfc"
[0x544d:1] Discovered endpoint information: SizePrefixedSimpleDescriptor(endpoint=1, profile=260, device_type=2080, device_version=2, input_clusters=[0, 3, 8, 4096, 64716], output_clusters=[3, 4, 6, 8, 4096, 64716])
[0x544D:1:0x0000] Sending request header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=False, is_reply=0, disable_default_response=0, reserved=0, *is_cluster=False, *is_general=True), tsn=41, command_id=<GeneralCommand.Read_Attributes: 0>, *is_reply=False)
[0x544D:1:0x0000] Sending request: Read_Attributes(attribute_ids=[4, 5])
[0x544d] Extending timeout for 0x29 request
[0x544D:1:0x0000] Received ZCL frame: b'\x18)\x01\x04\x00\x00B\nADUROLIGHT\x05\x00\x00B\x0eAdurolight_NCC'
[0x544D:1:0x0000] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=0, is_reply=1, disable_default_response=1, reserved=0, *is_cluster=False, *is_general=True), tsn=41, command_id=1, *is_reply=True)
[0x544D:1:0x0000] Decoded ZCL frame: Basic:Read_Attributes_rsp(status_records=[ReadAttributeRecord(attrid=0x0004, status=<Status.SUCCESS: 0>, value=TypeValue(type=CharacterString, value='ADUROLIGHT')), ReadAttributeRecord(attrid=0x0005, status=<Status.SUCCESS: 0>, value=TypeValue(type=CharacterString, value='Adurolight_NCC'))])
[0x544d] Read model 'Adurolight_NCC' and manufacturer 'ADUROLIGHT' from <Endpoint id=1 in=[basic:0x0000, identify:0x0003, level:0x0008, lightlink:0x1000, manufacturer_specific:0xFCCC] out=[identify:0x0003, groups:0x0004, on_off:0x0006, level:0x0008, lightlink:0x1000, manufacturer_specific:0xFCCC] status=<Status.ZDO_INIT: 1>>
[0x544d] Discovered basic device information for <Device model='Adurolight_NCC' manuf='ADUROLIGHT' nwk=0x544D ieee=00:15:8d:00:01:98:d3:94 is_initialized=True>
Device is initialized <Device model='Adurolight_NCC' manuf='ADUROLIGHT' nwk=0x544D ieee=00:15:8d:00:01:98:d3:94 is_initialized=True>
Checking quirks for ADUROLIGHT Adurolight_NCC (00:15:8d:00:01:98:d3:94)
Considering <class 'zhaquirks.aduro.adurolightncc.AdurolightNCC'>
Found custom device replacement for 00:15:8d:00:01:98:d3:94: <class 'zhaquirks.aduro.adurolightncc.AdurolightNCC'>
'button' component -> 'ZHAIdentifyButton' using ['identify']
'sensor' component -> 'RSSISensor' using ['basic']
'sensor' component -> 'LQISensor' using ['basic']
device - 0x544D:00:15:8d:00:01:98:d3:94 entering async_device_initialized - is_new_join: True
device - 0x544D:00:15:8d:00:01:98:d3:94 has joined the ZHA zigbee network
[0x544D](Adurolight_NCC): started configuration
[0x544D:ZDO](Adurolight_NCC): 'async_configure' stage succeeded
[0x544D:1:0x0003]: finished channel configuration
[0x544D:1:0x0000]: finished channel configuration
[0x544D:1:0x1000] Sending request header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.CLUSTER_COMMAND: 1>, is_manufacturer_specific=False, is_reply=0, disable_default_response=0, reserved=0, *is_cluster=True, *is_general=False), tsn=43, command_id=65, *is_reply=False)
[0x544D:1:0x1000] Sending request: get_group_identifiers(start_index=0)
[0x544d] Extending timeout for 0x2b request
[0x544d] Extending timeout for 0x2d request
[0x544d] Extending timeout for 0x2f request
Error handling '_save_attribute' event with (00:15:8d:00:01:98:d3:94, 1, 0, 4, 'ADUROLIGHT') params: FOREIGN KEY constraint failed
Error handling '_save_attribute' event with (00:15:8d:00:01:98:d3:94, 1, 0, 5, 'Adurolight_NCC') params: FOREIGN KEY constraint failed
[0x544D:1:0x1000] Received ZCL frame: b'\x18+\x0bA\x00'
[0x544D:1:0x1000] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=0, is_reply=1, disable_default_response=1, reserved=0, *is_cluster=False, *is_general=True), tsn=43, command_id=11, *is_reply=True)
[0x544D:1:0x1000] Decoded ZCL frame: LightLink:Default_Response(command_id=65, status=<Status.SUCCESS: 0>)
[0x544D:1:0x1000] Received ZCL frame: b'\x19+A\x01\x00\x01\x0c\xd2\x00'
[0x544D:1:0x1000] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.CLUSTER_COMMAND: 1>, is_manufacturer_specific=0, is_reply=1, disable_default_response=1, reserved=0, *is_cluster=True, *is_general=False), tsn=43, command_id=65, *is_reply=True)
[0x544D:1:0x1000] Decoded ZCL frame: LightLink:get_group_identifiers_rsp(total=1, start_index=0, group_info_records=[GroupInfoRecord(group_id=0xD20C, group_type=0)])
[0x544D:1:0x1000] Received command 0x41 (TSN 43): get_group_identifiers_rsp(total=1, start_index=0, group_info_records=[GroupInfoRecord(group_id=0xD20C, group_type=0)])
[0x544D:1:0x1000] No explicit handler for cluster command 0x41: get_group_identifiers_rsp(total=1, start_index=0, group_info_records=[GroupInfoRecord(group_id=0xD20C, group_type=0)])
[0x544D:1:0x0008]: bound 'level' cluster: Status.SUCCESS
[0x544D:1:0x0008]: finished channel configuration
[0x544D:1:0x0006]: bound 'on_off' cluster: Status.SUCCESS
[0x544D:1:0x0006]: finished channel configuration
[... 161 lines cut ...]
2022-04-14 14:10:57 DEBUG (MainThread) [zigpy_deconz.api] Command Command.aps_data_indication (1, 1)
[... cut ...]
2022-04-14 14:10:57 WARNING (MainThread) [zigpy.zcl] [0x544D:1:0xfccc] Unknown cluster command 0 b'\x00\x02\x00'
2022-04-14 14:10:57 DEBUG (MainThread) [zigpy.zcl] [0x544D:1:0xfccc] Received command 0x00 (TSN 41): b'\x00\x02\x00'
2022-04-14 14:10:57 DEBUG (MainThread) [zigpy.zcl] [0x544D:1:0xfccc] No explicit handler for cluster command 0x00: b'\x00\x02\x00'
2022-04-14 14:10:57 DEBUG (MainThread) [zigpy_deconz.api] 'aps_data_indication' response from <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0x544d>, ep: 1, profile: 0x0104, cluster_id: 0xfccc, data: b'112900000200'`

MattWestb, 2022-04-14:
If its the device automatons that is not working OK update HA to 2022.4.3 that if fixing some changed made with dimmer commands that was broken in 2022.4.0

jasondostal, 2022-04-14:
That did it - just updated to 2022.4.3 and as soon as it came up, tested out the switch and it works.  Thank you very much.

closed 2022-04-14 by jasondostal