zigpy/zha-device-handlers #1498: [Device Support Request] Aqara Motion Sensor P1 (lumi.motion.ac02)

madface303, 2022-04-15:
**Is your feature request related to a problem? Please describe.**
The Sensor is recognized by ZHA but there is no motion sensor (only power, lqi and rssi, power and rssi have state unknown)

**Link to product**
https://www.aqara.com/en/product/motion-sensor-p1

**Describe the solution you'd like**
Use it with home assistant ;)

**Device signature - this can be acquired by removing the device from ZHA and pairing it again from the add devices screen. Be sure to add the entire content of the log panel after pairing the device to a code block below this line.**

```
[0x0000:zdo] ZDO request ZDOCmd.Mgmt_Permit_Joining_rsp: [<Status.SUCCESS: 0>]
[0x0000:zdo] No handler for ZDO request:ZDOCmd.Mgmt_Permit_Joining_rsp([<Status.SUCCESS: 0>])
[0x0000](CC1352/CC2652, Z-Stack 3.30+ (build 20211217)): started initialization
[0x0000:ZDO](CC1352/CC2652, Z-Stack 3.30+ (build 20211217)): 'async_initialize' stage succeeded
[0x0000:1:0x0000]: initializing channel: from_cache: False
[0x0000:1:0x0000]: finished channel initialization
[0x0000:2:0x0000]: initializing channel: from_cache: False
[0x0000:2:0x0000]: finished channel initialization
[0x0000:1:0x0000]: 'async_initialize' stage succeeded
[0x0000:2:0x0000]: 'async_initialize' stage succeeded
[0x0000](CC1352/CC2652, Z-Stack 3.30+ (build 20211217)): power source: Mains
[0x0000](CC1352/CC2652, Z-Stack 3.30+ (build 20211217)): completed initialization
[0x3179:1:0xfcc0] Received ZCL frame: b'\x1C\x5F\x11\x5C\x0A\x12\x01\x23\x3C\x00\x01\x00'
[0x3179:1:0xfcc0] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=True, is_reply=1, disable_default_response=1, reserved=0, *is_cluster=False, *is_general=True), manufacturer=4447, tsn=92, command_id=10, *is_reply=True)
[0x3179:1:0xfcc0] Decoded ZCL frame: ManufacturerSpecificCluster:Report_Attributes(attribute_reports=[Attribute(attrid=0x0112, value=TypeValue(type=uint32_t, value=65596))])
[0x3179:1:0xfcc0] Received command 0x0A (TSN 92): Report_Attributes(attribute_reports=[Attribute(attrid=0x0112, value=TypeValue(type=uint32_t, value=65596))])
[0x3179:1:0xfcc0] Attribute report received: 0x0112=65596
[0x82E4:1:0x0101] Received ZCL frame: b'\x18\x63\x0A\x55\x00\x21\x01\x00'
[0x82E4:1:0x0101] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=0, is_reply=1, disable_default_response=1, reserved=0, *is_cluster=False, *is_general=True), tsn=99, command_id=10, *is_reply=True)
[0x82E4:1:0x0101] Decoded ZCL frame: MultistateInputCluster:Report_Attributes(attribute_reports=[Attribute(attrid=0x0055, value=TypeValue(type=uint16_t, value=1))])
[0x82E4:1:0x0101] Received command 0x0A (TSN 99): Report_Attributes(attribute_reports=[Attribute(attrid=0x0055, value=TypeValue(type=uint16_t, value=1))])
[0x82E4:1:0x0101] Attribute report received: present_value=1
[0x82E4:1:0x0500]: Updated alarm state: 1
[0x36FE:1:0x0400] Received ZCL frame: b'\x18\xE9\x0A\x00\x00\x21\x24\x5E'
[0x36FE:1:0x0400] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=0, is_reply=1, disable_default_response=1, reserved=0, *is_cluster=False, *is_general=True), tsn=233, command_id=10, *is_reply=True)
[0x36FE:1:0x0400] Decoded ZCL frame: IlluminanceMeasurement:Report_Attributes(attribute_reports=[Attribute(attrid=0x0000, value=TypeValue(type=uint16_t, value=24100))])
[0x36FE:1:0x0400] Received command 0x0A (TSN 233): Report_Attributes(attribute_reports=[Attribute(attrid=0x0000, value=TypeValue(type=uint16_t, value=24100))])
[0x36FE:1:0x0400] Attribute report received: measured_value=24100
light.sz_decke_1: polling current state
[0xAF1B:11:0x0006] Sending request header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=False, is_reply=0, disable_default_response=0, reserved=0, *is_cluster=False, *is_general=True), tsn=95, command_id=<GeneralCommand.Read_Attributes: 0>, *is_reply=False)
[0xAF1B:11:0x0006] Sending request: Read_Attributes(attribute_ids=[0])
[0xAF1B:11:0x0008] Sending request header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=False, is_reply=0, disable_default_response=0, reserved=0, *is_cluster=False, *is_general=True), tsn=96, command_id=<GeneralCommand.Read_Attributes: 0>, *is_reply=False)
[0xAF1B:11:0x0008] Sending request: Read_Attributes(attribute_ids=[0])
[0x3179:1:0xfcc0] Received ZCL frame: b'\x1C\x5F\x11\x5D\x0A\xFC\x00\x10\x00'
[0x3179:1:0xfcc0] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=True, is_reply=1, disable_default_response=1, reserved=0, *is_cluster=False, *is_general=True), manufacturer=4447, tsn=93, command_id=10, *is_reply=True)
[0x3179:1:0xfcc0] Decoded ZCL frame: ManufacturerSpecificCluster:Report_Attributes(attribute_reports=[Attribute(attrid=0x00FC, value=TypeValue(type=Bool, value=<Bool.false: 0>))])
[0x3179:1:0xfcc0] Received command 0x0A (TSN 93): Report_Attributes(attribute_reports=[Attribute(attrid=0x00FC, value=TypeValue(type=Bool, value=<Bool.false: 0>))])
[0x3179:1:0xfcc0] Attribute report received: 0x00FC=<Bool.false: 0>
Device 0x3179 (54:ef:44:10:00:43:e9:2c) left the network
Device 0xc906 (54:ef:44:10:00:43:e9:2c) joined the network
Device 54:ef:44:10:00:43:e9:2c changed id (0xc906 => 0xc906)
[0xc906] Skipping initialization, device is fully initialized
Device is initialized <Device model='lumi.motion.ac02' manuf='LUMI' nwk=0xC906 ieee=54:ef:44:10:00:43:e9:2c is_initialized=True>
Checking quirks for LUMI lumi.motion.ac02 (54:ef:44:10:00:43:e9:2c)
Considering <class 'zhaquirks.xbee.xbee_io.XBeeSensor'>
Fail because endpoint list mismatch: {232, 230} {1}
Considering <class 'zhaquirks.xbee.xbee3_io.XBee3Sensor'>
Fail because endpoint list mismatch: {232, 230} {1}
Considering <class 'zhaquirks.smartthings.tag_v4.SmartThingsTagV4'>
Fail because device_type mismatch on at least one endpoint
Considering <class 'zhaquirks.smartthings.multi.SmartthingsMultiPurposeSensor'>
Fail because device_type mismatch on at least one endpoint
Considering <class 'zhaquirks.netvox.z308e3ed.Z308E3ED'>
Fail because device_type mismatch on at least one endpoint
Considering <class 'zhaquirks.gledopto.soposhgu10.SoposhGU10'>
Fail because endpoint list mismatch: {11, 13} {1}
Considering <class 'bellows.zigbee.application.EZSPCoordinator'>
Fail because device_type mismatch on at least one endpoint
[... 83 lines cut ...]
[0xC906:1:0x0000]: finished channel initialization
[0xC906:1:0xfcc0]: initializing channel: from_cache: False
[0xC906:1:0xfcc0]: finished channel initialization
[0xC906:1:0x0019]: initializing channel: from_cache: False
[0xC906:1:0x0019]: finished channel initialization
[0xC906:1:0x0000] Received ZCL frame: b'\x18\x00\x0A\x01\x00\x20\x05\x05\x00\x42\x10\x6C\x75\x6D\x69\x2E\x6D\x6F\x74\x69\x6F\x6E\x2E\x61\x63\x30\x32'
[0xC906:1:0x0000] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=0, is_reply=1, disable_default_response=1, reserved=0, *is_cluster=False, *is_general=True), tsn=0, command_id=10, *is_reply=True)
[0xC906:1:0x0000] Decoded ZCL frame: Basic:Report_Attributes(attribute_reports=[Attribute(attrid=0x0001, value=TypeValue(type=uint8_t, value=5)), Attribute(attrid=0x0005, value=TypeValue(type=CharacterString, value='lumi.motion.ac02'))])
[0xC906:1:0x0000] Received command 0x0A (TSN 0): Report_Attributes(attribute_reports=[Attribute(attrid=0x0001, value=TypeValue(type=uint8_t, value=5)), Attribute(attrid=0x0005, value=TypeValue(type=CharacterString, value='lumi.motion.ac02'))])
[0xC906:1:0x0000] Attribute report received: app_version=5, model='lumi.motion.ac02'
[0xC906:1:0x0001] Received ZCL frame: b'\x18\x67\x01\x20\x00\x00\x20\x1F\x21\x00\x86'
[0xC906:1:0x0001] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=0, is_reply=1, disable_default_response=1, reserved=0, *is_cluster=False, *is_general=True), tsn=103, command_id=1, *is_reply=True)
[0xC906:1:0x0001] Decoded ZCL frame: PowerConfiguration:Read_Attributes_rsp(status_records=[ReadAttributeRecord(attrid=0x0020, status=<Status.SUCCESS: 0>, value=TypeValue(type=uint8_t, value=31)), ReadAttributeRecord(attrid=0x0021, status=<Statu [... cut ...]
[0xC906:1:0x0001] Decoded ZCL frame: PowerConfiguration:Read_Attributes_rsp(status_records=[ReadAttributeRecord(attrid=0x0031, status=<Status.UNSUPPORTED_ATTRIBUTE: 134>), ReadAttributeRecord(attrid=0x0033, status=<Status.UNSUPPORTED_ATTRIBUTE: 134>)])
[0xC906:1:0x0001]: finished channel initialization
[0xC906:1:0x0001]: 'async_initialize' stage succeeded
[0xC906:1:0x0003]: 'async_initialize' stage succeeded
[0xC906:1:0x0000]: 'async_initialize' stage succeeded
[0xC906:1:0xfcc0]: 'async_initialize' stage succeeded
```

Tropaion, 2022-04-15:
@madface303 Can you post the device signature? Without it, we can't really help much.

madface303, 2022-04-15:
Oh sorry, i'm really new to this.

```
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4447, maximum_buffer_size=127, maximum_incoming_transfer_size=100, server_mask=11264, maximum_outgoing_transfer_size=100, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0107",
      "in_clusters": [
        "0x0000",
        "0x0001",
        "0x0003",
        "0xfcc0"
      ],
      "out_clusters": [
        "0x0003",
        "0x0019",
        "0xfcc0"
      ]
    }
  },
  "manufacturer": "LUMI",
  "model": "lumi.motion.ac02",
  "class": "zigpy.device.Device"
}
```

Is it that what you need?

Tropaion, 2022-04-15:
Yes, that's what I need 😄.
Your device is a bit weird, may need a bit of testing.
Try if this custom quirk works for your device: [motion_ac02.zip](https://github.com/zigpy/zha-device-handlers/files/8497532/motion_ac02.zip)

madface303, 2022-04-16:
Thank you very much. Just tried but it doesn't work.
I have to pair it twice (the first time it hangs at the initialize stage in HA). There are now 2 new sensors (ias_zone and occupancy, i know the from my old aqara motion sensors), but non of them changes if there is motion.

Tropaion, 2022-04-16:
Did you check if the quirk applies? Did you get some error in the logs?

Tropaion, 2022-04-16:
@madface303 Does the battery status work?
@MattWestb Do you know something about this? The device is very weird, it doesn't have a ```OccupancySensing``` input cluster, could it be the information is handled via the manufacturer cluster ```0xFCC0```?

MattWestb, 2022-04-16:
Xiaomi / Aquara is having there own protocol they is using for some functions but i have dot digging in it but you can looking in the xiaomi and aquara quirk folder and i think you is finding most in the init and also in other devices how they have doing all things we knowing (Xiaomi was one large PITA before tuya was taking over with there MCU devices).

madface303, 2022-04-16:
@Tropaion
No, there is a battery sensor but it's state is unknown. I attached a picture of the entities in HA.
Debug Logs of the pairing process is attached too
[image]
[zha_log.zip](https://github.com/zigpy/zha-device-handlers/files/8500101/zha_log.zip)
.

Tropaion, 2022-04-16:
@madface303 I looks like this will be trial and error.
Since this device is similar to an already existing one, lets try this first: [motion_ac02.zip](https://github.com/zigpy/zha-device-handlers/files/8500686/motion_ac02.zip)

madface303, 2022-04-17:
Thank you very much, but  unfortunately it is  also not working.
Now i have another sensor with illumance which has 302lx and is not changing like other sensors. 
[image]

I found another thread for this device for Zigbee2MQTT. I do not know much about zigbee behind the scenes, maybe it can help you? [https://github.com/Koenkk/zigbee2mqtt/issues/11990](https://github.com/Koenkk/zigbee2mqtt/issues/11990)

And do you need logfiles from me again? 
Thanks for your help!

ScottG489, 2022-04-19:
Btw, this devices is working with zigbee2mqtt. Hopefully this issue can provide some information you'll find helpful:
https://github.com/Koenkk/zigbee2mqtt/issues/11990

sklosinski, 2022-04-29:
Noob question... How does this update get pulled in to zha?

TheJulianJES, 2022-04-30:
The sensor isn't supported yet. If it's supported eventually, the update will arrive through a normal Home Assistant Core update.

(It's also possible to temporarily install a custom quirk when it's done, but that can break stuff too.)

MrRagga-, 2022-05-01:
Hi,
what is the status? I've just bought the device and I am also familar with python. I could help develop it further, if needed?

dmulcahey, 2022-05-01:
There is a or open in draft. Everything works except the configuration

MrRagga-, 2022-05-01:
My signature looks a bit different, so I am posting it, too:

```
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4447, maximum_buffer_size=127, maximum_incoming_transfer_size=100, server_mask=11264, maximum_outgoing_transfer_size=100, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0107",
      "in_clusters": [
        "0x0000",
        "0x0001",
        "0x0003",
        "0x0400",
        "0x0406",
        "0x0500",
        "0xfcc0"
      ],
      "out_clusters": [
        "0x0003",
        "0x0019",
        "0xfcc0"
      ]
    }
  },
  "manufacturer": "LUMI",
  "model": "lumi.motion.ac02",
  "class": "motion_ac02.LumiMotionAC02"
}
```

dmulcahey, 2022-05-01:
Hrm… there must be a mode setting for these then. I tried the well known one and it didn’t work… maybe I did something wrong. Does yours just work??

MrRagga-, 2022-05-01:
I added the `motion_ac02.py`from your draft PR and added it to the /config/custom_zha_quirks/ folder, changed the configuration.yaml to load it and restarted hass.

```
zha:
  custom_quirks_path: /config/custom_zha_quirks/
```

See the screenshot in the PR. ias_zone, illuminance and occupancy looks ok, but the battery is not displayed properly.
Sensors are looking good and seem to work.

dmulcahey, 2022-05-01:
My quirk won’t apply to your device. So that means yours works ootb.

dmulcahey, 2022-05-01:
Never mind… you posted the signature after the quirk was applied

MrRagga-, 2022-05-02:
I am a bit confused now. Did I apply your quirk now or not?

MrRagga-, 2022-05-02:
Btw, battery is also displayed now. Looks good.

dmulcahey, 2022-05-02:
Yes you did.

LCMilstein, 2022-05-02:
FWIW, I installed the custom quirk, which is getting picked up by ZHA, and the device pairs, but doesn't work.
I see the following:

- Power: unknown
- ias_zone: Clear
- Occupancy: Clear

And this happens even if I press the physical button to wake it up, and no matter what I waive in front of the sensor itself.  There's also "Identify: Press" which does nothing, and 2 unshown diagnostic entities. I've paired and re-paired a few times with the same results. I have NOT cleared __pycache__ as I'm not sure if that will mess up the working devices, but will try if that's necessary.

LCMilstein, 2022-05-02:
@dmulcahey I'm using the Nortek HubZ USB stick (HUSBZB-1).  It's not perfect, but I have a lot of other Aqara devices paired with it successfully.

dmulcahey, 2022-05-02:
And what version of HA?

LCMilstein, 2022-05-02:
> And what version of HA?

2022.4.7

dmulcahey, 2022-05-02:
Have you done anything with custom deps or anything wacky? I’m also using a Nortek stick and it just works once quirked. I’m out of ideas…

LCMilstein, 2022-05-02:
Nope. Nothing. I'm running Home Assistant in Docker on Synology, and have
HACS with some repositories, but nothing that should impact ZHA. This is
the ONLY custom quirk I have installed. I have been trying to figure out if
the Nortek is on the latest firmware, but I've only had it a handful of
months and it said it was the latest when I got it... Should I be trying to
update that... would I use the instructions here (
https://github.com/walthowd/husbzb-firmware)?

LCMilstein, 2022-05-03:
UPDATE: I deleted the chain quirk and folder asking with the reference to the zha folder in the config. Then restarted ha twice. Then downloaded the quirk again and started fresh with install and pairing... It worked! It never reached the screen with the yellow and green icons, but the sensors are working when I look at the device or entities in developer tools or configuration devices. 

I can't set sensitivity levels, but this is already fantastic!

brettcrane, 2022-05-06:
I'm new to this process and have the Aqara P1 sitting unused on my desk, what are typical timelines for this to make it into HA Core? It sounds like the quirk is working at this point.

LCMilstein, 2022-05-06:
@dmulcahey Amazing work you're doing. Do you have a new zip of the custom quirk you'd like us to test?

Korni22, 2022-05-12:
The quirk from @Tropaion looks pretty good to me as well.

Even the illuminance sensor seems to work for me.

Just like @LCMilstein the sensitivity levels would be amazing, no idea what that takes to get to work though.

dmulcahey, 2022-05-12:
> @dmulcahey Amazing work you're doing. Do you have a new zip of the custom quirk you'd like us to test?

The linked PR has the current code. This device may require a HA change for the custom interval support… I have it working just not sure if I like it

LCMilstein, 2022-05-12:
> > @dmulcahey Amazing work you're doing. Do you have a new zip of the custom quirk you'd like us to test?
> 
> The linked PR has the current code. This device may require a HA change for the custom interval support… I have it working just not sure if I like it

Apologies for the dumb question, but I am only somewhat technical and not super familiar with the capabilities here on GitHub... I have gone to your pull request, but when I click on the second link, it just takes me to a description page; there's no file there to download. Am I supposed to copy the code and save it to a file using my editor directly? Has the .zip file you linked to above been updated so I can just download that again (even though it has the same name it did before your latest additions)? OR should I just wait until it gets incorporated into HA's update in a future release?

dmulcahey, 2022-05-16:
> I installed the linked P1 quirk and was successfully using the P1 to trigger LED lights when entering a room. This was with HA OS 7.6 installed.
> 
> This morning, I upgraded to HA OS 8.0 and now the P1 will not initiate the automations. The P1 still shows correctly detecting motion, but the automations no longer start. Nothing else in the system changed other than allowing the OS upgrade.
> 
> If I trigger the P1 such that it shows "detected", I can manually run the automations and they function properly. However, as mentioned above, the P1 will not trigger the automations "automatically". I edited the automations to re-search/reconnect with the P1 but no joy.
> 
> Any thoughts?

Nothing on our end can cause that.

MollyrH, 2022-05-16:
Acknowledging that I'm not even close to an expert, but it seems coincidental that the system has been working perfectly for days and even 5 minutes prior to updating to OS 8.0.  Upon the system reboot of OS 8.0, all 5 associated automations stopped working.

Anto79-ops, 2022-05-16:
hi, watching this thread closely cause I have a few of the P1s in the box, and waiting for the official integration in ZHA (sorry, I know that doesn't help). But have you tried checking automation trace to be sure if it’s a trigger or condition issue? Also, check the HA logs at around the time where the automation sould have worked.

MollyrH, 2022-05-16:
Anto79-ops - good suggestions on trace and David was correct, there was not anything in the update that caused the problem - just something dumb I did.  So much to learn - thank you both for responding so quickly.

MollyrH, 2022-05-16:
BTW - have you installed the quirk that David linked?  It works well and provides the motion, occupancy, battery functions correctly.  Cannot change the refresh time yet but I guess that’s coming..

Best regards,
Bobby Hobbs

Anto79-ops, 2022-05-17:
quick question for those using the quirk, there was a request for the light detection (lux) part to operate without the motion. Is this working in the quirk and if so, how often does the light sensor report? thanks!

LCMilstein, 2022-05-17:
After over a week of usage, I'm finding 2 challenges with the version of the device so far:

1. In my office, if I use this device to control lights, the lights go out no matter how much I'm moving in my seat, and I have to approach the device (get within a few feet) for the lights to be triggered to go back on. I'm assuming this has to do with the update interval and/or the range. I do NOT have this issue when using a Lutron Caseta Motion Sensor; and
2. In my hallway, I am seeing occupancy "detected" for a LONG time after I've left the hall, which I also assume has to do with the update interval (I guess they're different for motion and occupancy?).

diederik-maes, 2022-05-23:
I got 2 of these today. Added 1 to ZHA without issues but indeed it shows nothing, no motion and battery status unknown. Looking forward to seeing this been added to ZHA and configurable.

closed 2022-05-25 by dmulcahey

Jeppedy, 2022-06-03:
I removed and readded this device after the official release.  There are entities called opple_cluster, which I've rather determined what they are, but detection time doesn't seem to do anything, I don't know what the "switch" does (it auto resets after a fixed period of time) and no lux indictator.
Whassup?

TheJulianJES, 2022-06-03:
@Jeppedy There seems to be a bug where the lux sensor currently doesn't initialize, so it won't work for now.

Regarding changing the settings: you'll need to change the setting (detection interval, sensitivity or the trigger LED) and then immediately press the button on the motion sensor once, so it updates the setting.

You'll find out what configuration entity does what at the end of the entity name/id by default.

karthik-krishnan, 2022-06-03:
I found a workaround for the detection interval setting change not taking effect.  If I use the developer setting page to change the setting it stays. Yet to test if it’s taking effect or not. 

Best regards,
Karthik. 
https://karthiks.in

TheJulianJES, 2022-06-03:
It should also work that way. I'm able to properly use the configuration setting for the detection interval. Just have to press the button on the sensor shortly after hitting enter/clicking out of the field/changing the number.

philibu, 2022-06-17:
Hello,

I have an issue with the illuminance. It does not show up for the latest P1 that I have connected to my home assistant. It worked with a previous install. Is there a bug ?

Thanks

MollyrH, 2022-10-11:
Yup - you’re correct.  Something dumb on my part…. 😞.  So much to learn….

Sorry for the false alarm and thanks for all the effort you are putting into HA.  It’s a STEEP learning curve, but so much better than others products in the market.

Best regards,
Bobby Hobbs

karthik-krishnan, 2022-10-11:
Kind of defeats the purpose of having it use a web Ui without having to be physically next to the device when changing config 

Best regards,
Karthik. 
https://karthiks.in

aagit, 2023-04-18:
Hi,

Latest HA ZHA stock uses "Quirk: zhaquirks.xiaomi.aqara.motion_ac02.LumiMotionAC02" and everything worked fine with the original fw ver 6 that the device come with.

There was only a GUI issue where the GUI shows the led indicator off in the GUI, but it's always on after ZHA pairing. 

Writing "0" by hand in the motion indicator attribute in the "manage zigbee device" menu works around the led GUI issue.

Then I had the idea to test ZHA OTA,  sending the update request from zha-toolkit made ZHA upload the fw version 10, but with version 10 the whole OppleCluster seem to have stopped working so I cannot turn off the led indicator on motion anymore.

I had to commit the qcow2 image into its base, restart the VM with a new qcow2 cow, edit this file in /mnt/*merged*/ like this:

```
diff --git a/zigpy/zcl/clusters/general.py b/general.py
index 5cf05dc..a4d1b9f 100644
--- a/zigpy/zcl/clusters/general.py
+++ b/general.py
@@ -1873,6 +1873,7 @@ class Ota(Cluster):
             should_update = img.should_update(
                 manufacturer_id, image_type, current_file_version, hardware_version
             )
+            should_update=True
             self.debug(
                 "OTA image version: %s, size: %s. Update needed: %s",
                 img.version,
@@ -1887,7 +1888,7 @@ class Ota(Cluster):
                     foundation.Status.SUCCESS,
                     img.key.manufacturer_id,
                     img.key.image_type,
-                    img.version,
+                    11,
                     img.header.image_size,
                     tsn=tsn,
                 )
@@ -1933,7 +1934,7 @@ class Ota(Cluster):
         img = await self.endpoint.device.application.ota.get_ota_image(
             manufacturer_id, image_type, model
         )
-        if img is None or img.version != file_version:
+        if img is None:
             self.debug("OTA image is not available")
             await self.image_block_response(foundation.Status.ABORT, tsn=tsn)
             return
```

ha core restart

Then issue the ZHA Toolkit: ota_notify again.

Finally shutdown the VM and delete the qcow2 cow, create a new fresh cow from the untouched base, and start the VM,( to rollback the above patch).

The OppleCluster works again now on the fw ver 6, so I could turn off the led again which I really don't like to flash by night, the sensitivity setting also works again.

I guess one should be more careful with the  OTA updates, but at least I managed to downgrade.

SkyDroVV, 2024-04-04:
I can,t connect Aqara P1 with ZHA in HA with Conbee III

[Feeding watchdog
Received a packet: ZigbeePacket(timestamp=datetime.datetime(2024, 4, 4, 18, 35, 51, 960322, tzinfo=datetime.timezone.utc), src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x6D6F), src_ep=0, dst=AddrModeAddress(addr_mode=<AddrMode.Broadcast: 15>, address=<BroadcastAddress.ALL_DEVICES: 65535>), dst_ep=0, source_route=None, extended_timeout=False, tsn=None, profile_id=0, cluster_id=19, data=Serialized[b'<omJ*\xb6\x00\x10D\xefT\x80'], tx_options=<TransmitOptions.NONE: 0>, radius=0, non_member_radius=0, lqi=197, rssi=-59)
New device 0x6d6f (54:ef:44:10:00:b6:2a:4a) joined the network
[0x6d6f] Scheduling initialization
Received frame on uninitialized device <Device model=None manuf=None nwk=0x6D6F ieee=54:ef:44:10:00:b6:2a:4a is_initialized=False> from ep 0 to ep 0, cluster 19: Serialized[b'<omJ*\xb6\x00\x10D\xefT\x80']
[0x6d6f:zdo] ZDO request ZDOCmd.Device_annce: [0x6D6F, 54:ef:44:10:00:b6:2a:4a, 128]
Tries remaining: 5
[0x6d6f] Requesting 'Node Descriptor'
[0x6d6f] Extending timeout for 0x01 request
Feeding watchdog
Tries remaining: 4
[0x6d6f] Requesting 'Node Descriptor'
[0x6d6f] Extending timeout for 0x02 request
Tries remaining: 3
[0x6d6f] Requesting 'Node Descriptor'
[0x6d6f] Extending timeout for 0x03 request
Tries remaining: 2
[0x6d6f] Requesting 'Node Descriptor'
[0x6d6f] Extending timeout for 0x04 request
[0xFDF7](TS0202): Attempting to checkin with device - missed checkins: 1
[0xFDF7:1:0x0000] Sending request header: ZCLHeader(frame_control=FrameControl<0x00>(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=False, direction=<Direction.Client_to_Server: 0>, disable_default_response=0, reserved=0, *is_cluster=False, *is_general=True), tsn=1, command_id=<GeneralCommand.Read_Attributes: 0>, *direction=<Direction.Client_to_Server: 0>)
[0xFDF7:1:0x0000] Sending request: Read_Attributes(attribute_ids=[4])
[0xfdf7] Extending timeout for 0x01 request
Tries remaining: 1
[0x6d6f] Requesting 'Node Descriptor'
[0x6d6f] Extending timeout for 0x05 request
Feeding watchdog
[0x2886](TS130F): Device seen - marking the device available and resetting counter
[0x2886](TS130F): Update device availability -  device available: True - new availability: True - changed: False
[0xB4A7](50131): Device seen - marking the device available and resetting counter
[0xB4A7](50131): Update device availability -  device available: True - new availability: True - changed: False
Feeding watchdog
[0xFDF7](TS0202): Attempting to checkin with device - missed checkins: 2
[0xFDF7:1:0x0000] Sending request header: ZCLHeader(frame_control=FrameControl<0x00>(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=False, direction=<Direction.Client_to_Server: 0>, disable_default_response=0, reserved=0, *is_cluster=False, *is_general=True), tsn=2, command_id=<GeneralCommand.Read_Attributes: 0>, *direction=<Direction.Client_to_Server: 0>)
[0xFDF7:1:0x0000] Sending request: Read_Attributes(attribute_ids=[4])
[0xfdf7] Extending timeout for 0x02 request
Feeding watchdog
Feeding watchdog
[0x2886](TS130F): Device seen - marking the device available and resetting counter
[0x2886](TS130F): Update device availability -  device available: True - new availability: True - changed: False
[0xB4A7](50131): Device seen - marking the device available and resetting counter
[0xB4A7](50131): Update device availability -  device available: True - new availability: True - changed: False
[0xFDF7](TS0202): last_seen is 100188.31617498398 seconds ago and ping attempts have been exhausted, marking the device unavailable
[0xFDF7](TS0202): Update device availability -  device available: False - new availability: False - changed: False
Feeding watchdog
Feeding watchdog
Received a packet: ZigbeePacket(timestamp=datetime.datetime(2024, 4, 4, 18, 39, 12, 529634, tzinfo=datetime.timezone.utc), src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0xB4A7), src_ep=1, dst=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), dst_ep=1, source_route=None, extended_timeout=False, tsn=None, profile_id=260, cluster_id=8, data=Serialized[b'\x08,\n\x00\x00 \x00'], tx_options=<TransmitOptions.NONE: 0>, radius=0, non_member_radius=0, lqi=197, rssi=-59)
[0xB4A7:1:0x0008] Received ZCL frame: b'\x08,\n\x00\x00 \x00'
[0xB4A7:1:0x0008] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl<0x08>(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=0, direction=<Direction.Server_to_Client: 1>, disable_default_response=0, reserved=0, *is_cluster=False, *is_general=True), tsn=44, command_id=10, *direction=<Direction.Server_to_Client: 1>)
[0xB4A7:1:0x0008] Decoded ZCL frame: LevelControl:Report_Attributes(attribute_reports=[Attribute(attrid=0x0000, value=TypeValue(type=uint8_t, value=0))])
[0xB4A7:1:0x0008] Received command 0x0A (TSN 44): Report_Attributes(attribute_reports=[Attribute(attrid=0x0000, value=TypeValue(type=uint8_t, value=0))])
[0xB4A7:1:0x0008] Attribute report received: current_level=0
[0xB4A7:1:0x0008]: received attribute: 0 update with value: 0
[0xB4A7:1:0x0008] Sending reply header: ZCLHeader(frame_control=FrameControl<0x10>(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=False, direction=<Direction.Client_to_Server: 0>, disable_default_response=1, reserved=0, *is_cluster=False, *is_general=True), tsn=44, command_id=<GeneralCommand.Default_Response: 11>, *direction=<Direction.Client_to_Server: 0>)
[... cut ...]
[0x2886](TS130F): Update device availability -  device available: True - new availability: True - changed: False
[0xFDF7](TS0202): last_seen is 100248.31722521782 seconds ago and ping attempts have been exhausted, marking the device unavailable
[0xFDF7](TS0202): Update device availability -  device available: False - new availability: False - changed: False
Feeding watchdog
[0xB4A7](50131): Device seen - marking the device available and resetting counter
[0xB4A7](50131): Update device availability -  device available: True - new availability: True - changed: False
]()

PR #1541 (2022-05-25): Add support for Aqara P1 motion sensor
**TODO:**

- [x] motion sensitivity
- [x] detection interval
- [ ] illuminance update w/o motion

Fixes: #1498