zigpy/zha-device-handlers #1499: [Device Support Request] Move into production climate.py one code line change

sKalach, 2022-04-15:
I am using \custom_components\zha folder for LUX KONOZ Thermostat with modified climate.py (manifest.json with added version).
As you suggested in ticket #1054 I do modify climate.py script by changing line 
manufacturers="Zen Within",

onto

manufacturers={"Zen Within", "LUX"},

LUX thermostat works fine under "Zen Within" specification, not everything, but at least what I needed the most. This approach was fine from October 2021 till version 2022.4.0 (.0 - .4).

It looks like now ZHA Custom Component does not work anymore for me. I have several errors which I am not able to fix and I have to continue working on version 2022.3.X.

I am using RPi 4 and ConBee II.

**Describe the solution you'd like**
Can we do the permanent change in climate.py by updating line manufacturers="Zen Within", onto manufacturers={"Zen Within", "LUX"},? 

**Device signature - this can be acquired by removing the device from ZHA and pairing it again from the add devices screen. Be sure to add the entire content of the log panel after pairing the device to a code block below this line.**
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4610, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0301",
      "in_clusters": [
        "0x0000",
        "0x0001",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0020",
        "0x0201",
        "0x0202",
        "0x0204",
        "0x0b05"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "LUX",
  "model": "KONOZ",
  "class": "zigpy.device.Device"
}

**Additional context**
If this change is promoted to the production it should simplify error troubleshooting by removing one dependency on Custom Components and also with the new version installation process. Not to mention, it will add an additional new supported device. 

This ticket is not about errors I have for version 2022.4.X, but in case it can be useful, here are some errors I've observed:

--------------------------------------------------------------------------------------------------------------------------------------

Logger: homeassistant
Source: config_entries.py:750
First occurred: 9:41:46 AM (2 occurrences)
Last logged: 9:41:46 AM

Error doing job: Task exception was never retrieved
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 743, in async_create_flow
    integration.get_platform("config_flow")
  File "/usr/src/homeassistant/homeassistant/loader.py", line 572, in get_platform
    cache[full_name] = self._import_platform(platform_name)
  File "/usr/src/homeassistant/homeassistant/loader.py", line 589, in _import_platform
    return importlib.import_module(f"{self.pkg_path}.{platform_name}")
  File "/usr/local/lib/python3.9/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1030, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1007, in _find_and_load
  File "<frozen importlib._bootstrap>", line 972, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 228, in _call_with_frames_removed
  File "<frozen importlib._bootstrap>", line 1030, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1007, in _find_and_load
  File "<frozen importlib._bootstrap>", line 986, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 680, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 850, in exec_module
  File "<frozen importlib._bootstrap>", line 228, in _call_with_frames_removed
  File "/config/custom_components/zha/__init__.py", line 17, in <module>
    from . import api
  File "/config/custom_components/zha/api.py", line 63, in <module>
    from .core.group import GroupMember
  File "/config/custom_components/zha/core/__init__.py", line 4, in <module>
    from .device import ZHADevice
  File "/config/custom_components/zha/core/device.py", line 29, in <module>
    from . import channels, typing as zha_typing
  File "/config/custom_components/zha/core/channels/__init__.py", line 29, in <module>
    from .. import (
  File "/config/custom_components/zha/core/discovery.py", line 20, in <module>
    from .. import (  # noqa: F401 pylint: disable=unused-import,
  File "/config/custom_components/zha/alarm_control_panel.py", line 6, in <module>
    from homeassistant.components.alarm_control_panel import (
ImportError: cannot import name 'AlarmControlPanelEntityFeature' from 'homeassistant.components.alarm_control_panel' (/usr/src/homeassistant/homeassistant/components/alarm_control_panel/__init__.py)

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/data_entry_flow.py", line 203, in async_init
    flow, result = await task
  File "/usr/src/homeassistant/homeassistant/data_entry_flow.py", line 221, in _async_init
    flow = await self.async_create_flow(handler, context=context, data=data)
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 750, in async_create_flow
    raise data_entry_flow.UnknownHandler
homeassistant.data_entry_flow.UnknownHandler

Error occurred loading configuration flow for integration zha: cannot import name 'AlarmControlPanelEntityFeature' from 'homeassistant.components.alarm_control_panel' (/usr/src/homeassistant/homeassistant/components/alarm_control_panel/__init__.py)

Logger: homeassistant.config_entries
Source: config_entries.py:745
First occurred: 8:27:13 AM (2 occurrences)
Last logged: 8:27:13 AM

Error occurred loading configuration flow for integration zha: cannot import name 'CALLABLE_T' from 'homeassistant.core' (/usr/src/homeassistant/homeassistant/core.py)

Component error: zha - cannot import name 'AlarmControlPanelEntityFeature' from 'homeassistant.components.alarm_control_panel' (/usr/src/homeassistant/homeassistant/components/alarm_control_panel/__init__.py)

Logger: homeassistant.components.websocket_api.http.connection
Source: components/websocket_api/connection.py:96
Integration: Home Assistant WebSocket API (documentation, issues)
First occurred: 8:25:49 AM (4 occurrences)
Last logged: 8:25:49 AM

[548136933936] Received invalid command: zha/devices/reconfigure
[548136933936] Received invalid command: energy/info

Invalid config for [automation]: Integration 'zha' does not support device automation triggers. Got OrderedDict([('default_config', {}), ('homeassistant', OrderedDict([('allowlist_external_dirs', ['/config']), ('media_dirs', OrderedDict([('media', '/media')]))])), ('recorder', OrderedDict([('purge_keep_days', 30), ('commit_interval', 1)])), ('tts', [OrderedDict([('platform', 'google_translate')])]), ('sun', {}), ('python_script', {}), ('group', OrderedDict([('presence_home', OrderedDict([('name', 'Presence Home'), ('entities', ['person.slawa', 'person.larysa'])])), ('water_sensors', Ordered.... (See /config/configuration.yaml, line 24).

-------------------------------------------------------------------------------------------------------------------------------------- 

Thank you.

TheJulianJES, 2022-04-17:
Could you re-download https://github.com/home-assistant/core/tree/rc (rc branch), possibly update your ``manifest.json``, delete the old custom_components ZHA folder and copy the new one from ``components/zha``, then make the change in ``climate.py`` again and update to the latest Home Assistant 2022.4.5?
This would just be to confirm the patch still works, so it can be implemented into Home Assistant core itself.

sKalach, 2022-04-17:
Thank you for the quick response! I do not see any errors after I'd done below steps.

I did everything as you guided:
- Download Release Candidate ZHA component (before I was taking Dev. one) 
- Adde LUX into climat.py
- Added version 2021.9.1 to manifest.json
- Dropped existing ZHA folder and copied new RC one.
- update core to 2022.4.5

It works!

Thank you.

TheJulianJES, 2022-04-17:
Thank you for testing it again! I'll mark the PR with the changes as "ready".
For now, you should be able to continue your custom component until these changes (hopefully) make it into Home Assistant Core.

closed 2022-04-24 by dmulcahey