zigpy/zha-device-handlers #1500: [Device Support Request] _TZ3000_amdymr7l TS011F (BlitzWolf BW-SHP13)

EagleStClair, 2022-04-15:
**Is your feature request related to a problem? Please describe.**
The device is detected and adds on_off, electrical_measurement, rms_current, rms_voltage, smartenergy_metering summation_delivered. But nothing is updated and on_off can't communicate with the plug. After the default timeout the device becomes unavailable.
Uses zhaquirks.tuya.ts011_plug.Plug quirk by default.

**Describe the solution you'd like**
Device to be detected and report sensors with working on_off.

**Device signature - this can be acquired by removing the device from ZHA and pairing it again from the add devices screen. Be sure to add the entire content of the log panel after pairing the device to a code block below this line.**
`
[0x0214:11:0x0006] Sending request header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=False, is_reply=0, disable_default_response=0, reserved=0, *is_cluster=False, *is_general=True), tsn=2, command_id=<GeneralCommand.Read_Attributes: 0>, *is_reply=False)
[0x0214:11:0x0006] Sending request: Read_Attributes(attribute_ids=[0])
[0x0214:11:0x0006] Received ZCL frame: b'\x18\x02\x01\x00\x00\x00\x10\x00'
[0x0214:11:0x0006] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=0, is_reply=1, disable_default_response=1, reserved=0, *is_cluster=False, *is_general=True), tsn=2, command_id=1, *is_reply=True)
[0x0214:11:0x0006] Decoded ZCL frame: OnOff:Read_Attributes_rsp(status_records=[ReadAttributeRecord(attrid=0x0000, status=<Status.SUCCESS: 0>, value=TypeValue(type=Bool, value=<Bool.false: 0>))])
[0x0214:11:0x0008] Sending request header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=False, is_reply=0, disable_default_response=0, reserved=0, *is_cluster=False, *is_general=True), tsn=4, command_id=<GeneralCommand.Read_Attributes: 0>, *is_reply=False)
[0x0214:11:0x0008] Sending request: Read_Attributes(attribute_ids=[0])
[0x0214:11:0x0008] Received ZCL frame: b'\x18\x04\x01\x00\x00\x00 L'
[0x0214:11:0x0008] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=0, is_reply=1, disable_default_response=1, reserved=0, *is_cluster=False, *is_general=True), tsn=4, command_id=1, *is_reply=True)
[0x0214:11:0x0008] Decoded ZCL frame: LevelControl:Read_Attributes_rsp(status_records=[ReadAttributeRecord(attrid=0x0000, status=<Status.SUCCESS: 0>, value=TypeValue(type=uint8_t, value=76))])
[0x0214:11:0x0008]: received attribute: 0 update with value: 76
New device 0x9e01 (a4:c1:38:5e:4c:91:b3:0a) joined the network
[0x9e01] Scheduling initialization
Received frame on uninitialized device <Device model=None manuf=None nwk=0x9E01 ieee=a4:c1:38:5e:4c:91:b3:0a is_initialized=False> from ep 0 to ep 0, cluster 19: b'\x00\x01\x9e\n\xb3\x91L^8\xc1\xa4\x8e'
[0x9e01:zdo] ZDO request ZDOCmd.Device_annce: [0x9E01, a4:c1:38:5e:4c:91:b3:0a, 142]
Tries remaining: 3
[0x9e01] Requesting 'Node Descriptor'
Tries remaining: 2
[0x9e01] Extending timeout for 0x06 request
Received frame on uninitialized device <Device model=None manuf=None nwk=0x9E01 ieee=a4:c1:38:5e:4c:91:b3:0a is_initialized=False> from ep 0 to ep 0, cluster 32770: b'\x06\x00\x01\x9e\x01@\x8eA\x11BB\x00\x00*B\x00\x00'
[0x9e01] Got Node Descriptor: NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)
[0x9e01] Discovering endpoints
Tries remaining: 3
Received frame on uninitialized device <Device model=None manuf=None nwk=0x9E01 ieee=a4:c1:38:5e:4c:91:b3:0a is_initialized=False> from ep 0 to ep 0, cluster 32773: b'\x08\x00\x01\x9e\x02\x01\xf2'
[0x9e01] Discovered endpoints: [1, 242]
[0x9e01] Initializing endpoints [<Endpoint id=1 in=[] out=[] status=<Status.NEW: 0>>, <Endpoint id=242 in=[] out=[] status=<Status.NEW: 0>>]
[0x9e01:1] Discovering endpoint information
Tries remaining: 3
[0x9e01] Delivery error for seq # 0x0a, on endpoint id 0 cluster 0x0004: message send failure
Received frame on uninitialized device <Device model=None manuf=None nwk=0x9E01 ieee=a4:c1:38:5e:4c:91:b3:0a is_initialized=False> from ep 0 to ep 0, cluster 32772: b'\n\x00\x01\x9e\x1e\x01\x04\x01\n\x01\x01\t\x03\x00\x04\x00\x05\x00\x06\x00\x02\x07\x04\x0b\x01\xe0\x00\xe0\x00\x00\x02\x19\x00\n\x00'
[0x9e01:zdo] ZDO request ZDOCmd.Simple_Desc_rsp: [<Status.SUCCESS: 0>, 0x9E01, SizePrefixedSimpleDescriptor(endpoint=1, profile=260, device_type=266, device_version=1, input_clusters=[3, 4, 5, 6, 1794, 2820, 57345, 57344, 0], output_clusters=[25, 10])]
[0x9e01:zdo] No handler for ZDO request:ZDOCmd.Simple_Desc_rsp([<Status.SUCCESS: 0>, 0x9E01, SizePrefixedSimpleDescriptor(endpoint=1, profile=260, device_type=266, device_version=1, input_clusters=[3, 4, 5, 6, 1794, 2820, 57345, 57344, 0], output_clusters=[25, 10])])
Received frame on uninitialized device <Device model=None manuf=None nwk=0x9E01 ieee=a4:c1:38:5e:4c:91:b3:0a is_initialized=False> from ep 1 to ep 1, cluster 0: b'\x08g\n\x01\x00 J\xe2\xff 5\xe4\xff \x01'
Ignoring message (b'08670a0100204ae2ff2035e4ff2001') on cluster 0: unknown endpoint or cluster id: 'No cluster ID 0x0000 on (a4:c1:38:5e:4c:91:b3:0a, 1)'
Tries remaining: 2
Received frame on uninitialized device <Device model=None manuf=None nwk=0x9E01 ieee=a4:c1:38:5e:4c:91:b3:0a is_initialized=False> from ep 0 to ep 0, cluster 32772: b'\x0c\x00\x01\x9e\x1e\x01\x04\x01\n\x01\x01\t\x03\x00\x04\x00\x05\x00\x06\x00\x02\x07\x04\x0b\x01\xe0\x00\xe0\x00\x00\x02\x19\x00\n\x00'
[0x9e01:1] Discovered endpoint information: SizePrefixedSimpleDescriptor(endpoint=1, profile=260, device_type=266, device_version=1, input_clusters=[3, 4, 5, 6, 1794, 2820, 57345, 57344, 0], output_clusters=[25, 10])
Unknown cluster 0xE001
Unknown cluster 0xE000
[0x9e01:242] Discovering endpoint information
Tries remaining: 3
Received frame on uninitialized device <Device model=None manuf=None nwk=0x9E01 ieee=a4:c1:38:5e:4c:91:b3:0a is_initialized=False> from ep 0 to ep 0, cluster 32772: b'\x0e\x00\x01\x9e\n\xf2\xe0\xa1a\x00\x00\x00\x01!\x00'
[0x9e01:242] Discovered endpoint information: SizePrefixedSimpleDescriptor(endpoint=242, profile=41440, device_type=97, device_version=0, input_clusters=[], output_clusters=[33])
[0x9E01:1:0x0000] Sending request header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=False, is_reply=0, disable_default_response=0, reserved=0, *is_cluster=False, *is_general=True), tsn=16, command_id=<GeneralCommand.Read_Attributes: 0>, *is_reply=False)
[0x9E01:1:0x0000] Sending request: Read_Attributes(attribute_ids=[4, 5])
[0x9E01:1:0x0000] Received ZCL frame: b'\x18\x10\x01\x04\x00\x00B\x10_TZ3000_amdymr7l\x05\x00\x00B\x06TS011F'
[0x9E01:1:0x0000] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=0, is_reply=1, disable_default_response=1, reserved=0, *is_cluster=False, *is_general=True), tsn=16, command_id=1, *is_reply=True)
[0x9E01:1:0x0000] Decoded ZCL frame: Basic:Read_Attributes_rsp(status_records=[ReadAttributeRecord(attrid=0x0004, status=<Status.SUCCESS: 0>, value=TypeValue(type=CharacterString, value='_TZ3000_amdymr7l')), ReadAttributeRecord(attrid=0x0005, status=<Status.SUCCESS: 0>, value=TypeValue(type=CharacterString, value='TS011F'))])
[0x9e01] Read model 'TS011F' and manufacturer '_TZ3000_amdymr7l' from <Endpoint id=1 in=[identify:0x0003, groups:0x0004, scenes:0x0005, on_off:0x0006, smartenergy_metering:0x0702, electrical_measurement:0x0B04, None:0xE001, None:0xE000, basic:0x0000] out=[ota:0x0019, time:0x000A] status=<Status.ZDO_INIT: 1>>
[... cut ...]
Device is initialized <Plug model='TS011F' manuf='_TZ3000_amdymr7l' nwk=0xA6AD ieee=a4:c1:38:5e:4c:91:b3:0a is_initialized=True>
[0xa6ad:zdo] ZDO request ZDOCmd.Device_annce: [0xA6AD, a4:c1:38:5e:4c:91:b3:0a, 142]
device - 42669:a4:c1:38:5e:4c:91:b3:0a entering async_device_initialized - is_new_join: False
device - 42669:a4:c1:38:5e:4c:91:b3:0a has been reset and re-added or its nwk address changed
skipping discovery for previously discovered device - 42669:a4:c1:38:5e:4c:91:b3:0a
`

**Additional context**
Add any other context or screenshots about the feature request here.

Tropaion, 2022-04-15:
Device signature please.

EagleStClair, 2022-04-15:
> Device signature please.

Not the part pasted above?

This maybe?
`{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x010a",
      "in_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0702",
        "0x0b04",
        "0xe000",
        "0xe001"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZ3000_amdymr7l",
  "model": "TS011F",
  "class": "zhaquirks.tuya.ts011f_plug.Plug"
}`

Tropaion, 2022-04-15:
In an unreadable format but thats it 😄

MattWestb, 2022-04-15:
Its looks some TS011F is not reporting status changes that norm Zigbee device is doing (also tuya ones) and need being pulled as Philips HUE and i think its not implanted in Zigpy / ZHA doing from the quirk.

EagleStClair, 2022-04-15:
> In an unreadable format but thats it 😄

Sorry, complete noob regarding github. The "add code" changes the format.

Better like this?
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x010a",
      "in_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0702",
        "0x0b04",
        "0xe000",
        "0xe001"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZ3000_amdymr7l",
  "model": "TS011F",
  "class": "zhaquirks.tuya.ts011f_plug.Plug"
}

EagleStClair, 2022-04-15:
No idea if it helps but I found that the same device needed to be added to zigbee2mqtt here recently:
[https://github.com/Koenkk/zigbee2mqtt/issues/11703](url)

Tropaion, 2022-04-15:
@EagleStClair Better, but you can also just edit your previous comment instead of making a new one.
It looks like your problem is out of my league, I think @MattWestb should be able to help better.

javicalle, 2022-04-16:
@EagleStClair I can't even read the log. Please, edit your comment and put the logs inside 3 apostrophes (`\``):
* https://docs.github.com/get-started/writing-on-github/working-with-advanced-formatting/creating-and-highlighting-code-blocks#fenced-code-blocks

Any error on the logs could point to the problem, so review if there is something relevant that can be related to the device (it seems that yours is device `0x9E01`).

Regards.

closed 2022-10-20 by github-actions[bot]