zigpy/zha-device-handlers #1502: [Device Support Request] Tuya ZG-102ZL / _TZE200_pay2byax

centershock, 2022-04-16:
**Is your feature request related to a problem? Please describe.**
The device is not supported, I request a full support of this device.

**Describe the solution you'd like**
Full support for the device, primarily to check "open" and "close" of a window / door.

**Device signature - this can be acquired by removing the device from ZHA and pairing it again from the add devices screen. Be sure to add the entire content of the log panel after pairing the device to a code block below this line.**
`New device 0x0df7 (a4:c1:38:4d:82:df:a6:b0) joined the network
[0x0df7] Scheduling initialization
Received frame on uninitialized device <Device model=None manuf=None nwk=0x0DF7 ieee=a4:c1:38:4d:82:df:a6:b0 is_initialized=False> from ep 0 to ep 0, cluster 19: b'\x00\xf7\r\xb0\xa6\xdf\x82M8\xc1\xa4\x80'
[0x0df7:zdo] ZDO request ZDOCmd.Device_annce: [0x0DF7, a4:c1:38:4d:82:df:a6:b0, 128]
Tries remaining: 3
[0x0df7] Requesting 'Node Descriptor'
Tries remaining: 2
[0x0df7] Extending timeout for 0x77 request
Received frame on uninitialized device <Device model=None manuf=None nwk=0x0DF7 ieee=a4:c1:38:4d:82:df:a6:b0 is_initialized=False> from ep 0 to ep 0, cluster 32770: b'w\x00\xf7\r\x02@\x80A\x11BB\x00\x00*B\x00\x00'
[0x0df7] Got Node Descriptor: NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)
[0x0df7] Discovering endpoints
Tries remaining: 3
[0x0df7] Extending timeout for 0x79 request
light.hue_runner_spot_1_1: polling current state
[0x3808:11:0x0006] Sending request header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=False, is_reply=0, disable_default_response=0, reserved=0, *is_cluster=False, *is_general=True), tsn=123, command_id=<GeneralCommand.Read_Attributes: 0>, *is_reply=False)
[0x3808:11:0x0006] Sending request: Read_Attributes(attribute_ids=[0])
Received frame on uninitialized device <Device model=None manuf=None nwk=0x0DF7 ieee=a4:c1:38:4d:82:df:a6:b0 is_initialized=False> from ep 0 to ep 0, cluster 32773: b'y\x00\xf7\r\x01\x01'
[0x0df7] Discovered endpoints: [1]
[0x0df7] Initializing endpoints [<Endpoint id=1 in=[] out=[] status=<Status.NEW: 0>>]
[0x0df7:1] Discovering endpoint information
Tries remaining: 3
[0x0df7] Extending timeout for 0x7d request
[0x3808:11:0x0006] Received ZCL frame: b'\x18{\x01\x00\x00\x00\x10\x00'
[0x3808:11:0x0006] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=0, is_reply=1, disable_default_response=1, reserved=0, *is_cluster=False, *is_general=True), tsn=123, command_id=1, *is_reply=True)
[0x3808:11:0x0006] Decoded ZCL frame: OnOff:Read_Attributes_rsp(status_records=[ReadAttributeRecord(attrid=0x0000, status=<Status.SUCCESS: 0>, value=TypeValue(type=Bool, value=<Bool.false: 0>))])
[0x3808:11:0x0008] Sending request header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=False, is_reply=0, disable_default_response=0, reserved=0, *is_cluster=False, *is_general=True), tsn=127, command_id=<GeneralCommand.Read_Attributes: 0>, *is_reply=False)
[0x3808:11:0x0008] Sending request: Read_Attributes(attribute_ids=[0])
[0x3808:11:0x0008] Received ZCL frame: b'\x18\x7f\x01\x00\x00\x00 \xfe'
[0x3808:11:0x0008] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=0, is_reply=1, disable_default_response=1, reserved=0, *is_cluster=False, *is_general=True), tsn=127, command_id=1, *is_reply=True)
[0x3808:11:0x0008] Decoded ZCL frame: LevelControl:Read_Attributes_rsp(status_records=[ReadAttributeRecord(attrid=0x0000, status=<Status.SUCCESS: 0>, value=TypeValue(type=uint8_t, value=254))])
[0x3808:11:0x0008]: received attribute: 0 update with value: 254
[0x3808:11:0x0300] Sending request header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=False, is_reply=0, disable_default_response=0, reserved=0, *is_cluster=False, *is_general=True), tsn=129, command_id=<GeneralCommand.Read_Attributes: 0>, *is_reply=False)
[0x3808:11:0x0300] Sending request: Read_Attributes(attribute_ids=[8, 7, 3, 4, 16386])
Received frame on uninitialized device <Device model=None manuf=None nwk=0x0DF7 ieee=a4:c1:38:4d:82:df:a6:b0 is_initialized=False> from ep 0 to ep 0, cluster 32772: b'}\x00\xf7\r\x12\x01\x04\x01\x02\x04\x01\x03\x01\x00\x00\x05\x00\x00\x02\x19\x00\n\x00'
[0x0df7:1] Discovered endpoint information: SizePrefixedSimpleDescriptor(endpoint=1, profile=260, device_type=1026, device_version=1, input_clusters=[1, 1280, 0], output_clusters=[25, 10])
[0x0DF7:1:0x0000] Sending request header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=False, is_reply=0, disable_default_response=0, reserved=0, *is_cluster=False, *is_general=True), tsn=131, command_id=<GeneralCommand.Read_Attributes: 0>, *is_reply=False)
[0x0DF7:1:0x0000] Sending request: Read_Attributes(attribute_ids=[4, 5])
[0x0df7] Extending timeout for 0x83 request
[0x3808:11:0x0300] Received ZCL frame: b'\x18\x81\x01\x08\x00\x000\x02\x07\x00\x00!n\x01\x03\x00\x00!\x0fu\x04\x00\x00!\xf5h\x02@\x86'
[0x3808:11:0x0300] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=0, is_reply=1, disable_default_response=1, reserved=0, *is_cluster=False, *is_general=True), tsn=129, command_id=1, *is_reply=True)
[0x3808:11:0x0300] Decoded ZCL frame: Color:Read_Attributes_rsp(status_records=[ReadAttributeRecord(attrid=0x0008, status=<Status.SUCCESS: 0>, value=TypeValue(type=enum8, value=<enum8.undefined_0x02: 2>)), ReadAttributeRecord(attrid=0x0007, status=<Status.SUCCESS: 0>, value=TypeValue(type=uint16_t, value=366)), ReadAttributeRecord(attrid=0x0003, status=<Status.SUCCESS: 0>, value=TypeValue(type=uint16_t, value=29967)), ReadAttributeRecord(attrid=0x0004, status=<Status.SUCCESS: 0>, value=TypeValue(type=uint16_t, value=26869)), ReadAttributeRecord(attrid=0x4002, status=<Status.UNSUPPORTED_ATTRIBUTE: 134>)])
[0x0DF7:1:0x0000] Received ZCL frame: b'\x18\x83\x01\x04\x00\x00B\x10_TZE200_pay2byax\x05\x00\x00B\x06TS0601'
[0x0DF7:1:0x0000] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=0, is_reply=1, disable_default_response=1, reserved=0, *is_cluster=False, *is_general=True), tsn=131, command_id=1, *is_reply=True)
[0x0DF7:1:0x0000] Decoded ZCL frame: Basic:Read_Attributes_rsp(status_records=[ReadAttributeRecord(attrid=0x0004, status=<Status.SUCCESS: 0>, value=TypeValue(type=CharacterString, value='_TZE200_pay2byax')), ReadAttributeRecord(attrid=0x0005, status=<Status.SUCCESS: 0>, value=TypeValue(type=CharacterString, value='TS0601'))])
[0x0df7] Read model 'TS0601' and manufacturer '_TZE200_pay2byax' from <Endpoint id=1 in=[power:0x0001, ias_zone:0x0500, basic:0x0000] out=[ota:0x0019, time:0x000A] status=<Status.ZDO_INIT: 1>>
[0x0df7] Discovered basic device information for <Device model='TS0601' manuf='_TZE200_pay2byax' nwk=0x0DF7 ieee=a4:c1:38:4d:82:df:a6:b0 is_initialized=True>
Device is initialized <Device model='TS0601' manuf='_TZE200_pay2byax' nwk=0x0DF7 ieee=a4:c1:38:4d:82:df:a6:b0 is_initialized=True>
Checking quirks for _TZE200_pay2byax TS0601 (a4:c1:38:4d:82:df:a6:b0)
Considering <class 'zhaquirks.xbee.xbee_io.XBeeSensor'>
Fail because endpoint list mismatch: {232, 230} {1}
Considering <class 'zhaquirks.xbee.xbee3_io.XBee3Sensor'>
Fail because endpoint list mismatch: {232, 230} {1}
Considering <class 'zhaquirks.smartthings.tag_v4.SmartThingsTagV4'>
Fail because device_type mismatch on at least one endpoint
Considering <class 'zhaquirks.smartthings.multi.SmartthingsMultiPurposeSensor'>
[... cut ...]  *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0402",
      "in_clusters": [
        "0x0000",
        "0x0001",
        "0x0500"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZE200_pay2byax",
  "model": "TS0601",
  "class": "zigpy.device.Device"
}

Tropaion, 2022-04-16:
Do you have an link to the device?

centershock, 2022-04-16:
Can find it only in my AliExpress order:
https://de.aliexpress.com/item/1005003518715880.html?gatewayAdapt=glo2deu&spm=a2g0o.order_detail.0.0.328363687Th9Lw

Tropaion, 2022-04-16:
Normally, this device should work without a quirk because it uses standard clusters, did you try it? Do you have some problems with it?

centershock, 2022-04-16:
[image]
The power is always 0%, and there is not update whatsoever for ias_zone. I tried with three different devices of the same model, same result.

Tropaion, 2022-04-16:
Try if this quirk works: [ts0601_magnet.zip](https://github.com/zigpy/zha-device-handlers/files/8500583/ts0601_magnet.zip)

centershock, 2022-04-17:
Unfortunately, no success and the same behavior.
The log when pairing:
`[0x340B:1:0x0402] Received ZCL frame: b'\x18\x1b\n\x00\x00)\xe8\x08'
[0x340B:1:0x0402] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=0, is_reply=1, disable_default_response=1, reserved=0, *is_cluster=False, *is_general=True), tsn=27, command_id=10, *is_reply=True)
[0x340B:1:0x0402] Decoded ZCL frame: TemperatureMeasurementCluster:Report_Attributes(attribute_reports=[Attribute(attrid=0x0000, value=TypeValue(type=int16s, value=2280))])
[0x340B:1:0x0402] Received command 0x0A (TSN 27): Report_Attributes(attribute_reports=[Attribute(attrid=0x0000, value=TypeValue(type=int16s, value=2280))])
[0x340B:1:0x0402] Attribute report received: measured_value=2280
[0x340B:1:0x0405] Received ZCL frame: b'\x18\x1c\n\x00\x00!T\x0f'
[0x340B:1:0x0405] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=0, is_reply=1, disable_default_response=1, reserved=0, *is_cluster=False, *is_general=True), tsn=28, command_id=10, *is_reply=True)
[0x340B:1:0x0405] Decoded ZCL frame: RelativeHumidityCluster:Report_Attributes(attribute_reports=[Attribute(attrid=0x0000, value=TypeValue(type=uint16_t, value=3924))])
[0x340B:1:0x0405] Received command 0x0A (TSN 28): Report_Attributes(attribute_reports=[Attribute(attrid=0x0000, value=TypeValue(type=uint16_t, value=3924))])
[0x340B:1:0x0405] Attribute report received: measured_value=3924
[0x340B:1:0x0403] Received ZCL frame: b"\x18\x1d\n\x00\x00)\xfb\x03\x14\x00(\xff\x10\x00)\xd1'"
[0x340B:1:0x0403] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=0, is_reply=1, disable_default_response=1, reserved=0, *is_cluster=False, *is_general=True), tsn=29, command_id=10, *is_reply=True)
[0x340B:1:0x0403] Decoded ZCL frame: PressureMeasurementCluster:Report_Attributes(attribute_reports=[Attribute(attrid=0x0000, value=TypeValue(type=int16s, value=1019)), Attribute(attrid=0x0014, value=TypeValue(type=int8s, value=-1)), Attribute(attrid=0x0010, value=TypeValue(type=int16s, value=10193))])
[0x340B:1:0x0403] Received command 0x0A (TSN 29): Report_Attributes(attribute_reports=[Attribute(attrid=0x0000, value=TypeValue(type=int16s, value=1019)), Attribute(attrid=0x0014, value=TypeValue(type=int8s, value=-1)), Attribute(attrid=0x0010, value=TypeValue(type=int16s, value=10193))])
[0x340B:1:0x0403] Attribute report received: measured_value=1019, 0x0014=-1, 0x0010=10193
Device 0x877f (a4:c1:38:4d:82:df:a6:b0) joined the network
Device a4:c1:38:4d:82:df:a6:b0 changed id (0x877f => 0x877f)
[0x877f] Skipping initialization, device is fully initialized
Device is initialized <TuyaContactSensor model='TS0601' manuf='_TZE200_pay2byax' nwk=0x877F ieee=a4:c1:38:4d:82:df:a6:b0 is_initialized=True>
[0x877f:zdo] ZDO request ZDOCmd.Device_annce: [0x877F, a4:c1:38:4d:82:df:a6:b0, 128]
'binary_sensor' component -> 'IASZone' using ['ias_zone']
'sensor' component -> 'Battery' using ['power']
'sensor' component -> 'RSSISensor' using ['basic']
'sensor' component -> 'LQISensor' using ['basic']
device - 34687:a4:c1:38:4d:82:df:a6:b0 entering async_device_initialized - is_new_join: True
device - 34687:a4:c1:38:4d:82:df:a6:b0 has joined the ZHA zigbee network
[34687](TS0601): started configuration
[34687:ZDO](TS0601): 'async_configure' stage succeeded
[34687:1:0x0000]: finished channel configuration
[0x877f] Extending timeout for 0x27 request
[0x877F:1:0x0500] Sending request header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=False, is_reply=0, disable_default_response=0, reserved=0, *is_cluster=False, *is_general=True), tsn=41, command_id=<GeneralCommand.Read_Attributes: 0>, *is_reply=False)
[0x877F:1:0x0500] Sending request: Read_Attributes(attribute_ids=[1])
[0x877f] Extending timeout for 0x29 request
[34687:1:0x0019]: finished channel configuration
[34687:1:0x0001]: bound 'power' cluster: Status.SUCCESS
[0x877F:1:0x0001] Sending request header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=False, is_reply=0, disable_default_response=0, reserved=0, *is_cluster=False, *is_general=True), tsn=43, command_id=<GeneralCommand.Configure_Reporting: 6>, *is_reply=False)
[0x877F:1:0x0001] Sending request: Configure_Reporting(config_records=[AttributeReportingConfig(direction=0, attrid=0x0020, datatype=32, min_interval=3600, max_interval=10800, reportable_change=1), AttributeReportingConfig(direction=0, attrid=0x0021, datatype=32, min_interval=3600, max_interval=10800, reportable_change=1)])
[0x877f] Extending timeout for 0x2b request
[0x877F:1:0x0500] Received ZCL frame: b'\x18)\x01\x01\x00\x001(\x00'
[0x877F:1:0x0500] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=0, is_reply=1, disable_default_response=1, reserved=0, *is_cluster=False, *is_general=True), tsn=41, command_id=1, *is_reply=True)
[0x877F:1:0x0500] Decoded ZCL frame: IasZone:Read_Attributes_rsp(status_records=[ReadAttributeRecord(attrid=0x0001, status=<Status.SUCCESS: 0>, value=TypeValue(type=enum16, value=<enum16.undefined_0x0028: 40>))])
[34687:1:0x0500]: started IASZoneChannel configuration
[0x877f] Extending timeout for 0x2d request
[... cut ...]
Ignoring message (b'0917020024650200040000030a') on cluster 61184: unknown endpoint or cluster id: 'No cluster ID 0xef00 on (a4:c1:38:4d:82:df:a6:b0, 1)'
Ignoring message (b'091802002b0202000400000064') on cluster 61184: unknown endpoint or cluster id: 'No cluster ID 0xef00 on (a4:c1:38:4d:82:df:a6:b0, 1)'
Ignoring message (b'091802002b0202000400000064') on cluster 61184: unknown endpoint or cluster id: 'No cluster ID 0xef00 on (a4:c1:38:4d:82:df:a6:b0, 1)'
`

Happy Easter and thanks for your help!

Tropaion, 2022-04-17:
Thanks, also Happy Easter 😄 
The information you provide is ok, mabye you can look if there is something else interesting in the logs.
It looks like the device sends data on cluster ```0xEF00``` even though it isn't shown in the device signature.
Try this: [ts0601_magnet.zip](https://github.com/zigpy/zha-device-handlers/files/8501556/ts0601_magnet.zip)

centershock, 2022-04-17:
Again no change in behavior. The log when pairing:

`[0x877f] Delivery error for seq # 0x21, on endpoint id 0 cluster 0x0034: message send failure
Sending 'zdo_leave_req' failed: [0x877f:0:0x0034]: Message send failure
[0x3808:11:0x0008] Received ZCL frame: b'\x18\x03\n\x00\x00 ;'
[0x3808:11:0x0008] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=0, is_reply=1, disable_default_response=1, reserved=0, *is_cluster=False, *is_general=True), tsn=3, command_id=10, *is_reply=True)
[0x3808:11:0x0008] Decoded ZCL frame: LevelControl:Report_Attributes(attribute_reports=[Attribute(attrid=0x0000, value=TypeValue(type=uint8_t, value=59))])
[0x3808:11:0x0008] Received command 0x0A (TSN 3): Report_Attributes(attribute_reports=[Attribute(attrid=0x0000, value=TypeValue(type=uint8_t, value=59))])
[0x3808:11:0x0008] Attribute report received: current_level=59
[0x3808:11:0x0008]: received attribute: 0 update with value: 59
New device 0x035f (a4:c1:38:4d:82:df:a6:b0) joined the network
[0x035f] Scheduling initialization
Received frame on uninitialized device <Device model=None manuf=None nwk=0x035F ieee=a4:c1:38:4d:82:df:a6:b0 is_initialized=False> from ep 0 to ep 0, cluster 19: b'\x07_\x03\xb0\xa6\xdf\x82M8\xc1\xa4\x80'
[0x035f:zdo] ZDO request ZDOCmd.Device_annce: [0x035F, a4:c1:38:4d:82:df:a6:b0, 128]
Tries remaining: 3
[0x035f] Requesting 'Node Descriptor'
Tries remaining: 2
[0x035f] Extending timeout for 0x27 request
Received frame on uninitialized device <Device model=None manuf=None nwk=0x035F ieee=a4:c1:38:4d:82:df:a6:b0 is_initialized=False> from ep 0 to ep 0, cluster 32770: b"'\x00_\x03\x02@\x80A\x11BB\x00\x00*B\x00\x00"
[0x035f] Got Node Descriptor: NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)
[0x035f] Discovering endpoints
Tries remaining: 3
[0x035f] Extending timeout for 0x29 request
Received frame on uninitialized device <Device model=None manuf=None nwk=0x035F ieee=a4:c1:38:4d:82:df:a6:b0 is_initialized=False> from ep 0 to ep 0, cluster 32773: b')\x00_\x03\x01\x01'
[0x035f] Discovered endpoints: [1]
[0x035f] Initializing endpoints [<Endpoint id=1 in=[] out=[] status=<Status.NEW: 0>>]
[0x035f:1] Discovering endpoint information
Tries remaining: 3
[0x035f] Extending timeout for 0x2b request
Received frame on uninitialized device <Device model=None manuf=None nwk=0x035F ieee=a4:c1:38:4d:82:df:a6:b0 is_initialized=False> from ep 0 to ep 0, cluster 32772: b'+\x00_\x03\x12\x01\x04\x01\x02\x04\x01\x03\x01\x00\x00\x05\x00\x00\x02\x19\x00\n\x00'
[0x035f:1] Discovered endpoint information: SizePrefixedSimpleDescriptor(endpoint=1, profile=260, device_type=1026, device_version=1, input_clusters=[1, 1280, 0], output_clusters=[25, 10])
[0x035F:1:0x0000] Sending request header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=False, is_reply=0, disable_default_response=0, reserved=0, *is_cluster=False, *is_general=True), tsn=45, command_id=<GeneralCommand.Read_Attributes: 0>, *is_reply=False)
[0x035F:1:0x0000] Sending request: Read_Attributes(attribute_ids=[4, 5])
[0x035f] Extending timeout for 0x2d request
[0x035F:1:0x0000] Received ZCL frame: b'\x18-\x01\x04\x00\x00B\x10_TZE200_pay2byax\x05\x00\x00B\x06TS0601'
[0x035F:1:0x0000] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=0, is_reply=1, disable_default_response=1, reserved=0, *is_cluster=False, *is_general=True), tsn=45, command_id=1, *is_reply=True)
[0x035F:1:0x0000] Decoded ZCL frame: Basic:Read_Attributes_rsp(status_records=[ReadAttributeRecord(attrid=0x0004, status=<Status.SUCCESS: 0>, value=TypeValue(type=CharacterString, value='_TZE200_pay2byax')), ReadAttributeRecord(attrid=0x0005, status=<Status.SUCCESS: 0>, value=TypeValue(type=CharacterString, value='TS0601'))])
[0x035f] Read model 'TS0601' and manufacturer '_TZE200_pay2byax' from <Endpoint id=1 in=[power:0x0001, ias_zone:0x0500, basic:0x0000] out=[ota:0x0019, time:0x000A] status=<Status.ZDO_INIT: 1>>
[0x035f] Discovered basic device information for <Device model='TS0601' manuf='_TZE200_pay2byax' nwk=0x035F ieee=a4:c1:38:4d:82:df:a6:b0 is_initialized=True>
Device is initialized <Device model='TS0601' manuf='_TZE200_pay2byax' nwk=0x035F ieee=a4:c1:38:4d:82:df:a6:b0 is_initialized=True>
Checking quirks for _TZE200_pay2byax TS0601 (a4:c1:38:4d:82:df:a6:b0)
Considering <class 'ts0601_magnet.TuyaContactSensor'>
Found custom device replacement for a4:c1:38:4d:82:df:a6:b0: <class 'ts0601_magnet.TuyaContactSensor'>
'binary_sensor' component -> 'IASZone' using ['ias_zone']
'sensor' component -> 'Battery' using ['power']
'sensor' component -> 'RSSISensor' using ['basic']
'sensor' component -> 'LQISensor' using ['basic']
[... cut ...]
2022-04-17 20:32:16 DEBUG (MainThread) [zigpy.zcl] [0x035F:1:0xef00] Decoded ZCL frame: TuyaManufCluster:set_data_response(param=Command(status=0, tsn=78, command_id=257, function=0, data=[1, 0]))
2022-04-17 20:32:16 DEBUG (MainThread) [zigpy.zcl] [0x035F:1:0xef00] Received command 0x02 (TSN 63): set_data_response(param=Command(status=0, tsn=78, command_id=257, function=0, data=[1, 0]))
2022-04-17 20:32:16 DEBUG (MainThread) [zigpy.zcl] [0x035F:1:0xef00] No explicit handler for cluster command 0x02: set_data_response(param=Command(status=0, tsn=78, command_id=257, function=0, data=[1, 0]))`

Tropaion, 2022-04-17:
What is the NWK-Address of the device you used?
And can you put the logs in a more readable format please? It's really hard like this put it in (\`\`\`)

centershock, 2022-04-17:
NWK-Adress should be 0x035f.
Sorry about the logs, changed it for the last post.

centershock, 2022-04-26:
Anything I can help with to get a solution? 
Would really appreciate it, if a solution can be found. Thank you :)

Tropaion, 2022-04-26:
@centershock Sorry, the next two weeks I'm very busy and don't have much time.
@javicalle Should be able to help better in this issue.

centershock, 2022-04-26:
@Tropaion No need for apologies, thank you for forwarding the issue 😊

javicalle, 2022-04-26:
I believe that can be done.
Device reports 2 DP:
* DP1, `command_id=257` --> 0/1 values
* DP2, `command_id=514` --> constant 100 value

DP1 is the sensor status.
DP2 seems to be battery percentage remaining
Maybe there is also a tamper sensor (when battery closure is opened) and/or a low battery sensor (not clear in the vendor site).

I have followed a different approach than @Tropaion . Neither better nor worse, just my approach.
[ts0601_magnet.py.txt](https://github.com/zigpy/zha-device-handlers/files/8567527/ts0601_magnet.py.txt)

Download, rename (remove the `.txt` in filename) and replace your local quirk with this one.
Remove any local cache (`__pycache__` folder) and restart.

I'm 'coding' from secondary laptop and can't validate code. Lots of compile errors can be expected.

javicalle, 2022-04-27:
Sorry @centershock  but 752 lines of mostly non-device related log doesn't help me to help you.
Please try to include only the traces that are relevant to fixing the device, for example, those generated when you open the sensor and when you close it (better if they are in separate logs).

I am aware that it is not easy to know what information is relevant, but it is necessary to make the analysis of the information as easy as possible in order to reach more cases.

A can see the battery report. It's the device battery showing a value?
Your previous logs (with @Tropaion version) have a lot of sensor open/closed reports. In the last logs I can only see 2 reports (both open sensor). Can you include a log with multip`le open/close reports?

You can also try a code change, just in case that's the problem. In the local quirk modify the following part of the code (`zone_status` --> `zone_state`):
```python
    dp_to_attribute: Dict[int, DPToAttributeMapping] = {
        1: DPToAttributeMapping(
            ContactSwitchCluster.ep_attribute,
            # "zone_status",
            "zone_state",
            dp_type=TuyaDPType.BOOL,
        ),
        2: DPToAttributeMapping(
            DoublingPowerConfigurationCluster.ep_attribute,
            "battery_percentage_remaining",  # 0x0021
            dp_type=TuyaDPType.ENUM,
        ),
    }
```

From your logs, also we can see that there is another DP:
* DP101, command_id=257 --> reported values: 426, 526, ¿634?
We can keep here the new DP until we can go forward with the main problem.

centershock, 2022-04-28:
Hey @javicalle,
sorry about that, but as I don't really understand how zha works I did not know the important information.
In the following log there are some changes from  from "open" to "closed" and back to "open", however, I don't know if there is something to see.
The device battery shows always 0%.

I already integrated your proposed change from 'zone_status' -> 'zone_state'

```
2022-04-28 14:52:31 DEBUG (MainThread) [zigpy.zcl] [0xF273:1:0xef00] Received ZCL frame: b'\tw\x02\x00ae\x02\x00\x04\x00\x00\x03\x82'
2022-04-28 14:52:31 DEBUG (MainThread) [zigpy.zcl] [0xF273:1:0xef00] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.CLUSTER_COMMAND: 1>, is_manufacturer_specific=0, is_reply=1, disable_default_response=0, reserved=0, *is_cluster=True, *is_general=False), tsn=119, command_id=2, *is_reply=True)
2022-04-28 14:52:31 DEBUG (MainThread) [zigpy.zcl] [0xF273:1:0xef00] Decoded ZCL frame: ContactSwitchManufCluster:set_data_response(data=TuyaCommand(status=0, tsn=97, dp=101, data=TuyaData(dp_type=<TuyaDPType.VALUE: 2>, function=0, raw=b'\x82\x03\x00\x00', *payload=898)))
2022-04-28 14:52:31 DEBUG (MainThread) [zigpy.zcl] [0xF273:1:0xef00] Received command 0x02 (TSN 119): set_data_response(data=TuyaCommand(status=0, tsn=97, dp=101, data=TuyaData(dp_type=<TuyaDPType.VALUE: 2>, function=0, raw=b'\x82\x03\x00\x00', *payload=898)))
2022-04-28 14:52:31 DEBUG (MainThread) [zigpy.zcl] [0xF273:1:0xef00] No datapoint handler for TuyaCommand(status=0, tsn=97, dp=101, data=TuyaData(dp_type=<TuyaDPType.VALUE: 2>, function=0, raw=b'\x82\x03\x00\x00', *payload=898))
2022-04-28 14:52:31 WARNING (MainThread) [zigpy.zcl] [0xF273:1:0xef00] No 'handle_set_data_response' tuya handler found for set_data_response(data=TuyaCommand(status=0, tsn=97, dp=101, data=TuyaData(dp_type=<TuyaDPType.VALUE: 2>, function=0, raw=b'\x82\x03\x00\x00', *payload=898)))
2022-04-28 14:52:31 DEBUG (MainThread) [zigpy.zcl] [0xF273:1:0xef00] Sending reply header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=False, is_reply=1, disable_default_response=1, reserved=0, *is_cluster=False, *is_general=True), tsn=119, command_id=<GeneralCommand.Default_Response: 11>, *is_reply=True)
2022-04-28 14:52:31 DEBUG (MainThread) [zigpy.zcl] [0xF273:1:0xef00] Sending reply: Default_Response(command_id=2, status=<Status.UNSUP_CLUSTER_COMMAND: 129>)
2022-04-28 14:52:31 DEBUG (MainThread) [zigpy.zcl] [0xF273:1:0xef00] Received ZCL frame: b'\tx\x02\x00b\x01\x01\x00\x01\x01'
2022-04-28 14:52:31 DEBUG (MainThread) [zigpy.zcl] [0xF273:1:0xef00] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.CLUSTER_COMMAND: 1>, is_manufacturer_specific=0, is_reply=1, disable_default_response=0, reserved=0, *is_cluster=True, *is_general=False), tsn=120, command_id=2, *is_reply=True)
2022-04-28 14:52:31 DEBUG (MainThread) [zigpy.zcl] [0xF273:1:0xef00] Decoded ZCL frame: ContactSwitchManufCluster:set_data_response(data=TuyaCommand(status=0, tsn=98, dp=1, data=TuyaData(dp_type=<TuyaDPType.BOOL: 1>, function=0, raw=b'\x01', *payload=<Bool.true: 1>)))
2022-04-28 14:52:31 DEBUG (MainThread) [zigpy.zcl] [0xF273:1:0xef00] Received command 0x02 (TSN 120): set_data_response(data=TuyaCommand(status=0, tsn=98, dp=1, data=TuyaData(dp_type=<TuyaDPType.BOOL: 1>, function=0, raw=b'\x01', *payload=<Bool.true: 1>)))
2022-04-28 14:52:31 DEBUG (MainThread) [zigpy.zcl] [0xF273:1:0xef00] Sending reply header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=False, is_reply=1, disable_default_response=1, reserved=0, *is_cluster=False, *is_general=True), tsn=120, command_id=<GeneralCommand.Default_Response: 11>, *is_reply=True)
2022-04-28 14:52:31 DEBUG (MainThread) [zigpy.zcl] [0xF273:1:0xef00] Sending reply: Default_Response(command_id=2, status=<Status.SUCCESS: 0>)
2022-04-28 14:52:36 DEBUG (MainThread) [zigpy.zcl] [0xF273:1:0xef00] Received ZCL frame: b'\ty\x02\x00ce\x02\x00\x04\x00\x00\x03\x82'
2022-04-28 14:52:36 DEBUG (MainThread) [zigpy.zcl] [0xF273:1:0xef00] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.CLUSTER_COMMAND: 1>, is_manufacturer_specific=0, is_reply=1, disable_default_response=0, reserved=0, *is_cluster=True, *is_general=False), tsn=121, command_id=2, *is_reply=True)
2022-04-28 14:52:36 DEBUG (MainThread) [zigpy.zcl] [0xF273:1:0xef00] Decoded ZCL frame: ContactSwitchManufCluster:set_data_response(data=TuyaCommand(status=0, tsn=99, dp=101, data=TuyaData(dp_type=<TuyaDPType.VALUE: 2>, function=0, raw=b'\x82\x03\x00\x00', *payload=898)))
2022-04-28 14:52:36 DEBUG (MainThread) [zigpy.zcl] [0xF273:1:0xef00] Received command 0x02 (TSN 121): set_data_response(data=TuyaCommand(status=0, tsn=99, dp=101, data=TuyaData(dp_type=<TuyaDPType.VALUE: 2>, function=0, raw=b'\x82\x03\x00\x00', *payload=898)))
2022-04-28 14:52:36 DEBUG (MainThread) [zigpy.zcl] [0xF273:1:0xef00] No datapoint handler for TuyaCommand(status=0, tsn=99, dp=101, data=TuyaData(dp_type=<TuyaDPType.VALUE: 2>, function=0, raw=b'\x82\x03\x00\x00', *payload=898))
2022-04-28 14:52:36 WARNING (MainThread) [zigpy.zcl] [0xF273:1:0xef00] No 'handle_set_data_response' tuya handler found for set_data_response(data=TuyaCommand(status=0, tsn=99, dp=101, data=TuyaData(dp_type=<TuyaDPType.VALUE: 2>, function=0, raw=b'\x82\x03\x00\x00', *payload=898)))
[... cut ...]
2022-04-28 14:52:42 DEBUG (MainThread) [zigpy.zcl] [0xF273:1:0xef00] Sending reply header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=False, is_reply=1, disable_default_response=1, reserved=0, *is_cluster=False, *is_general=True), tsn=128, command_id=<GeneralCommand.Default_Response: 11>, *is_reply=True)
2022-04-28 14:52:42 DEBUG (MainThread) [zigpy.zcl] [0xF273:1:0xef00] Sending reply: Default_Response(command_id=2, status=<Status.UNSUP_CLUSTER_COMMAND: 129>)
```

javicalle, 2022-04-28:
There is something I want to check. Can you test with this version?
[ts0601_magnet.py.txt](https://github.com/zigpy/zha-device-handlers/files/8585363/ts0601_magnet.py.txt)

Delete cache, restart and test.

Also, can you check from HA the device clusters attributes?
From the device view, "Clusters administration" option:
* select `ContactSwitchCluster (Endpoint id: 1, ...)`, then select `zone_status (id: 0x0002)` and push the button "GET ZIGBEE ATTRIBUTE"
  *  this value is expected to change when the sensor is open/closed (0/1)
* select `DoublingPowerConfigurationCluster (Endpoint id: 1, ...)`, then select `battery_percentage_remaining (id: 0x0021)` and push the button "GET ZIGBEE ATTRIBUTE"
  * the expected value of this attribute is 100

javicalle, 2022-04-28:
And I want to ask you to put the logs from https://github.com/zigpy/zha-device-handlers/issues/1502#issuecomment-1111357062 in a collapsable section, this will help to scroll the conversation.

````
<details>
<summary>HA Log:</summary>

``` 
Paste your logs here keeping spaces and all the ` .
``` 

</details>
````

centershock, 2022-05-04:
Hey @javicalle,
sorry, took a while :/
For ContactSwitchCluster with zone_status the value is always "None", no matter, if sensor is open or closed.
For DoublingPowerConfigurationCluster no value occurs, simply nothing happens when I push "GET ZIGBEE ATTRIBUTE".

Do you need the logs or other information?
I did *not* repair the device...

javicalle, 2022-05-05:
Another iteration:
[ts0601_magnet.py.txt](https://github.com/zigpy/zha-device-handlers/files/8633825/ts0601_magnet.py.txt)

Enable logs, delete cache and restart.

Look in the logs for events when you open/close the sensor.
Also (from HA UI) get the clusters attributes and look at the logs, there must be sometraces when you ask for attributes.

centershock, 2022-05-26:
Hey, sorry for the delay, just got a newborn kid ;)
This is what happens when I try to get the `zone_status (id: 0x0002)` from `ContactSwitchCluster (Endpoint id: 1, ...)`.  Result shown in the UI is always `Bool.true`, no matter the physical state of the sensor.
´´´
2022-05-26 18:31:50 DEBUG (MainThread) [zigpy.zcl] [0x3CB8:1:0x0500] read_attributes --> attrs: [2]
2022-05-26 18:32:05 DEBUG (MainThread) [zigpy.zcl] [0x3CB8:1:0x0500] read_attributes --> attrs: [2]
´´´

There are no log entries whatsoever when I open / close the sensor in the same time span. Just to make sure my logger:
´´´
logger:
  default: info
  logs:
    zigpy: debug
    zigpy_znp: debug
    zhaquirks: debug
´´´
There are other log entries for the zha integration for other devices, therefore I think that should be correct.
Does not seem to be too much information to work with, if I can do more, just tell me.

javicalle, 2022-05-26:
Could you try to repeat the tests with this logs:
```
logger:
  default: info
  logs:
    bellows.zigbee.application: debug
    bellows.ezsp: debug
    zigpy: debug
    zigpy_deconz.zigbee.application: debug
    zigpy_deconz.api: debug
    zigpy_xbee.zigbee.application: debug
    zigpy_xbee.api: debug
    zigpy_zigate: debug
    zigpy_znp: debug
    zhaquirks: debug
```

> sorry for the delay, just got a newborn kid

Congratulations.
That's a very big and important project.

centershock, 2022-05-28:
Thank you very much javicalle :)

The result and the log for the query of the cluster is the same.

Here is the enhanced log for the opening and close action. Nwk is 0x711f.

```
2022-05-28 19:56:16 DEBUG (MainThread) [zigpy_deconz.zigbee.application] Sending Zigbee request with tsn 8 under 30 request id, data: b'18080b0281'
2022-05-28 19:56:16 DEBUG (MainThread) [zigpy_deconz.api] 'aps_data_indication' response from <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0x711f>, ep: 1, profile: 0x0104, cluster_id: 0xef00, data: b'090802000765020004000002c2'
2022-05-28 19:56:16 DEBUG (MainThread) [zigpy_deconz.api] Command Command.aps_data_request (20, 30, <DeconzSendDataFlags.NONE: 0>, <DeconzAddressEndpoint address_mode=2 address=0x711F endpoint=1>, 260, 61184, 1, b'\x18\x08\x0b\x02\x81', <DeconzTransmitOptions.USE_NWK_KEY_SECURITY: 2>, 0)
2022-05-28 19:56:16 DEBUG (MainThread) [zigpy_deconz.api] APS data request response: [2, <DeviceState.APSDE_DATA_REQUEST_SLOTS_AVAILABLE|2: 34>, 30]
2022-05-28 19:56:16 DEBUG (MainThread) [zigpy_deconz.api] Device state changed response: [<DeviceState.128|APSDE_DATA_REQUEST_SLOTS_AVAILABLE|APSDE_DATA_CONFIRM|2: 166>, 0]
2022-05-28 19:56:16 DEBUG (MainThread) [zigpy_deconz.api] Command Command.aps_data_confirm (0,)
2022-05-28 19:56:16 DEBUG (MainThread) [zigpy_deconz.api] APS data confirm response for request with id 30: 00
2022-05-28 19:56:16 DEBUG (MainThread) [zigpy_deconz.api] Request id: 0x1e 'aps_data_confirm' for <DeconzAddressEndpoint address_mode=ADDRESS_MODE.NWK address=0x711f endpoint=1>, status: 0x00
2022-05-28 19:56:16 DEBUG (MainThread) [zigpy_deconz.api] Device state changed response: [<DeviceState.128|APSDE_DATA_REQUEST_SLOTS_AVAILABLE|APSDE_DATA_INDICATION|2: 170>, 0]
2022-05-28 19:56:16 DEBUG (MainThread) [zigpy_deconz.api] Command Command.aps_data_indication (1, 1)
2022-05-28 19:56:16 DEBUG (MainThread) [zigpy_deconz.api] APS data indication response: [33, <DeviceState.APSDE_DATA_REQUEST_SLOTS_AVAILABLE|2: 34>, <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0x0000>, 1, <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0x711f>, 1, 260, 61184, b'\t\t\x02\x00\x08\x01\x01\x00\x01\x01', 0, 175, 255, 213, 181, 32, 0, -55]
2022-05-28 19:56:16 DEBUG (MainThread) [zigpy.zcl] [0x711F:1:0xef00] Received ZCL frame: b'\t\t\x02\x00\x08\x01\x01\x00\x01\x01'
2022-05-28 19:56:16 DEBUG (MainThread) [zigpy.zcl] [0x711F:1:0xef00] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.CLUSTER_COMMAND: 1>, is_manufacturer_specific=0, is_reply=1, disable_default_response=0, reserved=0, *is_cluster=True, *is_general=False), tsn=9, command_id=2, *is_reply=True)
2022-05-28 19:56:16 DEBUG (MainThread) [zigpy.zcl] [0x711F:1:0xef00] Decoded ZCL frame: ContactSwitchManufCluster:set_data_response(data=TuyaCommand(status=0, tsn=8, dp=1, data=TuyaData(dp_type=<TuyaDPType.BOOL: 1>, function=0, raw=b'\x01', *payload=<Bool.true: 1>)))
2022-05-28 19:56:16 DEBUG (MainThread) [zigpy.zcl] [0x711F:1:0xef00] Received command 0x02 (TSN 9): set_data_response(data=TuyaCommand(status=0, tsn=8, dp=1, data=TuyaData(dp_type=<TuyaDPType.BOOL: 1>, function=0, raw=b'\x01', *payload=<Bool.true: 1>)))
2022-05-28 19:56:16 DEBUG (MainThread) [zigpy.zcl] [0x711F:1:0x0500] _update_attribute '2': Bool.true
2022-05-28 19:56:16 DEBUG (MainThread) [zigpy.zcl] [0x711F:1:0xef00] Sending reply header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=False, is_reply=1, disable_default_response=1, reserved=0, *is_cluster=False, *is_general=True), tsn=9, command_id=<GeneralCommand.Default_Response: 11>, *is_reply=True)
2022-05-28 19:56:16 DEBUG (MainThread) [zigpy.zcl] [0x711F:1:0xef00] Sending reply: Default_Response(command_id=2, status=<Status.SUCCESS: 0>)
2022-05-28 19:56:16 DEBUG (MainThread) [zigpy_deconz.zigbee.application] Sending Zigbee request with tsn 9 under 31 request id, data: b'18090b0200'
2022-05-28 19:56:16 DEBUG (MainThread) [zigpy_deconz.api] 'aps_data_indication' response from <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0x711f>, ep: 1, profile: 0x0104, cluster_id: 0xef00, data: b'09090200080101000101'
2022-05-28 19:56:16 DEBUG (MainThread) [zigpy_deconz.api] Command Command.aps_data_request (20, 31, <DeconzSendDataFlags.NONE: 0>, <DeconzAddressEndpoint address_mode=2 address=0x711F endpoint=1>, 260, 61184, 1, b'\x18\t\x0b\x02\x00', <DeconzTransmitOptions.USE_NWK_KEY_SECURITY: 2>, 0)
2022-05-28 19:56:16 DEBUG (MainThread) [zigpy_deconz.api] APS data request response: [2, <DeviceState.APSDE_DATA_REQUEST_SLOTS_AVAILABLE|2: 34>, 31]
2022-05-28 19:56:16 DEBUG (MainThread) [zigpy_deconz.api] Device state changed response: [<DeviceState.128|APSDE_DATA_REQUEST_SLOTS_AVAILABLE|APSDE_DATA_CONFIRM|2: 166>, 0]
2022-05-28 19:56:16 DEBUG (MainThread) [zigpy_deconz.api] Command Command.aps_data_confirm (0,)
2022-05-28 19:56:16 DEBUG (MainThread) [zigpy_deconz.api] APS data confirm response for request with id 31: 00
2022-05-28 19:56:16 DEBUG (MainThread) [zigpy_deconz.api] Request id: 0x1f 'aps_data_confirm' for <DeconzAddressEndpoint address_mode=ADDRESS_MODE.NWK address=0x711f endpoint=1>, status: 0x00
2022-05-28 19:56:18 DEBUG (MainThread) [zigpy_deconz.api] Device state changed response: [<DeviceState.128|APSDE_DATA_REQUEST_SLOTS_AVAILABLE|APSDE_DATA_INDICATION|2: 170>, 0]
2022-05-28 19:56:18 DEBUG (MainThread) [zigpy_deconz.api] Command Command.aps_data_indication (1, 1)
[... cut ...]
2022-05-28 19:56:19 DEBUG (MainThread) [zigpy_deconz.api] Device state changed response: [<DeviceState.128|APSDE_DATA_REQUEST_SLOTS_AVAILABLE|APSDE_DATA_CONFIRM|2: 166>, 0]
2022-05-28 19:56:19 DEBUG (MainThread) [zigpy_deconz.api] Command Command.aps_data_confirm (0,)
2022-05-28 19:56:19 DEBUG (MainThread) [zigpy_deconz.api] APS data confirm response for request with id 35: 00
2022-05-28 19:56:19 DEBUG (MainThread) [zigpy_deconz.api] Request id: 0x23 'aps_data_confirm' for <DeconzAddressEndpoint address_mode=ADDRESS_MODE.NWK address=0x711f endpoint=1>, status: 0x00
```

javicalle, 2022-05-29:
That's what I can see from your logs:
```
2022-05-28 19:56:16 DEBUG (MainThread) [zigpy.zcl] [0x711F:1:0x0500] _update_attribute '2': Bool.true
2022-05-28 19:56:18 DEBUG (MainThread) [zigpy.zcl] [0x711F:1:0x0500] _update_attribute '2': Bool.false
2022-05-28 19:56:19 DEBUG (MainThread) [zigpy.zcl] [0x711F:1:0x0500] _update_attribute '2': Bool.true
```
So attribute `2` from `IasZone` cluster (0x0500) it's updating its value.

Maybe the data type is not good, let's try with a converter. Can you replace the `ContactSwitchManufCluster` with this version?:
```python
class ContactSwitchManufCluster(TuyaMCUCluster):
    """Tuya Manufacturer Cluster with ContactSwitch Sensor data points."""

    dp_to_attribute: Dict[int, DPToAttributeMapping] = {
        1: DPToAttributeMapping(
            ContactSwitchCluster.ep_attribute,
            "zone_status",
            dp_type=TuyaDPType.BOOL,
            converter=lambda x: 1 if x else 0,
        ),
        2: DPToAttributeMapping(
            DoublingPowerConfigurationCluster.ep_attribute,
            "battery_percentage_remaining",  # 0x0021
            dp_type=TuyaDPType.VALUE,
        ),
    }

    data_point_handlers = {
        1: "_dp_2_attr_update",
        2: "_dp_2_attr_update",
    }
```

Device is also reporting a `dp_101` with some values (718, 862, ...):
```
2022-05-28 19:56:18 DEBUG (MainThread) [zigpy.zcl] [0x711F:1:0xef00] No datapoint handler for TuyaCommand(status=0, tsn=9, dp=101, data=TuyaData(dp_type=<TuyaDPType.VALUE: 2>, function=0, raw=b'\xce\x02\x00\x00', *payload=718))
2022-05-28 19:56:18 DEBUG (MainThread) [zigpy.zcl] [0x711F:1:0xef00] No datapoint handler for TuyaCommand(status=0, tsn=11, dp=101, data=TuyaData(dp_type=<TuyaDPType.VALUE: 2>, function=0, raw=b'^\x03\x00\x00', *payload=862))
```

But I have no idea what these values can be.

centershock, 2022-05-29:
The update for ias_zone works now. Values are "on" for an "open window", "off" for a "closed window". Thank you very much so far!

For the battery_percentage_remaining. I tried to get `battery_percentage_remaining (id: 0x0021)` from `DoublingPowerConfigurationCluster (Endpoint id: 1, ...)` but there is no incoming value. Following the log, I can see an error but don't know what it means ;)

```
2022-05-29 20:20:50 DEBUG (MainThread) [zigpy.zcl] [0x711F:1:0x0001] Sending request header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=False, is_reply=0, disable_default_response=0, reserved=0, *is_cluster=False, *is_general=True), tsn=69, command_id=<GeneralCommand.Read_Attributes: 0>, *is_reply=False)
2022-05-29 20:20:51 DEBUG (MainThread) [zigpy.zcl] [0x711F:1:0x0001] Sending request: Read_Attributes(attribute_ids=[33])
2022-05-29 20:20:51 DEBUG (MainThread) [zigpy.device] [0x711f] Extending timeout for 0x45 request
2022-05-29 20:20:51 DEBUG (MainThread) [zigpy_deconz.zigbee.application] Sending Zigbee request with tsn 69 under 70 request id, data: b'0045002100'
2022-05-29 20:20:51 DEBUG (MainThread) [zigpy_deconz.api] Command Command.aps_data_request (20, 70, <DeconzSendDataFlags.NONE: 0>, <DeconzAddressEndpoint address_mode=2 address=0x711F endpoint=1>, 260, 1, 1, b'\x00E\x00!\x00', <DeconzTransmitOptions.USE_NWK_KEY_SECURITY: 2>, 0)
2022-05-29 20:20:51 DEBUG (MainThread) [zigpy_deconz.api] APS data request response: [2, <DeviceState.APSDE_DATA_REQUEST_SLOTS_AVAILABLE|2: 34>, 70]
2022-05-29 20:20:58 DEBUG (MainThread) [zigpy_deconz.api] Device state changed response: [<DeviceState.128|APSDE_DATA_REQUEST_SLOTS_AVAILABLE|APSDE_DATA_CONFIRM|2: 166>, 0]
2022-05-29 20:20:58 DEBUG (MainThread) [zigpy_deconz.api] Command Command.aps_data_confirm (0,)
2022-05-29 20:20:58 DEBUG (MainThread) [zigpy_deconz.api] APS data confirm response for request with id 70: f0
2022-05-29 20:20:58 DEBUG (MainThread) [zigpy_deconz.api] Request id: 0x46 'aps_data_confirm' for <DeconzAddressEndpoint address_mode=ADDRESS_MODE.NWK address=0x711f endpoint=1>, status: 0xf0
2022-05-29 20:20:58 DEBUG (MainThread) [zigpy_deconz.zigbee.application] Error while sending 70 req id frame: TXStatus.MAC_TRANSACTION_EXPIRED
2022-05-29 20:20:58 DEBUG (MainThread) [zigpy.device] [0x711f] Delivery error for seq # 0x45, on endpoint id 1 cluster 0x0001: message send failure
```

javicalle, 2022-05-29:
I can't say what is the reported error, but not seems related to the battery status.
The logs from a previous comment show several `dp_2` reports:
* https://github.com/zigpy/zha-device-handlers/issues/1502#issuecomment-1111357062

Maybe if battery is low, the report is less frequent, but just guessing.

centershock, 2022-05-29:
Maybe not related, but i tried some more, and there I produced a stacktrace, maybe that helps.

```
2022-05-29 20:44:22 DEBUG (MainThread) [zigpy_deconz.api] Device state changed response: [<DeviceState.128|APSDE_DATA_REQUEST_SLOTS_AVAILABLE|APSDE_DATA_CONFIRM|2: 166>, 0]
2022-05-29 20:44:22 DEBUG (MainThread) [zigpy_deconz.api] Command Command.aps_data_confirm (0,)
2022-05-29 20:44:22 DEBUG (MainThread) [zigpy_deconz.api] APS data confirm response for request with id 157: f0
2022-05-29 20:44:22 DEBUG (MainThread) [zigpy_deconz.api] Request id: 0x9d 'aps_data_confirm' for <DeconzAddressEndpoint address_mode=ADDRESS_MODE.NWK address=0x711f endpoint=1>, status: 0xf0
2022-05-29 20:44:22 DEBUG (MainThread) [zigpy_deconz.zigbee.application] Error while sending 157 req id frame: TXStatus.MAC_TRANSACTION_EXPIRED
2022-05-29 20:44:22 DEBUG (MainThread) [zigpy.device] [0x711f] Delivery error for seq # 0x9c, on endpoint id 1 cluster 0x0001: message send failure
2022-05-29 20:44:22 ERROR (MainThread) [homeassistant.components.websocket_api.http.connection] [547226071200] Error handling message: Unknown error (unknown_error)
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/websocket_api/decorators.py", line 27, in _handle_async_response
    await func(hass, connection, msg)
  File "/usr/src/homeassistant/homeassistant/components/zha/api.py", line 762, in websocket_read_zigbee_cluster_attributes
    success, failure = await cluster.read_attributes(
  File "/usr/local/lib/python3.9/site-packages/zigpy/zcl/__init__.py", line 441, in read_attributes
    result = await self.read_attributes_raw(to_read, manufacturer=manufacturer)
  File "/usr/local/lib/python3.9/site-packages/zigpy/quirks/__init__.py", line 212, in read_attributes_raw
    return await super().read_attributes_raw(
  File "/usr/local/lib/python3.9/site-packages/zigpy/device.py", line 315, in request
    raise zigpy.exceptions.DeliveryError(
zigpy.exceptions.DeliveryError: [0x711f:1:0x0001]: Message send failure
```

javicalle, 2022-05-29:
Ummm, there is a report from other quirk that `DoublingPowerConfigurationCluster` isn't working. Let's try with the same approach:

1. add a new `TuyaPowerConfigurationCluster` class (above the `ContactSwitchManufCluster` definition):
```python
class TuyaPowerConfigurationCluster(
    TuyaLocalCluster, DoublingPowerConfigurationCluster
):
    """PowerConfiguration cluster for battery-operated tuya devices reporting percentage."""

```

2. Use the new `TuyaPowerConfigurationCluster` class as replacement:
```python
    replacement = {
        ENDPOINTS: {
            1: {
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    TuyaPowerConfigurationCluster,
                    ContactSwitchCluster,
                    ContactSwitchManufCluster,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            }
        }
    }
```

centershock, 2022-05-29:
Did this, when i try to get the attribute for `TuyaPowerConfigurationCluster` and the battery_remaining_percentage I get a 0 (more than I got before ;)) but the 0 is constant...

javicalle, 2022-05-29:
Are you sure?
The proposed implementation for `TuyaPowerConfigurationCluster` is a new one that you must add in your quirk. Is not the existing from `zhaquirks.tuya`.

centershock, 2022-05-29:
[image]
[ts0601_magnet.py.txt](https://github.com/zigpy/zha-device-handlers/files/8794230/ts0601_magnet.py.txt)

I attached a screen of the Cluster-Management and attached my current `ts0601_magnet.py`. I tried to follow your instructions... ;)

domi190399, 2022-10-16:
I tested the qirk, posted [here](https://github.com/zigpy/zha-device-handlers/issues/1599) and at least the Occupancy works for me. Battery state is still listed as unavailable. Also as least my version of the sensor seems to have a luminance sensor which is not shown at all.

darouss99, 2022-12-11:
Hi, I use this quirks for my ZG-102ZL (TZE200_pay2byax), the door sensor work great, and have the battery level.  But I don't have the illuminance mesurement. I tried to modify the Quirks I but didn't get the illuminance working.  I'm a noob for this kind of script. Is someone have a solution and have the illuminance working?  Thank you
[ts0601_magnet.py.txt](https://github.com/zigpy/zha-device-handlers/files/10203080/ts0601_magnet.py.txt)

javicalle, 2022-12-11:
If the illuminance is the already reported `dp=101` (but I don't know) you could try to change this part in your local quirk:
```python
        101: DPToAttributeMapping(
            TuyaIlluminanceMeasurement.ep_attribute,
            "measured_value",
            dp_type=TuyaDPType.VALUE,
            converter=lambda x: 10000.0 * math.log10(x + 1.0),
        ),
```

darouss99, 2022-12-27:
Hi, Thank you for you help, but unfortunately that didn't work.

Don't know if it can help, there is the "reconfigure" result on Home Assistant on this sensor.

[image]

How can I find the DP number?

Thank you

javicalle, 2022-12-27:
You will need to enable the debug logs and analyze the ones from your device:

https://www.home-assistant.io/integrations/zha/#debug-logging

Look for the Nwk of your device. You should get the report messages from your device. The messages will include some `dp=101` (or whatever). Then you need to match the reported value with its meaning (iluminance, battery,...)

antoweb, 2022-12-28:
> Try if this quirk works: [ts0601_magnet.zip](https://github.com/zigpy/zha-device-handlers/files/8500583/ts0601_magnet.zip)

Hello, how to use this zip file?

antoweb, 2022-12-28:
> 

Helo same issue but my pir sensor wont work. Wich py file have used you?

antoweb, 2022-12-28:
> Hi, I use this quirks for my ZG-102ZL (TZE200_pay2byax), the door sensor work great, and have the battery level. But I don't have the illuminance mesurement. I tried to modify the Quirks I but [...]

Hello same issue here but my pir sensor wont work. Wich py file have used you?

javicalle, 2022-12-28:
> Hello, how to use this zip file?

You need to enable the local quirk folder:
* https://github.com/zigpy/zha-device-handlers/discussions/693#discussioncomment-857274

Once enabled create a new file with the proposed quirk, save changes, restart HA and repair the device again. Take the quirk version from:
* https://github.com/zigpy/zha-device-handlers/issues/1502#issuecomment-1118930342

> Hello same issue here but my pir sensor wont work.

The proposed quirk is for a door sensor (magnet) not a PIR sensor.

antoweb, 2022-12-28:
> > Hello, how to use this zip file?
> 
> You need to enable the local quirk folder:
> 
> * [Editing (ZHA) quirks in a (HA) container or local. #693 (comment)](https://github.com/zigpy/zha-device-handlers/discussions/693#discussioncomment-857274)
> 
> Once enabled create a new file with the proposed quirk, save changes, restart HA and repair the device again. Take the quirk version from:
> 
> * [[Device Support Request] Tuya ZG-102ZL / _TZE200_pay2byax #1502 (comment)](https://github.com/zigpy/zha-device-handlers/issues/1502#issuecomment-1118930342)
> 
> > Hello same issue here but my pir sensor wont work.
> 
> The proposed quirk is for a door sensor (magnet) not a PIR sensor.

Sorry mine is a door sensor. And works thanks

javicalle, 2022-12-28:
Could someone try with this version? :
```python
"""Tuya contact sensor device."""

from typing import Dict

from zigpy.profiles import zha
from zigpy.quirks import CustomDevice
from zigpy.zcl.clusters.general import Basic, Ota, PowerConfiguration, Time
from zigpy.zcl.clusters.security import IasZone

from zhaquirks import LocalDataCluster, DoublingPowerConfigurationCluster

from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODELS_INFO,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
    ZONE_STATUS,
    ZONE_TYPE,
)

from zhaquirks.tuya import TuyaIlluminanceMeasurement, TuyaLocalCluster

from zhaquirks.tuya.mcu import DPToAttributeMapping, TuyaAttributesCluster, TuyaMCUCluster, TuyaDPType

IASZONE_ZONE_STATUS_ATTR = 0x0002 

class ContactSwitchCluster(TuyaLocalCluster, IasZone):
    """Tuya ContactSwitch Sensor."""

    _CONSTANT_ATTRIBUTES = {ZONE_TYPE: IasZone.ZoneType.Contact_Switch}

    def _update_attribute(self, attrid, value):
        self.debug("_update_attribute '%s': %s", attrid, value)
        super()._update_attribute(attrid, value)

class ContactSwitchManufCluster(TuyaMCUCluster):
    """Tuya Manufacturer Cluster with ContactSwitch Sensor data points."""

    dp_to_attribute: Dict[int, DPToAttributeMapping] = {
        1: DPToAttributeMapping(
            ContactSwitchCluster.ep_attribute,
            "zone_status",
            dp_type=TuyaDPType.BOOL,
            converter=lambda x: IasZone.ZoneStatus.Alarm_1 if x else 0,
        ),
        2: DPToAttributeMapping(
            DoublingPowerConfigurationCluster.ep_attribute,
            "battery_percentage_remaining",  # 0x0021
            dp_type=TuyaDPType.VALUE,
        ),
        101: DPToAttributeMapping(
            TuyaIlluminanceMeasurement.ep_attribute,
            "measured_value",
            dp_type=TuyaDPType.VALUE,
            converter=lambda x: 10000.0 * math.log10(x + 1.0),
        ),
    }

    data_point_handlers = {
        1: "_dp_2_attr_update",
        2: "_dp_2_attr_update",
        101: "_dp_2_attr_update",
    }

class TuyaContactSensor(CustomDevice):
    """Tuya contact sensor device."""

    signature = {
        # NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)]
        # device_version=1
        # SizePrefixedSimpleDescriptor(endpoint=1, profile=260, device_type=1026, device_version=1,
        # input_clusters=[0, 1, 1280],
        # output_clusters=[10, 25])
        MODELS_INFO: [("_TZE200_pay2byax", "TS0601")],
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.IAS_ZONE,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    PowerConfiguration.cluster_id,
                    IasZone.cluster_id,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            }
        },
    }

    replacement = {
        ENDPOINTS: {
            1: {
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    DoublingPowerConfigurationCluster,
                    ContactSwitchCluster,
                    ContactSwitchManufCluster,
                    TuyaIlluminanceMeasurement, 
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            }
        }
    } 
```

darouss99, 2022-12-29:
@javicalle Yes it seem to work,  But the value seem a bit strange if I compare with my cellphone luxmeter.

4.0 lux = 10 lux on my cellphone
66 lux = 184 lux on my cellphone
145 lux = 407 lux on mycell phone

So the converter formula seem to be wrong.  But we have values now. Thank you.

javicalle, 2022-12-29:
You can try with this other version for the converter (but I keep thinking that is wrong):
```python
            converter=lambda x: 10000 * math.log10(x) + 1,
```

darouss99, 2022-12-29:
The converter seem to be a little bit better, but still wrong.

7.0 lux = 18 lux on my phone
95 lux = 241 lux on my phone
342 lux = 474 lux on my phone

`converter=lambda x: 10000 * math.log10(x) + 1,`

javicalle, 2022-12-29:
None of the conversions seem appropriate.
Can you get and post the raw values for several measures. Just comment the `converter` line and take some values as you have done until now.

darouss99, 2022-12-29:
Yes for sure.  I commented the converter. There is some exemple of value on the sensor.

1.0 lux = 15 lux on cellphone = 6 (Mesured_value (id:0x0000))
1.0 lux = 216 on cellphone = 180 (Mesured_value (id:0x0000))
1.1 lux = 485 lux on cellphone = 502 (Mesured_value (id:0x0000))

So the raw value seem about good without the converter, but homeassistant don't show the good value in the sensor.

[image]
[image]

domusonline, 2023-01-11:
> Hi. I bought this same sensor from AliExpress and I'm having issues with it. Keep in mind that I started using HA 2-3 weeks ago... I created the quirk file based on the above suggestions and it is able to identify the sensor and it's usable as a contact sensor. But although it shows the luminance entity it always shows "unknown". I tried to look at the file, but I never touched python... and I didn't try to look at the logs... Besides this the sensor behavior looks a bit strange: 1- If I remove it from HA it seems to enter pairing mode automatically 2- It seems to identify differently on different attempts. In most situations I end up with a sensor named "unk_manufacturer unk_model". I've found an issue reported some time ago that was fixed (and not related to this device specifically). 3- After some attempts of pairing I get the behavior above: contact and battery seem to work (contact I am sure, battery shows 100% which I assume is correct)
> 
> I connected the Tuya app to my gateway, paired it, and verified the luminance sensor is there and works (it responds to different light conditions with what seems to be reasonable values).
> 
> So I'm requesting what was the final "quirk" content? I'd like to be sure I'm using the correct values. If with the final one it doesn't work, what should I do? Collect debug info and share it here? Could it be that different sensor versions require different configurations? Thanks!

After some debug I noticed in the logs it was complaining about "math not found". I added "import math" at the top and I'm now able to get the three entities (open/close, battery and illuminance.
I'm still having weird behaviors in the pairing.
The lux value is updated when I open/close. Apart from that it seems to be updated but with a long interval (I assume that's normal in this type of sensors).

The quirk I have at this moment is (note the "import math" - newbie, no experience with Python - maybe that's not the best way to solve my issue) :

```
"""Tuya contact sensor device."""

import math

from typing import Dict

from zigpy.profiles import zha
from zigpy.quirks import CustomDevice
from zigpy.zcl.clusters.general import Basic, Ota, PowerConfiguration, Time
from zigpy.zcl.clusters.measurement import IlluminanceMeasurement
from zigpy.zcl.clusters.security import IasZone

from zhaquirks import LocalDataCluster, DoublingPowerConfigurationCluster

from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODELS_INFO,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
    ZONE_STATUS,
    ZONE_TYPE,
)

from zhaquirks.tuya import TuyaLocalCluster

from zhaquirks.tuya.mcu import DPToAttributeMapping, TuyaAttributesCluster, TuyaMCUCluster, TuyaDPType

IASZONE_ZONE_STATUS_ATTR = 0x0002

class ContactSwitchCluster(TuyaAttributesCluster, IasZone):
    """Tuya ContactSwitch Sensor."""

    _CONSTANT_ATTRIBUTES = {ZONE_TYPE: IasZone.ZoneType.Contact_Switch}

    def _update_attribute(self, attrid, value):
        self.debug("_update_attribute '%s': %s", attrid, value)
        super()._update_attribute(attrid, value)

class TuyaPowerConfigurationCluster(
    TuyaLocalCluster, DoublingPowerConfigurationCluster
):
    """PowerConfiguration cluster for battery-operated tuya devices reporting percentage."""

class TuyaIlluminanceMeasurement(IlluminanceMeasurement, TuyaLocalCluster):
    """Tuya local IlluminanceMeasurement cluster."""

class ContactSwitchManufCluster(TuyaMCUCluster):
    """Tuya Manufacturer Cluster with ContactSwitch Sensor data points."""

    dp_to_attribute: Dict[int, DPToAttributeMapping] = {
        1: DPToAttributeMapping(
            ContactSwitchCluster.ep_attribute,
            "zone_status",
            dp_type=TuyaDPType.BOOL,
        ),
[... 35 lines cut ...]
                    PowerConfiguration.cluster_id,
                    IasZone.cluster_id,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            }
        },
    }

    replacement = {
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.IAS_ZONE,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    TuyaPowerConfigurationCluster,
                    ContactSwitchCluster,
                    ContactSwitchManufCluster,
                    TuyaIlluminanceMeasurement,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            }
        }
    }
```

domusonline, 2023-01-11:
It seems I was over optimistic... After some more testing I noticed it apparently wasn't working in the dark...
Looking at the logs I got:

```
 File "/usr/local/lib/python3.10/site-packages/zhaquirks/tuya/init.py", line 1498, in _dp_2_attr_update
    value = dp_map.converter(value)
  File "/config/custom_zha_quirks/teste.py", line 67, in <lambda>
    converter=lambda x: 10000 * math.log10(x) + 1,
ValueError: math domain error

```
It seems that in darkness the values read cause a problem in the formula...

javicalle, 2023-01-12:
Can you try with this formula?
```python
converter=lambda x: 10000 * math.log10(x + 1)
```

domusonline, 2023-01-12:
> Can you try with this formula?
> 
> ```python
> converter=lambda x: 10000 * math.log10(x + 1)
> ```

Sorry... I should have updated the thread, but I finished this yesterday too late... I managed to solve it/workaround it by using this (I got it from another issue):
converter=lambda x: (10000.0 * math.log10(x) + 1.0 if x != 0 else 0),

not sure which one would be better.
The problem is that in fact the sensor returns 0 in dark conditions. And math.log10(0) raises an error. So I'm sure your suggestion would work. Which one would be better?
I was doing some tests and apparently the range of values I get are not between 1-1000 (max I've seen was 960).

Thank you!

domusonline, 2023-01-13:
>  I was doing some tests and apparently the range of values I get are not between 1-1000 (max I've seen was 960).
> 
> Thank you!

Sorry... (again). I meant that the values I get ARE between 1-1000 which sound reasonable...

javicalle, 2023-01-14:
> > I was doing some tests and apparently the range of values I get are not between 1-1000 (max I've seen was 960).
> > Thank you!
> 
> Sorry... (again). I meant that the values I get ARE between 1-1000 which sound reasonable...

But you are getting 0 values, if not this wouldn't be needed:
```
if x != 0 else 0
```

domusonline, 2023-01-14:
> But you are getting 0 values, if not this wouldn't be needed:
>
> ```
> if x != 0 else 0
> ```

I'm not making myself clear... what I intended to say was that yes, the "x" receives 0 (I assume this is the value returned by the sensor). And math.log10(0) raises an error.
But with the formula:

converter=lambda x: (10000.0 * math.log10(x) + 1.0 if x != 0 else 0),

what I get in HA when I check the entity are values between 1 and ~1000 (if I point a mobile phone flash light I get for example 960 or 980).
And it responds to different light conditions with values somewhere inside this range.
I was wondering which formula you would think it would be better (I'd say they would give ver similar values) and also if you think this range of values is reasonable.

Thanks for your support.

javicalle, 2023-01-14:
> also if you think this range of values is reasonable

I can't say it because these are not familiar to me.
I'm sure that both formulas would give a similar values/behavior. But the one I suggest wouldn't give negative values for illumination when 0<x<1 🤷🏻‍♂️ 

Anyone can create the PR with the proposed quirk and I'm sure that would be accepted as is.

GillesMattioli, 2023-02-10:
https://github.com/zigpy/zha-device-handlers/issues/1502#issuecomment-1378743535
Cette solution fonctionne avec un _TZE200_pay2byax
Merci de votre travail

MattWestb, 2023-02-10:
>https://github.com/zigpy/zha-device-handlers/issues/1502#issuecomment-1378743535
This solution works with a _TZE200_pay2byax
Thank you for your work

antoweb, 2023-03-02:
Hello, after last update HA to 2023.3 this quirks generate error and all Zigbee Integration cannot load. i have to remove from custom_zha_quirks any solution?

domusonline, 2023-03-02:
> Hello, after last update HA to 2023.3 this quirks generate error and all Zigbee Integration cannot load. i have to remove from custom_zha_quirks any solution?

I had issues with the quirk also and then some errors in some automations based on Zigbee buttons. I didn't try to remove the quirk to understand if the other issues were related....
I just restored the backup... meanwhile I see version 2023.3.1 but HA insists that the latest one is 2023.3.0.... I'll get back to this when 3.1 is possible to install...
The error complains about an import of some Tuya classes....

antoweb, 2023-03-02:
https://github.com/zigpy/zha-device-handlers/issues/2241#issuecomment-1452487362

[12 more replies]

closed 2024-12-16 by github-actions[bot]