zigpy/zha-device-handlers #1504: [Device Support Request] Show state change ACK in icon

Tropaion, 2022-04-16:
Not sure if this a problem for this repo but the bulb icon of all my tuya light switches show if the state change was ACK (on/off).
Now I have this device: https://www.amazon.de/gp/product/B08KZK69NR/ref=ppx_yo_dt_b_asin_title_o05_s00?ie=UTF8&psc=1
The quirk applies correctly and it also works: ```zhaquirks.tuya.ts011f_switch.Tuya_1G_Switch```.
But the icon of this device doesn't show if the device acknowledged the state change except for in the details?
[image]
[image]

Tropaion, 2022-04-17:
I now discovered that this isn't only a problem with the icon.
I added the device to my zigbee group where I have all lights in it.
Sending on/off commands via groups works without a problem, but the state isn't reported and with it, the group state is also not correct.
For example, when all device are off and I turn this one on, the groups still says all is off.

Tropaion, 2022-04-17:
@javicalle @MattWestb I managed to "fix" this behaviour by changing the device type, but I don't think it's intended to do it like that, it should work with the original device type the same.
<details>
<summary>ts011f_switch.py</summary>
```python3
"""Tuya TS011F Switches."""

from zigpy.profiles import zha
from zigpy.zcl.clusters.general import (
    Basic,
    GreenPowerProxy,
    Groups,
    Identify,
    OnOff,
    Ota,
    Scenes,
    Time,
)

from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODEL,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
)
from zhaquirks.tuya import (
    TuyaSwitch,
    TuyaZBE000Cluster,
    TuyaZBExternalSwitchTypeCluster,
    TuyaZBOnOffAttributeCluster,
)

class Tuya_2G_Switch(TuyaSwitch):
    """Tuya 2 gang light switch."""

    signature = {
        MODEL: "TS011F",
        ENDPOINTS: {
            # <SimpleDescriptor endpoint=1 profile=260 device_type=266
            # input_clusters=[0, 3, 4, 5, 6, 57344, 57345]
            # output_clusters=[10, 25]>
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_PLUG_IN_UNIT,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Identify.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    OnOff.cluster_id,
                    TuyaZBE000Cluster.cluster_id,
                    TuyaZBExternalSwitchTypeCluster.cluster_id,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            },
            # <SimpleDescriptor endpoint=2 profile=260 device_type=266
            # input_clusters=[3, 4, 5, 6, 57344, 57345]
            # output_clusters=[]>
            2: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_PLUG_IN_UNIT,
                INPUT_CLUSTERS: [
                    Identify.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    OnOff.cluster_id,
                    TuyaZBE000Cluster.cluster_id,
                    TuyaZBExternalSwitchTypeCluster.cluster_id,
                ],
                OUTPUT_CLUSTERS: [],
            },
            # <SimpleDescriptor endpoint=242 profile=41440 device_type=97
            # input_clusters=[]
            # output_clusters=[33]>
            242: {
                PROFILE_ID: 41440,
                DEVICE_TYPE: 97,
                INPUT_CLUSTERS: [],
                OUTPUT_CLUSTERS: [GreenPowerProxy.cluster_id],
            },
        },
    }
    replacement = {
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_PLUG_IN_UNIT,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Identify.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaZBOnOffAttributeCluster,
                    TuyaZBE000Cluster,
                    TuyaZBExternalSwitchTypeCluster,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            },
            2: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_PLUG_IN_UNIT,
                INPUT_CLUSTERS: [
                    Identify.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaZBOnOffAttributeCluster,
                    TuyaZBE000Cluster,
                    TuyaZBExternalSwitchTypeCluster,
                ],
                OUTPUT_CLUSTERS: [],
            },
            242: {
                PROFILE_ID: 41440,
                DEVICE_TYPE: 97,
                INPUT_CLUSTERS: [],
                OUTPUT_CLUSTERS: [GreenPowerProxy.cluster_id],
            },
        },
    }

class Tuya_1G_Switch(TuyaSwitch):
    """Tuya 1 gang light switch with RF433."""

    signature = {
        MODEL: "TS011F",
        ENDPOINTS: {
            # <SimpleDescriptor endpoint=1 profile=260 device_type=9
            # input_clusters=[0, 4, 5, 6,]
            # output_clusters=[10, 25]>
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.MAIN_POWER_OUTLET,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    OnOff.cluster_id,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            },
        },
    }
    replacement = {
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_LIGHT,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaZBOnOffAttributeCluster,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            },
        },
    }
```
</details>

TheJulianJES, 2022-04-17:
Switches do not color their icon based on their current state unless you turn on ``state_color`` in the frontend.
This is intended since Home Assistant 0.105: https://www.home-assistant.io/blog/2020/02/05/release-105/#icons-no-longer-change-color-based-on-state

Tropaion, 2022-04-17:
@TheJulianJES I know, but as far as I understood, in this case the icon color changes according to "HA state" and not the device state, so when I turn the device on in the UI the icon changes even when the device doesn't send an ACK response.
An this doesn't cover this: https://github.com/zigpy/zha-device-handlers/issues/1504#issuecomment-1100875917

TheJulianJES, 2022-04-17:
Ah, so the issue only seems to exist when using groups then?
And if you replace the plug ``device_type`` with a light using a custom quirk, groups update their state correctly (and also the single "plug" state)?

MattWestb, 2022-04-17:
Changing the device profile can being problematic with some device that is not liking it and is not recommended by the maintainers. Sadly have HA taking away customize eateries for some version go that was possible making the same thing / eateries  in HA and not mixing the underlying system. 

I have mounting one Aqara door / window sensor in one normal smoke detector and changing its OnOff to fire and its working great but not possible making in the GUI for the moment only editing yaml files.

But you thing is in the ZHA then you is using Zignee groups i think ( = not knowing).

Tropaion, 2022-04-17:
@TheJulianJES No, these two are separate issues, but I thought they have the same underlying reason.
I know that I can enable the icon color change, but I'm not sure if it represents the state which HA thinks the device has or the real state what ZHA says the device has. I thought it was the first one, that's why I created this issue and wanted to make sure.
The second one with the groups was discovered later and I thought it is similar to this, but it was probably offtopic.

@MattWestb So you are not sure either what the problem with ZigBee Groups is?
Currently it looks like changing device type didn't make any problems until now, would rather say it works better with groups.
Don't know if its allowed to make an PR for this so I will just continue using it with a custom quirk.

dmulcahey, 2022-04-17:
What is the issue with groups… I’m not sure I understand what is wrong

Tropaion, 2022-04-17:
> I added the device to my zigbee group where I have all lights in it.
> Sending on/off commands via groups works without a problem, but the state isn't reported and with it, the group state is also not [...]

This

dmulcahey, 2022-04-17:
What is the state of all of the members in the group? Group state is derived from members

Tropaion, 2022-04-17:
I know, didn't I write in the commend?
For example, when all devices in the group are off and I turn this device on, the groups state still says off.

TheJulianJES, 2022-04-17:
Just to confirm, can you check if the switches are still added to the groups in Home Assistant (check if all switches still show up under Settings -> Integrations -> ZHA -> Groups -> select your switch group)?

(There was an issue a couple of versions back where ZHA sometimes lost which devices were in a group -- even though the devices themselves still responded)

Tropaion, 2022-04-17:
Yes they all show up and I can also switch them all.
[image]

dmulcahey, 2022-04-17:
> I know, didn't I write in the commend?
> For example, when all devices in the group are off and I turn this device on, the groups state still says off.

Look at light.py in zha. You can modify turn on and turn off to log the responses from the commands. This should help determine what’s wrong.

Tropaion, 2022-04-17:
What do you mean with modify? What should I modify?

dmulcahey, 2022-04-17:
Actually we already log the response if not successful. Turn debug logging on and look for these logs: https://github.com/home-assistant/core/blob/42c448c422c1e7f129f0a693c1757972cd87abec/homeassistant/components/zha/light.py#L246

closed 2022-10-21 by github-actions[bot]