zigpy/zha-device-handlers #1505: [Device Support Request] Alchemy VIII AL8TC13W-AP downlight GW-ZLTH10W01 no temperature control

drothacker, 2022-04-17:
The same issue as reported and resolved for zigbee2mqtt here https://github.com/Koenkk/zigbee2mqtt/issues/10966

The following device only has Colour temperature control and brightness but after adding it to ZHA it will allow RGB Colour that doesn't work and color temperature is missing.
https://urneeds.com.au/13w-90mm-smart-zigbee-tri-colour-led-downlight-kit-for-echo-plus-smartthings-hub-hue/

Zigbee device signature

{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=0, maximum_buffer_size=80, maximum_incoming_transfer_size=160, server_mask=0, maximum_outgoing_transfer_size=160, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "11": {
      "profile_id": 49246,
      "device_type": "0x0220",
      "in_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0008",
        "0x0300"
      ],
      "out_clusters": []
    },
    "13": {
      "profile_id": 49246,
      "device_type": "0xe15e",
      "in_clusters": [
        "0x1000"
      ],
      "out_clusters": [
        "0x1000"
      ]
    }
  },
  "manufacturer": "ALCHEMY_VIII",
  "model": "AL8TC13W-AP",
  "class": "zigpy.device.Device"
}

TheJulianJES, 2022-04-17:
Can you open the device settings in Home Assistant (where the device signature button is), then click on "manage clusters", select the ``Color`` cluster in the dropdown menu. For "cluster attribute", select ``color_capabilities``, then click on "Read zigbee attribute" and send the output of what's in the text field here.

drothacker, 2022-04-17:
Hi,
Color cluster > "cluster attribute", > color_capabilities > "Get zigbee attribute"
color_capabilities = ColorCapabilities.Color_temperature

[image]

closed 2022-10-21 by github-actions[bot]