zigpy/zha-device-handlers #1509: [Device Support Request] Immax Neo Smart Thermostatic Head 07732L

cornelsavu, 2022-04-18:
*Hello all,
Is it possible to add support in zigbeetomqtt for immax neo smart thermostat model 07732L.
On the device compatibility list i see it has support in ZHA.
ZigbeeToMqtt detects it successfully but it says unsupported and cannot control it in anyway.
Please help.
Big thanks

MattWestb, 2022-04-18:
If you device have the manufacture ID `_TZE200_4eeyebrt` you shall getting basic functions and the rest on cluster https://github.com/zigpy/zha-device-handlers/issues/1199.
Its also one extended quirk that have more function in the ZHA GUI if you like trying it but its need being installed as local quirk.

cornelsavu, 2022-04-19:
Hi Matt,
Big thx for your quick reply.
I dont have ZHA, only ZigbeeToMqtt will this quirk still work?

MattWestb, 2022-04-19:
No its ZHA-Quirks is only working with ZHA.

cornelsavu, 2022-04-19:
Understood
Btw, also a big thx for all your work here on github. Your solutions helped me with many other issues.

Tropaion, 2022-04-19:
@cornelsavu If your issue is solved, please close it, thank you 😄

closed 2022-04-20 by cornelsavu