zigpy/zha-device-handlers #1512: [Device Support Request] _TZE200_c2fmom5z", "TS0601

Hogesyx, 2022-04-18:
This new Tuya Air Box seems to be similar to the existing TuyaCO2SensorGPP class(ts0601_air_quality.py), I could get the sensor up and working just by adding the new model info entry.

        MODELS_INFO: [
            ("_TZE200_ryfmq5rl", "TS0601"),
            ("_TZE200_yvx5lh6k", "TS0601"),
            ("_TZE200_dwcarsat", "TS0601"),
            ("_TZE200_c2fmom5z", "TS0601"),
        ],

However it seems that the VOC and HCHO sensor values position are swapped? Unless the VOC sensor excludes HCHO values in the calculation, I get impossible HCHO to VOC ratio.

For example
VOC: 8 µg/m³ 
HCHO: 19ppm (assuming unit is wrong and should be ppb, 19 ppb of HCHO is 0.023 mg/m3 which is way above the 0.008 mg/m3

Tropaion, 2022-04-19:
You should post the device signature so that people can better help.

padadof2, 2022-05-27:
"node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0051",
      "in_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0xef00"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "242": {
      "profile_id": 41440,
      "device_type": "0x0061",
      "in_clusters": [],
      "out_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "_TZE200_c2fmom5z",
  "model": "TS0601",
  "class": "zigpy.device.Device"
}

egelmex, 2022-08-04:
I also have this device, device signatures seem to match 
```json
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0051",
      "in_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0xef00"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "242": {
      "profile_id": 41440,
      "device_type": "0x0061",
      "in_clusters": [],
      "out_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "_TZE200_c2fmom5z",
  "model": "TS0601",
  "class": "zigpy.device.Device"
}
```

What other information would you need to get this added?

zatricky, 2022-11-30:
New to HA and also have this device. I have an identical signature to that which @egelmex provided. The device is detected and added but of course shows no useful sensor information from HA's perspective.

Please let me know if there's any additional steps I should take or information I could provide. :-)

zatricky, 2023-07-06:
Looks like this was fixed in #2073

closed 2024-01-09 by github-actions[bot]