zigpy/zha-device-handlers #1519: [BUG] ZBMini-L delayed response after minutes of pairing. Workaround found on Zigbee2MQTT

pulento, 2022-04-23:
**Describe the bug**
After minutes ZBMini-L switch is paired it shows a delayed response that seems to get worse with time.

**To Reproduce**
Steps to reproduce the behavior:
1. Pair the ZBMini-L with ZHA
2. It responds fast at the beginning
3. Wait 30 minutes or more
4. Switch it ON/OFF and note the delay
5. It gets workse with time

**Expected behavior**
No delay. But it seems to be a workaround for this behavior since it was fixed in Z2M on this commit:

https://github.com/Koenkk/zigbee-herdsman-converters/commit/ac7a095463c8b4dec3d6a6b34eb1e221a0f37d90

Bug was reported here:

https://github.com/Koenkk/zigbee2mqtt/issues/11676

TIA

hevili, 2022-04-23:
also experiencing this issue

MattWestb, 2022-04-23:
Very likely is not one problem with the host system more the device not working as intended.
Read little of the problem our Z2M friends is having and all reclamations of devices they have doing to SonOff.
https://github.com/Koenkk/zigbee2mqtt/issues/10282

Hedda, 2022-04-25:
If the Zigbee2MQTT developers could implement a workaround for this slow delay problem in a Zigbee converter for zigbee-herdsman-converters then maybe it is also possible to do the same in a ZHA Device Handlers quirk for the ZHA integration too?

https://github.com/Koenkk/zigbee2mqtt/issues/11676

Sound as if as a workaround they chose to unbind genPollCtrl to prevent the device from sending checkin message which causes Sonoff Zigbee End-Devices poll slower about 1-hour after powering the device?

https://github.com/Koenkk/zigbee-herdsman-converters/commit/ac7a095463c8b4dec3d6a6b34eb1e221a0f37d90

Hedda, 2022-04-25:
> Very likely is not one problem with the host system more the device not working as intended.

ITead SONOFF Product Manager @Daniel-zhan-itead confirmed that their SONOFF engineer analyses found that the root cause for Sonoff ZBMINI-L delays and slow response in Home Assistant is a firmware issue, but hope problem can be worked around in application.

"**_Our engineer analyzes the reason for this may be: After ZBMINI-L joins HA, HA will configure the Check-in message of the device. When HA receives the Check-in message from the device, it will modify the wake-up time of the device to 6S, which will cause the device to receive the control command delay. And our check-in period is 1 hour, so the control delay problem will occur after one hour. Is it possible for HA to handle polling time for ZBMINI-L especially?_**"

For reference; the Sonoff ZBMINI-L is a Zigbee End Device, (not a Zigbee Router ). Still, the product is a switch with a relay so it needs to respond quickly to control commands sent by the Zigbee application.

MattWestb, 2022-04-25:
One Zigbee 3 end device shall supporting pull control for not loosing messages the host system like sending it with out pulling its parent in short intervals all the time but doing check in to the host system..

Fast pull shall timing out if the hoist is not sending one fast pull stop that ZHA is sending but its looks the device is not being Zigbee compatible and / or not tested of the manufacture.

Its looks like TI have implanting many nice buts in the SDK and the companies need learning how to working then around or getting TI fixing then. 

Disabling chicken is not recommend but if the manufacture cant fixing the bug i think its the last resort for getting the device working or returning them to the seller (at least EU it shall being no problem if the seller cant fixing the device working OK) and perhaps some manufactures is learning how to do working devices and not only selling components in one plastic box with firmware that is not working.

caroku, 2022-05-06:
I have the same problem, my ZBMINI-L shows a delayed response after 1 hour, then appears offline the next day. I am from Chile and i bought the product from aliexpress, and returning to China is not an option. Great customer service. Hope you can help us.

DieserKirsche, 2022-05-06:
Expierencing exactly same problem with HA, ConBee II and ZHA integration. After initial setting up everthing works fine and later some wiered behavior of ZBMINI-L switch. All other devices (in same automation) work as expected.
ZBMINI-L as itself is perfect device for my purpose, as long as this issue will / can be fixed with my setup (HA, ConBee II and ZHA).

pulento, 2022-05-06:
Strangely I changed my coordinator to a Sonoff Dongle USB 3.0 from a ZBBridge and I've no experiencing big delays.

But what I'm having now is randoms turn off of the ZBMini-L not random turn on just random turn off on all of my 4 switches. I noted that other people reported the same problem on iTead page.

Daniel-zhan-itead, 2022-05-07:
We will optimize the firmware to fix the delay issue of ZBMINI-L running on ZHA and release the OTA update image for you.
The firmware is under testing, once the test is completed, it will be published.

dmulcahey, 2022-05-07:
> We will optimize the firmware to fix the delay issue of ZBMINI-L running on ZHA and release the OTA update image for you.
> The firmware is under testing, once the test is completed, it will be published.

Is the issue caused by something we are doing?

Daniel-zhan-itead, 2022-05-07:
> > We will optimize the firmware to fix the delay issue of ZBMINI-L running on ZHA and release the OTA update image for you.
> > The firmware is under testing, once the test is completed, it will be published.
> 
> Is the issue caused by something we are doing?

After ZBMINI-L is added to ZHA, ZHA will enable the device poll control cluster function, and then the device interval of sleep & wake-up will be set to 6 seconds as default,  We will disable the poll control function in the device firmware so the interval will be kept at 500ms as default.

jds11111, 2022-05-07:
> After ZBMINI-L is added to ZHA, ZHA will enable the device poll control cluster function, and then the device interval of sleep & wake-up will be set to 6 seconds as default, We will disable the [...]

Will this also fix the random shutdowns?

DieserKirsche, 2022-05-07:
> Expierencing exactly same problem with HA, ConBee II and ZHA integration. After initial setting up everthing works fine and later some strange behavior of ZBMINI-L switch. All other devices (in same automation) work as expected. ZBMINI-L as itself is perfect device for my purpose, as long as this issue will / can be fixed with my setup (HA, ConBee II and ZHA).

I had the same problem with the random turn off, not very often, sometimes after half an hour and sometimes immediatly after turning on (10s).

dmulcahey, 2022-05-07:
> After ZBMINI-L is added to ZHA, ZHA will enable the device poll control cluster function, and then the device interval of sleep & wake-up will be set to 6 seconds as default, We will disable the [...]

Why Is this cluster even part of this device. This is really meant for battery powered devices that may need to accept configuration commands from an application so that they don't go to sleep while they are in the middle of being configured no? seems odd to me that this cluster is on a mains powered relay.

pulento, 2022-05-07:
Maybe powering without Neutral hasn't much power either. If not why iTead didn't implemented it as Router devices ?

Hedda, 2022-05-09:
> Why Is this cluster even part of this device? This is really meant for battery powered end devices
> 
> > Maybe powering without Neutral hasn't much power either. If not why iTead didn't implemented it as Router devices ?

IIRC believe someone in Home Assistant community forum posted a link 
reference to a technical article that explained why power to non-neutral devices is unreliable/unstable or should at least not be assumed to be guaranteed, and therefore they should at least not be Zigbee Router devices.

https://github.com/Koenkk/zigbee2mqtt/issues/11676

Daniel-zhan-itead, 2022-05-09:
> Will this also fix the random shutdowns?

Yup, this random shutdown issue will also be fixed together, Pls wait for the new firmware release. Thanks for your understanding

pulento, 2022-05-09:
Great news 👍 

Now we have to figure how to apply OTA on these with HA

MattWestb, 2022-05-09:
Im having one tuya `TS110F _TZ3000_92chsky7` its one dual dimmer switch  without neutral that works great and is one router device so its depends of the hardware implementation and its no need having external capacitor for making it working stable.
If getting one OTA files from Sonoff / itead its only configure local OTA in HA and putting the file in the folder and restarting HA https://github.com/zigpy/zigpy/wiki/OTA-Device-Firmware-Updates if the original firmware have implanting working OTA function.

Also if its shall being one end device (not real sleeper) is better deleting the poll control cluster then making some crazy half implanting then its need pulling its parent very oft for not getting delay then sending commands to it. If it shall being one sleeper it shall having working pull control as one Zigbee 3 device for not loosing commands sent to it like all battery powered remotes is doing (that is Zigbee certificated).

Or the best is fixing the bug in the TI SDK or in the code made of the manufacture as shall being done before releasing the product if it was good tested.

Hedda, 2022-05-10:
> Now we have to figure how to apply OTA on these with HA

@Daniel-zhan-itead FYI, if you can add images to a public repo then Itead Sonoff OTA provider for it could be added to zigpy:

https://github.com/zigpy/zigpy/blob/dev/README.md#zigbee-device-ota-updates

https://github.com/zigpy/zigpy/tree/dev/zigpy/ota

https://github.com/zigpy/zigpy/wiki/OTA-Device-Firmware-Updates

Once your OTA provider is in zigpy then OTA support in Home Assistant's ZHA can be extended to support Itead Sonoff devices:

https://www.home-assistant.io/integrations/zha#ota-firmware-updates

PS: That will by the way also be similar for Zigbee2MQTT which will need a new OTA provider for zigbee-herdsman-converters:

https://github.com/Koenkk/zigbee-herdsman-converters/tree/master/lib/ota

https://www.zigbee2mqtt.io/guide/usage/ota_updates.html

DieserKirsche, 2022-05-20:
> Yup, this random shutdown issue will also be fixed together, Pls wait for the new firmware release. Thanks for your understanding

Do you have any updates about the timeschedule or much better a pre-release? Otherwise I have to refund my devices today.

Daniel-zhan-itead, 2022-05-20:
We have raised the pull request for the OTA update image on github https://github.com/Koenkk/zigbee-OTA/pulls.
Once it is approved, you can download the image and update the ZBMINI-L devices.

MattWestb, 2022-05-20:
Use https://github.com/Koenkk/zigbee-OTA/blob/a26ec426d0c744f63e6ee534bcee7875434d1008/images/Sonoff/zbmini-l_v1.1.1.ota and clicking on download and putting the file in your OTA folder you have reconfigured in Z/HA and restart HA and it shall being possible updating the device firmware.
Configuring and triggering OTA updated is in in the wiki https://github.com/zigpy/zigpy/wiki/OTA-Device-Firmware-Updates.

dmulcahey, 2022-05-20:
> We have raised the pull request for the OTA update image on github https://github.com/Koenkk/zigbee-OTA/pulls.
> Once it is approved, you can download the image and update the ZBMINI-L devices.

Do you have release notes for the firmware?

pulento, 2022-05-20:
OTA is aborting on my case:

```
2022-05-20 09:34:16 DEBUG (MainThread) [zigpy.zcl] [0xF9E4:1:0x0019] OTA query_next_image handler for 'SONOFF ZBMINI-L': field_control=0, manufacture_id=4742, image_type=1, current_file_version=4101, hardware_version=None, model=ZBMINI-L

2022-05-20 09:34:16 DEBUG (MainThread) [zigpy.zcl] [0xF9E4:1:0x0019] OTA image version: 4353, size: 131086. Update needed: True

2022-05-20 09:35:12 DEBUG (MainThread) [zigpy.zcl] [0xF9E4:1:0x0019] OTA upgrade_end handler for 'SONOFF ZBMINI-L': status=Status.ABORT, manufacturer_id=4742, image_type=1, file_version=4353

2022-05-20 09:35:16 DEBUG (MainThread) [zigpy.zcl] [0xF9E4:1:0x0019] OTA query_next_image handler for 'SONOFF ZBMINI-L': field_control=0, manufacture_id=4742, image_type=1, current_file_version=4101, hardware_version=None, model=ZBMINI-L
```

m2r57700, 2022-05-20:
Hi
First time I tried an OTA from ZHA and finally seems ok.

Previous Firmware: 0x00001004
Now : Firmware: 0x00001101

FYI

I used the following commands 

[image]

pulento, 2022-05-20:
> Hi First time I tried an OTA from ZHA and finally seems ok.
> 
> Previous Firmware: 0x00001004 Now : Firmware: 0x00001101
> 
> [image]
> 
> FYI
> 
> I used the following commands
> 
> [image]
> 
> I hope response delay will disappear.

My fault. I needed to upgrade latest HA, surely had mixed dependencies.

Now I see OTA file being uploaded. Will report if the firmware fixed the problems.

m2r57700, 2022-05-21:
I did the update 21h ago and device seems working without response delay.

jds11111, 2022-05-21:
For anyone seeking more details on accomplishing the OTA flash, I found this link the most helpful (also mentioned by Hedda): [https://github.com/zigpy/zigpy/wiki/OTA-Device-Firmware-Updates](https://github.com/zigpy/zigpy/wiki/OTA-Device-Firmware-Updates).  It worked fine for me, and took several minutes.

Daniel-zhan-itead, 2022-05-23:
> > We have raised the pull request for the OTA update image on github https://github.com/Koenkk/zigbee-OTA/pulls.
> > Once it is approved, you can download the image and update the ZBMINI-L devices.
> 
> Do you have release notes for the firmware?

Feature: Added Poll Control Cluster private attribute poll control 0xFF00 enable/disable configuration, the default Poll Control Cluster initial configuration is disabled
Bugfix: Random shutdown issue

Hedda, 2022-05-23:
> We have raised the pull request for the OTA update image on github https://github.com/Koenkk/zigbee-OTA/pulls.
> Once it is approved, you can download the image and update the ZBMINI-L devices.

@Daniel-zhan-itead Great to see you uploading OTA files (thanks!), however, can you please consider creating a new separate dedicated repository called "Sonoff_Zigbee_OTA" or similar under ITead's own official organization on GitHub and use that as your primary publishing location for all officially released Zigbee OTA files for Sonoff branded Zigbee devices?

https://github.com/itead

(That would for example be https://github.com/itead/Sonoff_Zigbee_OTA or someting like that).

Preferably also copying the concept of `index.json` with metadata on all files like Koenkk repository has to make scripting easier:

https://github.com/Koenkk/zigbee-OTA/blob/master/index.json

And of course `README.md` file which among other things could contain some basic information on how to update those:

https://github.com/Koenkk/zigbee-OTA/blob/master/README.md

Perhaps also as keeping some kind of `CHANGELOG.md` with changelog release history for each device firmware version:

https://github.com/Koenkk/Z-Stack-firmware/blob/master/coordinator/Z-Stack_3.x.0/CHANGELOG.md

For ex. `zbmini-l_v1.1.1_release_notes.md` and so on with release notes for each version released + compounded changelog.

That way it will be a single trusted and official primary source of official Zigbee OTA files released by ITead.

jds11111, 2022-05-24:
> > > We have raised the pull request for the OTA update image on github https://github.com/Koenkk/zigbee-OTA/pulls.
> > > Once it is approved, you can download the image and update the ZBMINI-L devices.
> > 
> > 
> > Do you have release notes for the firmware?
> 
> Feature: Added Poll Control Cluster private attribute 0xFF00 (enable/disable configuration), the default Poll Control Cluster initial configuration is disabled Bugfix: Random shutdown issue

The random shutdown issue is not fixed by this firmware for my device.

DieserKirsche, 2022-05-24:
My device seeems to work properly since firmware update 4 days ago.

jds11111, 2022-05-24:
With this firmware, I got two random shutdowns this morning:
[image]
[image]

(timezone is central US) This is the latest, no?

pulento, 2022-05-24:
That meas that last time the device was seen on the network. Not the last time it went off.

jds11111, 2022-05-24:
Of course. I witnessed it shut off twice.

pulento, 2022-05-24:
> Of course. I witnessed it shut off twice.

Can you post the history to see if just "Turn off" which will mean bug still there.

In my case I haven't see a random shut off yet.

jds11111, 2022-05-25:
Sure.  The last two here appear to be just random.  OTOH, I found an odd automation that might be causing problems. I have disabled it to see what happens.
[image]

dtalens, 2022-06-09:
I have same issue with random shut off. In my case, device turns on or off for a second and returns to previous state. Issue occurs approximately 1 or 2 every 30 minutes. It has same behavior whether it is off or on.

langejos, 2022-06-20:
I managed to upgrade to version 1.1.1 but it still is not working more than maybe half a day. Giving up for now.

mbarcia, 2022-07-30:
Glad to see a new firmware has been provided to deal with these issues, thanks to the maintainers for that!

I've 4 of these ZBMINI-L installed, and one of them is exhibiting the bug described here (random turn on and off behaviour). I've followed this thread and managed to install the new firmware OTA, but unfortunately it didn't fix the bug, as you can see in the screen capture below:

[image]

Is there anything else required to make this work? Thanks,
--mariano

dtalens, 2022-07-30:
Same behaviour with last firmware 1.1.1. There seems to be more random shutdowns when I turn light on from home assistant switch.

maxime1992, 2022-12-25:
Not sure if I witness the same issue or not but my light just went absolutely insane:

[image]

When this happen the only outcome for me is rebooting all my docker compose stack (including HA and Z2M) until it happens again.

myref, 2023-05-22:
I have the same behaviour

caroku, 2023-05-22:
I gave up, my zbmini-l is in a drawer.

closed 2023-11-25 by github-actions[bot]