zigpy/zha-device-handlers #1520: [BUG] Upgrading Home Assistant to 2022.4.4 breaks Elko Super TR thermostat.

falense, 2022-04-23:
**Describe the bug**

Upgrading Home Assistant to 2022.4.4 breaks Elko Super TR thermostat.

**To Reproduce**

Upgrade Home Assistant from an earlier version to 2022.4.4 no longer allows control of the thermostat. Commands fail with "Failed to call service climate/set_hvac_mode. 'system_mode'" 

**Expected behavior**

Pressing the on button on the default thermostat card should enable the heater. Fails with error message.

<details>
<summary>Device signature </summary>

```
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4919, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0301",
      "in_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0201",
        "0x0204",
        "0x0b04"
      ],
      "out_clusters": [
        "0x0003",
        "0x0019"
      ]
    }
  },
  "manufacturer": "ELKO",
  "model": "Super TR",
  "class": "zhaquirks.elko.smart_super_thermostat.ElkoSuperTRThermostat"
}
```

</details>

<details>
<summary>Additional logs</summary>

```File "/usr/src/homeassistant/homeassistant/components/zha/climate.py", line 425, in async_set_hvac_mode
if await self._thrm.async_set_operation_mode(HVAC_MODE_2_SYSTEM[hvac_mode]):
File "/usr/src/homeassistant/homeassistant/components/zha/core/channels/hvac.py", line 243, in async_set_operation_mode
if not await self.write_attributes({"system_mode": mode}):
File "/usr/src/homeassistant/homeassistant/components/zha/core/channels/hvac.py", line 292, in write_attributes
res = await self.cluster.write_attributes(data, **kwargs)
File "/usr/local/lib/python3.9/site-packages/zhaquirks/elko/smart_super_thermostat.py", line 110, in write_attributes
return await super().write_attributes(attributes, manufacturer=manufacturer)
File "/usr/local/lib/python3.9/site-packages/zigpy/zcl/__init__.py", line 535, in write_attributes
attrs = self._write_attr_records(attributes)
File "/usr/local/lib/python3.9/site-packages/zigpy/zcl/__init__.py", line 502, in _write_attr_records
attr_def = self.find_attribute(attrid)
File "/usr/local/lib/python3.9/site-packages/zigpy/zcl/__init__.py", line 186, in find_attribute
return self.attributes_by_name[name_or_id]
KeyError: 'system_mode'
```

</details>

thgjerde, 2022-05-03:
This is still not working. All the custom attributes are returning UNSUP_MANUF_GENERAL_COMMAND
I need to set is_manufacturer_specific to False to get this to work.

falense, 2022-06-07:
As of Home Assisant 2022.6.2 with HAOS 8.1 this is still not working. Fails with set_hvac_mode uint8_t is not iterable.

falense, 2022-07-08:
Still broken in Home Assistant 2022.7.1. Same error.

PR #1526 (2022-05-25): Verify quirks subclassing ZCL clusters extend them
Fixes #1520

Currently a few unit tests are failing:

- [ ] `zhaquirks.tuya.mcu.TuyaOnOff`
- [ ] `zhaquirks.tuya.ts0601_dimmer.TuyaOnOffNM`
- [ ] `zhaquirks.xbee.XBeeCommon.DigitalIOCluster`

@javicalle for the Tuya clusters, them subclassing `zcl.OnOff` but then exposing only a single `on_off` attribute seems like it will break code expecting the other attributes to exist. Can the `OnOff` subclass be replaced with just `CustomCluster` or is this intentional?