zigpy/zha-device-handlers #1527: [Device Support Request] Moes Zigbee + RF Smart Curtain Switch

mdiazgoncalves, 2022-04-24:
https://www.moeshouse.com/collections/new-arrivals/products/zigbee-rf-smart-touch-curtain-roller-blinds-shutter-switch-tuya-smart-life-app-wireless-control-works-with-alexa-and-google-home-eu?variant=39793319411793

**Is your feature request related to a problem? Please describe.**
Device is detected but no cover entity is exposed only basic_lqi and basic_rssi

**Describe the solution you'd like**
Add support for Moes Zigbee + RF Smart Curtain Switch

<details>
<summary>Device signature - this can be acquired by clicking on the "Zigbee Device Signature" button in the device settings</summary>

```
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0051",
      "in_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0xef00"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZE200_nhyj64w2",
  "model": "TS0601",
  "class": "zigpy.device.Device"
}
```

</details>

<details>
<summary>Diagnostic information - this can be acquired by clicking on the "Download Diagnostics" button in the device settings</summary>

```
{
  "home_assistant": {
    "installation_type": "Home Assistant Supervised",
    "version": "2022.4.6",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.9.9",
    "docker": true,
    "arch": "x86_64",
    "timezone": "Europe/Lisbon",
    "os_name": "Linux",
    "os_version": "5.10.0-13-amd64",
    "supervisor": "2022.04.0",
    "host_os": "Debian GNU/Linux 11 (bullseye)",
    "docker_version": "20.10.14",
    "chassis": "",
    "run_as_root": true
  },
  "custom_components": {
    "authenticated": {
      "version": "21.9.0",
      "requirements": []
    },
    "webrtc": {
      "version": "v2.3.0",
      "requirements": []
    },
[... 298 lines cut ...]
    "endpoint_names": [
      {
        "name": "SMART_PLUG"
      }
    ],
    "user_given_name": "estore_sala_estar",
    "device_reg_id": "b865462fe1e36765183e03c79e6c0067",
    "area_id": "8655c42078554434b66a10c1b0c6baaa"
  }
}
```

</details>

<details>
<summary>Additional logs</summary>

```
Paste any additional debug logs here.
```

</details>

**Additional context**
I tried using TuyaMoesCover0601 quirk but doesn't work.

javicalle, 2022-04-25:
The device signature seems to match the `TuyaMoesCover0601` one (just some diferences in `node_descriptor`).
Is the device loading the quirk? (you will see at the device signature)
Any error log when quirk is loaded and you interacts with the cover (via remote for example)?

mdiazgoncalves, 2022-04-25:
Hi javicalle

Yes, it loads the quirk and with the new quirk it exposes a cover entity.
When I interact with the cover, I get "failed to call service cover/open _TZE200_nhyj64w2"

In the logs:
```
2022-04-25 17:45:46 DEBUG (MainThread) [zhaquirks.tuya] 2c:11:65:ff:fe:9e:60:8b Sending Tuya Cluster Command.. Manufacturer is _TZE200_nhyj64w2 Cluster Command is 0x0000, Arguments are ()
2022-04-25 17:45:46 ERROR (MainThread) [homeassistant.components.websocket_api.http.connection] [140411244942384] '_TZE200_nhyj64w2'
KeyError: '_TZE200_nhyj64w2'
```

MattWestb, 2022-04-25:
The device need being added in the INIT for working OK and also if needed as inverted.
https://github.com/zigpy/zha-device-handlers/blob/ae8377749383648bc8b806de0b3e1e258b6ef85a/zhaquirks/tuya/__init__.py#L104
https://github.com/zigpy/zha-device-handlers/blob/ae8377749383648bc8b806de0b3e1e258b6ef85a/zhaquirks/tuya/__init__.py#L129

mdiazgoncalves, 2022-04-25:
The buttons don't work, but the slider works. It is not calibrated, but it works.

MattWestb, 2022-04-25:
The buttons cant working if not adding the device in the button map in the tuya INIT i was pointing at and also it can being needed adding it in the inverted section if its inverted.

Also if making changes in tuya INIT it must being made in the HA container if not patching the local quirk.

Also trying up/down and stop for the buttons so its right mapped or its not working OK (pressing up and its going down or somthing like that).

mdiazgoncalves, 2022-04-25:
It is defenitly not inverted. 

The buttons aren't working but the position slider works.

What is the meaning of the values in TUYA_COVER_COMMAND?

"_TZE200_nhyj64w2": {0x0000: 0x0000, 0x0001: 0x0002, 0x0002: 0x0001},

Maybe something to do with the values?

MattWestb, 2022-04-25:
From:
https://github.com/zigpy/zha-device-handlers/blob/ae8377749383648bc8b806de0b3e1e258b6ef85a/zhaquirks/tuya/__init__.py#L103
Its mapping buttons to function then standard is 0 up , 1 stop and 3 down.

mdiazgoncalves, 2022-04-25:
It is strange because the position slider is working but not the buttons
Should I try different values for the buttons in TUYA_COVER_COMMAND?

MattWestb, 2022-04-25:
Ýes

javicalle, 2022-04-25:
> Should I try different values for the buttons in TUYA_COVER_COMMAND?

Or you can began with the normal values (`"_TZE200_nhyj64w2": {0x0000: 0x0000, 0x0001: 0x0001, 0x0002: 0x0002},`) and then try to figure which buttons are switched. 
If open/close are switched, then the cover is inverted.
If there are the open/stop, switch the `0x0000` and `0x0001`
If there are the stop/close, switch the `0x0001` and `0x0002`

mdiazgoncalves, 2022-04-25:
It's working now. Somehow, I was not editing the correct __init__.py
The correct command values are: 
`("_TZE200_nhyj64w2": {0x0000: 0x0000, 0x0001: 0x0002, 0x0002: 0x0001},) `
Now I just have to figure how to calibrate it

Thanks guys for your support

closed 2022-04-25 by mdiazgoncalves

Cas7or, 2024-06-23:
> It's working now. Somehow, I was not editing the correct **init**.py The correct command values are: `("_TZE200_nhyj64w2": {0x0000: 0x0000, 0x0001: 0x0002, 0x0002: 0x0001},) ` Now I just have to figure how to calibrate it
> 
> Thanks guys for your support

Hey,

Did you ever figure out how to calibrate?

mdiazgoncalves, 2024-06-23:
Nope. I think the only way is to connect it to Tuya and calibrate.
I don't have a Tuya ZigBee bridge.

Cas7or, 2024-06-24:
> Nope. I think the only way is to connect it to Tuya and calibrate. I don't have a Tuya ZigBee bridge.

Nope, did not work. I just tried it. Interestingly enough within the smart life app it works fine once calibrated.
The other curious bit of information is that I have now seen two options to calibrate and sometimes one of the option is missing on some of the devices.

Option 1: set time in seconds to Open/close
Option 2: detailed calibration - asks you to set the blinds to fully open, then pause, then fully close then pause and that calibrates the blinds.

For anyone working on this, I am willing to help with anything I can so do let me know if I can be of assistance.