zigpy/zha-device-handlers #1528: Tuya Tubular Roller Blind motor

smarthomemakers, 2022-04-24:
Is your feature request related to a problem? Please describe.

Request and placehoder for information about the Tuya Tubular Roller Blind motor > specfically:

https://www.ajaxonline.co.uk/product/zignito-zigbee-roller-blind-motor/

Describe the solution you'd like

ZHA support :)

```{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0051",
      "in_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0xef00"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZE200_uzinxci0",
  "model": "TS0601",
  "class": "zigpy.device.Device"
}
```

javicalle, 2022-04-25:
Have you tried to add your device to the `TuyaMoesCover0601` quirk?

If you are not familiar, there are some guides about it. Maybe the most popular can be:
* https://github.com/zigpy/zha-device-handlers/discussions/693#discussioncomment-857274

smarthomemakers, 2022-04-26:
Hiya @javicalle ! 

Thanks for commenting my understanding is the following:

- Add custom quirk folder in HA config files
- Create a .py file inside my custom folder
- Do I need to copy & paste the TuyaMoesCover0601
- Change the manufacturer? 
- Save & restart?

javicalle, 2022-04-26:
Mostly, yes.

Copy all the `ts0601_cover.py` file in the `custom_zha_quirks` folder:
* https://raw.githubusercontent.com/zigpy/zha-device-handlers/dev/zhaquirks/tuya/ts0601_cover.py

Better add your manufacturer in the `TuyaMoesCover0601` list:
```python
        MODELS_INFO: [
            ("_TZE200_zah67ekd", "TS0601"),
            ("_TZE200_xuzcvlku", "TS0601"),
            ("_TZE200_rddyvrci", "TS0601"),
            ("_TZE200_nueqqe6k", "TS0601"),
            ("_TZE200_gubdgai2", "TS0601"),
            ("_TZE200_yenbr4om", "TS0601"),
            ("_TZE200_5sbebbzs", "TS0601"),
            ("_TZE200_xaabybja", "TS0601"),
            ("_TZE200_hsgrhjpf", "TS0601"),
            ("_TZE200_68nvbio9", "TS0601"),
            ("_TZE200_zuz7f94z", "TS0601"),
            ("_TZE200_ergbiejo", "TS0601"),
            ("_TZE200_uzinxci0", "TS0601"),
        ],
```

Save & restart.

Then check if device signature changes, that would mean that the device loads the quirk.

MattWestb, 2022-04-26:
Need adding the key map in tuya INIT (in the HA docker container) for the device or you is getting errors then loading the quirk or using it.

smarthomemakers, 2022-04-26:
Hiya @MattWestb  I'm running HA with supervisor where can I get access to the Tuya init file? Have you got example of the key map change I need to make? Thanks a lot!

MattWestb, 2022-04-26:
Install SSH access add-on for getting CLI to the HA container if you is running OS and if running not blocked supervisor install portainer and you can open CLI to the HA container.

One option is patching the quirk for reading the INIT from the custom quirk folder but i dont knowing how to doing it but im sure @javicalle can helping patching that.

javicalle, 2022-04-26:
> Need adding the key map in tuya INIT (in the HA docker container) for the device or you is getting errors then loading the quirk or using it.

Ouuch! Yes, that's true.

You can try to copy the `tuya.__init__.py` file in your `custom_zha_quirks` folder:
* https://raw.githubusercontent.com/zigpy/zha-device-handlers/dev/zhaquirks/tuya/__init__.py

Add your manufacturer to the array (in the `__init__.py` file):
```python
# For most tuya devices 0 = Up/Open, 1 = Stop, 2 = Down/Close
TUYA_COVER_COMMAND = {
    "_TZE200_zah67ekd": {0x0000: 0x0000, 0x0001: 0x0002, 0x0002: 0x0001},
    "_TZE200_fzo2pocs": {0x0000: 0x0002, 0x0001: 0x0000, 0x0002: 0x0001},
    "_TZE200_xuzcvlku": {0x0000: 0x0000, 0x0001: 0x0002, 0x0002: 0x0001},
    "_TZE200_rddyvrci": {0x0000: 0x0002, 0x0001: 0x0001, 0x0002: 0x0000},
    "_TZE200_3i3exuay": {0x0000: 0x0000, 0x0001: 0x0002, 0x0002: 0x0001},
    "_TZE200_nueqqe6k": {0x0000: 0x0000, 0x0001: 0x0002, 0x0002: 0x0001},
    "_TZE200_gubdgai2": {0x0000: 0x0000, 0x0001: 0x0002, 0x0002: 0x0001},
    "_TZE200_zpzndjez": {0x0000: 0x0000, 0x0001: 0x0002, 0x0002: 0x0001},
    "_TZE200_cowvfni3": {0x0000: 0x0002, 0x0001: 0x0000, 0x0002: 0x0001},
    "_TYST11_wmcdj3aq": {0x0000: 0x0000, 0x0001: 0x0002, 0x0002: 0x0001},
    "_TZE200_yenbr4om": {0x0000: 0x0000, 0x0001: 0x0002, 0x0002: 0x0001},
    "_TZE200_5sbebbzs": {0x0000: 0x0000, 0x0001: 0x0002, 0x0002: 0x0001},
    "_TZE200_xaabybja": {0x0000: 0x0000, 0x0001: 0x0002, 0x0002: 0x0001},
    "_TZE200_hsgrhjpf": {0x0000: 0x0000, 0x0001: 0x0002, 0x0002: 0x0001},
    "_TZE200_iossyxra": {0x0000: 0x0000, 0x0001: 0x0002, 0x0002: 0x0001},
    "_TZE200_68nvbio9": {0x0000: 0x0000, 0x0001: 0x0002, 0x0002: 0x0001},
    "_TZE200_zuz7f94z": {0x0000: 0x0000, 0x0001: 0x0002, 0x0002: 0x0001},
    "_TZE200_ergbiejo": {0x0000: 0x0000, 0x0001: 0x0002, 0x0002: 0x0001},
    "_TZE200_nhyj64w2": {0x0000: 0x0000, 0x0001: 0x0002, 0x0002: 0x0001},
    "_TZE200_uzinxci0": {0x0000: 0x0000, 0x0001: 0x0002, 0x0002: 0x0001},
}
```

In your `ts0601_cover.py` file, replace all the `zhaquirks.tuya` imports (at the beginning of file) with:
```python
# from zhaquirks.tuya import (
from . import (
    TuyaManufacturerWindowCover,
    TuyaManufCluster,
    TuyaWindowCover,
    TuyaWindowCoverControl,
)
```

I believe that this would be enought.

And before restart, delete any `__pycache__` subfolder in your `custom_zha_quirks`.

closed 2022-10-30 by github-actions[bot]