zigpy/zha-device-handlers #1533: [Device Support Request] Aqara FP1 Presence Detection

Danny2100, 2022-04-26:
**Is your feature request related to a problem? Please describe.**
New Aqara FP1 sensor is only partially compatible with zigbee2mqtt at this time. I say partially because they haven't figured out how to expose the zone part of the sensor to zigbee2mqtt. I would rather stick with zha so it would be nice if we can get it to work with zha potentially piggybacking off what zigbee2mqtt did as a start.

https://github.com/Koenkk/zigbee2mqtt/issues/11019

https://www.aqara.com/cn/productDetail/Aqara-Presence-Detector-FP1

I don't have the device physically to paste anything here.

EverythingSmartHome, 2022-04-26:
Let me know if I can provide anything from this sensor to help make a quirk for it - I can try and give the quirk a go myself but I've only done more basic ones myself before, this one is probably a bit out of my league!

Tropaion, 2022-04-26:
@EverythingSmartHome You have to post pairing logs and device signature.
Your issue could probably be solved together with this one: https://github.com/zigpy/zha-device-handlers/issues/1498

dmulcahey, 2022-04-26:
> Let me know if I can provide anything from this sensor to help make a quirk for it - I can try and give the quirk a go myself but I've only done more basic ones before, this one is probably a bit out of my league!

Do you have a reliable way to acquire them? I have been trying for a couple months to get my hands on the. Finally got an order placed on Ali Express a couple weeks ago and they cancelled it on me!

EverythingSmartHome, 2022-04-26:
> > Let me know if I can provide anything from this sensor to help make a quirk for it - I can try and give the quirk a go myself but I've only done more basic ones before, this one is probably a bit out of my league!
> 
> Do you have a reliable way to acquire them? I have been trying for a couple months to get my hands on them. Finally got an order placed on Ali Express a couple weeks ago and they cancelled it on me!

I don't unfortunately, I wish I did - I would have bought a few more! I'll do some investigating see if I can get some more info from Aqara directly

EverythingSmartHome, 2022-04-26:
> @EverythingSmartHome You have to post pairing logs and device signature. Your issue could probably be solved together with this one: #1498

Here is the signature:
```
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered: 140>, manufacturer_code=4660, maximum_buffer_size=108, maximum_incoming_transfer_size=127, server_mask=11264, maximum_outgoing_transfer_size=127, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0xfff0",
      "in_clusters": [
        "0x0000",
        "0x0003",
        "0xfcc0"
      ],
      "out_clusters": [
        "0x0003",
        "0x0019"
      ]
    }
  },
  "manufacturer": "aqara",
  "model": "lumi.motion.ac01",
  "class": "zigpy.device.Device"
}
```

Logs:
```
[0x0000:zdo] ZDO request ZDOCmd.Mgmt_Permit_Joining_rsp: [<Status.SUCCESS: 0>]
[0x0000:zdo] No handler for ZDO request:ZDOCmd.Mgmt_Permit_Joining_rsp([<Status.SUCCESS: 0>])
[0xa732:zdo] ZDO request ZDOCmd.Mgmt_Permit_Joining_rsp: [<Status.SUCCESS: 0>]
[0xa732:zdo] No handler for ZDO request:ZDOCmd.Mgmt_Permit_Joining_rsp([<Status.SUCCESS: 0>])
[0xa732:zdo] ZDO request ZDOCmd.Mgmt_Permit_Joining_rsp: [<Status.SUCCESS: 0>]
[0xa732:zdo] No handler for ZDO request:ZDOCmd.Mgmt_Permit_Joining_rsp([<Status.SUCCESS: 0>])
New device 0xf306 (54:ef:44:10:00:47:1a:5c) joined the network
[0xf306] Scheduling initialization
Received frame on uninitialized device <Device model=None manuf=None nwk=0xF306 ieee=54:ef:44:10:00:47:1a:5c is_initialized=False> from ep 0 to ep 0, cluster 19: b'\x01\x06\xf3\\\x1aG\x00\x10D\xefT\x8c'
[0xf306:zdo] ZDO request ZDOCmd.Device_annce: [0xF306, 54:ef:44:10:00:47:1a:5c, 140]
Tries remaining: 3
[0xf306] Requesting 'Node Descriptor'
Tries remaining: 2
[0xf306] Extending timeout for 0xae request
Received frame on uninitialized device <Device model=None manuf=None nwk=0xF306 ieee=54:ef:44:10:00:47:1a:5c is_initialized=False> from ep 1 to ep 1, cluster 10: b'\x00\x00\x00\x00\x00'
[0xf306] Received ZCL while uninitialized on endpoint id 1, cluster 0x000a id, hdr: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=0, is_reply=0, disable_default_response=0, reserved=0, *is_cluster=False, *is_general=True), tsn=0, command_id=0, *is_reply=False), payload: b'\x00\x00'
[0xf306] Uninitialized device command '0' params: Read_Attributes(attribute_ids=[0])
Received frame on uninitialized device <Device model=None manuf=None nwk=0xF306 ieee=54:ef:44:10:00:47:1a:5c is_initialized=False> from ep 1 to ep 1, cluster 0: b'\x18\x01\x0A\x05\x00\x42\x10\x6C\x75\x6D\x69\x2E\x6D\x6F\x74\x69\x6F\x6E\x2E\x61\x63\x30\x31\x01\x00\x20\x35'
[0xf306] Received ZCL while uninitialized on endpoint id 1, cluster 0x0000 id, hdr: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=0, is_reply=1, disable_default_response=1, reserved=0, *is_cluster=False, *is_general=True), tsn=1, command_id=10, *is_reply=True), payload: b'\x05\x00B\x10lumi.motion.ac01\x01\x00 5'
[0xf306] Uninitialized device command '10' params: Report_Attributes(attribute_reports=[Attribute(attrid=0x0005, value=TypeValue(type=CharacterString, value='lumi.motion.ac01')), Attribute(attrid=0x0001, value=TypeValue(type=uint8_t, value=53))])
Received frame on uninitialized device <Device model=None manuf=None nwk=0xF306 ieee=54:ef:44:10:00:47:1a:5c is_initialized=False> from ep 1 to ep 1, cluster 64704: b'\x18\x02\x0A\xF7\x00\x41\x2D\x03\x28\x17\x05\x21\x01\x00\x08\x21\x35\x01\x0A\x21\x96\x90\x0C\x20\x14\x10\x20\x01\x12\x20\x00\x65\x20\xFF\x66\x20\x03\x67\x20\x00\x68\x20\x00\x69\x20\x01\x6A\x20\x01\x6B\x20\x03'
[0xf306] Received ZCL while uninitialized on endpoint id 1, cluster 0xfcc0 id, hdr: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=0, is_reply=1, disable_default_response=1, reserved=0, *is_cluster=False, *is_general=True), tsn=2, command_id=10, *is_reply=True), payload: b'\xf7\x00A-\x03(\x17\x05!\x01\x00\x08!5\x01\n!\x96\x90\x0c \x14\x10 \x01\x12 \x00e \xfff \x03g \x00h \x00i \x01j \x01k \x03'
[0xf306] Uninitialized device command '10' params: Report_Attributes(attribute_reports=[Attribute(attrid=0x00F7, value=TypeValue(type=LVBytes, value=b'\x03(\x17\x05!\x01\x00\x08!5\x01\n!\x96\x90\x0c \x14\x10 \x01\x12 \x00e \xfff \x03g \x00h \x00i \x01j \x01k \x03'))])
Received frame on uninitialized device <Device model=None manuf=None nwk=0xF306 ieee=54:ef:44:10:00:47:1a:5c is_initialized=False> from ep 0 to ep 0, cluster 32770: b'\xae\x00\x06\xf3\x02@\x8c4\x12l\x7f\x00\x00,\x7f\x00\x00'
Received frame on uninitialized device <Device model=None manuf=None nwk=0xF306 ieee=54:ef:44:10:00:47:1a:5c is_initialized=False> from ep 1 to ep 1, cluster 64704: b'\x18\x03\x0A\xF7\x00\x41\x2D\x03\x28\x17\x05\x21\x01\x00\x08\x21\x35\x01\x0A\x21\x96\x90\x0C\x20\x14\x10\x20\x01\x12\x20\x00\x65\x20\xFF\x66\x20\x03\x67\x20\x00\x68\x20\x00\x69\x20\x01\x6A\x20\x01\x6B\x20\x03'
[... cut ...]
[0xF306:1:0x0019]: initializing channel: from_cache: False
[0xF306:1:0x0019]: finished channel initialization
[0xF306:1:0x0000]: 'async_initialize' stage succeeded
[0xF306:1:0x0003]: 'async_initialize' stage succeeded
[0xF306:1:0xfcc0]: 'async_initialize' stage succeeded
[0xF306:1:0x0019]: 'async_initialize' stage succeeded
[0xF306](lumi.motion.ac01): power source: Mains
[0xF306](lumi.motion.ac01): completed initialization
```

I've got a basic quirk loaded at the moment but not any further than that

Tropaion, 2022-04-26:
@dmulcahey I also wanted to buy one, but according to what I read, the everywhere sold out even in change.
It looks like they have production problems because of the massive covid lockdowns and it will probably take a month+ for resupply.

dmulcahey, 2022-04-26:
> @dmulcahey I also wanted to buy one, but according to what I read, they are sold out everywhere even in china.
> It looks like they have production problems because of the massive covid lockdowns and it will probably take a month+ for resupply.

I managed to order them… thing is they aren’t supposed to ship to the USA. My assumption was they messed up the Ali listing and it wasn’t supposed to be available for USA residents. Who knows though… the Covid issue could definitely be a factor

dmulcahey, 2022-04-27:
> Here is the signature:
>
> ```
> {
>   "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, [...]

When you have time / if you want to: hit me up on discord and we can reverse engineer the z2m support for it. I can help but you will have to test it.

dmulcahey, 2022-04-27:
> I've got a basic quirk loaded at the moment but not any further than that

Stay tuned… working on quirk stub generation from diagnostics data.

EverythingSmartHome, 2022-04-27:
> > I've got a basic quirk loaded at the moment but not any further than that
> 
> Stay tuned… working on quirk stub generation from diagnostics data.

Message incoming..

Tropaion, 2022-04-27:
@dmulcahey A bit offtopic but of you are interested in mmWave presence detection, I found a similar device: https://de.aliexpress.com/item/1005003796368221.html
Though pretty expensive.

dmulcahey, 2022-04-28:
> @dmulcahey A bit offtopic but if you are interested in mmWave presence detection, I found a similar device: https://de.aliexpress.com/item/1005003796368221.html
> Though pretty expensive.

Thx! Although I don’t wanna support the mess tuya makes. I’ll take a look at how it’s implemented 😃

Tropaion, 2022-04-28:
Nice, then I can save the money until I know from you that it works 😄

Hedda, 2022-04-28:
> Everything Smart Home YouTube Video on this new device: https://youtu.be/Leru0rCS8b0

@EverythingSmartHome When using Aqara FP1 with their gateway/app does it have sensor configuration to make it ignore pets?

(Some motion sensors have settings allowing sensitivity configurations based on size or similar if you have cats or other animals).

EverythingSmartHome, 2022-04-29:
> @EverythingSmartHome When using Aqara FP1 with their gateway/app does it have sensor configuration to make it ignore pets?
>
> (Some motion sensors have settings allowing sensitivity configurations [...]

No option for that at least at the moment!

diederik-maes, 2022-05-23:
I just got 2 pieces of FP1 today. Added 1 to ZHA but no luck. No motion detection nor battery status.
Hoping to see this added soon to ZHA. WOuld love to play around with the reset time

BradleyShaner, 2022-05-28:
They are in-stock right now on Aliexpress - https://www.aliexpress.com/item/3256803596744231.html?spm=a2g0o.order_list.0.0.21ef1802r8ZVZH

Tropaion, 2022-05-29:
Dawn, I was too slow -.-

SeraphimSerapis, 2022-05-29:
> Dawn, I was too slow -.-

Check again. I was just able to order 2 of them!

Hedda, 2022-05-30:
FYI, Tuya also has several "Zigbee Human Presence Detector" using similar microwave/millimeter wave motion sensor tenology.

These are used for either smart human body presence (a.k.a. occupancy/vacancy detection), breathing sensor, or fall-down sensor.

See example Tuya product models: MSA050-Z1, MIR-HE200-ZB, MYQ-ZHFD02, MSA042D-Z1, MSA035-Z1, and MYQ-ZHFD01:

https://expo.tuya.com/smart/sensors/infrared-sensor/occupancy---vacancy-sensor/CT9cn94ki9rh36-CT9cn9la1lzi8j-CT9cndvh3x4xle

https://expo.tuya.com/product/842246

https://expo.tuya.com/product/952001

https://expo.tuya.com/product/1031059

https://expo.tuya.com/product/902056

https://expo.tuya.com/product/804011

https://expo.tuya.com/product/1031058

PS: Note that those are listed for global worldwide sale, unlike this competing Aqara FP1 product.

blakadder, 2022-06-03:
New stock available: https://s.click.aliexpress.com/e/_AmTlud (*affiliate link)

dmulcahey, 2022-06-03:
> New stock available: https://s.click.aliexpress.com/e/_AmTlud (*affiliate link)

Awesome. Thanks. I have one arriving today and just grabbed 4 more!

Sebas-, 2022-06-03:
> Are they that good?

Yes (in my opinion) , currently using them in a few rooms, no more need to create "smart" automations (for me at least). Using PIR sensor for initial detection, and then the FP1 to keep the state of the room to occupied until it says otherwise. No more problem with sitting still in front of the tv or when I'm working in front of a computer (pretty much sitting still).

dmulcahey, 2022-06-03:
> > Are they that good?
> 
> Yes (in my opinion) , currently using them in a few rooms, no more need to create "smart" automations (for me at least). Using PIR sensor for initial detection, and then the FP1 to keep the state of the room as occupied until it says otherwise. No more problem with sitting still in front of the tv or when I'm working in front of a computer (pretty much sitting still).

This is the promise right here… true presence

Tropaion, 2022-06-03:
I'm not sure if I should order one since I'm in a small financial crisis and I'm not sure if it will be supported by ZHA.
@dmulcahey Are you working on an implementation?

dmulcahey, 2022-06-03:
Just pushed the first parts of support

Tropaion, 2022-06-03:
@dmulcahey nice, what does this implement? Since you didn't write anything in your pull request.
What I'm interested in, as far as I know, in the tuya app it's possible to define "action zones", how will this look like in ZHA/HA?

eatoff8, 2022-06-04:
> Just pushed the first parts of support

Well, seeing that made the purchase decision easy for me. Thanks @blakadder  too for the link, some reason it didn't come up in a search.

Looking forward to seeing it supported in zha

Hedda, 2022-06-04:
> @dmulcahey nice, what does this implement? Since you didn't write anything in your pull request.
> What I'm interested in, as far as I know, in the tuya app it's possible to define "action zones", how will this look like in ZHA/HA?

See this PR from dmulcahey for ZHA to HA’s core -> https://github.com/home-assistant/core/pull/72466

eatoff8, 2022-06-16:
How can we see what version of HA this is merged into? The tags just say home-assistant:dev

eatoff8, 2022-06-17:
> > How can we see what version of HA this is merged into? The tags just say home-assistant:dev
> > I just had a couple delivered, so keen to find out when support will be arriving
> 
> It hasn't been released yet. You can download the .py and put it in a custom quirks file and add the path to your configuration. With that you will only have the ability to use the device until the configuration stuff is merged into core.

So how can I tell which HA version it will go into?

This just says dev, is there a way to see ahead of time what version it's going into? If it's going into 2022.6.7 I'll wait, but if it's not until 2022.11.0 or whatever I'll workaround 
https://github.com/home-assistant/core/pull/72466

eatoff8, 2022-06-17:
> 
> Put this
> 
> https://github.com/zigpy/zha-device-handlers/blob/851e784207160645b108e78c577cde2b5704a722/zhaquirks/xiaomi/aqara/motion_ac01.py
> 
> Into /config/zha_quirks Then in your configuration.yaml add
> 
> ```
> zha:
>   custom_quirks: true
>   custom_quirks_path: /config/zha_quirks
> ```

Thanks for the pointers, but this doesnt work. Just gets stuck in "interviewing device". I put the file you linked in the zha_quirks folder, and no love for me.

[image]

And the zha custom quirks is actually "enable_quirks: true" just for anyone else thats reading along, or you can remove it completely as the default is true anyway. It gives me an error using "custom_quirks".

Hedda, 2022-06-17:
> How can we see what version of HA this is merged into? The tags just say home-assistant:dev
> 
> I just had a couple delivered, so keen to find out when support will be arriving

If you are asking when Aqara FP1 support will be available in the stable release of Home Assistant then the answer is that you have to first wait until pull request https://github.com/zigpy/zha-device-handlers/pull/1600 with the FP1 "quirk" is merged into zha-device-handlers before waiting until there is a new release:

https://github.com/zigpy/zha-device-handlers/releases 

(for dev of zha-device-handlers / zha-quirks see https://github.com/zigpy/zha-device-handlers/pulls?q=is%3Apr+is%3Aclosed ).

Then you have to wait until that "zha-quirks" (zha-device-handlers) is [bumped in Home Assistant core (as a separate pull request)](https://github.com/home-assistant/core/pulls) and pull request https://github.com/home-assistant/core/pull/73027 is merged into Home Assistant core:

https://github.com/home-assistant/core/releases

(for dev of Home Assistant core see https://github.com/home-assistant/core/pulls?q=is%3Apr+is%3Aclosed ).

After that you have to wait for next major release of Home Assistant which contain both "zha-quirks" with FP1 quirk and that PR.

eatoff8, 2022-06-20:
> Just to clarify the file is a .py file that you added?

Yes, it was. I did some other reading, and it was the same procedure recommended elsewhere too, but doesnt work for me.

I put the .py file in a folder called /config/zha_quirks (no subfolders within that).

eatoff8, 2022-06-20:
> After that you have to wait for next major release of Home Assistant which contain both "zha-quirks" with FP1 quirk and that PR.

Thanks for the super detailed response. Looks like its going to be a few weeks away, so I'll try get the custom quirks working in the meantime.

zeltak, 2022-07-01:
Hi all

I have added the custom quick as @Danny2100 suggested, it does pick up the sensor but the sensor dosent seem to do anything and looks strange:
[image]

am I missing a step (I also added of course the entries to my YAML):

```

# ZHA
zha:
  enable_quirks: true
  custom_quirks_path: /config/zha_quirks

```

thx

Z

TheJulianJES, 2022-07-01:
If you're using Home Assistant OS, you can join the beta under Settings -> System -> Updates -> click on the three dots at the top right corner -> join beta channel.

Then, you can check for updates and (backup +) install the latest update.

zeltak, 2022-07-01:
cool thx!

zeltak, 2022-07-07:
@TheJulianJES !

I updated today to 2022.7 and the fp-1 device is still not working. is there a new beta I should try it with or was this support planed for 2022.7?

thx

Z

Draghmar, 2022-07-07:
> @TheJulianJES !
> 
> I updated today to 2022.7 and the fp-1 device is still not working. is there a new beta I should try it with or was this support planed for 2022.7?
> 
> thx
> 
> Z

Not sure if you already try that but [Z2M says](https://www.zigbee2mqtt.io/devices/RTCZCGQ11LM.html) there are some additional things FP1 requires in order to work (channel and firmware). Maybe it's the same for ZHA?

TheJulianJES, 2022-07-08:
@zeltak What stick are you using? Do you also know what firmware version is on it? Like Draghmar mentioned, there are some minimum firmware requirements: https://www.zigbee2mqtt.io/devices/RTCZCGQ11LM.html
Otherwise, try to re-pair the device.

zeltak, 2022-07-08:
hi again @TheJulianJES 

im suing the Sonoff zigbee stick ( Sonoff3pro) and flashed a few weeks ago when I got it with the latest firmware, ie 

```C1352P2_CC2652P_launchpad_coordinator_20220219.zip```

which according to the link should work?

as with @TyzzyT  I get the `stuck in 'Starting Interview'.` thing

thx for the help!

Z

closed 2022-07-08 by dmulcahey

WeterPeter, 2022-08-11:
I am running HA 2022.7 using ZHA and the FP1 pairs immediately and is now running fine in my office.
Upgraded two days ago to HA 2022.8.2, still running fine.

Hedda, 2022-08-11:
@zeltak Make sure to read and follow best practices by using a long USB extension cable, etc. to try to avoid interference, see -> https://github.com/zigpy/zigpy/wiki/Generic-best-practice-tips-on-improving-Zigbee-network-range-and-general-stability

jwall87, 2022-09-02:
> > > Put this
> > > https://github.com/zigpy/zha-device-handlers/blob/851e784207160645b108e78c577cde2b5704a722/zhaquirks/xiaomi/aqara/motion_ac01.py
> > > Into /config/zha_quirks Then in your configuration.yaml add
> > > ```
> > > zha:
> > >   custom_quirks: true
> > >   custom_quirks_path: /config/zha_quirks
> > > ```
> > 
> > 
> > Thanks for the pointers, but this doesnt work. Just gets stuck in "interviewing device". I put the file you linked in the zha_quirks folder, and no love for me.
> > [image]
> 
> I'm having the same issue. Sonoff Zigbee Bridge with Tasmota firmware, using ZHA to connect to the bridge. But Aqara FP1 is stuck in 'Starting Interview'. Any clue what to try/do?

Did you get this working? I’m in the same scenario with the same hub.

TyzzyT, 2022-09-02:
I ended up buying a Sonoff Zigbee 3.0 stick and flashed it with Zigbee2MQTT. Z2M seems far more reliable than ZHA, and way more options. I'll migrate everything over the coming weeks.

jwall87, 2022-09-06:
> > I ended up buying a Sonoff Zigbee 3.0 stick and flashed it with Zigbee2MQTT. Z2M seems far more reliable than ZHA, and way more options. I'll migrate everything over the coming weeks.
> 
> You don't flash a coordinator with either zha or z2m... Zha is extremely reliable for me. Did you try using zha with the Sonoff Zigbee 3.0 ? That's what I'm using and I have 2 fp1's working without issue.

Thanks I just made the switch. Zigbee2MQTT seems so much better.