zigpy/zha-device-handlers #1534: [Device Support Request] OSRAM Flex Outdoor RGBW

mtilstra, 2022-04-26:
**Is your feature request related to a problem? Please describe.**
I have the OSRAM Flex RGBW and OSRAM Flex Outdoor RGBW light strips.  The OSRAM Flex RGBW strips work with the flexrgbw quirk as expected.  The OSRAM Flex Outdoor RGBW strip has the same Zigbee device signature, but a different model name so it does not work currently.

**Describe the solution you'd like**
The OSRAM Flex Outdoor RGBW quirk should be exactly the same as the OSRAM Flex RGBW quirk, but with this new model name:
"LIGHTIFY FLEX OUTDOOR RGBW"

I have tested this with an edited local copy of the flexrgbw.py quirk on my HA installation and can verify it works correctly.

<details>
<summary>Device signature - this can be acquired by clicking on the "Zigbee Device Signature" button in the device settings</summary>

{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=48042, maximum_buffer_size=64, maximum_incoming_transfer_size=0, server_mask=0, maximum_outgoing_transfer_size=0, descriptor_capability_field=<DescriptorCapability.ExtendedSimpleDescriptorListAvailable|ExtendedActiveEndpointListAvailable: 3>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "3": {
      "profile_id": 260,
      "device_type": "0x0102",
      "in_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0008",
        "0x0300",
        "0xfc0f"
      ],
      "out_clusters": [
        "0x0019"
      ]
    }
  },
  "manufacturer": "OSRAM",
  "model": "LIGHTIFY FLEX OUTDOOR RGBW",
  "class": "zigpy.device.Device"
}

</details>

javicalle, 2022-04-27:
If the quirk as been tested, you can create the PR yourself. It is quite straight and can be done from the web.
Go to the file page and edit the content (the pencil icon):
* https://github.com/zigpy/zha-device-handlers/blob/dev/zhaquirks/osram/flexrgbw.py

Follow the indications and you can get the PR in a few minutes.

mtilstra, 2022-04-27:
I edited the file and created PR #1536

javicalle, 2022-04-27:
Great!
Thanks for contributing.

Could you add in the first post message the reference to this issue?
Like this:
```
- Fixes: #1534
```

This way, this issue will be closed when PR accepted.

mtilstra, 2022-04-27:
I edited the first post in the PR to include the reference to this issue

PR #1536 (2022-04-27): Add Flex Outdoor RGBW Model info
Added model info for OSRAM Flex Outdoor RGBW
fixed model comment in line 1

- Fixes: #1534