zigpy/zha-device-handlers #1535: [Device Support Request] TZE200_k6jhsr0q TS0601 Tuya

caslubas, 2022-04-27:
Hi, sorry, my bad inglish!

When adding a device TZE200_k6jhsr0q, with 4 switches, the HA recognized only two entities, which are related to the signal of the zigbee network, they are:

sensor.tze200_k6jhsr0q_ts0601_595213fe_basic_lqi

sensor.tze200_k6jhsr0q_ts0601_595213fe_basic_rssi

I expected the 4 switches to be recognized.

By checking the list of devices that are supported found here: https://zigbee.blakadder.com/

I noticed that the device is not yet supported, is there a way to add it? What information do I need to provide to help?

I can help with getting the logs, although I have a lot of difficulty with the English language, and I'm still new to Home Assistant, but if they point the way I can get everything they need given my limitations.

Please, if I'm missing any information, I'll be happy to answer.

Here's the link where I bought the device:

https://www.aliexpress.com/item/4001181765345.html?spm=a2g0o.order_list.0.0.79c2caa4QvJ4og&gatewayAdapt=glo2bra

Paste the device signature here.
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0051",
      "in_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0xef00"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZE200_k6jhsr0q",
  "model": "TS0601",
  "class": "zigpy.device.Device"
}

</details>

<details>
<summary>Diagnostic information - this can be acquired by clicking on the "Download Diagnostics" button in the device settings</summary>
{
  "home_assistant": {
    "installation_type": "Home Assistant Supervised",
    "version": "2022.4.7",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.9.9",
    "docker": true,
    "arch": "x86_64",
    "timezone": "America/Campo_Grande",
    "os_name": "Linux",
    "os_version": "5.10.0-13-amd64",
    "supervisor": "2022.04.0",
    "host_os": "Debian GNU/Linux 11 (bullseye)",
    "docker_version": "20.10.14",
    "chassis": "desktop",
    "run_as_root": true
  },
  "custom_components": {
    "alexa_media": {
      "version": "3.11.2",
      "requirements": [
        "alexapy==1.25.3",
        "packaging>=20.3",
        "wrapt>=1.12.1"
      ]
    },
    "hacs": {
      "version": "1.24.5",
      "requirements": [
        "aiogithubapi>=21.11.0"
      ]
    },
    "nodered": {
      "version": "1.0.8",
      "requirements": []
    }
  },
  "integration_manifest": {
    "domain": "zha",
    "name": "Zigbee Home Automation",
    "config_flow": true,
    "documentation": "https://www.home-assistant.io/integrations/zha",
    "requirements": [
      "bellows==0.29.0",
      "pyserial==3.5",
      "pyserial-asyncio==0.6",
      "zha-quirks==0.0.72",
      "zigpy-deconz==0.14.0",
      "zigpy==0.44.2",
      "zigpy-xbee==0.14.0",
      "zigpy-zigate==0.7.4",
      "zigpy-znp==0.7.0"
    ],
    "usb": [
      {
        "vid": "10C4",
        "pid": "EA60",
        "description": "*2652*",
        "known_devices": [
          "slae.sh cc2652rb stick"
        ]
      },
      {
        "vid": "10C4",
        "pid": "EA60",
        "description": "*sonoff*plus*",
        "known_devices": [
          "sonoff zigbee dongle plus"
        ]
      },
      {
        "vid": "10C4",
        "pid": "EA60",
        "description": "*tubeszb*",
        "known_devices": [
          "TubesZB Coordinator"
        ]
      },
      {
        "vid": "1A86",
        "pid": "7523",
        "description": "*tubeszb*",
        "known_devices": [
          "TubesZB Coordinator"
        ]
      },
      {
        "vid": "1A86",
        "pid": "7523",
        "description": "*zigstar*",
        "known_devices": [
          "ZigStar Coordinators"
        ]
      },
      {
        "vid": "1CF1",
        "pid": "0030",
        "description": "*conbee*",
        "known_devices": [
          "Conbee II"
        ]
      },
      {
        "vid": "10C4",
        "pid": "8A2A",
        "description": "*zigbee*",
        "known_devices": [
          "Nortek HUSBZB-1"
        ]
      },
      {
        "vid": "10C4",
        "pid": "8B34",
        "description": "*bv 2010/10*",
        "known_devices": [
          "Bitron Video AV2010/10"
        ]
      }
    ],
    "codeowners": [
      "@dmulcahey",
      "@adminiuga"
    ],
    "zeroconf": [
      {
        "type": "_esphomelib._tcp.local.",
        "name": "tube*"
      }
    ],
    "after_dependencies": [
      "usb",
      "zeroconf"
    ],
    "iot_class": "local_polling",
    "loggers": [
      "aiosqlite",
      "bellows",
      "crccheck",
      "pure_pcapy3",
      "zhaquirks",
      "zigpy",
      "zigpy_deconz",
      "zigpy_xbee",
      "zigpy_zigate",
      "zigpy_znp"
    ],
    "is_built_in": true
  },
  "data": {
    "ieee": "**REDACTED**",
    "nwk": 18231,
    "manufacturer": "_TZE200_k6jhsr0q",
    "model": "TS0601",
    "name": "_TZE200_k6jhsr0q TS0601",
    "quirk_applied": false,
    "quirk_class": "zigpy.device.Device",
    "manufacturer_code": 4098,
    "power_source": "Mains",
    "lqi": 148,
    "rssi": -63,
    "last_seen": "2022-04-26T19:45:56",
    "available": true,
    "device_type": "Router",
    "signature": {
      "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
      "endpoints": {
        "1": {
          "profile_id": 260,
          "device_type": "0x0051",
          "in_clusters": [
            "0x0000",
            "0x0004",
            "0x0005",
            "0xef00"
          ],
          "out_clusters": [
            "0x000a",
            "0x0019"
          ]
        }
      }
    },
    "entities": [],
    "neighbors": [
      {
        "device_type": "Coordinator",
        "rx_on_when_idle": "On",
        "relationship": "Parent",
        "extended_pan_id": "**REDACTED**",
        "ieee": "**REDACTED**",
        "nwk": "0x0000",
        "permit_joining": "Unknown",
        "depth": "0",
        "lqi": "162"
      }
    ],
    "endpoint_names": [
      {
        "name": "SMART_PLUG"
      }
    ],
    "user_given_name": null,
    "device_reg_id": "f547be0e4823f3c1e3e6056196aaa1dd",
    "area_id": "area_gourmet"
  }
}
```

</details>

<details>
<summary>Additional logs</summary>

TS0601
por _TZE200_k6jhsr0q
Conectado via [Zigbee Coordinator](..../config/devices/device/dbc82715842d60b4b5c92c2754ab9969)
Zigbee info
IEEE: cc:86:ec:ff:fe:13:52:59
Nwk: 0x4737
Device Type: Router
LQI: 148
RSSI: -63
Visto pela Última Vez: 2022-04-26T19:43:06
Fonte de Energia: Mains
Paste any additional debug logs here.
```

</details>

**Additional context**
Add any other context or screenshots about the feature request here.

caslubas, 2022-04-27:
Logs when adding device:

[0x0000:zdo] ZDO request ZDOCmd.Mgmt_Permit_Joining_req: [60, <Bool.false: 0>]
New device 0x4737 (cc:86:ec:ff:fe:13:52:59) joined the network
[0x4737] Scheduling initialization
Tries remaining: 3
[0x4737] Requesting 'Node Descriptor'
Tries remaining: 2
[0x4737] Extending timeout for 0x18 request
Device 0x4737 (cc:86:ec:ff:fe:13:52:59) joined the network
[0x4737] Scheduling initialization
[0x4737] Canceling old initialize call
Received frame on uninitialized device <Device model=None manuf=None nwk=0x4737 ieee=cc:86:ec:ff:fe:13:52:59 is_initialized=False> from ep 0 to ep 0, cluster 19: b'\x927GYR\x13\xfe\xff\xec\x86\xcc\x8e'
[0x4737:zdo] ZDO request ZDOCmd.Device_annce: [0x4737, cc:86:ec:ff:fe:13:52:59, 142]
Tries remaining: 3
[0x4737] Requesting 'Node Descriptor'
Tries remaining: 2
[0x4737] Extending timeout for 0x1a request
Received frame on uninitialized device <Device model=None manuf=None nwk=0x4737 ieee=cc:86:ec:ff:fe:13:52:59 is_initialized=False> from ep 0 to ep 0, cluster 32770: b'\x1a\x007G\x01@\x8e\x02\x10RR\x00\x00,R\x00\x00'
[0x4737] Got Node Descriptor: NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)
[0x4737] Discovering endpoints
Tries remaining: 3
Received frame on uninitialized device <Device model=None manuf=None nwk=0x4737 ieee=cc:86:ec:ff:fe:13:52:59 is_initialized=False> from ep 0 to ep 0, cluster 32773: b'\x1c\x007G\x01\x01'
[0x4737] Discovered endpoints: [1]
[0x4737] Initializing endpoints [<Endpoint id=1 in=[] out=[] status=<Status.NEW: 0>>]
[0x4737:1] Discovering endpoint information
Tries remaining: 3
Received frame on uninitialized device <Device model=None manuf=None nwk=0x4737 ieee=cc:86:ec:ff:fe:13:52:59 is_initialized=False> from ep 0 to ep 0, cluster 32772: b'\x1e\x007G\x14\x01\x04\x01Q\x00\x01\x04\x00\x00\x04\x00\x05\x00\x00\xef\x02\x19\x00\n\x00'
[0x4737:1] Discovered endpoint information: SizePrefixedSimpleDescriptor(endpoint=1, profile=260, device_type=81, device_version=1, input_clusters=[0, 4, 5, 61184], output_clusters=[25, 10])
Unknown cluster 0xEF00
[0x4737:1:0x0000] Sending request header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=False, is_reply=0, disable_default_response=0, reserved=0, *is_cluster=False, *is_general=True), tsn=32, command_id=<GeneralCommand.Read_Attributes: 0>, *is_reply=False)
[0x4737:1:0x0000] Sending request: Read_Attributes(attribute_ids=[4, 5])
[0x4737:1:0x0000] Received ZCL frame: b'\x18 \x01\x04\x00\x00B\x10_TZE200_k6jhsr0q\x05\x00\x00B\x06TS0601'
[0x4737:1:0x0000] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=0, is_reply=1, disable_default_response=1, reserved=0, *is_cluster=False, *is_general=True), tsn=32, command_id=1, *is_reply=True)
[0x4737:1:0x0000] Decoded ZCL frame: Basic:Read_Attributes_rsp(status_records=[ReadAttributeRecord(attrid=0x0004, status=<Status.SUCCESS: 0>, value=TypeValue(type=CharacterString, value='_TZE200_k6jhsr0q')), ReadAttributeRecord(attrid=0x0005, status=<Status.SUCCESS: 0>, value=TypeValue(type=CharacterString, value='TS0601'))])
[0x4737] Read model 'TS0601' and manufacturer '_TZE200_k6jhsr0q' from <Endpoint id=1 in=[basic:0x0000, groups:0x0004, scenes:0x0005, None:0xEF00] out=[ota:0x0019, time:0x000A] status=<Status.ZDO_INIT: 1>>
[0x4737] Discovered basic device information for <Device model='TS0601' manuf='_TZE200_k6jhsr0q' nwk=0x4737 ieee=cc:86:ec:ff:fe:13:52:59 is_initialized=True>
Device is initialized <Device model='TS0601' manuf='_TZE200_k6jhsr0q' nwk=0x4737 ieee=cc:86:ec:ff:fe:13:52:59 is_initialized=True>
Checking quirks for _TZE200_k6jhsr0q TS0601 (cc:86:ec:ff:fe:13:52:59)
Considering <class 'zhaquirks.xbee.xbee_io.XBeeSensor'>
Fail because endpoint list mismatch: {232, 230} {1}
Considering <class 'zhaquirks.xbee.xbee3_io.XBee3Sensor'>
Fail because endpoint list mismatch: {232, 230} {1}
Considering <class 'zhaquirks.smartthings.tag_v4.SmartThingsTagV4'>
Fail because device_type mismatch on at least one endpoint
Considering <class 'zhaquirks.smartthings.multi.SmartthingsMultiPurposeSensor'>
Fail because device_type mismatch on at least one endpoint
Considering <class 'zhaquirks.netvox.z308e3ed.Z308E3ED'>
Fail because device_type mismatch on at least one endpoint
Considering <class 'zhaquirks.gledopto.soposhgu10.SoposhGU10'>
Fail because endpoint list mismatch: {11, 13} {1}
Considering <class 'bellows.zigbee.application.EZSPCoordinator'>
Fail because device_type mismatch on at least one endpoint
'sensor' component -> 'RSSISensor' using ['basic']
'sensor' component -> 'LQISensor' using ['basic']
device - 0x4737:cc:86:ec:ff:fe:13:52:59 entering async_device_initialized - is_new_join: True
[... cut ...]
[0x4737:1:0xef00] Received ZCL frame: b'\t\x1f\x11\x00\x9f\xd0'
[0x4737:1:0xef00] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.CLUSTER_COMMAND: 1>, is_manufacturer_specific=0, is_reply=1, disable_default_response=0, reserved=0, *is_cluster=True, *is_general=False), tsn=31, command_id=17, *is_reply=True)
[0x4737:1:0xef00] Unknown cluster command 17 b'\x00\x9f\xd0'
[0x4737:1:0xef00] Received command 0x11 (TSN 31): b'\x00\x9f\xd0'
[0x4737:1:0xef00] No explicit handler for cluster command 0x11: b'\x00\x9f\xd0'

javicalle, 2022-04-27:
Have you tried to add your device to the `TuyaQuadrupleSwitchTO` quirk?

If you are not familiar, there are some guides about it. Maybe the most popular can be:
* https://github.com/zigpy/zha-device-handlers/discussions/693#discussioncomment-857274

You will need to copy the `ts0601_switch.py` file in your local `custom_zha_quirks` folder:
* https://raw.githubusercontent.com/zigpy/zha-device-handlers/dev/zhaquirks/tuya/ts0601_switch.py

Add your manufacturer in the `TuyaQuadrupleSwitchTO` MODELS_INFO:
```python
        MODELS_INFO: [
            ("_TZE200_1ozguk6x", "TS0601"),
            ("_TZE200_aqnazj70", "TS0601"),
            ("_TZE200_k6jhsr0q", "TS0601"),
        ],
```

Save & restart.

Then check if device signature changes, that would mean that the device loads the quirk.

javicalle, 2022-04-27:
And please, wrap the logs with 3 apostrophes ```. Or even better inside a `<details>` tag:

> \<details>
> \<summary>Logs when adding device</summary>
> 
> \``` 
> Paste your logs here.
> \``` 
> 
> \</details>

caslubas, 2022-04-28:
Thanks for the answer!

I will perform all procedures and report if successful!

Grateful!

caslubas, 2022-04-30:
Good news

Today I had time to work on the switch

As I had never created a quirks before, I tried to follow the tutorials that @javicalle indicated, but the tutorials are for old versions of Home Assistant, and the new versions already have the quirks enabled, which led me to errors, but gave me the idea of ​​what was happening, so I will describe the steps I followed, to have the switch working correctly:

We must consider, that I didn't have any quirks working on the home assistant, it may be that some steps are not necessary, skip it, if that's the case.

First, I edited the configuratiom.yaml, adding the lines:

zha:
  custom_quirks_path: /config/custom_zha_quirks/

I knew I didn't have the basics, as most tutorials are old and refer to old versions of HA, I followed these tutorials and when saving the configuration.yaml, it gave an error...
But the HA of the last versions is already configured for the use of quirks (I don't know what the first version is), so I had to edit the configuration.yaml and inform the folder of the quirks, it looked like this:

zha:
  custom_quirks_path: /config/custom_zha_quirks/
  
Then:
  
I created the directory "custom_zha_quirks" inside "config"
I created the file "ts0601_switch.py" inside "custom_zha_quirks"
I pasted the content from the link https://raw.githubusercontent.com/zigpy/zha-device-handlers/dev/zhaquirks/tuya/ts0601_switch.py, posted by colleague @javicalle ,
I added my device in the TuyaQuadrupleSwitchTO MODELS_INFO session, as described by the colleague, looking like this:

MODELS_INFO: [
            ("_TZE200_aqnazj70", "TS0601"),
            ("_TZE200_1ozguk6x", "TS0601"),
            ("_TZE200_k6jhsr0q", "TS0601"),
        ],
        
Saved, restarted Home Assistant.

Now I have all 4 switches working!

Thanks to @javicalle  for showing me the way to go.
Thanks to @koying , for posting information on how to enable the use of quirks in Home Assistant at the following link:
https://community.home-assistant.io/t/how-to-setup-local-zha-quirks/341226
Thanks to @MattWestb 
Thanks to the Home Assistant community.

closed 2022-04-30 by caslubas

javicalle, 2022-04-30:
Great!
May I suggest you to create a PR with your device?

You can edit the file directly from the web page:
* https://github.com/zigpy/zha-device-handlers/blob/dev/zhaquirks/tuya/ts0601_switch.py

Follow the indications and done.

This way the device will be included in next version for other users.

MattWestb, 2022-04-30:
Our brave user have doing it  https://github.com/zigpy/zha-device-handlers/pull/1542 :-)))