zigpy/zha-device-handlers #1539: Migrate Tuya's `PowerOnState` attribute to `start_up_on_off` OnOff cluster attribute

javicalle, 2022-04-28:
Now that @dmulcahey has added support for configuring from HA the `power on state` of ZHA devices, I've realized that the implementation followed with Tuya devices is not ideal.

In Tuya devices, the `power_on_state (id: 0x8002)` attribute has been added to manage this property instead of using the `start_up_on_off (id: 0x4003)` attribute, but if the standard Zigbee property is used, we can take advantage of the implementation made to also control those Tuya devices from HA without any extra coding.

I am in favor of quirks implementing standard Zigbee attributes/functionalities whenever possible, and for me this is a clear candidate to improve the implementation.

However, there are some aspects to review before tackling the task:

1. There is no match between the Tuya attribute values and Zigbee attribute:

| Status | Zigbee | Tuya |
| --- | --- | --- |
| `Off` | 0x00| 0x00 | 
| `On` | 0x01 | 0x01 |
| `Toggle` | 0x02 | _don't exists_ |
| `PreviousValue` | 0xFF | 0x02|

2. In Tuya's MCU devices this attribute is just a quirk attribute (_fake_) to read and set the value, but I am not sure about the behavior in Tuya's (real) Zigbee devices. There is a real 0x8002 attribute in these devices?

In function of the answer to point 2. I see two approachs:
a. remove the `power_on_state (id: 0x8002)` attribute and make use of a wrapper class to 'translate' values (¿?)
b. overwritte the read/write `start_up_on_off (id: 0x4003)` attribute to make use of (and translate) the get/set `power_on_state (id: 0x8002)`

Any comment or help on this is welcome.

I can take care of the implementation and I can test with some MCU type devices that I have, but I think that the solution for the Tuya Zigbee devices can condition the implementation of the MCUs.

References:
* https://github.com/home-assistant/core/pull/70110
* https://github.com/zigpy/zigpy/blob/cab1032f7e8315682dcbaf4906bc644a3f99b4ee/zigpy/zcl/clusters/general.py#L606-L610
* https://github.com/zigpy/zha-device-handlers/blob/0163c7ca3dff3508544bca0d57b57d0156342c80/zhaquirks/tuya/__init__.py#L841-L846

javicalle, 2022-04-28:
Also, in devices with multiple switches, is possible to keep diferents behaviors in every switch? What is the behavior on these devices when setting the attribute on different switches?

On MCU devices the behavior can be managed for all switches at once, but not differently on each endpoint/switch.

MattWestb, 2022-04-28:
Great thinking @javicalle  !!
Is it possible "tagging" the attribute as manufacture attribute and ZHA is not using the normal variables and is using it from the manufacture code is linked 2  instead ? 
Also many tuya "normal Zigbee device" is not  needing any quirk so its not working for them so perhaps need one logic with manufacture number red from the device for using the tuya ones ?

javicalle, 2022-04-28:
I'm thinking in all those devices using the `TuyaZBOnOffAttributeCluster`. There are switches, plugs,... not sure if all them are having the `power_on_state` functional.

Yes, it's possible to made use of any manufacturer attribute, but if you want to configure it in from the HA UI, must be implemented. You need to configure the attribute and a select options for all possible values. I'm looking at the implemented ones, but it's away from my knowledge for now.
These features are starting to be implemented in HA (🎉 thanks David!), but I don't think we should try to implement non-standard attributes for now.

MattWestb, 2022-04-28:
The LIDL 3AC  4USB we have implanting the attribute on all EPs but i have not  testing is its working OK (only working on old firmware then we have not fixing casting the magic spell on the new firmware) but i cant testing it then i have working over 12 hours and need sleeping.
https://github.com/zigpy/zha-device-handlers/blob/0163c7ca3dff3508544bca0d57b57d0156342c80/zhaquirks/tuya/ts011f_plug.py#L154

javicalle, 2022-04-29:
I wasn't aware that my LIDL multi socket is one 3AC 4USB device 😓 
I have performed some tests and attribute `power_on_state (id: 0x8002)` from any endpoint manages all devices behaviour. Value is updated for all endpoints.

I believe that this allows to easily quirk the `start_up_on_off (id: 0x4003)` attribute.

I will try to get a version in the weekend.

javicalle, 2022-04-29:
> Do you have one tuya ZBGW then you can updating it and loosing the functionality on EP 2 and 3 and and only EP one is working for all (need tuya magic being casted on it for being "normal" and PR is made for fixing it).

I'm not sure if I want this improvement 😅 

> The problem is "normal" tuya devices that is not using / needing quirks for getting the "tuya standard Zigbee attributes" without quirking all of them.

Let's take the `Plug_3AC_4USB` example.
We can use it as a normal Zigbee device. But if we want to make use of any of the manufacturer attributes (`child_lock`, `backlight_mode`, `power_on_state`, `switch_mode`) we need to configure the quirk for the device, otherwise we could not use these attributes.

If the device uses the standard attributes, it can be controlled in HA. If the attribute uses manufacturer attributes, it will first need a quirk to expose it in HA and then it could be controlled from HA. 
If we already have to do the quirk, as long as is possible, let's make it behave like the standard. I understand that this is the objective of the that library, in front of exposing the particularities of each manufacturer.

Surely in the long run it will be necessary to see how those attributes that do not have a counterpart in the standard are managed, but I see it more as the natural evolution of what is done with the standard attributes.
Because, can you imagine how has to be to implement the controls for TRV devices? 

But all this is still _my personal opinion_. I only speak for myself. Each one is free to interpret the needs and priorities that they consider.
And whenever I can, I will help in any way possible.

MattWestb, 2022-04-29:
tuya Zigbee devices is not so bad but they have only certificated some of them also the LIDL 3AC 4 USB is one of them but **not** the updated firmware !!!

All tuya DP/MCU shall not being converted to Zigbee ZCL commands / attributes then its distorting ZHA and its making the device being very bad implemented.
Best wold being only piping the tuya cluster to one integration in HA that is using the DP for implanting the function in HA together with tuya WiFi device and other that is using DP commands.

I have helping our tuya RTV devs so i knowing how much tuya have making there magic things but in the end all of them have moving on then we was not allowed implanting description on cluster that was custom tagged and other devices have getting own custom code implanted for fixing some extras that the "needing".

For fixing normal tuya devices extra attributes shall not being so large thing only need reading the attributes so they is saved in the zigbee.db and the system is knowing its have them and if they is there with or without manufacture code tagged its possible making the same as the ZCL standard but using one "patched" version.

ZCL standard power on state shall not being configurated in ZHA if its not saved in Zigbee.db with one value that is not non = the device is not supporting it and can having other attributes that is doing  the same / similar function like tuya or Philips HUE devices.

dmulcahey, 2022-04-29:
> I'm not sure if I want this improvement 😅
>
>
> Let's take the `Plug_3AC_4USB` example. We can use it as a normal Zigbee device. But if we want to make use of any of the manufacturer attributes [...]

We can target manufacturer / model / attribute name together in ZHA. What I would say is this: if it is as easy as extending the entity class in ZHA and overriding the attribute name it probably makes more sense to do it in ZHA. If it would drastically change how the support in ZHA works we should probably quirk the device to make it match the spec if it is functionality that the spec offers. For functionality that isn't even in the spec it probably makes more sense to just do the work in ZHA. 

A good example may be the new innovelli switch. We will probably have a ton of innovelli specific configuration entities in ZHA because there is a ton of stuff that isn't part of the spec that the device can be configured to do.

All of this said... we need to evaluate what needs to be done to make this sustainable moving forward. if we follow the approach above things could get unruly very quickly. 

@puddly @Adminiuga do either of you have any thoughts?

MattWestb, 2022-04-29:
ZHA-Tools can scanning attributes and if ZHA is getting them and its storing them in zigbee.db so you is having all you need for see is its need some special treatment.
Making Quirks for all known and unknown tuya device is like being Don Quixote. 
For router device like light and plugs is only reading the attribute after the device is being paired and you is getting the attributes in zigbee.db and if you is reading the standard ZCL attributes for power on state you is getting `none` that i hope is not being saved in the DB as one valid state.

javicalle, 2022-04-30:
Not as simple as I imagine.
I can't overwrite the `write_attribue` and made it work as I was planning.

Adminiuga, 2022-05-01:
> We can target manufacturer / model / attribute name together in ZHA. What I would say is this: if it is as easy as extending the entity class in ZHA and overriding the attribute name it probably makes more sense to do it in ZHA

I also prefer this approach. 
ZHA was developed with the thought that vendors would stick to the specs and quirks were supposed to address some "minor" incompatibilities. 
Well, Tuya buried this idea and I think implementing Tuya in ZHA is more flexible approach in the long run.

javicalle, 2022-05-01:
As far as I can tell, I haven't found a simple and easy way to (for example) capture the reading of the `start_up_on_off` attribute and invoke (and translate) the call to the `power_on_state` attribute to return the proper value.

For my part, I am going to reject the proposal and try to focus it on the idea of exposing in HA the configuration attributes that are convenient.

Although I haven't understood the implementation used yet, I have found some entry about it:
* https://github.com/zigpy/zigpy/discussions/934

Thanks to all for your comments and guidance.

closed 2022-05-01 by javicalle