zigpy/zha-device-handlers #1540: [Device Support Request] Tuya Bling motor battery

thib5, 2022-04-29:
**Describe the solution you'd like**
Hi, Just wondering if it's possible to have battery status on this device

<details>
<summary>Device signature - this can be acquired by clicking on the "Zigbee Device Signature" button in the device settings</summary>

```
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0202",
      "in_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0x0102",
        "0xef00"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZE200_zah67ekd",
  "model": "TS0601",
  "class": "ts0601_cover.TuyaMoesCover0601"
}
```

thib5, 2022-05-01:
I just check the log and I don't think there is a battery status ... could it be do able to "hack" one .. flash the code on it or something ? 

Log for info : 

`{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2022.4.7",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.9.9",
    "docker": true,
    "arch": "x86_64",
    "timezone": "America/Toronto",
    "os_name": "Linux",
    "os_version": "5.10.108",
    "supervisor": "2022.04.0",
    "host_os": "Home Assistant OS 7.6",
    "docker_version": "20.10.9",
    "chassis": "vm",
    "run_as_root": true
  },
  "custom_components": {
    "climate_group": {
      "version": "0.4.2",
      "requirements": []
    },
    "iphonedetect": {
      "version": "1.4.0",
      "requirements": []
    },
    "deebot": {
      "version": "1.3.0",
      "requirements": [
        "deebot-client==1.3.0",
        "numpy>=1.21.2"
      ]
    },
    "scheduler": {
      "version": "v0.0.0",
      "requirements": []
    },
    "tuya_custom": {
      "version": "0.1.7",
      "requirements": []
    },
    "browser_mod": {
      "version": "1.5.3",
      "requirements": []
    },
    "tapo_control": {
      "version": "3.5.0",
      "requirements": [
        "pytapo==2.1",
        "onvif-zeep-async==1.2.0"
      ]
    },
    "hilo": {
      "version": "2022.4.2",
      "requirements": [
        "python-hilo>=2022.3.1"
      ]
    },
    "sonoff": {
      "version": "v2.4.7",
      "requirements": [
        "pycryptodome>=3.6.6"
      ]
    },
[... 161 lines cut ...]
          ],
          "out_clusters": [
            "0x000a",
            "0x0019"
          ]
        }
      }
    },
    "entities": [
      {
        "entity_id": "cover.tze200_zah67ekd_ts0601_64ce07fe_window_covering",
        "name": "_TZE200_zah67ekd TS0601"
      }
    ],
    "neighbors": [],
    "endpoint_names": [
      {
        "name": "WINDOW_COVERING_DEVICE"
      }
    ],
    "user_given_name": null,
    "device_reg_id": "1f6264a7b82e0d703d6a3a2d9e71468b",
    "area_id": null
  }
}`

thib5, 2022-05-01:
oups I accidentally close the

javicalle, 2022-05-01:
These aren't the application logs.

Look at:
* https://www.home-assistant.io/integrations/zha/#debug-logging

Then at home-assistant.log search for info related to your device. You will see logs when you operate your device similar to:
```
2022-05-01 21:50:49 INFO (MainThread) [zigpy.zcl] [0xd37d:1:0xef00] some trace about your device operation ....
```

The `[0xd37d:1:0x0006]` part is important because will identify your device.
Look at any trace for your device that refers to 'unhandled' commands. Usually they came after operate the device.

There's no any other hack or code flash posible.

thib5, 2022-05-04:
Hi, 

I see trace of my device  (witch in my case is : 0x452f:1:0x0000 ) but there is non unhandled commands...

does it mean the device have no battery status ? can you double check my log ? 

[home-assistant.log](https://github.com/zigpy/zha-device-handlers/files/8622127/home-assistant.log)

Thanks

closed 2022-11-08 by github-actions[bot]