zigpy/zha-device-handlers #1544: [Device Support Request] Calibration for Moes tuya cover devices

mdiazgoncalves, 2022-05-02:
I'm trying to calibrate a cover switch from Moes, but it's well over my knowledge. Can someone help to implement this feature? It is already in zigbee2mqtt.

https://github.com/Koenkk/zigbee2mqtt/issues/7128#issuecomment-895928027

Thanks

javicalle, 2022-05-02:
As many Tuya devices there must be a DP for setting the value.
If I have interpreted it correctly from `zigbee2mqtt` it could be the DP=0xF003

What is the quirk that the cover currently uses?
Could you capture the traces when you power up the device? The Tuya devices usually publishes the values of the configuration DPs when is power up. There should be some message related to DP 0xF003

mdiazgoncalves, 2022-05-02:
Yes, I also think the calibration is cluster attribute moesCalibrationTime: {ID: 0xf003, type: dataType_1.default.uint16 }
The quirk is ts0601_cover.TuyaMoesCover0601.
How do I capture the traces when you power up the device?

javicalle, 2022-05-02:
Enable the debug log, remove the power from device and power up again.
To enable logs:
* https://www.home-assistant.io/integrations/zha#debug-logging
* 
Most Tuya devices broadcast initial status on power up. Maybe you can get the DP (and value).
Also, you can look at the logs when device is operated. Some devices broadcast the current status when are operated from the physical device.

Once you verify the DP and value format, must implement the functions in quirk:
1. you will need anothe attribute for the `calibration_time`:
* https://github.com/zigpy/zha-device-handlers/blob/0163c7ca3dff3508544bca0d57b57d0156342c80/zhaquirks/tuya/__init__.py#L102
2. add another handle for the DP:
https://github.com/zigpy/zha-device-handlers/blob/0163c7ca3dff3508544bca0d57b57d0156342c80/zhaquirks/tuya/__init__.py#L1034-L1041
3. another command for the new attribute:
https://github.com/zigpy/zha-device-handlers/blob/0163c7ca3dff3508544bca0d57b57d0156342c80/zhaquirks/tuya/__init__.py#L1141-L1146

I'm not familiar with this kind of implementation, but I think that this is more or less what you need.

mdiazgoncalves, 2022-05-05:
I'll give it a try.
Thanks

closed 2022-11-08 by github-actions[bot]