zigpy/zha-device-handlers #1548: _TZ3000_msl6wxk9 Sensitivity and "On" Timer

foureight84, 2022-05-05:
I added the TS0202 (_TZ3000_msl6wxk9) to ZHA and it seems to work as expected. But I noticed that the PIR motion sensor doesn't detect unless I'm 3 feet away. It seems that there is a way to change the sensitivity. The default is set on low sensitivity with a timer to clear presence detected after 60 seconds. I was able to figure this out based on the information here: https://smarthomescene.com/reviews/cheapest-zigbee-motion-sensor-for-your-smart-home/. But this article refers to settings based on Zigbee2MQTT, and not ZHA.

I was wondering if there's a way to change sensitivity or perhaps query the PIR sensor to get the possible values to override?

[image]

[image]

javicalle, 2022-05-05:
Have you tried to get the attribute value for `num_zone_sensitivity_levels_supported`?

Also, the attribute value for `current_zone_sensitivity_level` can give you a clue about the possible values (0-10, 0-100, 0-255,...)

foureight84, 2022-05-05:
I've tried to get the values for `num_zone_sensitivity_levels_supported` but nothing returns in the fields below. 

As for `current_zone_sensitivity_level ` it's a bit flaky. Sometimes a value of 1 returns, sometimes nothing. It makes me unsure what the ranges would be.

Is there any way to directly query the device? I'm a bit new to this and currently using a Sonoff ZBBridge flashed with tasmota. I guess I lose out on querying the Zigbee in tasmota after changing the config to connect to ZHA.

javicalle, 2022-05-06:
I think that trhere is no way to ask to device for the proper values. Is a matter of trial and error.

According to '[zigbee-herdsman-converters](https://github.com/Koenkk/zigbee-herdsman-converters/blob/a97649c1a6fb57eee5d90c3e8edbd848a619e246/converters/fromZigbee.js#L7395-L7402)' the posible values for the attributes are:
* currentZoneSensitivityLevel: 
  * 'low': 0
  * 'medium': 1
  * 'high': 2
* 61441 (¿`keep_time`?):
  * 30: 0
  * 60: 1
  * 120: 2

If you attach the device signature, maybe we can propose a quirk for the `keep_time` attribute.

foureight84, 2022-05-06:
> I think that trhere is no way to ask to device for the proper values. Is a matter of trial and error.
>
> According to [...]

Thanks for that. I will give those values a try. As requested, here is the device signature:

```
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0402",
      "in_clusters": [
        "0x0000",
        "0x0001",
        "0x0003",
        "0x0500"
      ],
      "out_clusters": [
        "0x0006",
        "0x000a",
        "0x0019",
        "0x1000"
      ]
    }
  },
  "manufacturer": "_TZ3000_msl6wxk9",
  "model": "TS0202",
  "class": "zigpy.device.Device"
}
```

foureight84, 2022-05-06:
I'm not sure if this is expected behavior but I've tried changing the value to 2 for the sensitivity but it doesn't seem to stick. After seeing the green checkmark from pressing "Set Zigbee Attribute" I press "Get Zigbee Attribute" and the value 1 overrides the value input field where I had entered '2'

javicalle, 2022-05-07:
> What is the function of 'Manufacturer Code Override'?

This allows to override the manufacturer code in the Zigbee message. That's needed for some devices (maybe also yours).

> I'm not sure if this is expected behavior

No, it is not. Here the logs of the command would help to see what happens here:
* https://www.home-assistant.io/integrations/zha#debug-logging

javicalle, 2022-05-07:
Here you have a proposed quirk for this device:

<details>
<summary>ts0202.py</summary>

```python
"""Tuya PIR motion sensor."""

from zigpy.profiles import zha
from zigpy.quirks import CustomDevice
import zigpy.types as t
from zigpy.zcl.clusters.general import (
    Basic,
    Identify,
    OnOff,
    Ota,
    PowerConfiguration,
    Time,
)
from zigpy.zcl.clusters.lightlink import LightLink
from zigpy.zcl.clusters.security import IasZone

from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODELS_INFO,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
)

class SensitivityLevel(t.enum8):
    """Sensitivity level enum."""

    LOW = 0x00
    MEDIUM = 0x01
    HIGH = 0x02

class OnTimeValues(t.enum8):
    """Sensitivity level enum."""

    SEG_30 = 0x00
    SEC_60 = 0x01
    SEC_120 = 0x02

class PirSensor(IasZone):
    """IasZone with extra attributes."""

    attributes = IasZone.attributes.copy()
    # Wrap `current_zone_sensitivity_level` posible values
    attributes.update({0x0013: ("current_zone_sensitivity_level", SensitivityLevel)})
    attributes.update({0xF001: ("on_time", OnTimeValues)})

class PirMotion(CustomDevice):
    """Tuya PIR motion sensor."""

    signature = {
        MODELS_INFO: [("_TZ3000_msl6wxk9", "TS0202")],
        # endpoints=1 profile=260 device_type=0x0402
        # in_clusters=[0x0000, 0x0001, 0x0003, 0x0500],
        # out_clusters=[0x0006, 0x000a, 0x0019, 0x1000]
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.IAS_ZONE,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    PowerConfiguration.cluster_id,
                    Identify.cluster_id,
                    IasZone.cluster_id,
                ],
                OUTPUT_CLUSTERS: [
                    OnOff.cluster_id,
                    Time.cluster_id,
                    Ota.cluster_id,
                    LightLink.cluster_id,
                ],
            }
        },
    }

    replacement = {
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.IAS_ZONE,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    PowerConfiguration.cluster_id,
                    Identify.cluster_id,
                    PirSensor,
                ],
                OUTPUT_CLUSTERS: [
                    OnOff.cluster_id,
                    Time.cluster_id,
                    Ota.cluster_id,
                    LightLink.cluster_id,
                ],
            }
        }
    }
```

</details>

This quirk just add the 2 known attributes and wrap the possible values.

If you are not familiar with how to test a local quirk, take a look at this entry:
* https://github.com/zigpy/zha-device-handlers/discussions/693#discussioncomment-857274

There's no guarantee that the new `on_time` attribute will work either, but we don't lose anything by testing.

The traces of the executions of the commands may give us some information.

foureight84, 2022-05-16:
Thanks for this. The quirk does work. However, time and sensitivity do not do anything. It makes sense why I was not able to set the values. I verified this with Zigbee2MQTT as well. Even though I am able to set the sensitivity to high and time to 120 seconds, there's no functional difference on the device.  I don't think the advertised features are functional at all.

javicalle, 2022-05-17:
Just to be sure, do properties work in Z2M?
I just verified that Z2M has a peculiarity when sending the commands `sendWhen: 'active'` almost certainly because it is a battery-powered device that goes to sleep mode. I haven't seen that functionality in ZHA.

Could you try updating the attributes while keeping the device 'active'? Normally this can be forced by pressing the physical button on the device, which prevents it from going to sleep.

When you update the values, what do you get back when you hit 'get zigbee attributes'? does it return the new value or the old value?

foureight84, 2022-05-17:
That's what I thought as well, that the device functions differently in a sleep state. However, regardless of "sleeping" or "active" sending new sensitivity values, or time active values does not save. I could never get the existing time active value and sensitivity always returns 1.

javicalle, 2022-05-17:
Well, no more ideas from my side 🤷🏻‍♂️ 
Maybe the device is using a specific command or is using another DP, but without a sniff of the device's messaging it's hard to tell.

foureight84, 2022-05-18:
How do I go about sniffing the device's messaging? I'm definitely open to help with that

KrOnAsK, 2022-09-22:
I would also love to change those values. So if I can help just tell me what to do

javicalle, 2022-09-22:
Enable the debug logs and attach the logs from your device.
Once enabled, remove and put batteries, this way device will report its configuration and maybe we can see any relevant info.

KrOnAsK, 2022-09-22:
okay as I am quiet new to this all, how do I enable debug logs ?

javicalle, 2022-09-22:
> okay as I am quiet new to this all, how do I enable debug logs ?

https://www.home-assistant.io/integrations/zha#debug-logging

closed 2023-07-10 by github-actions[bot]