zigpy/zha-device-handlers #1550: [Device Support Request] - URC4460BC0-X-R by Universal Electronics Inc or 4460 Comcast or UEI XHS2-UE Door/Window Sensor. Xfinity Home Contact Sensor

lonejeeper, 2022-05-05:
**Is your feature request related to a problem? Please describe.**
The devices have a power entity (in Home Assistant), but it is unknown.  I'd like to have the battery information available.  This is my first device request, so if there's anything I
[image]
've missed, I'd be glad to provide it. 

<details>
<summary>Device signature - this can be acquired by clicking on the "Zigbee Device Signature" button in the device settings</summary>

```
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4335, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=0, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0402",
      "in_clusters": [
        "0x0000",
        "0x0001",
        "0x0003",
        "0x0020",
        "0x0402",
        "0x0500",
        "0x0b05"
      ],
      "out_clusters": [
        "0x0019"
      ]
    }
  },
  "manufacturer": "Universal Electronics Inc",
  "model": "URC4460BC0-X-R",
  "class": "zigpy.device.Device"
}
```

</details>

<details>
<summary>Diagnostic information - this can be acquired by clicking on the "Download Diagnostics" button in the device settings</summary>

```
{
  "home_assistant": {
    "installation_type": "Home Assistant Supervised",
    "version": "2022.5.0",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.9.9",
    "docker": true,
    "arch": "x86_64",
    "timezone": "America/New_York",
    "os_name": "Linux",
    "os_version": "5.10.0-13-amd64",
    "supervisor": "2022.05.0",
    "host_os": "Debian GNU/Linux 11 (bullseye)",
    "docker_version": "20.10.15",
    "chassis": "laptop",
    "run_as_root": true
  },
  "custom_components": {
    "blitzortung": {
      "version": "1.0.1",
      "requirements": [
        "paho-mqtt>=1.5.0"
      ]
    },
    "nodered": {
      "version": "1.0.8",
      "requirements": []
    },
    "fullykiosk": {
      "version": "1.0.1",
      "requirements": [
        "python-fullykiosk==0.0.11"
      ]
    },
    "powercalc": {
      "version": "v0.19.15",
      "requirements": [
        "numpy>=1.21.1"
      ]
    },
    "adaptive_lighting": {
      "version": "1.0.14",
      "requirements": []
    },
    "hacs": {
      "version": "1.24.5",
      "requirements": [
        "aiogithubapi>=21.11.0"
      ]
    },
    "deebot": {
      "version": "1.3.0",
      "requirements": [
        "deebot-client==1.3.0",
        "numpy>=1.21.2"
      ]
    },
    "nws_alerts": {
      "version": "2.4",
      "requirements": []
    },
    "alexa_media": {
      "version": "3.11.3",
      "requirements": [
        "alexapy==1.25.5",
        "packaging>=20.3",
        "wrapt>=1.12.1"
      ]
    },
    "alarmo": {
      "version": "v1.9.0",
      "requirements": []
    },
    "tryfi": {
      "version": "0.0.12",
      "requirements": [
        "pytryfi>=0.0.14.1"
      ]
    },
    "samsungtv_encrypted": {
      "version": "2022.3.1",
      "requirements": [
        "wakeonlan>=1.1.6",
        "beautifulsoup4>=4.6.0",
        "netdisco>=2.8.3"
      ]
    },
    "garbage_collection": {
      "version": "4.7.3",
      "requirements": [
        "python-dateutil>=2.8.2"
      ]
    },
    "dreamscreen": {
      "version": "2020.8.5",
      "requirements": [
        "PyDreamScreen>=0.0.14"
      ]
    },
    "emscrss": {
      "version": "1.1",
      "requirements": [
        "georss_emsc_csem_earthquakes_client==0.3"
      ]
    }
  },
  "integration_manifest": {
    "domain": "zha",
    "name": "Zigbee Home Automation",
    "config_flow": true,
    "documentation": "https://www.home-assistant.io/integrations/zha",
    "requirements": [
      "bellows==0.29.0",
      "pyserial==3.5",
      "pyserial-asyncio==0.6",
      "zha-quirks==0.0.73",
      "zigpy-deconz==0.16.0",
      "zigpy==0.45.1",
      "zigpy-xbee==0.14.0",
      "zigpy-zigate==0.7.4",
      "zigpy-znp==0.7.0"
    ],
    "usb": [
      {
        "vid": "10C4",
        "pid": "EA60",
        "description": "*2652*",
        "known_devices": [
          "slae.sh cc2652rb stick"
        ]
      },
      {
        "vid": "10C4",
        "pid": "EA60",
        "description": "*sonoff*plus*",
        "known_devices": [
          "sonoff zigbee dongle plus"
        ]
      },
      {
        "vid": "10C4",
        "pid": "EA60",
        "description": "*tubeszb*",
        "known_devices": [
          "TubesZB Coordinator"
        ]
      },
      {
        "vid": "1A86",
        "pid": "7523",
        "description": "*tubeszb*",
        "known_devices": [
          "TubesZB Coordinator"
        ]
      },
      {
        "vid": "1A86",
        "pid": "7523",
        "description": "*zigstar*",
        "known_devices": [
          "ZigStar Coordinators"
        ]
      },
      {
        "vid": "1CF1",
        "pid": "0030",
        "description": "*conbee*",
        "known_devices": [
          "Conbee II"
        ]
      },
      {
        "vid": "10C4",
        "pid": "8A2A",
        "description": "*zigbee*",
        "known_devices": [
          "Nortek HUSBZB-1"
        ]
      },
      {
        "vid": "10C4",
        "pid": "8B34",
        "description": "*bv 2010/10*",
        "known_devices": [
          "Bitron Video AV2010/10"
        ]
      }
    ],
    "codeowners": [
      "@dmulcahey",
      "@adminiuga"
    ],
    "zeroconf": [
      {
        "type": "_esphomelib._tcp.local.",
        "name": "tube*"
      }
    ],
    "after_dependencies": [
      "usb",
      "zeroconf"
    ],
    "iot_class": "local_polling",
    "loggers": [
      "aiosqlite",
      "bellows",
      "crccheck",
      "pure_pcapy3",
      "zhaquirks",
      "zigpy",
      "zigpy_deconz",
      "zigpy_xbee",
      "zigpy_zigate",
      "zigpy_znp"
    ],
    "is_built_in": true
  },
  "data": {
    "ieee": "**REDACTED**",
    "nwk": 30150,
    "manufacturer": "Universal Electronics Inc",
    "model": "URC4460BC0-X-R",
    "name": "Universal Electronics Inc URC4460BC0-X-R",
    "quirk_applied": false,
    "quirk_class": "zigpy.device.Device",
    "manufacturer_code": 4335,
    "power_source": "Battery or Unknown",
    "lqi": 255,
    "rssi": -15,
    "last_seen": "2022-05-05T19:20:39",
    "available": true,
    "device_type": "EndDevice",
    "signature": {
      "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4335, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=0, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
      "endpoints": {
        "1": {
          "profile_id": 260,
          "device_type": "0x0402",
          "in_clusters": [
            "0x0000",
            "0x0001",
            "0x0003",
            "0x0020",
            "0x0402",
            "0x0500",
            "0x0b05"
          ],
          "out_clusters": [
            "0x0019"
          ]
        }
      }
    },
    "entities": [
      {
        "entity_id": "button.door_bathroom_kids_identify",
        "name": "Universal Electronics Inc URC4460BC0-X-R"
      },
      {
        "entity_id": "sensor.door_bathroom_kids_power",
        "name": "Universal Electronics Inc URC4460BC0-X-R"
      },
      {
        "entity_id": "sensor.door_bathroom_kids_temperature",
        "name": "Universal Electronics Inc URC4460BC0-X-R"
      },
      {
        "entity_id": "binary_sensor.door_bathroom_contact",
        "name": "Universal Electronics Inc URC4460BC0-X-R"
      }
    ],
    "neighbors": [],
    "endpoint_names": [
      {
        "name": "IAS_ZONE"
      }
    ],
    "user_given_name": "Door - Bathroom - Kids",
    "device_reg_id": "9e3b1d0c9094f6596a515ea63b721a17",
    "area_id": null
  }
}
```

</details>

**Additional context**
Add any other context or screenshots about the feature request here.

javicalle, 2022-05-07:
It seems that there are multiple implementations for this device.

You can try with that quirk:

<details>
<summary>ts0202.py</summary>

```python
"""XHS2-UE Door/Window Sensor."""
from zigpy.profiles import zha
from zigpy.quirks import CustomDevice
from zigpy.zcl.clusters.general import Basic, Identify, Ota, PollControl
from zigpy.zcl.clusters.homeautomation import Diagnostic
from zigpy.zcl.clusters.measurement import TemperatureMeasurement
from zigpy.zcl.clusters.security import IasZone

from zhaquirks import PowerConfigurationCluster
from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODELS_INFO,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
)

class CustomPowerConfigurationCluster(PowerConfigurationCluster):
    """Custom PowerConfigurationCluster."""

    cluster_id = PowerConfigurationCluster.cluster_id
    MIN_VOLTS = 2.1
    MAX_VOLTS = 3.0

class ContactSensor(CustomDevice):
    """XHS2-UE Door/Window Sensor."""

    signature = {
        #  <SimpleDescriptor endpoint=1 profile=260 device_type=1026
        # device_version=0
        # input_clusters=[0, 1, 3, 1026, 1280, 32, 2821]
        # output_clusters=[25]>
        MODELS_INFO: [("Universal Electronics Inc", "model": "URC4460BC0-X-R")],
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.IAS_ZONE,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    CustomPowerConfigurationCluster.cluster_id,
                    Identify.cluster_id,
                    TemperatureMeasurement.cluster_id,
                    IasZone.cluster_id,
                    PollControl.cluster_id,
                    Diagnostic.cluster_id,
                ],
                OUTPUT_CLUSTERS: [Ota.cluster_id],
            }
        },
    }

    replacement = {
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    CustomPowerConfigurationCluster,
                    Identify.cluster_id,
                    TemperatureMeasurement.cluster_id,
                    IasZone.cluster_id,
                    PollControl.cluster_id,
                    Diagnostic.cluster_id,
                ],
                OUTPUT_CLUSTERS: [Ota.cluster_id],
            }
        }
    }
```

</details>

lonejeeper, 2022-05-08:
Hi, thanks for the help.  I'm stuck with an error I can't seem to suss out.  

`Logger: homeassistant.config_entries
Source: custom_zha_quirks/uei.py:28
First occurred: 1:45:30 PM (1 occurrences)
Last logged: 1:45:30 PM

Error setting up entry HubZ Smart Home Controller, s/n: 61201F63 - Silicon Labs for zha
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 335, in async_setup
    result = await component.async_setup_entry(hass, self)
  File "/usr/src/homeassistant/homeassistant/components/zha/__init__.py", line 99, in async_setup_entry
    setup_quirks(config)
  File "/usr/local/lib/python3.9/site-packages/zhaquirks/__init__.py", line 409, in setup
    importer.find_module(modname).load_module(modname)
  File "<frozen importlib._bootstrap_external>", line 529, in _check_name_wrapper
  File "<frozen importlib._bootstrap_external>", line 1029, in load_module
  File "<frozen importlib._bootstrap_external>", line 854, in load_module
  File "<frozen importlib._bootstrap>", line 274, in _load_module_shim
  File "<frozen importlib._bootstrap>", line 711, in _load
  File "<frozen importlib._bootstrap>", line 680, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 850, in exec_module
  File "<frozen importlib._bootstrap>", line 228, in _call_with_frames_removed
  File "/config/custom_zha_quirks/uei.py", line 28, in <module>
    class ContactSensor(CustomDevice):
  File "/usr/local/lib/python3.9/site-packages/zigpy/quirks/__init__.py", line 65, in __init_subclass__
    _DEVICE_REGISTRY.add_to_registry(cls)
  File "/usr/local/lib/python3.9/site-packages/zigpy/quirks/registry.py", line 35, in add_to_registry
    for manuf, model in models_info:
ValueError: too many values to unpack (expected 2)`

javicalle, 2022-05-08:
Uppsss, my fault. I have fixed the code.
Just replace this part:
```python
        MODELS_INFO: [("Universal Electronics Inc", "URC4460BC0-X-R")],
```

lonejeeper, 2022-05-10:
Not a problem, I appreciate the help!   I've replaced the line.  That appears to have done it!   

I was troubleshooting in the ballpark but never thought to remove the "model" part, and it was staring me in the face with the "too many values" error.  Gah.

Thanks!

closed 2022-05-12 by lonejeeper

HVR88, 2022-06-05:
Is there an additional secret to making this work?  My device says the quirk is loaded but no power info, just says "unknown" the same as before the quirk.

```
URC4460BC0-X-R
by Universal Electronics Inc
Connected via [Sonoff Zigbee Controller](http://homeassistant.local:8123/config/devices/device/f956cdbe7dda7d87d4cf1dcead89ef4f)
Firmware: 0x20160921

Zigbee info

IEEE: 00:0d:6f:00:10:c2:e9:4b
Nwk: 0xef1e
Device Type: EndDevice
LQI: 69
RSSI: Unknown
Last Seen: 2022-06-05T17:55:18
Power Source: Battery or Unknown
Quirk: uei.ContactSensor
```

javicalle, 2022-06-05:
Have you tried to repair the device?
Any errors in your logs?

HVR88, 2022-06-05:
**So, I just checked and battery percentage is working now**.  Go figure. I just left the system since I posted above (about two hours ago) and now it's reporting a value. Weird.

While previously testing, there were no errors that I've found.  And to get this sensor working I had to add it a couple of times before it was stable and open/close worked reliably. Restarted a few times over a period of some 20 minutes after adding the quirk and no joy.

MattWestb, 2022-06-06:
Some device is not getting the reporting configured OK then paring and need being reconfigured from the device card (wake up sleepers then doing it) like all IKEA controllers.
If it working OK letting it and if problem try reconfig.

pasyn, 2023-01-01:
Since this fix has not yet been added to the source, can we reopen this issue until a pull request is merged?