zigpy/zha-device-handlers #1554: [BUG]Unexpected interaction with New Dimmer quirk and Dimmer switch TS0601 _TZE200_1agwnems

yftj, 2022-05-08:
Thanks to the great work by the contributors here I have full control of my [dimmer switches](https://www.ozsmartthings.com.au/products/zigbee-dimmer-switch). (dimmer switch TS0601 _TZE200_1agwnems)! Thank you very much! 
There is however a small problem, changing the brightness slider causes the light to momentarily turn off almost like a reset for a fraction of a second (time it takes to blink almost) and then from full brightness goes down to the selected brightness. Expected result would be for the dimmer to change from eg. 40% to 50% **instead** it goes from 40% to off(barely noticeable) to 100% and slowly dims to 50%. 

**To Reproduce**
Steps to reproduce the behavior:
1. With the dimmer either on or off change the brightness value of dimmer using either automation or via the entity
2. Become blinded for a time until the light slowly dims to the set value from its max value.

**Expected behavior**
Sliding the brightness slider from eg. 0% to 40% should slowly step through the levels of brightness from 0 to 40% and sliding it bac from 40% to 0% should smoothly transition from 40% brightness to 0%

<details>
<summary>Device signature - this can be acquired by clicking on the "Zigbee Device Signature" button in the device settings</summary>

```
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0100",
      "in_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0008",
        "0xef00"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZE200_1agwnems",
  "model": "TS0601",
  "class": "zhaquirks.tuya.ts0601_dimmer.TuyaSingleSwitchDimmer"
}
```

</details>

<details>
<summary>Diagnostic information - this can be acquired by clicking on the "Download Diagnostics" button in the device settings</summary>

```
{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2022.5.2",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.9.9",
    "docker": true,
    "arch": "x86_64",
    "timezone": "Australia/Sydney",
    "os_name": "Linux",
    "os_version": "5.10.108",
    "supervisor": "2022.05.0",
    "host_os": "Home Assistant OS 7.6",
    "docker_version": "20.10.9",
    "chassis": "vm",
    "run_as_root": true
  },
  "custom_components": {
    "xiaomi_miot_raw": {
      "version": "v1.3.1",
      "requirements": [
[... 281 lines cut ...]
        "rx_on_when_idle": "On",
        "relationship": "Sibling",
        "extended_pan_id": "**REDACTED**",
        "ieee": "**REDACTED**",
        "nwk": "0xC54C",
        "permit_joining": "Unknown",
        "depth": "15",
        "lqi": "102"
      }
    ],
    "endpoint_names": [
      {
        "name": "ON_OFF_LIGHT"
      }
    ],
    "user_given_name": "REDACTEDS's Room",
    "device_reg_id": "1bac9fa15b21a8d52d1e563874e74754",
    "area_id": null
  }
}
```

</details>

 Please let me know if there is any further information you require! Thank you!

javicalle, 2022-05-08:
Please, can you attach a debug log from the device operation?
We need to see if there are multiple commands when light is dimmed or just one command send.

yftj, 2022-05-09:
> Please, can you attach a debug log from the device operation? We need to see if there are multiple commands when light is dimmed or just one command send.

Sorry, I am having trouble finding the debug logs for the device, are you looking for the /config/homeassistant.log file?
I've set the logger to debug in configuration.yaml but there is a ton of text so sift through do I need to filter it? and if so how? Your help is appreciated!

javicalle, 2022-05-09:
Have you enabled the debug logs for `zhaquirks`?
It is expected that you will have traces of the type `Sending Tuya Cluster Command... Cluster Command is %x, Arguments are %s` when turn on/off the device or of the type `Sending Tuya Cluster Command. Cluster Command is %x, Arguments are %s` when dimming the device.

In addition there are traces of at least 5 different devices (the first value of the tuple `[0x8073:1:0xef00]` defines the device). 
We need to get the correct info to be able to understand what is happening.

I am aware that it is not an easy task if you are not used to it, but it is necessary for someone with the device to do this analysis in order to give a correct answer.

yftj, 2022-05-10:
> Have you enabled the debug logs for `zhaquirks`? It is expected that you will have traces of the type `Sending Tuya Cluster Command... Cluster Command is %x, Arguments are %s` when turn on/off the [...]

<details>
<summary>Diagnostic information - I enabled debug logs zhaquriks and found some of the "Sending Tuya Cluster Command" lines you were looking for, Please let me know if you need more info,  I greatly appreciate your help</summary>

```
2022-05-10 13:30:23 DEBUG (MainThread) [homeassistant.components.websocket_api.http.connection] [139696786098016] Received {'type': 'call_service', 'domain': 'light', 'service': 'turn_on', 'service_data': {'entity_id': 'light.youssefs_room_level_on_off', 'brightness_pct': 61}, 'id': 89}
2022-05-10 13:30:23 DEBUG (Recorder) [homeassistant.components.recorder] Processing task: EventTask(event=<Event call_service[L]: domain=light, service=turn_on, service_data=entity_id=light.youssefs_room_level_on_off, brightness_pct=61>)
2022-05-10 13:30:23 DEBUG (MainThread) [zigpy.zcl] [0xFE77:1:0x0008] Sending Tuya Cluster Command. Cluster Command is 4, Arguments are (156, 1)
2022-05-10 13:30:23 DEBUG (MainThread) [zigpy.zcl] [0xFE77:1:0xef00] tuya_mcu_command: cluster_data=TuyaClusterData(endpoint_id=1, cluster_attr='on_off', attr_value=1, expect_reply=True)
2022-05-10 13:30:23 DEBUG (MainThread) [zigpy.zcl] [0xFE77:1:0xef00] get_dp_mapping --> found DP: 1
2022-05-10 13:30:23 DEBUG (MainThread) [zigpy.zcl] [0xFE77:1:0xef00] from_cluster_data: 1, DPToAttributeMapping(ep_attribute='on_off', attribute_name='on_off', dp_type=<TuyaDPType.BOOL: 1>, converter=None, dp_converter=None, endpoint_id=None)
2022-05-10 13:30:23 DEBUG (MainThread) [zigpy.zcl] [0xFE77:1:0xef00] ztype: Bool.true
2022-05-10 13:30:23 DEBUG (MainThread) [zigpy.zcl] [0xFE77:1:0xef00] from_value: [1, 1]
2022-05-10 13:30:23 DEBUG (MainThread) [zigpy.zcl] [0xFE77:1:0xef00] raw: b'\x01'
2022-05-10 13:30:23 DEBUG (MainThread) [zigpy.zcl] [0xFE77:1:0xef00] tuya_command: TuyaCommand(status=0, tsn=34, dp=1, data=TuyaData(dp_type=<TuyaDPType.BOOL: 1>, function=0, raw=b'\x01', *payload=<Bool.true: 1>))
2022-05-10 13:30:23 DEBUG (Thread-3) [aiosqlite] operation functools.partial(<built-in method commit of sqlite3.Connection object at 0x7f0daf3d9e40>) completed
2022-05-10 13:30:23 DEBUG (MainThread) [zigpy.zcl] [0xFE77:1:0xef00] Sending request header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.CLUSTER_COMMAND: 1>, is_manufacturer_specific=True, is_reply=0, disable_default_response=0, reserved=0, *is_cluster=True, *is_general=False), manufacturer=4098, tsn=35, command_id=0, *is_reply=False)
2022-05-10 13:30:23 DEBUG (MainThread) [zigpy.zcl] [0xFE77:1:0xef00] Sending request: set_data(data=TuyaCommand(status=0, tsn=34, dp=1, data=TuyaData(dp_type=<TuyaDPType.BOOL: 1>, function=0, raw=b'\x01', *payload=<Bool.true: 1>)))
2022-05-10 13:30:23 DEBUG (MainThread) [zigpy.zcl] [0xFE77:1:0xef00] tuya_mcu_command: cluster_data=TuyaClusterData(endpoint_id=1, cluster_attr='current_level', attr_value=156, expect_reply=True)
2022-05-10 13:30:23 DEBUG (MainThread) [zigpy.zcl] [0xFE77:1:0xef00] get_dp_mapping --> found DP: 2
2022-05-10 13:30:23 DEBUG (MainThread) [zigpy.zcl] [0xFE77:1:0xef00] from_cluster_data: 2, DPToAttributeMapping(ep_attribute='level', attribute_name='current_level', dp_type=<TuyaDPType.VALUE: 2>, converter=<function TuyaLevelControlManufCluster.<lambda> at 0x7f0daf5f4ca0>, dp_converter=<function TuyaLevelControlManufCluster.<lambda> at 0x7f0daf5f4d30>, endpoint_id=None)
2022-05-10 13:30:23 DEBUG (MainThread) [zigpy.zcl] [0xFE77:1:0xef00] converted: 611
2022-05-10 13:30:23 DEBUG (MainThread) [zigpy.zcl] [0xFE77:1:0xef00] ztype: 611
2022-05-10 13:30:23 DEBUG (MainThread) [zigpy.zcl] [0xFE77:1:0xef00] from_value: [4, 0, 0, 2, 99]
2022-05-10 13:30:23 DEBUG (MainThread) [zigpy.zcl] [0xFE77:1:0xef00] raw: b'\x00\x00\x02c'
2022-05-10 13:30:23 DEBUG (MainThread) [zigpy.zcl] [0xFE77:1:0xef00] tuya_command: TuyaCommand(status=0, tsn=36, dp=2, data=TuyaData(dp_type=<TuyaDPType.VALUE: 2>, function=0, raw=b'\x00\x00\x02c', *payload=1661075456))
2022-05-10 13:30:23 DEBUG (MainThread) [zigpy.zcl] [0xFE77:1:0xef00] Sending request header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.CLUSTER_COMMAND: 1>, is_manufacturer_specific=True, is_reply=0, disable_default_response=0, reserved=0, *is_cluster=True, *is_general=False), manufacturer=4098, tsn=37, command_id=0, *is_reply=False)
2022-05-10 13:30:23 DEBUG (bellows.thread_0) [bellows.uart] Data frame: b'200db1f1bb7c897a886ab5cf9fc56fef9c140f7e'
2022-05-10 13:30:23 DEBUG (bellows.thread_0) [bellows.uart] Sending: b'83401b7e'
2022-05-10 13:30:23 DEBUG (MainThread) [zigpy.zcl] [0xFE77:1:0xef00] Sending request: set_data(data=TuyaCommand(status=0, tsn=36, dp=2, data=TuyaData(dp_type=<TuyaDPType.VALUE: 2>, function=0, raw=b'\x00\x00\x02c', *payload=1661075456)))
2022-05-10 13:30:23 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0xFE77:1:0x0008]: executed 'move_to_level_with_on_off' command with args: '(156, 1)' kwargs: '{}' result: Default_Response(command_id=4, status=<Status.SUCCESS: 0>)
2022-05-10 13:30:23 DEBUG (MainThread) [homeassistant.components.zha.entity] light.youssefs_room_level_on_off: turned on: {'move_to_level_with_on_off': Default_Response(command_id=4, status=<Status.SUCCESS: 0>)}
[... cut ...] om level, on_off, supported_features=33 @ 2022-05-10T13:29:08.224809+10:00>>)
2022-05-10 13:30:23 DEBUG (Recorder) [homeassistant.components.recorder.pool.RecorderPool] Connection <sqlite3.Connection object at 0x7f0db4d82e40> checked out from pool
2022-05-10 13:30:23 DEBUG (bellows.thread_0) [bellows.uart] Data frame: b'600db1ed542e14b2b6954b25ab55929670e8c8fd12316a828cc76b8bff7c3fa3ebcdde6e8db4667e'
2022-05-10 13:30:23 DEBUG (bellows.thread_0) [bellows.uart] Sending: b'87009f7e'

```
</details>

yftj, 2022-05-10:
Your work is greatly appreciated and I understand making it work for a broad range of devices is the correct path. Is it possible however to add the individual dim up and dim down entities? This would certainly help.
Again I really appreciate your work!
Cheers.

willrollo, 2022-08-11:
Great work from @javicalle and others contributing to this.

I have the exact same issue with this dimmer. Has anyone worked out a solution?

I have not had any success writing a custom quirk but my experience is pretty limited.

This dimmer is commonly available in Australia.

I'm happy to provide logs etc. Not sure if that's necessary as it's the same behaviour described by @yftj.

Thanks!

willrollo, 2022-08-17:
<details><summary>Device Signature</summary>

```
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0100",
      "in_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0008",
        "0xef00"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZE200_1agwnems",
  "model": "TS0601",
  "class": "zhaquirks.tuya.ts0601_dimmer.TuyaSingleSwitchDimmer"
}

```
</details>

<details><summary>Diagnostic Information</summary>

{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2022.8.5",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.10.5",
    "docker": true,
    "arch": "aarch64",
    "timezone": "Australia/Sydney",
    "os_name": "Linux",
    "os_version": "5.15.32-v8",
    "supervisor": "2022.08.3",
    "host_os": "Home Assistant OS 8.4",
    "docker_version": "20.10.14",
    "chassis": "embedded",
    "run_as_root": true
  },
  "custom_components": {
    "hacs": {
      "version": "1.26.2",
      "requirements": [
        "aiogithubapi>=22.2.4"
      ]
    },
    "garbage_collection": {
      "version": "4.8.3",
      "requirements": [
        "python-dateutil>=2.8.2"
      ]
    },
    "browser_mod": {
      "version": "1.5.3",
      "requirements": []
    },
    "dwains_dashboard": {
[... 491 lines cut ...]
            "endpoint_attribute": "ota",
            "attributes": {},
            "unsupported_attributes": {}
          }
        }
      }
    }
  }
}
</details>

<details><summary>Logs</summary>

Logger: zigpy.zcl
Source: runner.py:119
First occurred: 1:46:03 PM (69 occurrences)
Last logged: 1:49:08 PM

[0x762B:1:0xef00] No 'handle_get_data' tuya handler found for get_data(data=TuyaCommand(status=0, tsn=186, dp=6, data=TuyaData(dp_type=<TuyaDPType.VALUE: 2>, function=0, raw=b'\x00\x00\x00\x00', *payload=0)))
[0x762B:1:0xef00] No 'handle_get_data' tuya handler found for get_data(data=TuyaCommand(status=0, tsn=187, dp=14, data=TuyaData(dp_type=<TuyaDPType.ENUM: 4>, function=0, raw=b'\x00', *payload=<enum8.undefined_0x00: 0>)))
[0x762B:1:0xef00] No 'handle_get_data' tuya handler found for get_data(data=TuyaCommand(status=0, tsn=160, dp=5, data=TuyaData(dp_type=<TuyaDPType.VALUE: 2>, function=0, raw=b'\x18\x00\x00\x00', *payload=24)))
[0x762B:1:0xef00] No 'handle_get_data' tuya handler found for get_data(data=TuyaCommand(status=0, tsn=161, dp=6, data=TuyaData(dp_type=<TuyaDPType.VALUE: 2>, function=0, raw=b'\x00\x00\x00\x00', *payload=0)))
[0x762B:1:0xef00] No 'handle_get_data' tuya handler found for get_data(data=TuyaCommand(status=0, tsn=162, dp=14, data=TuyaData(dp_type=<TuyaDPType.ENUM: 4>, function=0, raw=b'\x00', *payload=<enum8.undefined_0x00: 0>)))

</details>

willrollo, 2022-08-17:
> Great work from @javicalle and others contributing to this.
>
> I have the exact same issue with this dimmer. Has anyone worked out a solution?
>
> I have not had any success writing a custom quirk but my [...]

Here is some more information:

<details><summary>Device Signature</summary>

```
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0100",
      "in_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0008",
        "0xef00"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZE200_1agwnems",
  "model": "TS0601",
  "class": "zhaquirks.tuya.ts0601_dimmer.TuyaSingleSwitchDimmer"
}

```
</details>

<details><summary>Diagnostic Information</summary>

{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2022.8.5",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.10.5",
    "docker": true,
    "arch": "aarch64",
    "timezone": "Australia/Sydney",
    "os_name": "Linux",
    "os_version": "5.15.32-v8",
    "supervisor": "2022.08.3",
    "host_os": "Home Assistant OS 8.4",
    "docker_version": "20.10.14",
    "chassis": "embedded",
    "run_as_root": true
  },
  "custom_components": {
    "hacs": {
      "version": "1.26.2",
      "requirements": [
        "aiogithubapi>=22.2.4"
      ]
    },
    "garbage_collection": {
      "version": "4.8.3",
      "requirements": [
[... 499 lines cut ...]
            "endpoint_attribute": "ota",
            "attributes": {},
            "unsupported_attributes": {}
          }
        }
      }
    }
  }
}
</details>

<details><summary>Logs</summary>

Logger: zigpy.zcl
Source: runner.py:119
First occurred: 1:46:03 PM (69 occurrences)
Last logged: 1:49:08 PM

[0x762B:1:0xef00] No 'handle_get_data' tuya handler found for get_data(data=TuyaCommand(status=0, tsn=186, dp=6, data=TuyaData(dp_type=<TuyaDPType.VALUE: 2>, function=0, raw=b'\x00\x00\x00\x00', *payload=0)))
[0x762B:1:0xef00] No 'handle_get_data' tuya handler found for get_data(data=TuyaCommand(status=0, tsn=187, dp=14, data=TuyaData(dp_type=<TuyaDPType.ENUM: 4>, function=0, raw=b'\x00', *payload=<enum8.undefined_0x00: 0>)))
[0x762B:1:0xef00] No 'handle_get_data' tuya handler found for get_data(data=TuyaCommand(status=0, tsn=160, dp=5, data=TuyaData(dp_type=<TuyaDPType.VALUE: 2>, function=0, raw=b'\x18\x00\x00\x00', *payload=24)))
[0x762B:1:0xef00] No 'handle_get_data' tuya handler found for get_data(data=TuyaCommand(status=0, tsn=161, dp=6, data=TuyaData(dp_type=<TuyaDPType.VALUE: 2>, function=0, raw=b'\x00\x00\x00\x00', *payload=0)))
[0x762B:1:0xef00] No 'handle_get_data' tuya handler found for get_data(data=TuyaCommand(status=0, tsn=162, dp=14, data=TuyaData(dp_type=<TuyaDPType.ENUM: 4>, function=0, raw=b'\x00', *payload=<enum8.undefined_0x00: 0>)))

</details>

willrollo, 2022-08-20:
This is the manufacturer provided device handler for SmartThings. I wonder if this can be translated over to ZHA.

https://drive.google.com/file/d/1eqLByV3P_9wEDrZV9bQvIWGg2MIT5ySB/view

javicalle, 2022-09-12:
In the recent #1748 the behavior for some Tuya dimmers will change.
I believe that this issue can be related.

In case someone wants to test the new version before next release, can enable the local quirks and copy the file from here:
* https://github.com/zigpy/zha-device-handlers/blob/dev/zhaquirks/tuya/mcu/__init__.py

newtsy, 2022-09-27:
Hi @javicalle I've tried to load that quirk file with the prior change but didn't notice any change to the behaviour of these dimmers.  

When checking the device it wasn't showing that a different quirk had been loaded though.

They still flicker off and go full brightness before transitioning to the brightness setting requested.

javicalle, 2022-09-27:
I believe that you will need to copy also the `ts0601_dimmer.py` file and made the `mcu` imports relative to your `custom_quirk_folder`:
* https://github.com/zigpy/zha-device-handlers/blob/dev/zhaquirks/tuya/ts0601_dimmer.py

If you don't mind, you can wait for next week's version of HA or you can also try tomorrow's beta release.

newtsy, 2022-09-28:
Happy to wait for new version.  Thank you for working on these issues.

newtsy, 2022-10-11:
Has this change made it to the stable HA release @javicalle ?  I have browsed the changelog but I haven't seen anything.   Thank you

notabene00, 2022-10-11:
> Has this change made it to the stable HA release @javicalle ? I have browsed the changelog but I haven't seen anything. Thank you

Yep. Works like a charm

javicalle, 2022-10-11:
The dimming action from HA has a bug in some quirks (mainly Tuya `ts0601_dimmer` devices).
The latest HA version (2022.10.X) have the new implementation.

Double check that device is not loading any local quirk or that you have local changes in your installation. In your "Diagnostic Information" you must have the `"zha-quirks==0.0.812"` or `0.0.81`.

If everything is correct, please enable the logs, reproduce the behavior and attach the logs.

yftj, 2022-10-12:
> Double check that device is not loading any local quirk or that you have local changes in your installation. In your "Diagnostic Information" you must have the `"zha-quirks==0.0.812"` or `0.0.81`.
> 
 Thanks again for your help, I've just looked at my diagnostic info and it shows t version to be 
zha-quirks==0.0.82. 
I've also enabled the enhanced brightness slider and temperature transition from off-state options in the Zha configuration.
Physically adjusting the brightness from the upper and lower buttons on the switch works as expected, stepping through the brightness levels. However, the same behaviour I explained at the beginning of the post is occurring when adjusting through HomeAssistant

<details>
<summary>Logs with reproduced behaviour</summary>
 
```
2022-10-12 16:32:20.900 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x2D5B:1:0x0006]: executed 'on' command with args: '()' kwargs: '{}' result: Default_Response(command_id=1, status=<Status.SUCCESS: 0>)
2022-10-12 16:32:20.900 DEBUG (MainThread) [homeassistant.components.zha.entity] light.youssefs_room_level_on_off: starting transitioning timer for 1.25
2022-10-12 16:32:20.900 DEBUG (MainThread) [homeassistant.components.zha.entity] light.youssefs_room_level_on_off: turned on: {'on_off': Default_Response(command_id=1, status=<Status.SUCCESS: 0>)}
2022-10-12 16:32:20.929 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x2D5B:1:0x0008]: received attribute: 61186 update with value: enum8.undefined_0x00
2022-10-12 16:32:21.362 DEBUG (MainThread) [homeassistant.components.zha.entity] light.youssefs_room_level_on_off: received onoff Bool.true while transitioning - skipping update
2022-10-12 16:32:21.837 DEBUG (MainThread) [homeassistant.components.zha.core.device] [0x566A](TS0012): Device seen - marking the device available and resetting counter
2022-10-12 16:32:21.838 DEBUG (MainThread) [homeassistant.components.zha.core.device] [0x566A](TS0012): Update device availability - device available: True - new availability: True - changed: False
2022-10-12 16:32:21.956 DEBUG (MainThread) [homeassistant.components.zha.core.device] [0x812F](TS004F): Attempting to checkin with device - missed checkins: 2
2022-10-12 16:32:22.153 DEBUG (MainThread) [homeassistant.components.zha.entity] light.youssefs_room_level_on_off: transition complete - future attribute reports will write HA state
2022-10-12 16:32:22.618 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x2D5B:1:0x0008]: received attribute: 0 update with value: 223
2022-10-12 16:32:22.722 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x2D5B:1:0x0008]: received attribute: 61185 update with value: 0
2022-10-12 16:32:22.826 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x2D5B:1:0x0008]: received attribute: 61186 update with value: enum8.undefined_0x00
2022-10-12 16:32:23.199 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x5FEE:1:0x0008]: received attribute: 0 update with value: 74
2022-10-12 16:32:23.302 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x5FEE:1:0x0008]: received attribute: 61185 update with value: 0
2022-10-12 16:32:23.317 DEBUG (MainThread) [homeassistant.components.zha.entity] light.youssefs_room_level_on_off: setting transitioning flag to True
2022-10-12 16:32:23.331 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x2D5B:1:0x0008]: executed 'move_to_level_with_on_off' command with args: '()' kwargs: '{'level': 97, 'transition_time': 0}' result: Default_Response(command_id=4, status=<Status.SUCCESS: 0>)
2022-10-12 16:32:23.332 DEBUG (MainThread) [homeassistant.components.zha.entity] light.youssefs_room_level_on_off: starting transitioning timer for 0.25
2022-10-12 16:32:23.332 DEBUG (MainThread) [homeassistant.components.zha.entity] light.youssefs_room_level_on_off: turned on: {'move_to_level_with_on_off': Default_Response(command_id=4, status=<Status.SUCCESS: 0>)}
2022-10-12 16:32:23.459 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x5FEE:1:0x0008]: received attribute: 61186 update with value: enum8.undefined_0x00
2022-10-12 16:32:23.583 DEBUG (MainThread) [homeassistant.components.zha.entity] light.youssefs_room_level_on_off: transition complete - future attribute reports will write HA state
2022-10-12 16:32:23.725 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x2D5B:1:0x0008]: received attribute: 0 update with value: 96
2022-10-12 16:32:25.660 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x2D5B:1:0x0008]: received attribute: 0 update with value: 95
2022-10-12 16:32:25.787 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x2D5B:1:0x0008]: received attribute: 61185 update with value: 0
2022-10-12 16:32:25.895 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x2D5B:1:0x0008]: received attribute: 61186 update with value: enum8.undefined_0x00
2022-10-12 16:32:25.925 DEBUG (MainThread) [homeassistant.components.zha.core.device] [0xAF36](lumi.sensor_motion.aq2): Device seen - marking the device available and resetting counter
2022-10-12 16:32:25.925 DEBUG (MainThread) [homeassistant.components.zha.core.device] [0xAF36](lumi.sensor_motion.aq2): Update device availability - device available: True - new availability: True - changed: False
[... cut ...]
2022-10-12 16:32:40.832 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x2D5B:1:0x0008]: received attribute: 61186 update with value: enum8.undefined_0x00
2022-10-12 16:32:41.079 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x5FEE:1:0x0008]: received attribute: 0 update with value: 74
2022-10-12 16:32:41.195 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x5FEE:1:0x0008]: received attribute: 61185 update with value: 0
2022-10-12 16:32:41.309 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x5FEE:1:0x0008]: received attribute: 61186 update with value: enum8.undefined_0x00
```

javicalle, 2022-10-22:
Can you enable also the `zigpy` and the `zhaquirks` debug logs?
There is something extrange with the 'transitioning timer'.

yftj, 2022-10-23:
> Can you enable also the `zigpy` and the `zhaquirks` debug logs? There is something extrange with the 'transitioning timer'.

sure thing, thanks for looking into this again!

<details>
<summary>Logs , ctrl f "Youssef" to see the entity im using</summary>

```
2022-10-23 13:57:33.686 DEBUG (MainThread) [zigpy.zcl] [0x45B5:1:0xef00] Decoded ZCL frame: TuyaLevelControlManufCluster:get_data(data=TuyaCommand(status=9, tsn=5, datapoints=[TuyaDatapointData(dp=4, data=TuyaData(dp_type=<TuyaDPType.ENUM: 4>, function=0, raw=b'\x00', *payload=<enum8.undefined_0x00: 0>))]))
2022-10-23 13:57:33.686 DEBUG (MainThread) [zigpy.zcl] [0x45B5:1:0xef00] Received command 0x01 (TSN 96): get_data(data=TuyaCommand(status=9, tsn=5, datapoints=[TuyaDatapointData(dp=4, data=TuyaData(dp_type=<TuyaDPType.ENUM: 4>, function=0, raw=b'\x00', *payload=<enum8.undefined_0x00: 0>))]))
2022-10-23 13:57:33.686 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x45B5:1:0x0008]: received attribute: 61186 update with value: enum8.undefined_0x00
2022-10-23 13:57:33.686 DEBUG (MainThread) [zigpy.zcl] [0x45B5:1:0xef00] Sending reply header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=False, direction=<Direction.Client_to_Server: 1>, disable_default_response=1, reserved=0, *is_cluster=False, *is_general=True, *is_reply=True), tsn=96, command_id=<GeneralCommand.Default_Response: 11>, *direction=<Direction.Client_to_Server: 1>, *is_reply=True)
2022-10-23 13:57:33.686 DEBUG (MainThread) [zigpy.zcl] [0x45B5:1:0xef00] Sending reply: Default_Response(command_id=1, status=<Status.SUCCESS: 0>)
2022-10-23 13:57:33.818 DEBUG (MainThread) [homeassistant.components.zha.entity] light.youssefs_room_level_on_off: setting transitioning flag to True
2022-10-23 13:57:33.818 DEBUG (MainThread) [zigpy.zcl] [0x6F8F:1:0x0006] Sending Tuya Cluster Command... Cluster Command is 0, Arguments are ()
2022-10-23 13:57:33.818 DEBUG (MainThread) [zigpy.zcl] [0x6F8F:1:0xef00] tuya_mcu_command: cluster_data=TuyaClusterData(endpoint_id=1, cluster_attr='on_off', attr_value=0, expect_reply=True)
2022-10-23 13:57:33.818 DEBUG (MainThread) [zigpy.zcl] [0x6F8F:1:0xef00] get_dp_mapping --> found DP: 1
2022-10-23 13:57:33.818 DEBUG (MainThread) [zigpy.zcl] [0x6F8F:1:0xef00] from_cluster_data: 1, DPToAttributeMapping(ep_attribute='on_off', attribute_name='on_off', dp_type=<TuyaDPType.BOOL: 1>, converter=None, dp_converter=None, endpoint_id=None)
2022-10-23 13:57:33.818 DEBUG (MainThread) [zigpy.zcl] [0x6F8F:1:0xef00] ztype: Bool.false
2022-10-23 13:57:33.818 DEBUG (MainThread) [zigpy.zcl] [0x6F8F:1:0xef00] from_value: [1, 0]
2022-10-23 13:57:33.818 DEBUG (MainThread) [zigpy.zcl] [0x6F8F:1:0xef00] raw: b'\x00'
2022-10-23 13:57:33.819 DEBUG (MainThread) [zigpy.zcl] [0x6F8F:1:0xef00] tuya_command: TuyaCommand(status=0, tsn=45, datapoints=[TuyaDatapointData(dp=1, data=TuyaData(dp_type=<TuyaDPType.BOOL: 1>, function=0, raw=b'\x00', *payload=<Bool.false: 0>))])
2022-10-23 13:57:33.819 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x6F8F:1:0x0006]: executed 'off' command with args: '()' kwargs: '{}' result: Default_Response(command_id=0, status=<Status.SUCCESS: 0>)
2022-10-23 13:57:33.819 DEBUG (MainThread) [homeassistant.components.zha.entity] light.youssefs_room_level_on_off: starting transitioning timer for 1.25
2022-10-23 13:57:33.819 DEBUG (MainThread) [homeassistant.components.zha.entity] light.youssefs_room_level_on_off: turned off: Default_Response(command_id=0, status=<Status.SUCCESS: 0>)
2022-10-23 13:57:33.819 DEBUG (MainThread) [zigpy.zcl] [0x6F8F:1:0xef00] Sending request header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.CLUSTER_COMMAND: 1>, is_manufacturer_specific=True, direction=<Direction.Server_to_Client: 0>, disable_default_response=0, reserved=0, *is_cluster=True, *is_general=False, *is_reply=False), manufacturer=4098, tsn=46, command_id=0, *direction=<Direction.Server_to_Client: 0>, *is_reply=False)
2022-10-23 13:57:33.820 DEBUG (MainThread) [zigpy.zcl] [0x6F8F:1:0xef00] Sending request: set_data(data=TuyaCommand(status=0, tsn=45, datapoints=[TuyaDatapointData(dp=1, data=TuyaData(dp_type=<TuyaDPType.BOOL: 1>, function=0, raw=b'\x00', *payload=<Bool.false: 0>))]))
2022-10-23 13:57:33.828 DEBUG (MainThread) [zigpy.application] Received a packet: ZigbeePacket(src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x45B5), src_ep=1, dst=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), dst_ep=1, source_route=None, extended_timeout=False, tsn=152, profile_id=260, cluster_id=61184, data=Serialized[b'\ta\x01\t\x06\x05\x02\x00\x04\x00\x00\x00\x18'], tx_options=<TransmitOptions.NONE: 0>, radius=0, non_member_radius=0, lqi=235, rssi=-68)
2022-10-23 13:57:33.829 DEBUG (MainThread) [zigpy.zcl] [0x45B5:1:0xef00] Received ZCL frame: b'\ta\x01\t\x06\x05\x02\x00\x04\x00\x00\x00\x18'
2022-10-23 13:57:33.829 DEBUG (MainThread) [zigpy.zcl] [0x45B5:1:0xef00] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.CLUSTER_COMMAND: 1>, is_manufacturer_specific=0, direction=<Direction.Client_to_Server: 1>, disable_default_response=0, reserved=0, *is_cluster=True, *is_general=False, *is_reply=True), tsn=97, command_id=1, *direction=<Direction.Client_to_Server: 1>, *is_reply=True)
[... cut ...] , *is_general=True, *is_reply=True), tsn=100, command_id=<GeneralCommand.Default_Response: 11>, *direction=<Direction.Client_to_Server: 1>, *is_reply=True)
2022-10-23 13:57:42.173 DEBUG (MainThread) [zigpy.zcl] [0x45B5:1:0xef00] Sending reply: Default_Response(command_id=1, status=<Status.SUCCESS: 0>)
2022-10-23 13:57:42.184 DEBUG (MainThread) [homeassistant.components.zha.entity] light.youssefs_room_level_on_off: transition complete - future attribute reports will write HA state

```
</details>

newtsy, 2022-12-02:
Hi @javicalle just wondering if this was still on your radar and whether there was anything I can do to help?

javicalle, 2022-12-02:
Well yes, I had it off my radar, but let's see if it's possible to move something forward.

The truth is that it is a difficult problem to analyze remotely.
The logs don't seem to indicate anything strange, but probably the timing is important in this case.

This device also implements a power on/off transition time that generates additional traces (and may be the reason for #1854).

I'll try to check in the weekend.

javicalle, 2022-12-03:
I have found a bug in the payload part. Not sure if related or why there aren't many other faulty devices but there are something wrong.

Someone in the rrom that could be able to quirk the `__init__.py` ***and*** the `ts0601_dimmer.py`?

newtsy, 2022-12-05:
I have tried to quirk this @javicalle but after looking at your commits above I might have used the wrong _init_.py .  I haven't done this before, do I need to preserve the folder structure for the files that I'm quirking?

javicalle, 2022-12-08:
The fix is already in the latest HA version.
Can someone confirm if the issue persist? If is the case, please attach the logs from the device erratic operation.

dobromir-hristov, 2022-12-08:
Just outdated to latest 12.0 version and it's pretty bad. Dimmer works on/off most of the time but dragging the slider to adjust brightness is completely random.

newtsy, 2022-12-08:
Apologies @javicalle I had tried to quirk this earlier but I'm not sure if I was using the correct init.py.  Also do I need to preserve folder structure in the zhaquirks folder?

newtsy, 2022-12-08:
I can agree that the latest change appears to have broken the dimmer function to some extent and the result of adjusting the slider now gives a somewhat random result.

javicalle, 2022-12-08:
Can someone attach the logs of dimming the device?

dobromir-hristov, 2022-12-09:
I will upload some in a few minutes

yftj, 2022-12-09:
I can confirm that the dimmer function is broken and outputs random values or just maxes at 100. here are the logs:

<details>
<summary>Logs with reproduced behaviour</summary>
 
```
2022-12-09 18:30:58.019 DEBUG (MainThread) [zigpy.zcl] [0x830B:1:0xef00] Received command 0x01 (TSN 34): get_data(data=TuyaCommand(status=59, tsn=39, datapoints=[TuyaDatapointData(dp=6, data=TuyaData(dp_type=<TuyaDPType.VALUE: 2>, function=0, raw=b'\x00\x00\x00\x00', *payload=0))]))
2022-12-09 18:30:58.019 DEBUG (MainThread) [zigpy.zcl] [0x830B:1:0xef00] No datapoint handler for TuyaDatapointData(dp=6, data=TuyaData(dp_type=<TuyaDPType.VALUE: 2>, function=0, raw=b'\x00\x00\x00\x00', *payload=0))
2022-12-09 18:30:58.020 DEBUG (MainThread) [zigpy.zcl] [0x830B:1:0xef00] Sending reply header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=False, direction=<Direction.Client_to_Server: 1>, disable_default_response=1, reserved=0, *is_cluster=False, *is_general=True, *is_reply=True), tsn=34, command_id=<GeneralCommand.Default_Response: 11>, *direction=<Direction.Client_to_Server: 1>, *is_reply=True)
2022-12-09 18:30:58.021 DEBUG (MainThread) [zigpy.zcl] [0x830B:1:0xef00] Sending reply: Default_Response(command_id=1, status=<Status.UNSUPPORTED_ATTRIBUTE: 134>)
2022-12-09 18:30:58.115 DEBUG (MainThread) [homeassistant.components.zha.entity] light.youssefs_room_light_2: setting transitioning flag to True
2022-12-09 18:30:58.115 DEBUG (MainThread) [zigpy.zcl] [0x30FC:1:0x0008] Sending Tuya Cluster Command. Cluster Command is 4, Arguments are (), {'level': 196, 'transition_time': 0}
2022-12-09 18:30:58.116 DEBUG (MainThread) [zigpy.zcl] [0x30FC:1:0xef00] tuya_mcu_command: cluster_data=TuyaClusterData(endpoint_id=1, cluster_attr='on_off', attr_value=True, expect_reply=True)
2022-12-09 18:30:58.116 DEBUG (MainThread) [zigpy.zcl] [0x30FC:1:0xef00] get_dp_mapping --> found DP: 1
2022-12-09 18:30:58.116 DEBUG (MainThread) [zigpy.zcl] [0x30FC:1:0xef00] from_cluster_data: {1: DPToAttributeMapping(ep_attribute='on_off', attribute_name='on_off', dp_type=<TuyaDPType.BOOL: 1>, converter=None, dp_converter=None, endpoint_id=None)}
2022-12-09 18:30:58.116 DEBUG (MainThread) [zigpy.zcl] [0x30FC:1:0xef00] ztype: Bool.true
2022-12-09 18:30:58.116 DEBUG (MainThread) [zigpy.zcl] [0x30FC:1:0xef00] from_value: [1, 1]
2022-12-09 18:30:58.116 DEBUG (MainThread) [zigpy.zcl] [0x30FC:1:0xef00] raw: b'\x01'
2022-12-09 18:30:58.116 DEBUG (MainThread) [zigpy.zcl] [0x30FC:1:0xef00] tuya_commands: [TuyaCommand(status=0, tsn=68, datapoints=[TuyaDatapointData(dp=1, data=TuyaData(dp_type=<TuyaDPType.BOOL: 1>, function=0, raw=b'\x01', *payload=<Bool.true: 1>))])]
2022-12-09 18:30:58.117 DEBUG (MainThread) [zigpy.zcl] [0x30FC:1:0xef00] tuya_mcu_command: cluster_data=TuyaClusterData(endpoint_id=1, cluster_attr='current_level', attr_value=196, expect_reply=True)
2022-12-09 18:30:58.117 DEBUG (MainThread) [zigpy.zcl] [0x30FC:1:0xef00] get_dp_mapping --> found DP: 2
2022-12-09 18:30:58.117 DEBUG (MainThread) [zigpy.zcl] [0x30FC:1:0xef00] from_cluster_data: {2: DPToAttributeMapping(ep_attribute='level', attribute_name='current_level', dp_type=<TuyaDPType.VALUE: 2>, converter=<function TuyaLevelControlManufCluster.<lambda> at 0x7fea4fb44a60>, dp_converter=<function TuyaLevelControlManufCluster.<lambda> at 0x7fea4fb44af0>, endpoint_id=None)}
2022-12-09 18:30:58.117 DEBUG (MainThread) [zigpy.zcl] [0x30FC:1:0xef00] converted: 768
2022-12-09 18:30:58.117 DEBUG (MainThread) [zigpy.zcl] [0x30FC:1:0xef00] ztype: 768
2022-12-09 18:30:58.117 DEBUG (MainThread) [zigpy.zcl] [0x30FC:1:0xef00] from_value: [4, 0, 0, 3, 0]
2022-12-09 18:30:58.117 DEBUG (MainThread) [zigpy.zcl] [0x30FC:1:0xef00] raw: b'\x00\x03\x00\x00'
2022-12-09 18:30:58.117 DEBUG (MainThread) [zigpy.zcl] [0x30FC:1:0xef00] tuya_commands: [TuyaCommand(status=0, tsn=69, datapoints=[TuyaDatapointData(dp=2, data=TuyaData(dp_type=<TuyaDPType.VALUE: 2>, function=0, raw=b'\x00\x03\x00\x00', *payload=768))])]
2022-12-09 18:30:58.118 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x30FC:1:0x0008]: executed 'move_to_level_with_on_off' command with args: '()' kwargs: '{'level': 196, 'transition_time': 0}' result: Default_Response(command_id=4, status=<Status.SUCCESS: 0>)
2022-12-09 18:30:58.118 DEBUG (MainThread) [homeassistant.components.zha.entity] light.youssefs_room_light_2: starting transitioning timer for 0.25
2022-12-09 18:30:58.118 DEBUG (MainThread) [homeassistant.components.zha.entity] light.youssefs_room_light_2: turned on: {'move_to_level_with_on_off': Default_Response(command_id=4, status=<Status.SUCCESS: 0>)}
2022-12-09 18:30:58.120 DEBUG (MainThread) [zigpy.zcl] [0x30FC:1:0xef00] Sending request header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.CLUSTER_COMMAND: 1>, is_manufacturer_specific=True, direction=<Direction.Server_to_Client: 0>, disable_default_response=0, reserved=0, *is_cluster=True, *is_general=False, *is_reply=False), manufacturer=4098, tsn=70, command_id=0, *direction=<Direction.Server_to_Client: 0>, *is_reply=False)
2022-12-09 18:30:58.120 DEBUG (MainThread) [zigpy.zcl] [0x30FC:1:0xef00] Sending request: set_data(data=TuyaCommand(status=0, tsn=68, datapoints=[TuyaDatapointData(dp=1, data=TuyaData(dp_type=<TuyaDPType.BOOL: 1>, function=0, raw=b'\x01', *payload=<Bool.true: 1>))]))
[... cut ...] Client_to_Server: 1>, disable_default_response=0, reserved=0, *is_cluster=True, *is_general=False, *is_reply=True), tsn=36, command_id=1, *direction=<Direction.Client_to_Server: 1>, *is_reply=True)
2022-12-09 18:30:59.841 DEBUG (MainThread) [zigpy.zcl] [0x30FC:1:0xef00] Decoded ZCL frame: TuyaLevelControlManufCluster:get_data(data=TuyaCommand(status=0, tsn=86, datapoints=[TuyaDatapointData(dp=1, data=TuyaData(dp_type=<TuyaDPType.BOOL: 1>, function=0, raw=b'\x01', *payload=<Bool.true: 1>))]))
```

dobromir-hristov, 2022-12-09:
Here is a short video of how it behaves - 

[image]

javicalle, 2022-12-11:
Ok, it's clear to me that there is a bug with the `TuyaDPType.VALUE` values.

I have another version and need someone to test it.
You will need to copy 3 files in your local quirk folder and perform some modifications.

1. copy in your local quirk folder the current `tuya/__init__.py` file:
* https://github.com/zigpy/zha-device-handlers/blob/dev/zhaquirks/tuya/__init__.py

2. modify the `TuyaData` class like this:
```python
class TuyaData(t.Struct):
    """Tuya Data type."""
    dp_type: TuyaDPType
    function: t.uint8_t
    raw: t.LVBytes
    @classmethod
    def deserialize(cls, data: bytes) -> Tuple["TuyaData", bytes]:
        """Deserialize data."""
        _LOGGER.debug("data: %s", data)
        res = cls()
        res.dp_type, data = TuyaDPType.deserialize(data)
        res.function, data = t.uint8_t.deserialize(data)
        res.raw, data = t.LVBytes.deserialize(data)
        # if res.dp_type not in (TuyaDPType.BITMAP, TuyaDPType.STRING, TuyaDPType.ENUM):
        #     res.raw = res.raw[::-1]
        return res, data

    @property
    def payload(self) -> Union[t.Bool, t.CharacterString, t.uint32_t, t.data32]:
        """Payload accordingly to data point type."""
        if self.dp_type == TuyaDPType.VALUE:
            return t.uint32_t.deserialize(self.raw[::-1])[0]  # waiting to https://github.com/zigpy/zigpy/pull/1124
        elif self.dp_type == TuyaDPType.BOOL:
            return t.Bool.deserialize(self.raw)[0]
        elif self.dp_type == TuyaDPType.STRING:
            return self.raw.decode("utf8")
        elif self.dp_type == TuyaDPType.ENUM:
            return t.enum8.deserialize(self.raw)[0]
        elif self.dp_type == TuyaDPType.BITMAP:
            bitmaps = {1: t.bitmap8, 2: t.bitmap16, 4: t.bitmap32}
            try:
                return bitmaps[len(self.raw)].deserialize(self.raw)[0]
            except KeyError as exc:
                raise ValueError(f"Wrong bitmap length: {len(self.raw)}") from exc
        raise ValueError(f"Unknown {self.dp_type} datapoint type")
```

3. copy the current `tuya/mcu/__init__.py` file inside an `mcu` folder (would stay as `custom_zha_quirks/mcu/__init__.py`):
* https://github.com/zigpy/zha-device-handlers/blob/dev/zhaquirks/tuya/mcu/__init__.py

4. modify all the `zhaquirks.tuya` imports like this:
```python
from custom_zha_quirks import (
    TUYA_MCU_COMMAND,
    TUYA_MCU_VERSION_RSP,
    TUYA_SET_DATA,
    TUYA_SET_TIME,
    Data,
    NoManufacturerCluster,
    PowerOnState,
    TuyaCommand,
    TuyaData,
    TuyaDatapointData,
    TuyaLocalCluster,
    TuyaNewManufCluster,
    TuyaTimePayload,
)
```

5. copy in your local quirk folder the current `tuya/ts0601_dimmer.py` file:
* https://github.com/zigpy/zha-device-handlers/blob/dev/zhaquirks/tuya/ts0601_dimmer.py

6. modify all the `zhaquirks.tuya` imports and `zhaquirks.tuya.mcu` imports like this:
```python
from custom_zha_quirks import NoManufacturerCluster, TuyaDimmerSwitch
from custom_zha_quirks.mcu import (
    TuyaInWallLevelControl,
    TuyaLevelControlManufCluster,
    TuyaOnOff,
    TuyaOnOffNM,
)
```

Save all the changes, delete any `__pycache__` folder, remove the device from HA, restart HA and pair the device again.
Enable de debug logs and attach the dimming logs from the physical device and from HA

javicalle, 2022-12-11:
@newtsy I really appreciate your offers. Any help will be welcomed but I really need of someone that could progress with the errors and contribute with ideas, tests and information.

yftj, 2022-12-12:
> Ok, it's clear to me that there is a bug with the `TuyaDPType.VALUE` values.
> 
> I have another version and need someone to test it. You will need to copy 3 files in your local quirk folder and perform some modifications.
> 
> 1. copy in your local quirk folder the current `tuya/__init__.py` file:

I'm trying to follow this guide to help out but I have a question, first of all here is what my directory looks like: 
[image]

Where is my custom zha quirk folder located at? and did you want to completely replace my __init__.py file with the new one on git? also just making sure my localtuya addon has nothing to do with this also right?

javicalle, 2022-12-12:
There's a guide for configure the local quirk in your instalation:
https://github.com/zigpy/zha-device-handlers/discussions/693#discussioncomment-857274

Shulyaka, 2022-12-12:
@javicalle were you able to reproduce the issue with a test?

javicalle, 2022-12-12:
> @javicalle were you able to reproduce the issue with a test?

No, I can't imagine how can it be reproduced with a test.
I have some logs and a lot of suppositions.
Also I'm afraid to fix one thing and break two news. 😨

Shulyaka, 2022-12-12:
That's exactly why we need to reproduce. `_TZE200_1agwnems` from this issue and `_TZE200_la2c2uo9` from #1996 use the same quirk `zhaquirks.tuya.ts0601_dimmer.TuyaSingleSwitchDimmer`, yet they behave differently. We have a test for `zhaquirks.tuya.ts0601_dimmer.TuyaDoubleSwitchDimmer` which uses same `TuyaLevelControlManufCluster` and `TuyaInWallLevelControl` clusters, so we shouldn't have had this issue.

yftj, 2022-12-14:
> @yftj may I ask you to try to change the `move_to_level_with_on_off` to `move_to_level` in the following line: https://github.com/home-assistant/core/blob/dev/homeassistant/components/zha/light.py#L272 Then restart HA and repeat your test?

I'm still trying to figure out how to get my custom quirks folder to appear in my config, I added 

```
zha:
  custom_quirks_path: /config/custom_zha_quirks/
```
To my config.yaml file and created that folder but apparently I need to input a quirk file in it and restart to initalise? I copied over my localtuya custom compnent init.py file into the new folder in the hopes of getting initialised but I'm not able to.

[image]

Once I figure this out ill do all I can to help out!

MattWestb, 2022-12-14:
ZHA_Custom_Quirks shall being relative to your HA configure path = where HA is having its config file.

From that do one folder with the name `custom_zha_quirks` and putting your local quirk in it and restart HA.

If you is editing the **in** the HA container (that you looks doing) its `/config/` that is mapped to `/usr/share/hassio/homeassistant/` in the OS you is running.

yftj, 2022-12-14:
> ZHA_Custom_Quirks shall being relative to your HA configure path = where HA is having its config file.
>
> From that do one folder with the name `custom_zha_quirks` and putting your local quirk in it [...]

This is my current directory: 
[image]
[image]
[image]

I've restarted, how do I know it worked? Do I place files in the custom zha quirk folder and would they be read by zha? Does the folder populate with quirks? Should I just delete the init.py file I placed and add the file mentioned here?

> @yftj may I ask you to try to change the `move_to_level_with_on_off` to `move_to_level` in the following line: https://github.com/home-assistant/core/blob/dev/homeassistant/components/zha/light.py#L272 Then restart HA and repeat your test?

MattWestb, 2022-12-14:
If you have configure debug logging you can see it in HA log like this after ZHA have loaded the standard quirks:
```
2022-12-08 09:33:42.556 DEBUG (MainThread) [zhaquirks] Loading custom quirks from /config/custom_zha_quirks
2022-12-08 09:33:42.597 DEBUG (MainThread) [zhaquirks] Loading custom quirks module ts011f_plug
````
You can also see if the system is loading one quirk or is its cant matching it for your device.

Your question of the INI file its up to the dev that have making the local quirk but normally all init things is loaded from the system integrated so you need reading what you shall installing.

yftj, 2022-12-15:
> Could the issue be that ZHA invokes cluster command 0x04 ("move_to_level_with_on_off") instead of 0x00 ("move_to_level")?
> 
> @yftj may I ask you to try to change the `move_to_level_with_on_off` to `move_to_level` in the following line: https://github.com/home-assistant/core/blob/dev/homeassistant/components/zha/light.py#L272 Then restart HA and repeat your test?

I don't know whether it was the update or the custom quirk that fixed the random values that was previously broken but brightness works correctly apart from original issue in this thread. Here are the updated logs, 

<Logs>

```
2022-12-15 18:59:40.246 DEBUG (MainThread) [zhaquirks] Loading quirks module zhaquirks.zhongxing
2022-12-15 18:59:40.248 DEBUG (MainThread) [zhaquirks] Loading quirks module zhaquirks.zhongxing.motion
2022-12-15 18:59:40.248 DEBUG (MainThread) [zhaquirks] Loading custom quirks from /config/custom_zha_quirks
2022-12-15 18:59:40.249 DEBUG (MainThread) [homeassistant.components.zha] ZHA storage file does not exist or was already removed

2022-12-15 18:59:41.499 DEBUG (MainThread) [zigpy.appdb] Current database version is v11
2022-12-15 18:59:47.924 DEBUG (MainThread) [zigpy.appdb] Loading application state
<TuyaDPType.VALUE: 2>, function=0, raw=b'\x00\x00\x01\xb7', *payload=3070296064))])]
2022-12-15 19:02:30.687 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x30FC:1:0x0008]: executed 'move_to_level_with_on_off' command with args: '()' kwargs: '{'level': 112, 'transition_time': 0}' result: Default_Response(command_id=4, status=<Status.SUCCESS: 0>)
2022-12-15 19:02:30.687 DEBUG (MainThread) [homeassistant.components.zha.entity] light.youssefs_room_light_2: turned on: {'move_to_level_with_on_off': Default_Response(command_id=4, status=<Status.SUCCESS: 0>)}
2022-12-15 19:02:30.688 DEBUG (MainThread) [zigpy.zcl] [0x30FC:1:0xef00] Sending request header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.CLUSTER_COMMAND: 1>, is_manufacturer_specific=True, direction=<Direction.Server_to_Client: 0>, disable_default_response=0, reserved=0, *is_cluster=True, *is_general=False, *is_reply=False), manufacturer=4098, tsn=145, command_id=0, *direction=<Direction.Server_to_Client: 0>, *is_reply=False)
2022-12-15 19:02:30.688 DEBUG (MainThread) [zigpy.zcl] [0x30FC:1:0xef00] Sending request: set_data(data=TuyaCommand(status=0, tsn=143, datapoints=[TuyaDatapointData(dp=1, data=TuyaData(dp_type=<TuyaDPType.BOOL: 1>, function=0, raw=b'\x01', *payload=<Bool.true: 1>))]))
2022-12-15 19:02:30.690 DEBUG (MainThread) [zigpy.zcl] [0x30FC:1:0xef00] Sending request header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.CLUSTER_COMMAND: 1>, is_manufacturer_specific=True, direction=<Direction.Server_to_Client: 0>, disable_default_response=0, reserved=0, *is_cluster=True, *is_general=False, *is_reply=False), manufacturer=4098, tsn=147, command_id=0, *direction=<Direction.Server_to_Client: 0>, *is_reply=False)
2022-12-15 19:02:30.690 DEBUG (MainThread) [zigpy.zcl] [0x30FC:1:0xef00] Sending request: set_data(data=TuyaCommand(status=0, tsn=144, datapoints=[TuyaDatapointData(dp=2, data=TuyaData(dp_type=<TuyaDPType.VALUE: 2>, function=0, raw=b'\x00\x00\x01\xb7', *payload=3070296064))]))
2022-12-15 19:02:30.739 DEBUG (MainThread) [zigpy.application] Received a packet: ZigbeePacket(src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x446A), src_ep=1, dst=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), dst_ep=1, source_route=None, extended_timeout=False, tsn=159, profile_id=260, cluster_id=61184, data=Serialized[b'\tL\x01\x03R\x03\x02\x00\x04\x00\x00\x00\x01'], tx_options=<TransmitOptions.NONE: 0>, radius=0, non_member_radius=0, lqi=176, rssi=-56)
2022-12-15 19:02:30.739 DEBUG (MainThread) [zigpy.zcl] [0x446A:1:0xef00] Received ZCL frame: b'\tL\x01\x03R\x03\x02\x00\x04\x00\x00\x00\x01'
2022-12-15 19:02:30.740 DEBUG (MainThread) [zigpy.zcl] [0x446A:1:0xef00] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.CLUSTER_COMMAND: 1>, is_manufacturer_specific=0, direction=<Direction.Client_to_Server: 1>, disable_default_response=0, reserved=0, *is_cluster=True, *is_general=False, *is_reply=True), tsn=76, command_id=1, *direction=<Direction.Client_to_Server: 1>, *is_reply=True)
2022-12-15 19:02:30.740 DEBUG (MainThread) [zigpy.zcl] [0x446A:1:0xef00] Decoded ZCL frame: TuyaLevelControlManufCluster:get_data(data=TuyaCommand(status=3, tsn=82, datapoints=[TuyaDatapointData(dp=3, data=TuyaData(dp_type=<TuyaDPType.VALUE: 2>, function=0, raw=b'\x01\x00\x00\x00', *payload=1))]))
2022-12-15 19:02:30.741 DEBUG (MainThread) [zigpy.zcl] [0x446A:1:0xef00] Received command 0x01 (TSN 76): get_data(data=TuyaCommand(status=3, tsn=82, datapoints=[TuyaDatapointData(dp=3, data=TuyaData(dp_type=<TuyaDPType.VALUE: 2>, function=0, raw=b'\x01\x00\x00\x00', *payload=1))]))
2022-12-15 19:02:30.741 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x446A:1:0x0008]: received attribute: 61185 update with value: 0
2022-12-15 19:02:30.742 DEBUG (MainThread) [zigpy.zcl] [0x446A:1:0xef00] Sending reply header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=False, direction=<Direction.Client_to_Server: 1>, disable_default_response=1, reserved=0, *is_cluster=False, *is_general=True, *is_reply=True), tsn=76, command_id=<GeneralCommand.Default_Response: 11>, *direction=<Direction.Client_to_Server: 1>, *is_reply=True)
[... cut ...]
2022-12-15 19:02:38.291 DEBUG (MainThread) [zigpy.application] Received a packet: ZigbeePacket(src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x30FC), src_ep=1, dst=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), dst_ep=1, source_route=None, extended_timeout=False, tsn=216, profile_id=260, cluster_id=61184, data=Serialized[b'\x0c\x02\x10\xcd\x0b\x00\x00'], tx_options=<TransmitOptions.NONE: 0>, radius=0, non_member_radius=0, lqi=192, rssi=-52)
2022-12-15 19:02:38.291 DEBUG (MainThread) [zigpy.zcl] [0x30FC:1:0xef00] Received ZCL frame: b'\x0c\x02\x10\xcd\x0b\x00\x00'

```

Shulyaka, 2022-12-15:
I can still see `move_to_level_with_on_off` in the logs, so it's definitely not this change:)

Do I understand it correctly that the brightness level is correct now, but it still has weird transition (flashes, then dims)?

yftj, 2022-12-15:
> I can still see `move_to_level_with_on_off` in the logs, so it's definitely not this change:)
> 
> Do I understand it correctly that the brightness level is correct now, but it still has a weird transition (flashes, then dims)?
 
Weird I changed the line in the custom zha quriks and it was loaded. 
[image]
and yes you are correct in that understanding.

yftj, 2022-12-15:
> No, that is the zha file, if you copy it to the custom quirk and change it there, it won't work... Ok, new plan! In your `custom_zha_quirks/mcu/__init__.py`, find the `TuyaLevelControl` class, then find the following lines:

Sorry I don't have that file in my custom_zha_quriks folder, I created that folder and placed the path into my config,yaml. where do you get all these files from in the first place?

here's a look at my directory:

[image]
[image]
[image]
The last image with the init.py file I found online and pasted into it because there was no other obvious way for me to get it. Before I am able to do all this I need to get access to the files you are talking about, can you help me?

javicalle, 2022-12-15:
That would need to copy several files and replace the imports to the new paths.

yftj, 2022-12-16:
> That would need to copy several files and replace the imports to the new paths.

I'm more than willing to do that if it means this issue can be fixed

Shulyaka, 2022-12-16:
Yes, you can take the files online:
https://github.com/zigpy/zha-device-handlers/tree/master/zhaquirks/tuya

javicalle, 2022-12-16:
I don't know which is your current `__init__.py` file but what you need to do is:
* copy in your local quirk folder the current version of this files:
  * https://github.com/zigpy/zha-device-handlers/blob/master/zhaquirks/tuya/ts0601_dimmer.py
  * https://github.com/zigpy/zha-device-handlers/blob/master/zhaquirks/tuya/mcu/__init__.py
* edit in the `ts0601_dimmer.py` file the `zhaquirks.tuya.mcu` import statement like this:
```python
from custom_zha_quirks import (
    TuyaInWallLevelControl,
    TuyaLevelControlManufCluster,
    TuyaOnOff,
    TuyaOnOffNM,
)
```

Save changes and restart HA.
Test at this point that device is working the same way until now.
Check that the device signature has changed.

If everything is OK, perform the @Shulyaka proposed changes and tests.

yftj, 2022-12-17:
> I don't know which is your current `__init__.py` file but what you need to do is:
>
> * copy in your local quirk folder the current version of this files:
>
>   * [...]

Absolute Legend, thanks for making all that clear for me! Alright I've double checked the devices response to these changes and it has the same response. Now after editing what @Shulyaka asked for and with these files 
[image]
[image]
[image]
^this import settings was causing ZHA to crash tried first without commenting old import and with, both caused ZHA to fail to load on restart so didn't include the custom_zha import for these logs.

here are the logs:

<details>
<summary>Logs with and without Enable enhanced brightness slider during light transition
</summary>

```

[0x30FC:1:0xef00] Traceback (most recent call last): File "/usr/local/lib/python3.10/site-packages/zigpy/zcl/__init__.py", line 358, in reply return await self._endpoint.reply( File "/usr/local/lib/python3.10/site-packages/zigpy/endpoint.py", line 256, in reply return await self.device.reply( File "/usr/local/lib/python3.10/site-packages/zigpy/device.py", line 377, in reply return await self.request( File "/usr/local/lib/python3.10/site-packages/zigpy/device.py", line 296, in request await self._application.request( File "/usr/local/lib/python3.10/site-packages/zigpy/application.py", line 672, in request await self.send_packet( File "/usr/local/lib/python3.10/site-packages/bellows/zigbee/application.py", line 744, in send_packet status, _ = await self._ezsp.sendUnicast( AttributeError: 'NoneType' object has no attribute 'sendUnicast'
7:34:30 PM – (ERROR) runner.py
Discarding _save_device_relays event
7:34:24 PM – (WARNING) runner.py - message first occurred at 7:34:24 PM and shows up 34 times
Discarding _save_attribute event
7:34:24 PM – (WARNING) components/zha/core/channels/base.py - message first occurred at 7:34:24 PM and shows up 2 times

2022-12-17 19:33:07.458 DEBUG (MainThread) [zhaquirks] Loading custom quirks from /config/custom_zha_quirks
2022-12-17 19:33:07.459 DEBUG (MainThread) [zhaquirks] Loading custom quirks module mcu
2022-12-17 19:33:07.467 DEBUG (MainThread) [zhaquirks] Loading custom quirks module ts0601_dimmer
2022-12-17 19:33:07.473 DEBUG (MainThread) [homeassistant.components.zha] ZHA storage file does not exist or was already removed
2022-12-17 19:33:15.085 DEBUG (MainThread) [zigpy.appdb] Current database version is v11
2022-12-17 19:33:15.102 DEBUG (MainThread) [zigpy.appdb] Loading application state

2022-12-17 19:33:46.294 DEBUG (MainThread) [zigpy.zcl] [0x446A:1:0xef00] No datapoint handler for TuyaDatapointData(dp=14, data=TuyaData(dp_type=<TuyaDPType.ENUM: 4>, function=0, raw=b'\x00', *payload=<enum8.undefined_0x00: 0>))
2022-12-17 19:33:46.295 DEBUG (MainThread) [zigpy.zcl] [0x446A:1:0xef00] Sending reply header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=False, direction=<Direction.Client_to_Server: 1>, disable_default_response=1, reserved=0, *is_cluster=False, *is_general=True, *is_reply=True), tsn=54, command_id=<GeneralCommand.Default_Response: 11>, *direction=<Direction.Client_to_Server: 1>, *is_reply=True)
2022-12-17 19:33:46.295 DEBUG (MainThread) [zigpy.zcl] [0x446A:1:0xef00] Sending reply: Default_Response(command_id=1, status=<Status.UNSUPPORTED_ATTRIBUTE: 134>)
2022-12-17 19:33:46.827 DEBUG (MainThread) [homeassistant.components.zha.entity] light.youssefs_room_light_2: setting transitioning flag to True
2022-12-17 19:33:46.827 DEBUG (MainThread) [zigpy.zcl] [0x30FC:1:0x0008] Sending Tuya Cluster Command. Cluster Command is 4, Arguments are (), {'level': 51, 'transition_time': 0}
2022-12-17 19:33:46.828 DEBUG (MainThread) [zigpy.zcl] [0x30FC:1:0xef00] tuya_mcu_command: cluster_data=TuyaClusterData(endpoint_id=1, cluster_attr='on_off', attr_value=True, expect_reply=True)
2022-12-17 19:33:46.828 DEBUG (MainThread) [zigpy.zcl] [0x30FC:1:0xef00] get_dp_mapping --> found DP: 1
2022-12-17 19:33:46.828 DEBUG (MainThread) [zigpy.zcl] [0x30FC:1:0xef00] from_cluster_data: {1: DPToAttributeMapping(ep_attribute='on_off', attribute_name='on_off', dp_type=<TuyaDPType.BOOL: 1>, converter=None, dp_converter=None, endpoint_id=None)}
2022-12-17 19:33:46.828 DEBUG (MainThread) [zigpy.zcl] [0x30FC:1:0xef00] ztype: Bool.true
2022-12-17 19:33:46.828 DEBUG (MainThread) [zigpy.zcl] [0x30FC:1:0xef00] from_value: [1, 1]
2022-12-17 19:33:46.828 DEBUG (MainThread) [zigpy.zcl] [0x30FC:1:0xef00] raw: b'\x01'
2022-12-17 19:33:46.828 DEBUG (MainThread) [zigpy.zcl] [0x30FC:1:0xef00] tuya_commands: [TuyaCommand(status=0, tsn=155, datapoints=[TuyaDatapointData(dp=1, data=TuyaData(dp_type=<TuyaDPType.BOOL: 1>, function=0, raw=b'\x01', *payload=<Bool.true: 1>))])]
2022-12-17 19:33:46.829 DEBUG (MainThread) [zigpy.zcl] [0x30FC:1:0xef00] tuya_mcu_command: cluster_data=TuyaClusterData(endpoint_id=1, cluster_attr='current_level', attr_value=51, expect_reply=True)
2022-12-17 19:33:46.829 DEBUG (MainThread) [zigpy.zcl] [0x30FC:1:0xef00] get_dp_mapping --> found DP: 2
2022-12-17 19:33:46.829 DEBUG (MainThread) [zigpy.zcl] [0x30FC:1:0xef00] from_cluster_data: {2: DPToAttributeMapping(ep_attribute='level', attribute_name='current_level', dp_type=<TuyaDPType.VALUE: 2>, converter=<function TuyaLevelControlManufCluster.<lambda> at 0x7f613294c1f0>, dp_converter=<function TuyaLevelControlManufCluster.<lambda> at 0x7f613294c280>, endpoint_id=None)}
[... cut ...] ent_to_Server: 1>, disable_default_response=0, reserved=0, *is_cluster=True, *is_general=False, *is_reply=True), tsn=99, command_id=1, *direction=<Direction.Client_to_Server: 1>, *is_reply=True)
2022-12-17 19:34:45.263 DEBUG (MainThread) [zigpy.zcl] [0xA4C3:1:0xef00] Decoded ZCL frame: TuyaLevelControlManufCluster:get_data(data=TuyaCommand(status=0, tsn=81, datapoints=[TuyaDatapointData(dp=2, data=TuyaData(dp_type=<TuyaDPType.VALUE: 2>, function=0, raw=b'\xa0\x01\x00\x00', *payload=416))]))

```

yftj, 2022-12-17:
Never mind I figured out what caused the crash and applied the custom zha settings. And the lights work exactly as expected! the original issue is solved! smooth transition without light flickering on and off! However they fail to turn on when brightness is increased from 0.  here are the new logs:

<details>
<summary>LOGS</summary>

```

2022-12-17 19:45:18.390 DEBUG (MainThread) [zigpy.zcl] [0x30FC:1:0xef00] No datapoint handler for TuyaDatapointData(dp=14, data=TuyaData(dp_type=<TuyaDPType.ENUM: 4>, function=0, raw=b'\x00', *payload=<enum8.undefined_0x00: 0>))
2022-12-17 19:45:18.392 DEBUG (MainThread) [zigpy.zcl] [0x30FC:1:0xef00] Sending reply header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=False, direction=<Direction.Client_to_Server: 1>, disable_default_response=1, reserved=0, *is_cluster=False, *is_general=True, *is_reply=True), tsn=105, command_id=<GeneralCommand.Default_Response: 11>, *direction=<Direction.Client_to_Server: 1>, *is_reply=True)
2022-12-17 19:45:18.393 DEBUG (MainThread) [zigpy.zcl] [0x30FC:1:0xef00] Sending reply: Default_Response(command_id=1, status=<Status.UNSUPPORTED_ATTRIBUTE: 134>)
2022-12-17 19:45:18.974 DEBUG (MainThread) [homeassistant.components.zha.core.device] [0x625E](lumi.magnet.acn001): Device seen - marking the device available and resetting counter
2022-12-17 19:45:18.974 DEBUG (MainThread) [homeassistant.components.zha.core.device] [0x625E](lumi.magnet.acn001): Update device availability - device available: True - new availability: True - changed: False
2022-12-17 19:45:19.069 DEBUG (MainThread) [homeassistant.components.zha.core.device] [0x446A](TS0601): Device seen - marking the device available and resetting counter
2022-12-17 19:45:19.070 DEBUG (MainThread) [homeassistant.components.zha.core.device] [0x446A](TS0601): Update device availability - device available: True - new availability: True - changed: False
2022-12-17 19:45:19.082 DEBUG (MainThread) [zigpy.zcl] [0x30FC:1:0x0008] Sending Tuya Cluster Command. Cluster Command is 4, Arguments are (), {'level': 240, 'transition_time': 0}
2022-12-17 19:45:19.082 DEBUG (MainThread) [zigpy.zcl] [0x30FC:1:0xef00] tuya_mcu_command: cluster_data=TuyaClusterData(endpoint_id=1, cluster_attr='current_level', attr_value=240, expect_reply=True)
2022-12-17 19:45:19.082 DEBUG (MainThread) [zigpy.zcl] [0x30FC:1:0xef00] get_dp_mapping --> found DP: 2
2022-12-17 19:45:19.082 DEBUG (MainThread) [zigpy.zcl] [0x30FC:1:0xef00] from_cluster_data: {2: DPToAttributeMapping(ep_attribute='level', attribute_name='current_level', dp_type=<TuyaDPType.VALUE: 2>, converter=<function TuyaLevelControlManufCluster.<lambda> at 0x7f5bd5f82710>, dp_converter=<function TuyaLevelControlManufCluster.<lambda> at 0x7f5bd5f827a0>, endpoint_id=None)}
2022-12-17 19:45:19.082 DEBUG (MainThread) [zigpy.zcl] [0x30FC:1:0xef00] converted: 941
2022-12-17 19:45:19.082 DEBUG (MainThread) [zigpy.zcl] [0x30FC:1:0xef00] ztype: 941
2022-12-17 19:45:19.083 DEBUG (MainThread) [zigpy.zcl] [0x30FC:1:0xef00] from_value: [4, 0, 0, 3, 173]
2022-12-17 19:45:19.083 DEBUG (MainThread) [zigpy.zcl] [0x30FC:1:0xef00] raw: b'\x00\x00\x03\xad'
2022-12-17 19:45:19.083 DEBUG (MainThread) [zigpy.zcl] [0x30FC:1:0xef00] tuya_commands: [TuyaCommand(status=0, tsn=151, datapoints=[TuyaDatapointData(dp=2, data=TuyaData(dp_type=<TuyaDPType.VALUE: 2>, function=0, raw=b'\x00\x00\x03\xad', *payload=2902654976))])]
2022-12-17 19:45:19.083 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x30FC:1:0x0008]: executed 'move_to_level_with_on_off' command with args: '()' kwargs: '{'level': 240, 'transition_time': 0}' result: Default_Response(command_id=4, status=<Status.SUCCESS: 0>)
2022-12-17 19:45:19.083 DEBUG (MainThread) [homeassistant.components.zha.entity] light.youssefs_room_light_2: turned on: {'move_to_level_with_on_off': Default_Response(command_id=4, status=<Status.SUCCESS: 0>)}
2022-12-17 19:45:19.084 DEBUG (MainThread) [zigpy.zcl] [0x30FC:1:0xef00] Sending request header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.CLUSTER_COMMAND: 1>, is_manufacturer_specific=True, direction=<Direction.Server_to_Client: 0>, disable_default_response=0, reserved=0, *is_cluster=True, *is_general=False, *is_reply=False), manufacturer=4098, tsn=152, command_id=0, *direction=<Direction.Server_to_Client: 0>, *is_reply=False)
2022-12-17 19:45:19.085 DEBUG (MainThread) [zigpy.zcl] [0x30FC:1:0xef00] Sending request: set_data(data=TuyaCommand(status=0, tsn=151, datapoints=[TuyaDatapointData(dp=2, data=TuyaData(dp_type=<TuyaDPType.VALUE: 2>, function=0, raw=b'\x00\x00\x03\xad', *payload=2902654976))]))
2022-12-17 19:45:19.146 DEBUG (MainThread) [zigpy.application] Received a packet: ZigbeePacket(src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x446A), src_ep=1, dst=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), dst_ep=1, source_route=None, extended_timeout=False, tsn=115, profile_id=260, cluster_id=61184, data=Serialized[b'\t=\x01\x002\x01\x01\x00\x01\x00'], tx_options=<TransmitOptions.NONE: 0>, radius=0, non_member_radius=0, lqi=180, rssi=-55)
2022-12-17 19:45:19.147 DEBUG (MainThread) [zigpy.zcl] [0x446A:1:0xef00] Received ZCL frame: b'\t=\x01\x002\x01\x01\x00\x01\x00'
2022-12-17 19:45:19.147 DEBUG (MainThread) [zigpy.zcl] [0x446A:1:0xef00] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.CLUSTER_COMMAND: 1>, is_manufacturer_specific=0, direction=< [... cut ...] ient_to_Server: 1>, disable_default_response=0, reserved=0, *is_cluster=True, *is_general=False, *is_reply=True), tsn=68, command_id=1, *direction=<Direction.Client_to_Server: 1>, *is_reply=True)
2022-12-17 19:45:28.155 DEBUG (MainThread) [zigpy.zcl] [0x446A:1:0xef00] Decoded ZCL frame: TuyaLevelControlManufCluster:get_data(data=TuyaCommand(status=0, tsn=56, datapoints=[TuyaDatapointData(dp=1, data=TuyaData(dp_type=<TuyaDPType.BOOL: 1>, function=0, raw=b'\x00', *payload=<Bool.false: 0>))]))

```

Shulyaka, 2022-12-17:
@yftj Thanks for testing!
I hope that #2015 will solve that as well.

PR #2015 (2022-12-29): `move_to_level_with_on_off`: don't send `on_off` when already on
Fixes #1554 
See also: #1748, #1754

The problem:
ZCL defines the `move_to_level_with_on_off` which HA `zha` integration uses to control the brightness.
The Tuya dimmers don't have this command. The quirk has implemented this command by subsequently issuing `on_off` and `move_to_level` commands.
However it was found out that not all Tuya dimmers behave same on the `on_off` command: some of them switch brightness to 100% upon it, that means that changing brightness from 50% to 55% will first briefly transition through 100%.

This  [... cut]
closed 2022-12-29 by dmulcahey