zigpy/zha-device-handlers #1556: [Device Support Request] Tuya TS0601 - TZE200_81isopgh - Garden Watering System

simonbaudart, 2022-05-10:
Hello, first of all, thanks for your future help :p

**Is your feature request related to a problem? Please describe.**
I buy this product to manage my garden watering system : https://www.domadoo.fr/fr/peripheriques/5886-saswell-electrovanne-et-programmateur-d-arrosage-connecte-zigbee-mesure-de-consommation.html

It's not recognized by ZHA and I started a quirks, with some tries, I managed to have the on_off working ! Yeah !
I would like to go further, because I know that it reports the consummed water (I hope because it was specified)

<details>
<summary>Actual Quirks
</summary>

```
"""Tuya Garden Watering"""

from zigpy.profiles import zha
from zigpy.quirks import CustomDevice
from zigpy.zcl.clusters.general import Basic, Groups, Ota, Scenes, Time

from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODELS_INFO,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
)

from zhaquirks.tuya import TuyaManufCluster
from zhaquirks.tuya.mcu import MoesSwitchManufCluster, TuyaOnOff, TuyaOnOffManufCluster

class TuyaGardenWatering(CustomDevice):
    """Tuya Garden Watering"""

    signature = {
        MODELS_INFO: [("_TZE200_81isopgh", "TS0601")],
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.SMART_PLUG,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaOnOffManufCluster.cluster_id,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            }
        },
    }

    replacement = {
        ENDPOINTS: {
            1: {
                DEVICE_TYPE: zha.DeviceType.SMART_PLUG,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaManufCluster.cluster_id,
                    TuyaOnOffManufCluster,
                    TuyaOnOff,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            }
        }
    }
```
</details>

[... 247 lines cut ...]
    "endpoint_names": [
      {
        "name": "SMART_PLUG"
      }
    ],
    "user_given_name": "Arrosage",
    "device_reg_id": "c0fcf22479dc276b1786a1351bdd7c8c",
    "area_id": "garage"
  }
}
```

</details>

<details>
<summary>Additional logs</summary>

```
Don't know what I can send, but ask me, I'll post what you need :)
```

</details>

**Additional context**
Haha ... I understand the principle of quirks and I can read the python code ... I don't know how to find the right value without try and try and try and try again ...

javicalle, 2022-05-10:
The debug logs from the device are needed to get the necesary info.

Usually when you operate the physical device it reports its state. Maybe we can get some clues about the DPs that report the info and values.
Also, when the batteries are removed and put again, the device would report its status.

To activate the debug logs:
* https://www.home-assistant.io/integrations/zha/#debug-logging
It can be very verbose. Take just the relevant info from your device.

simonbaudart, 2022-05-10:
Hello

Thank you for your answers.

@MattWestb : thanks, I checked and I made some changes on my quirks and I get the battery percent ... but I cannot get more ...

@javicalle : here is the log :

</details>

<details>
<summary>Additional logs</summary>

```
2022-05-10 22:44:19 DEBUG (MainThread) [homeassistant.components.zha.core.gateway] [0x873E](_TZE200_81isopgh TS0601) restored as 'available', last seen: 0:01:15 ago, consider_unavailable_time: 21600 seconds
2022-05-10 22:44:19 DEBUG (MainThread) [homeassistant.components.zha.core.device] [0x873E](TS0601): started initialization
2022-05-10 22:44:19 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x873E:ZDO](TS0601): 'async_initialize' stage succeeded
2022-05-10 22:44:19 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x873E:1:0x0006]: initializing channel: from_cache: True
2022-05-10 22:44:19 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x873E:1:0x0006]: initializing cached channel attributes: ['start_up_on_off']
2022-05-10 22:44:19 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x873E:1:0x0006]: Reading attributes in chunks: ['start_up_on_off']
2022-05-10 22:44:19 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x873E:1:0x0001]: initializing channel: from_cache: True
2022-05-10 22:44:19 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x873E:1:0x0001]: initializing uncached channel attributes: ['battery_voltage', 'battery_percentage_remaining'] - from cache[True]
2022-05-10 22:44:19 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x873E:1:0x0001]: Reading attributes in chunks: ['battery_voltage', 'battery_percentage_remaining']
2022-05-10 22:44:19 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x873E:1:0x0001]: Performing channel specific initialization: ['battery_voltage', 'battery_percentage_remaining']
2022-05-10 22:44:19 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x873E:1:0x0001]: Reading attributes in chunks: ['battery_size', 'battery_quantity']
2022-05-10 22:44:19 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x873E:1:0x0001]: finished channel initialization
2022-05-10 22:44:19 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x873E:1:0x0000]: initializing channel: from_cache: True
2022-05-10 22:44:19 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x873E:1:0x0000]: finished channel initialization
2022-05-10 22:44:19 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x873E:1:0x0019]: initializing channel: from_cache: True
2022-05-10 22:44:19 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x873E:1:0x0019]: finished channel initialization
2022-05-10 22:44:19 WARNING (MainThread) [homeassistant.components.zha.core.channels.base] [0x873E:1:0x0006]: 'async_initialize' stage failed: 'start_up_on_off'
2022-05-10 22:44:19 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x873E:1:0x0001]: 'async_initialize' stage succeeded
2022-05-10 22:44:19 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x873E:1:0x0000]: 'async_initialize' stage succeeded
2022-05-10 22:44:19 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x873E:1:0x0019]: 'async_initialize' stage succeeded
2022-05-10 22:44:19 DEBUG (MainThread) [homeassistant.components.zha.core.device] [0x873E](TS0601): power source: Battery or Unknown
2022-05-10 22:44:19 DEBUG (MainThread) [homeassistant.components.zha.core.device] [0x873E](TS0601): completed initialization
2022-05-10 22:44:33 DEBUG (MainThread) [zigpy.zcl] [0x873E:1:0x0006] Sending Tuya Cluster Command... Cluster Command is 1, Arguments are ()
2022-05-10 22:44:33 DEBUG (MainThread) [zigpy.zcl] [0x873E:1:0xef00] tuya_mcu_command: cluster_data=TuyaClusterData(endpoint_id=1, cluster_attr='on_off', attr_value=1, expect_reply=True)
2022-05-10 22:44:33 DEBUG (MainThread) [zigpy.zcl] [0x873E:1:0xef00] get_dp_mapping --> found DP: 1
2022-05-10 22:44:33 DEBUG (MainThread) [zigpy.zcl] [0x873E:1:0xef00] from_cluster_data: 1, DPToAttributeMapping(ep_attribute='on_off', attribute_name='on_off', dp_type=<TuyaDPType.BOOL: 1>, converter=None, dp_converter=None, endpoint_id=None)
2022-05-10 22:44:33 DEBUG (MainThread) [zigpy.zcl] [0x873E:1:0xef00] ztype: Bool.true
2022-05-10 22:44:33 DEBUG (MainThread) [zigpy.zcl] [0x873E:1:0xef00] from_value: [1, 1]
2022-05-10 22:44:33 DEBUG (MainThread) [zigpy.zcl] [0x873E:1:0xef00] raw: b'\x01'
2022-05-10 22:44:33 DEBUG (MainThread) [zigpy.zcl] [0x873E:1:0xef00] tuya_command: TuyaCommand(status=0, tsn=68, dp=1, data=TuyaData(dp_type=<TuyaDPType.BOOL: 1>, function=0, raw=b'\x01', *payload=<Bool.true: 1>))
2022-05-10 22:44:33 DEBUG (MainThread) [zigpy.zcl] [0x873E:1:0xef00] Sending request header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.CLUSTER_COMMAND: 1>, is_manufacturer_specific=True, is_reply=0, disable_default_response=0, reserved=0, *is_cluster=True, *is_general=False), manufacturer=4098, tsn=69, command_id=0, *is_reply=False)
2022-05-10 22:44:33 DEBUG (MainThread) [zigpy.zcl] [0x873E:1:0xef00] Sending request: set_data(data=TuyaCommand(status=0, tsn=68, dp=1, data=TuyaData(dp_type=<TuyaDPType.BOOL: 1>, function=0, raw=b'\x01', *payload=<Bool.true: 1>)))
[... cut ...]
2022-05-10 23:05:03 DEBUG (MainThread) [zigpy_znp.api] Sending request: AF.DataRequestExt.Req(DstAddrModeAddress=AddrModeAddress(mode=<AddrMode.NWK: 2>, address=0x873E), DstEndpoint=1, DstPanId=0x0000, SrcEndpoint=1, ClusterId=61184, TSN=44, Options=<TransmitOptions.SUPPRESS_ROUTE_DISC_NETWORK: 32>, Radius=30, Data=b'\x18\x2C\x0B\x01\x81')
    
```

</details>

I think the data point (correct for DP ?) are right in the pull request but I cannot manage to have the value displayed in Home Assistant :'( I think I miss something ...

simonbaudart, 2022-05-10:
Maybe something more relevant : 
- dp=12; payload=1 -> Open
- dp=12; payload=1 -> Closed
- dp=7; payload=100 -> Battery
- dp=11; payload=... -> Remaining time because it's only open for 10 minutes
- dp=5; payload=0 -> According to https://github.com/zigpy/zha-device-handlers/pull/1446, it should be volume but I don"t know why it's 0, and what is the magic number 33.8140226

Can you help me to make these number appear in HA ?

```
2022-05-10 23:36:39 WARNING (MainThread) [zigpy.zcl] [0x873E:1:0xef00] No 'handle_get_data' tuya handler found for get_data(data=TuyaCommand(status=0, tsn=87, dp=12, data=TuyaData(dp_type=<TuyaDPType.ENUM: 4>, function=0, raw=b'\x01', *payload=<enum8.undefined_0x01: 1>)))
2022-05-10 23:36:39 WARNING (MainThread) [zigpy.zcl] [0x873E:1:0xef00] No 'handle_get_data' tuya handler found for get_data(data=TuyaCommand(status=0, tsn=88, dp=11, data=TuyaData(dp_type=<TuyaDPType.VALUE: 2>, function=0, raw=b'X\x02\x00\x00', *payload=600)))
2022-05-10 23:37:26 WARNING (MainThread) [zigpy.zcl] [0x873E:1:0xef00] No 'handle_get_data' tuya handler found for get_data(data=TuyaCommand(status=0, tsn=90, dp=11, data=TuyaData(dp_type=<TuyaDPType.VALUE: 2>, function=0, raw=b'\x1c\x02\x00\x00', *payload=540)))
2022-05-10 23:38:26 WARNING (MainThread) [zigpy.zcl] [0x873E:1:0xef00] No 'handle_get_data' tuya handler found for get_data(data=TuyaCommand(status=0, tsn=92, dp=11, data=TuyaData(dp_type=<TuyaDPType.VALUE: 2>, function=0, raw=b'\xe0\x01\x00\x00', *payload=480)))
2022-05-10 23:38:28 WARNING (MainThread) [zigpy.zcl] [0x873E:1:0xef00] No 'handle_get_data' tuya handler found for get_data(data=TuyaCommand(status=0, tsn=93, dp=7, data=TuyaData(dp_type=<TuyaDPType.VALUE: 2>, function=0, raw=b'd\x00\x00\x00', *payload=100)))
2022-05-10 23:39:26 WARNING (MainThread) [zigpy.zcl] [0x873E:1:0xef00] No 'handle_get_data' tuya handler found for get_data(data=TuyaCommand(status=0, tsn=95, dp=11, data=TuyaData(dp_type=<TuyaDPType.VALUE: 2>, function=0, raw=b'\xa4\x01\x00\x00', *payload=420)))
2022-05-10 23:40:26 WARNING (MainThread) [zigpy.zcl] [0x873E:1:0xef00] No 'handle_get_data' tuya handler found for get_data(data=TuyaCommand(status=0, tsn=97, dp=11, data=TuyaData(dp_type=<TuyaDPType.VALUE: 2>, function=0, raw=b'h\x01\x00\x00', *payload=360)))
2022-05-10 23:41:26 WARNING (MainThread) [zigpy.zcl] [0x873E:1:0xef00] No 'handle_get_data' tuya handler found for get_data(data=TuyaCommand(status=0, tsn=99, dp=11, data=TuyaData(dp_type=<TuyaDPType.VALUE: 2>, function=0, raw=b',\x01\x00\x00', *payload=300)))
2022-05-10 23:42:26 WARNING (MainThread) [zigpy.zcl] [0x873E:1:0xef00] No 'handle_get_data' tuya handler found for get_data(data=TuyaCommand(status=0, tsn=101, dp=11, data=TuyaData(dp_type=<TuyaDPType.VALUE: 2>, function=0, raw=b'\xf0\x00\x00\x00', *payload=240)))
2022-05-10 23:43:26 WARNING (MainThread) [zigpy.zcl] [0x873E:1:0xef00] No 'handle_get_data' tuya handler found for get_data(data=TuyaCommand(status=0, tsn=103, dp=11, data=TuyaData(dp_type=<TuyaDPType.VALUE: 2>, function=0, raw=b'\xb4\x00\x00\x00', *payload=180)))
2022-05-10 23:44:26 WARNING (MainThread) [zigpy.zcl] [0x873E:1:0xef00] No 'handle_get_data' tuya handler found for get_data(data=TuyaCommand(status=0, tsn=105, dp=11, data=TuyaData(dp_type=<TuyaDPType.VALUE: 2>, function=0, raw=b'x\x00\x00\x00', *payload=120)))
2022-05-10 23:45:26 WARNING (MainThread) [zigpy.zcl] [0x873E:1:0xef00] No 'handle_get_data' tuya handler found for get_data(data=TuyaCommand(status=0, tsn=107, dp=11, data=TuyaData(dp_type=<TuyaDPType.VALUE: 2>, function=0, raw=b'<\x00\x00\x00', *payload=60)))
2022-05-10 23:46:28 WARNING (MainThread) [zigpy.zcl] [0x873E:1:0xef00] No 'handle_get_data' tuya handler found for get_data(data=TuyaCommand(status=0, tsn=110, dp=15, data=TuyaData(dp_type=<TuyaDPType.VALUE: 2>, function=0, raw=b'X\x02\x00\x00', *payload=600)))
2022-05-10 23:46:28 WARNING (MainThread) [zigpy.zcl] [0x873E:1:0xef00] No 'handle_get_data' tuya handler found for get_data(data=TuyaCommand(status=0, tsn=111, dp=6, data=TuyaData(dp_type=<TuyaDPType.VALUE: 2>, function=0, raw=b'\x00\x00\x00\x00', *payload=0)))
2022-05-10 23:46:28 WARNING (MainThread) [zigpy.zcl] [0x873E:1:0xef00] No 'handle_get_data' tuya handler found for get_data(data=TuyaCommand(status=0, tsn=112, dp=5, data=TuyaData(dp_type=<TuyaDPType.VALUE: 2>, function=0, raw=b'\x00\x00\x00\x00', *payload=0)))
2022-05-10 23:46:29 WARNING (MainThread) [zigpy.zcl] [0x873E:1:0xef00] No 'handle_get_data' tuya handler found for get_data(data=TuyaCommand(status=0, tsn=113, dp=12, data=TuyaData(dp_type=<TuyaDPType.ENUM: 4>, function=0, raw=b'\x02', *payload=<enum8.undefined_0x02: 2>)))
```

javicalle, 2022-05-11:
I haven't had time to go through the logs in detail, but what I see is:
* DP12 it's an ENUM. It suggest a few discrete values (more than open/closed that could be a BOOLEAN like DP1)
* DP15 open_time? the total time the valve will be open. Maybe this value can be setted

If DP5 it's the volume, maybe it is not reported and must be obtained from the device. Have you tried to get its value from the device view?

javicalle, 2022-05-12:
@simonbaudart before continuing with the PR, can we verify what works and what doesn't?
* IMO `ts0601_valve.py` would be a more appropiate filename
* why the x / 60 conversion? I think that will be more usefull the remaining time in seconds.

Also I have a few code suggestions (not tested, can be code errors):

<details>
<summary>ts0601_valve.py</summary>

```python
"""Tuya Garden Watering."""

import logging
from typing import Dict

from zigpy.profiles import zha
from zigpy.quirks import CustomDevice
import zigpy.types as t
from zigpy.zcl.clusters.general import Basic, Groups, Ota, Scenes, Time
from zigpy.zcl.clusters.measurement import FlowMeasurement

from zhaquirks import DoublingPowerConfigurationCluster
from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODELS_INFO,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
)
from zhaquirks.tuya import TuyaLocalCluster
from zhaquirks.tuya.mcu import (
    DPToAttributeMapping,
    TuyaAttributesCluster,
    TuyaDPType,
    TuyaMCUCluster,
    TuyaOnOff,
)

_LOGGER = logging.getLogger(__name__)

class TuyaGardenWateringTimer(TuyaAttributesCluster):
    """Timer cluster."""

    cluster_id = 0x043E
    name = "Timer"
    ep_attribute = "timer"

    attributes = {
        0x000C: ("state", t.uint16_t),
        0x000B: ("time_left", t.uint16_t),
        0x000F: ("last_valve_open_duration", t.uint16_t),
    }

class TuyaGardenWateringWaterConsumed(FlowMeasurement, TuyaLocalCluster):
    """Tuya Water consumed cluster."""

class TuyaGardenManufCluster(TuyaMCUCluster):
    """On/Off Tuya cluster with extra device attributes."""

    dp_to_attribute: Dict[int, DPToAttributeMapping] = {
        1: DPToAttributeMapping(
            TuyaOnOff.ep_attribute,
            "on_off",
            dp_type=TuyaDPType.BOOL,
        ),
        5: DPToAttributeMapping(
            TuyaGardenWateringWaterConsumed.ep_attribute,
            "measured_value",
            dp_type=TuyaDPType.VALUE,
        ),
        7: DPToAttributeMapping(
            DoublingPowerConfigurationCluster.ep_attribute,
            "battery_percentage_remaining",
            TuyaDPType.VALUE,
        ),
        11: DPToAttributeMapping(
            TuyaGardenWateringTimer.ep_attribute,
            "time_left",
            TuyaDPType.VALUE,
            lambda x: x / 60,
        ),
        12: DPToAttributeMapping(
            TuyaGardenWateringTimer.ep_attribute,
            "state",
            TuyaDPType.VALUE,
        ),
        15: DPToAttributeMapping(
            TuyaGardenWateringTimer.ep_attribute,
            "last_valve_open_duration",
            TuyaDPType.VALUE,
            lambda x: x / 60,
        ),
    }

    data_point_handlers = {
        1: "_dp_2_attr_update",
        5: "_dp_2_attr_update",
        7: "_dp_2_attr_update",
        11: "_dp_2_attr_update",
        12: "_dp_2_attr_update",
        15: "_dp_2_attr_update",
    }

class TuyaGardenWatering(CustomDevice):
    """Tuya Garden Watering."""

    signature = {
        MODELS_INFO: [("_TZE200_81isopgh", "TS0601")],
        # SizePrefixedSimpleDescriptor(endpoint=1, profile=260, device_type=81, device_version=1,
        # input_clusters=[0, 4, 5, 61184], output_clusters=[25, 10])
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.SMART_PLUG,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaGardenManufCluster.cluster_id,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            }
        },
    }

    replacement = {
        ENDPOINTS: {
            1: {
                DEVICE_TYPE: zha.DeviceType.SMART_PLUG,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaOnOff,
                    TuyaGardenWateringWaterConsumed,
                    DoublingPowerConfigurationCluster,
                    TuyaGardenWateringTimer,
                    TuyaGardenManufCluster,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            }
        }
    }
```

</details>

And before going further with the PR can we review what is working and what is not?
Are you able to open/close the valve from the HA device (light, plug, switch,...)?

javicalle, 2022-05-12:
> is FlowMeasurement a right Cluster to report the water volume ?

I can't tell you. I'm not familiar with all the ZHA <--> HA conversions, but I know that not all clusters generates an HA entity.

My guessing will be that the `Metering` cluster will create an HA sensor:
* https://github.com/zigpy/zigpy/blob/8fefb7c8cf327858c35002a66c3e3cac31fdafc4/zigpy/zcl/clusters/smartenergy.py#L24
* https://github.com/home-assistant/core/blob/ae89a1243a563ea757a8251310101c6956c19976/homeassistant/components/zha/sensor.py#L423

Probably needs to configure the 'unit of measurement' and maybe other attributes. Will need to investigate if there are any quirk implementation and a lot of trial and error

> * The DP 5 always send 0 to me, some other users (in Jeedom) reported as working ... maybe an issue on my side

I don't think it's an implementation issue.
The device logs show that the value is always 0, so it's not a problem in the quirk.
Perhaps, although it looks like the same device, it is different and in yours it has a different functionality.
Of the functions announced by the manufacturer, are there any that have not been implemented? Maybe that can give us a clue to the DP5 function.

simonbaudart, 2022-05-13:
Ok, I switched to a Metering, it's a better idea because the valve should report a total volume not a flow.
And yes, it's supported by HA

I found the unit of measurement, I cloned the homeassistant-core repo and dig to find the right values :)

The function announced are : 
- On/Off
- Volume reporting

In the meantime I found that DP6 sometimes raise a value at 0 ... I've mapped to a fake cluster (I found the same usage in ts0601_motion.

javicalle, 2022-05-13:
Nice work!

Maybe I would replace the Garden and GardenWatering for a mire generic 'Valve' names. 😅

simonbaudart, 2022-05-16:
> Nice work!
> 
> Maybe I would replace the Garden and GardenWatering for a more generic 'Valve' names. 😅

Done !

rforro, 2022-05-16:
I was asking in the PR about the Timer cluster because I don't see any advantage in the approach to separate the timer values into made up cluster.

```
class TuyaValveTimer(TuyaAttributesCluster):
    """Timer cluster."""

    cluster_id = 0x043E
    name = "Timer"
    ep_attribute = "timer"

    attributes = {
        0x000C: ("state", t.uint16_t),
        0x000B: ("time_left", t.uint16_t),
        0x000F: ("last_valve_open_duration", t.uint16_t),
    }
```

In HA frontend there are not entities which made to match this cluster. Of course you can create them by yourself, but it is not standard so anybody in the future implementing a timer has to find this quirk.

I was advised to use different approach. Do not create a new made up cluster, but add those values into your `TuyaValveManufCluster` cluster. Like here:
```
class TuyaWaterValveManufCluster(TuyaMCUCluster):

    attributes = TuyaMCUCluster.attributes.copy()
    attributes.update(
        {
            0xef01: ("timer", t.uint32_t, True),
            0xef02: ("timer_time_left", t.uint32_t, True),
            0xef03: ("frost_lock", t.Bool, True),
            0xef04: ("frost_lock_reset", t.Bool, True),  # 0 resets frost lock
        }
    )

    dp_to_attribute: Dict[int, DPToAttributeMapping] = {
            5: DPToAttributeMapping(
                TuyaMCUCluster.ep_attribute,
                "timer",
                TuyaDPType.VALUE,
            ),
    }
```

This will reduce bit from the code complexity and after small change to HA frontend (which I have do for my lidl valve) the timer entities will be there.

simonbaudart, 2022-05-16:
Yes, this could be another way to do it. I'm pretty new in ZHA contribution, so, I don't know what is the best.

@javicalle : your advice ?

dmulcahey, 2022-05-16:
> I was asking in the PR about the Timer cluster because I don't see any advantage in the approach to separate the timer values into made up cluster.
>
> ```
> class [...]

This is the correct way to do this

simonbaudart, 2022-05-16:
OK, I'll make the changes !
Thanks for you feedback.

simonbaudart, 2022-05-16:
Ok, it's updated !
I've checked and I see the value in the cluster attributes in HA

[image]

closed 2022-05-23 by dmulcahey

Gilles2senior, 2023-06-12:
From this issue, it looks like a lot of problems have been resolved and most entities have been made available in ZHA.
Nevertheless, my device still doesn't report battery level for instance. The quick listed in the integration in ZHA is the following :  zhaquirks.tuya.ts0601_valve.TuyaValve

rubendura, 2023-06-12:
> From this issue, it looks like a lot of problems have been resolved and most entities have been made available in ZHA. Nevertheless, my device still doesn't report battery level for instance. The quick listed in the integration in ZHA is the following : zhaquirks.tuya.ts0601_valve.TuyaValve

Same. It works, but just as a basic switch. The battery level, water flows, and timer entities aren't working properly for me.

almendesb, 2024-01-10:
Hello!

Initially, I am new here and also a beginner in the world of HA.

I have a TS0601 water valve and it is already active on my HA, however, I can only turn it on or off, as well as see the battery level.

My main objective with this device is to measure water consumption and add it to the DashBoard Energy, however, the water measurement functionality is not immediately active.

From what I read here in this thread, there seems to be a configuration solution for this, but I really couldn't understand it.

I tried to change the cluster, but from what I saw in my HASS there is only way to see the cluster.

I apologize if this is a very beginner's question or request, but what would be the step-by-step process to implement the solution presented in this topic and be able to implement water measurement?