zigpy/zha-device-handlers #1562: [Device Support Request] Bosch Smoke Alarm II

mnadvornik, 2022-05-14:
**Is your feature request related to a problem? Please describe.**
Pairing works but no controls or sensors are detected

**Describe the solution you'd like**
Would love if we could get this device to work. I think it should be possible to read the alarm status, the battery status and also trigger the alarm.

<details>
<summary>Device signature - this can be acquired by clicking on the "Zigbee Device Signature" button in the device settings</summary>

```
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4617, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0402",
      "in_clusters": [
        "0x0000",
        "0x0001",
        "0x0003",
        "0x0020",
        "0x0500",
        "0x0b05"
      ],
      "out_clusters": [
        "0x0019"
      ]
    }
  },
  "manufacturer": "Bosch",
  "model": "RBSH-SD-ZB-EU",
  "class": "zigpy.device.Device"
}
```

</details>

<details>
<summary>Diagnostic information - this can be acquired by clicking on the "Download Diagnostics" button in the device settings</summary>

```
{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2022.5.4",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.9.9",
    "docker": true,
    "arch": "x86_64",
    "timezone": "Europe/Vienna",
    "os_name": "Linux",
    "os_version": "5.10.108",
    "supervisor": "2022.05.1",
    "host_os": "Home Assistant OS 7.6",
    "docker_version": "20.10.9",
    "chassis": "vm",
    "run_as_root": true
  },
  "custom_components": {
    "hacs": {
      "version": "1.24.5",
      "requirements": [
        "aiogithubapi>=21.11.0"
      ]
    },
    "opensprinkler": {
      "version": "1.1.8",
      "requirements": [
        "pyopensprinkler==0.7.2"
      ]
    },
    "alexa_media": {
      "version": "3.11.3",
      "requirements": [
        "alexapy==1.25.5",
        "packaging>=20.3",
        "wrapt>=1.12.1"
      ]
    }
  },
  "integration_manifest": {
    "domain": "zha",
    "name": "Zigbee Home Automation",
    "config_flow": true,
    "documentation": "https://www.home-assistant.io/integrations/zha",
    "requirements": [
      "bellows==0.29.0",
      "pyserial==3.5",
      "pyserial-asyncio==0.6",
      "zha-quirks==0.0.73",
      "zigpy-deconz==0.16.0",
      "zigpy==0.45.1",
      "zigpy-xbee==0.14.0",
      "zigpy-zigate==0.7.4",
      "zigpy-znp==0.7.0"
    ],
    "usb": [
      {
        "vid": "10C4",
        "pid": "EA60",
        "description": "*2652*",
        "known_devices": [
          "slae.sh cc2652rb stick"
        ]
      },
      {
        "vid": "10C4",
        "pid": "EA60",
        "description": "*sonoff*plus*",
        "known_devices": [
          "sonoff zigbee dongle plus"
        ]
      },
      {
        "vid": "10C4",
        "pid": "EA60",
        "description": "*tubeszb*",
        "known_devices": [
          "TubesZB Coordinator"
        ]
      },
      {
        "vid": "1A86",
        "pid": "7523",
        "description": "*tubeszb*",
        "known_devices": [
          "TubesZB Coordinator"
        ]
      },
      {
        "vid": "1A86",
        "pid": "7523",
        "description": "*zigstar*",
        "known_devices": [
          "ZigStar Coordinators"
        ]
      },
      {
        "vid": "1CF1",
        "pid": "0030",
        "description": "*conbee*",
        "known_devices": [
          "Conbee II"
        ]
      },
      {
        "vid": "10C4",
        "pid": "8A2A",
        "description": "*zigbee*",
        "known_devices": [
          "Nortek HUSBZB-1"
        ]
      },
      {
        "vid": "10C4",
        "pid": "8B34",
        "description": "*bv 2010/10*",
        "known_devices": [
          "Bitron Video AV2010/10"
        ]
      }
    ],
    "codeowners": [
      "@dmulcahey",
      "@adminiuga"
    ],
    "zeroconf": [
      {
        "type": "_esphomelib._tcp.local.",
        "name": "tube*"
      }
    ],
    "after_dependencies": [
      "usb",
      "zeroconf"
    ],
    "iot_class": "local_polling",
    "loggers": [
      "aiosqlite",
      "bellows",
      "crccheck",
      "pure_pcapy3",
      "zhaquirks",
      "zigpy",
      "zigpy_deconz",
      "zigpy_xbee",
      "zigpy_zigate",
      "zigpy_znp"
    ],
    "is_built_in": true
  },
  "data": {
    "ieee": "**REDACTED**",
    "nwk": 55672,
    "manufacturer": "Bosch",
    "model": "RBSH-SD-ZB-EU",
    "name": "Bosch RBSH-SD-ZB-EU",
    "quirk_applied": false,
    "quirk_class": "zigpy.device.Device",
    "manufacturer_code": 4617,
    "power_source": "Battery or Unknown",
    "lqi": 148,
    "rssi": -63,
    "last_seen": "2022-05-14T02:29:09",
    "available": true,
    "device_type": "EndDevice",
    "signature": {
      "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4617, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
      "endpoints": {
        "1": {
          "profile_id": 260,
          "device_type": "0x0402",
          "in_clusters": [
            "0x0000",
            "0x0001",
            "0x0003",
            "0x0020",
            "0x0500",
            "0x0b05"
          ],
          "out_clusters": [
            "0x0019"
          ]
        }
      }
    },
    "entities": [],
    "neighbors": [],
    "endpoint_names": [
      {
        "name": "IAS_ZONE"
      }
    ],
    "user_given_name": null,
    "device_reg_id": "f1b27eb989277500e68560e9bc2f6747",
    "area_id": null
  }
}
```

</details>

**Additional context**
This is a very newly released product. What I can tell is that it uses Zigbee 3.0. The IEEE Device ID as well as the install code is printed on the inside of the device. Pairing worked fine by invoking zha.permit and then connecting the battery. However no controls or sensor are available.
I am happy to perform any testing. I just don't have any in-depth knoweldge of the zigbee protocol / zigpy so I don't know where to start

Schnabulation, 2022-10-27:
Has there been any update regarding this? I'm trying to make the Bosch Smoke Alarm II work with deCONZ but it won't even connect.

On another note: were you able to integrate the Bosch Smoke Alarm II into Home Assistant using Boschs smart home coordinator maybe?

cs224, 2022-12-04:
I would be interested in this feature, too.

mnadvornik, 2023-01-20:
@Schnabulation as I don't have a Bosch Smart Home Controller I dont't know if all features would be supported that way. 
But zigpy does currently support reading smoke alarm status and battery status. I don't know which zigpy update added the basic compatiblity but it was not long after opening this issue.
It would be very nice if we could be able to trigger the smoke alarm via zigbee as this is a feature available when using the official controller and app from Bosch

011V32, 2023-04-05:
Hello @mnadvornik,

how should it look like?
How my device looks:
[image]
I get no Battery status and if i press the test button, the sensor stays at off..
Thx.

kfroeschl, 2023-04-10:
@011V32 :
Hi,
I've also 7 of those at home connected through ZHA.
It seems if you don't see a battery your pairing was not 100% successful.

I've "Factory Reset" those according to the manual and started the pairing again.

Afterwards they worked fine.

011V32, 2023-04-12:
Hello @kfroeschl,

Thx. Now I see the battery status. 👍
What shoud "Identify" and "laszone" do?
If I press the smoke alarm, i can hear the alarm, but no status is raised in HA?
(If I press Identity nothing happend?)
[image]
Thx

kfroeschl, 2023-04-12:
@011V32 :
"Identify": Lets the red light of the sensor blink twice.
"iaszone": "Clear": No Smoke, "Dedected": Smoke. I've used it for an automation to send a Notification to the phone if it would detect anything. Though I've never triggered them with real smoke so far.

Schnabulation, 2023-04-12:
Just so you know: pressing the smoke alarm manually does not trigger the sensor. This is the same if using the Bosch SHC.

kfroeschl, 2023-04-12:
I was able to make them sound the alarm by calling the IasZone/init_test_mode command 
[image]

This could also be called from a service if required.
Though I would need to read the actual Zigbee IasZone Cluster Specification to know what SHC is calling to sound the alarm.

011V32, 2023-04-12:
> 

I do not have init_test_mode, but `zone_status` with the same ID?
[image]

kfroeschl, 2023-04-12:
wrong tab:
[image]

011V32, 2023-04-12:
My mistake. Sry

Something is still not right. The Last Seen is when I paired the device. When I run Identify, there is something in the log, but nothing flashes on the device.
[image]

When I run the test command I get the following error:
[image]

I've also tried putting the device right next to the stick... with no success

kfroeschl, 2023-04-12:
I would guess your paring was still not successful. 

I had a similar behavior for some of my Devices. The device needs to "CheckIn" once an hour and update the "Last Seen".
Like it is now in your screenshot you will loose the information in some hours.

You may try again:
1.  Reset the Device
1.1 Remove Battery
1.2 Wait for 1 minute or press the button (to remove remaining voltage
1.3 Hold the button
1.4 Insert the Battery
1.5 Wait for Device to make a sound
2. Wait for the device to start (at least 10 seconds)
3. Call the Permit Service in HomeAssistant
4. Press the Device Button until it sounds
5. Release the Device Button
6. Press it again for a short time
7. Finish the Setup in HA (like for any other Zigbee Device)

I hope i remembered all steps correctly.

011V32, 2023-04-16:
All good things are 3. :P
LQI and RSSI is only displayed under Device Info and not under Diagnostic?

[image]

kfroeschl, 2023-04-16:
you can disable/enable specific entities.

enabling those would also keep their history which normally makes little sense.
(except error diagnosis maybe)

closed 2023-10-20 by github-actions[bot]

hksthff, 2024-03-26:
May I ask guys: were you able to integrate with or without Bosch Smart Home Controller?

kfroeschl, 2024-03-26:
without.

with "Bosch Smart Home Controller" would require a different integration (Bosch SHC i guess)

hksthff, 2024-03-26:
Ok thanks. Then I'll try the smoke alarm with my Conbee III and ZHA later. Will provide feedback here.

hksthff, 2024-03-26:
Not successful, pairing didn't work since ZHA was not able to find the smoke alarm. Tested with Conbee III stick.

mnadvornik, 2024-03-26:
I am using a Sonoff Zigbee Bridge connected to Home Assistant ZHA without any issues. The only thing missing is manually triggering the alarm

hksthff, 2024-03-26:
are you using the bridge pro or stick? 
if bridge pro: just directly connected to your computer running HA via usb?

mnadvornik, 2024-03-26:
> are you using the bridge pro or stick? 
> if bridge pro: just directly connected to your computer running HA via usb? 

This one: https://sonoff.tech/product/gateway-and-sensors/zbbridge/
Using Tasmota as Firmware and EZSP protocol

imo the smoke detector should theoretically work with any hardware supported by zigpy / zha.
I don't know if the Bosch smoke detector supports Zigbee 2 but since your Conbee III seems to support Zigbee 3.0 this shouldn't be the issue.
Are you sure you tried the paring with the correct install code printed on the smoke detector?

hksthff, 2024-03-26:
can you maybe please elaborate on this: 

_Pairing worked fine by invoking zha.permit and then connecting the battery_

How do I invoke zha.permit?

mnadvornik, 2024-03-26:
> can you maybe please elaborate on this: 
> 
> _Pairing worked fine by invoking zha.permit and then connecting the battery_
> 
> How do I invoke zha.permit?

It is explained in the documentation: https://www.home-assistant.io/integrations/zha/#service-zhapermit

I haven't found a simple tutorial but it really isn't more than calling the service via the developer tools using the data on the device (or packaging, can't remember). Seems like there is a QR code feature now so if there is a qr code on the smoke detector you can use that instead of the address and install code (don't want to unmount mine from the ceiling to check)

hksthff, 2024-03-27:
After filling in the details, the service won't start: Fehler beim Aufrufen des Diensts zha.permit. Unknown error
👎

kfroeschl, 2024-03-27:
can you paste the ha-logs in the chat?

btw. i don't remember filing out 2 IEEE back then but only one.
(Only "Quelle IEEE" and "Installationscode")

I also did't use the dashes ("-") in the "Installationscode" 
(since i pasted the values from the QR code scanner and deleted the rest)

hksthff, 2024-03-27:
I tested also with qr code only (without quelle IEEE and installationscode). There were several error logs then:

> Logger: homeassistant.helpers.script.websocket_api_script
Quelle: helpers/script.py:1805
Erstmals aufgetreten: 07:40:25 (15 Vorkommnisse)
Zuletzt protokolliert: 07:53:02

websocket_api script: Error executing script. Invalid data for call_service at pos 1: some but not all values in the same group of inclusion 'install_code' @ data[<install_code>]

other error log:

> [547600176832] Error handling message: some but not all values in the same group of inclusion 'install_code' @ data[<install_code>]. Got None (invalid_format) SH9 from 127.0.0.1 (Mozilla/5.0 (Macintosh; Intel Mac OS X 10_15_7) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/123.0.0.0 Safari/537.36)

hksthff, 2024-03-27:
After reading through some github sites, I noted that Conbee sticks simply do not support adding devices via sha.permit yet. 

For further details I link to the threads I found helpful:

https://github.com/dresden-elektronik/deconz-serial-protocol/issues/20#issuecomment-1431599456
https://github.com/Koenkk/zigbee2mqtt/issues/17492

Will order a Sonoff 3.0 USB Dongle Plus (P-version) now to check if the problem is solved then.

puddly, 2024-03-28:
Conbee III doesn't support install codes. Conbee II I believe does.

HaraldGithub, 2024-05-10:
I can't confirm, that Bosch smoke detector are working with ZHA&SkyConnect coordinator.
I've tried two smoke detectors in my HA-installation. ZHA detects the device and the interview starts, but won't finish, so I couldn't use them in my home. Now I'm looking for other Zigbee-Smoke-Detectors which will working with ZHA&SkyConnect.

NightDragon1, 2024-07-08:
I know this one is closed, but for all trying to pair it and land here. I had success pairing it by using the "QR-Code" only.

How I did it: 
1. Scan the QR code at the bottom of the device (the bigger and longer one. If you only see one, it's probably the right one).
2. I went into the developer tools, services and choose "zha.permit".
3. Leave the duration as is at 60 secs and checked
4. checked "QR-Code" and pasted the code 1:1 the way the scan result gave it (no dashes, no blanks, just as a whole long string)
5. Pressed execute --> Resulting in "success"
6. Went to Settings - Devices & Services and got into the "ZHA"
7. Press "Add Device" - the same way as I would any other Zigbee device
8. The smoke detector was found right away and after a few seconds it went to "initializing" That took maybe another 2 min until all was fine and set up. 

I hope that helps others...

Smarthome-Creator, 2024-11-24:
@NightDragon1 , can you tell me about your experience with the smoke detector and ZHA? Are you still using the smoke detector with ZHA and what is the command to query the battery status?

I'm quite new to Zigbee and currently only use a Lidl smart blind with ZHA.

NightDragon1, 2024-11-24:
@Smarthome-Creator Yes I'm still using it. due to it's price I only have one. But plan to buy more asap.
I set it up exactly as I wrote and had no issues so far. Stable connection, but it's not very verbose in general. Guess it's a normal behavior to save battery. In regards of the command - I'm not sure if my value is correct, but it's not that old, so could be it's still 100%. It never had to go off... so...
[image]

You can issue a lot of commands, and it does response to them quite well. Just for now I was looking for the polling time. Don't mask me for the unit. Could be hours - as said it's not very verbose and I see it every here and there sending a "all good"
[image]

NightDragon1, 2024-11-24:
@Smarthome-Creator 
I think this is the command you had been looking for:
[image]

EinzellerXD, 2025-04-23:
While writing a qirk for the Smoke Alarm II, I had problem with the way it exposes its data. Using this [zigbee-herdsman-converter](https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/src/devices/bosch.ts) as a template I tried to implement the quirk for ZHA.
But the smoke detector seems to have a lot of its data in the IasZone Cluster. Not only that, but everything is in the zone_status attribute (ID 0x0002). It seems to be a bitmap with most of the Information we need. But I couldn't figure out how to get the data from this attribute to be exposed in multiple entities in HA at the end.

geeuz, 2025-11-21:
Thanks @NightDragon1, it worked for me too.

Question, is there a way to set the device to silence via Home Assistant?

NightDragon1, 2025-12-24:
> Thanks [@NightDragon1](https://github.com/NightDragon1), it worked for me too.
> 
> Question, is there a way to set the device to silence via Home Assistant?

Thankfully none of my detectors (In total I'm now running already 4) had an alert so far. And until I trigger it or have some real alert and then do some research on the commands, I can't tell you for sure. But I would guess there is a command to do so.

kitor, 2025-12-24:
> Is this normal I don't have Iaszone sensor but rather on/off (binary) sensor with just device name? I tried pairing multiple times, I have battery status, identify button works and I can read IasZone cluster fine (zone type returns FireSensor, status is 48, state - enrolled, etc)

Just a follow-up, no sure how / when this happened but HA now properly sees it as a smoke sensor:

[image]