zigpy/zha-device-handlers #1563: [Device Support Request] Tuya Smart Air Detector 6 in 1

julianjwong, 2022-05-14:
**Is your feature request related to a problem? Please describe.**
Current device support for the Tuya Air Detector 6 in 1 is limited to only temperature and VOC sensors. Other sensors such as PM2.5 and CO2 are missing.
This is the device
https://zigbee.blakadder.com/Tuya_DCR-KQG.html
Supposedly working on Z2M

**Describe the solution you'd like**
Would be great if other sensor entities could be added especially the PM2.5 sensor

<details>
<summary>Device signature - this can be acquired by clicking on the "Zigbee Device Signature" button in the device settings</summary>

```
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0100",
      "in_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0x0402",
        "0x0405",
        "0x040d",
        "0x042b",
        "0x042e",
        "0xef00"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "242": {
      "profile_id": 41440,
      "device_type": "0x0061",
      "in_clusters": [],
      "out_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "_TZE200_dwcarsat",
  "model": "TS0601",
  "class": "zhaquirks.tuya.air.ts0601_air_quality.TuyaCO2SensorGPP"
}
```

</details>

<details>
<summary>Diagnostic information - this can be acquired by clicking on the "Download Diagnostics" button in the device settings</summary>

```
{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2022.5.4",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.9.9",
    "docker": true,
    "arch": "x86_64",
    "timezone": "Australia/Sydney",
    "os_name": "Linux",
    "os_version": "5.10.108",
    "supervisor": "2022.05.1",
[... 453 lines cut ...]
      },
      {
        "name": "unknown 97 device_type of 0xa1e0 profile id"
      }
    ],
    "user_given_name": "Air quality sensor",
    "device_reg_id": "7746e4f45890650bf6cf8d5f6db31638",
    "area_id": "toy_room"
  }
}
```

</details>

<details>
<summary>Additional logs</summary>

```
Paste any additional debug logs here.
```

</details>

**Additional context**
Add any other context or screenshots about the feature request here.

javicalle, 2022-05-15:
The device has been tunned in `zigbee-herdsman-converters` library:
* https://github.com/Koenkk/zigbee-herdsman-converters/blob/ce119a6ca14033880dca8bf83cd01bc928ab5657/converters/fromZigbee.js#L4334

This could be a quirk for the device:
<details>
<summary>ts0601_air_quality.py</summary>

```python
"""Tuya Air sensors."""

from typing import Dict

import zigpy.types as t
from zigpy.zcl.clusters.measurement import (
    CarbonDioxideConcentration,
    FormaldehydeConcentration,
    PM25,
    RelativeHumidity,
    TemperatureMeasurement,
)

from zhaquirks.tuya import DPToAttributeMapping, TuyaLocalCluster, TuyaNewManufCluster

class TuyaAirQualityVOC(TuyaLocalCluster):
    """Tuya VOC level cluster."""

    cluster_id = 0x042E
    name = "VOC Level"
    ep_attribute = "voc_level"

    attributes = {
        0x0000: ("measured_value", t.Single),  # fraction of 1 (one)
        0x0001: ("min_measured_value", t.Single),
        0x0002: ("max_measured_value", t.Single),
        0x0003: ("tolerance", t.Single),
    }

    server_commands = {}
    client_commands = {}

class TuyaAirQualityTemperature(TemperatureMeasurement, TuyaLocalCluster):
    """Tuya temperature measurement."""

class TuyaAirQualityHumidity(RelativeHumidity, TuyaLocalCluster):
    """Tuya relative humidity measurement."""

class TuyaAirQualityCO2(CarbonDioxideConcentration, TuyaLocalCluster):
    """Tuya Carbon Dioxide concentration measurement."""

class TuyaAirQualityFormaldehyde(FormaldehydeConcentration, TuyaLocalCluster):
    """Tuya Formaldehyde concentration measurement."""

class TuyaAirQualityPM25(PM25, TuyaLocalCluster):
    """Tuya PM25 concentration measurement."""

class TuyaCO2ManufCluster(TuyaNewManufCluster):
    """Tuya with Air quality data points."""

    dp_to_attribute: Dict[int, DPToAttributeMapping] = {
        2: DPToAttributeMapping(
            TuyaAirQualityCO2.ep_attribute,
            "measured_value",
            lambda x: x * 1e-6,
        ),
        18: DPToAttributeMapping(
            TuyaAirQualityTemperature.ep_attribute, "measured_value", lambda x: x * 10
        ),
        19: DPToAttributeMapping(
            TuyaAirQualityHumidity.ep_attribute, "measured_value", lambda x: x * 10
        ),
        21: DPToAttributeMapping(
            TuyaAirQualityVOC.ep_attribute, "measured_value", lambda x: x * 1e-6
        ),
        22: DPToAttributeMapping(
            TuyaAirQualityFormaldehyde.ep_attribute,
            "measured_value",
            lambda x: x * 1e-6,
        ),
    }

    data_point_handlers = {
        2: "_dp_2_attr_update",
        18: "_dp_2_attr_update",
        19: "_dp_2_attr_update",
        21: "_dp_2_attr_update",
        22: "_dp_2_attr_update",
    }

class TuyaCO2ManufCluster_2(TuyaNewManufCluster):
    """Tuya with Air quality device v2 with data points."""

    dp_to_attribute: Dict[int, DPToAttributeMapping] = {
        2: DPToAttributeMapping(
            TuyaAirQualityPM25.ep_attribute,
            "measured_value",
            lambda x: None if (value === 0xaaac || value === 0xaaab) else x,
        ),
        18: DPToAttributeMapping(
            TuyaAirQualityTemperature.ep_attribute, "measured_value", lambda x: x * 10
        ),
        19: DPToAttributeMapping(
            TuyaAirQualityHumidity.ep_attribute, "measured_value", lambda x: x * 10
        ),
        20: DPToAttributeMapping(
            TuyaAirQualityFormaldehyde.ep_attribute,
            "measured_value",
            lambda x: x * 1e-6,
        ),
        21: DPToAttributeMapping(
            TuyaAirQualityVOC.ep_attribute, "measured_value", lambda x: x * 1e-6
        ),
        22: DPToAttributeMapping(
            TuyaAirQualityCO2.ep_attribute,
            "measured_value",
            lambda x: x * 1e-6,
        ),
    }

    data_point_handlers = {
        2: "_dp_2_attr_update",
        18: "_dp_2_attr_update",
        19: "_dp_2_attr_update",
        20: "_dp_2_attr_update",
        21: "_dp_2_attr_update",
        22: "_dp_2_attr_update",
    }
```

</details>

julianjwong, 2022-05-15:
Sorry looks like the PM2.5 sensor code is not right.

```
Logger: homeassistant.config_entries
Source: components/zha/__init__.py:99
First occurred: 12:15:54 AM (1 occurrences)
Last logged: 12:15:54 AM

Error setting up entry Sonoff Zigbee 3.0 USB Dongle Plus, s/n: 0a28bcebc512ec11892b21c7bd930c07 - ITead for zha
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 335, in async_setup
    result = await component.async_setup_entry(hass, self)
  File "/usr/src/homeassistant/homeassistant/components/zha/__init__.py", line 99, in async_setup_entry
    setup_quirks(config)
  File "/usr/local/lib/python3.9/site-packages/zhaquirks/__init__.py", line 409, in setup
    importer.find_module(modname).load_module(modname)
  File "<frozen importlib._bootstrap_external>", line 529, in _check_name_wrapper
  File "<frozen importlib._bootstrap_external>", line 1029, in load_module
  File "<frozen importlib._bootstrap_external>", line 854, in load_module
  File "<frozen importlib._bootstrap>", line 274, in _load_module_shim
  File "<frozen importlib._bootstrap>", line 711, in _load
  File "<frozen importlib._bootstrap>", line 680, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 846, in exec_module
  File "<frozen importlib._bootstrap_external>", line 983, in get_code
  File "<frozen importlib._bootstrap_external>", line 913, in source_to_code
  File "<frozen importlib._bootstrap>", line 228, in _call_with_frames_removed
  File "/config/quirks/ts0601_air_quality.py", line 96
    lambda x: None if (value === 0xaaac || value === 0xaaab) else x,
                               ^
SyntaxError: invalid syntax

```

javicalle, 2022-05-15:
Can you try replacing this expression:
```python
lambda x: None if (value == 0xaaac || value == 0xaaab) else x,
```

julianjwong, 2022-05-16:
Thanks but new issue

```
Logger: homeassistant.config_entries
Source: components/zha/__init__.py:99
First occurred: 11:49:22 AM (1 occurrences)
Last logged: 11:49:22 AM

Error setting up entry Sonoff Zigbee 3.0 USB Dongle Plus, s/n: 0a28bcebc512ec11892b21c7bd930c07 - ITead for zha
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 335, in async_setup
    result = await component.async_setup_entry(hass, self)
  File "/usr/src/homeassistant/homeassistant/components/zha/__init__.py", line 99, in async_setup_entry
    setup_quirks(config)
  File "/usr/local/lib/python3.9/site-packages/zhaquirks/__init__.py", line 409, in setup
    importer.find_module(modname).load_module(modname)
  File "<frozen importlib._bootstrap_external>", line 529, in _check_name_wrapper
  File "<frozen importlib._bootstrap_external>", line 1029, in load_module
  File "<frozen importlib._bootstrap_external>", line 854, in load_module
  File "<frozen importlib._bootstrap>", line 274, in _load_module_shim
  File "<frozen importlib._bootstrap>", line 711, in _load
  File "<frozen importlib._bootstrap>", line 680, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 846, in exec_module
  File "<frozen importlib._bootstrap_external>", line 983, in get_code
  File "<frozen importlib._bootstrap_external>", line 913, in source_to_code
  File "<frozen importlib._bootstrap>", line 228, in _call_with_frames_removed
  File "/config/quirks/ts0601_air_quality.py", line 96
    lambda x: None if (value == 0xaaac || value == 0xaaab) else x,
                                        ^
SyntaxError: invalid syntax

```

julianjwong, 2022-05-16:
Ok so I tried replacing || with or and that seems to have gotten it out of the errors. But for some reason the device isn't using the custom quirk.
Have this in my config and I know that works hence why it was throwing errors previously

```
zha:
  enable_quirks: true
  custom_quirks_path: /config/quirks
```

javicalle, 2022-05-16:
I have made a mess trying to put everything in a single file.
Could you try with the new version (from the previous comment)?

Delete any `__pycache__` folder in your `custom_quirks_path` and restart HA.

julianjwong, 2022-05-16:
Cool thanks looks like it loaded this time as denoted by Quirk: ts0601_air_quality.TuyaCO2SensorGPP_2.
Even the previous standard quirk loaded VOC, temperature, CO2, formaldehyde, and humidity (but didn't when I first looked) but unfortunately this custom quirk doesnt add a PM2.5 sensor entity. Should it show up automatically? Or do I need to remove the device and pair it again?

javicalle, 2022-05-16:
> formaldehyde dropped from 360ppm to 4ppm

You can try to change in your quirk this part:
```python
        20: DPToAttributeMapping(
            TuyaAirQualityFormaldehyde.ep_attribute,
            "measured_value",
            lambda x: x * 1e-3,
        ),
```
There's any way to validate wich values are the correct ones?

The conversion factors are probably need to be revised. Tell me if there is any other value that seems strange to you.

I think that I saw that PM2.5 does not generate an entity in HA, but you should be able to query it as attributes of the `TuyaAirQualityPM25` cluster.

> crazy co2 spikes may have been eliminated which is good

Well, actually these spikes seems that aren't CO2 values but PM2.5 values (and had been filtered).

The modifications would be:
* temp, hum & VOC keep unchanged
* the previous CO2 readings (with spikes) readings were actually PM2.5 readings
* the previous Formaldehyde readings were actually CO2 readings
* the current Formaldehyde readings weren't previouslly readed

In Z2M there are a table with units and range values for this device:
* https://github.com/Koenkk/zigbee2mqtt/issues/11033#issuecomment-1124056947

julianjwong, 2022-05-16:
Ok here's my values vs the table. Looks like some mismatched ones. 
[image]
[image]

afaucogney, 2022-05-16:
Hey guys, I bought this device too, but it even does not pair ! Have you got some issue about that by your side ?

```
Info 2022-05-16 21:50:00Device '0xa4c13821495fcb38' joined
Info 2022-05-16 21:50:01MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xa4c13821495fcb38","ieee_address":"0xa4c13821495fcb38"},"type":"device_joined"}'
Info 2022-05-16 21:50:01MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"friendly_name":"0xa4c13821495fcb38"},"type":"device_connected"}'
Info 2022-05-16 21:50:01Starting interview of '0xa4c13821495fcb38'
Info 2022-05-16 21:50:02MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xa4c13821495fcb38","ieee_address":"0xa4c13821495fcb38","status":"started"},"type":"device_interview"}'
Info 2022-05-16 21:50:02MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_started","meta":{"friendly_name":"0xa4c13821495fcb38"},"type":"pairing"}'
Error 2022-05-16 21:52:19Failed to interview '0xa4c13821495fcb38', device has not successfully been paired
```

javicalle, 2022-05-17:
Apart from the temperature and humidity, the rest of the values I have no way of validating them.
Values for CO2 may be in the correct range values.

It seems that the units for VOC would not be correct (the device measures in ppm and HA shows ug/m3). Maybe needs to be adjusted here:
* https://github.com/home-assistant/core/blob/513e276bbaba4c1d750cad4253db5c79fb341ec5/homeassistant/components/zha/sensor.py#L570-L576

The rest of the measures also seem to need to be adjusted. 
According to these tables:
* https://github.com/Koenkk/zigbee2mqtt/issues/11033#issuecomment-1110717475

values for Formaldehyde seems to be too high, and also with the diferent units. There is no alternative sensor in HA, so maybe a units conversion can be needed:
* https://github.com/home-assistant/core/blob/513e276bbaba4c1d750cad4253db5c79fb341ec5/homeassistant/components/zha/sensor.py#L586-L594

julianjwong, 2022-05-17:
So I have another sensor in a different room in the same house but presumably similar readings.
Here's what it shows
[image]
Ignore temperature and humidity values as the heater is on in that room

javicalle, 2022-05-17:
Ummm, it seems that CO2 and VOC values are fine (VOC: 15µg/m3 ~ 0,012mg/m3)

The measuring range of the device seems to be 0-10µg/m3 (possibly without decimals)
HA assumes that Formaldehyde units are PPM with a multiplier of 1e6, and displays values without decimals:
* https://github.com/home-assistant/core/blob/513e276bbaba4c1d750cad4253db5c79fb341ec5/homeassistant/components/zha/sensor.py#L587-L594

Here I don't know what should be the correct solution.
We can try to do a ug/m3 --> PPM conversion in the quirk so that the information displayed in HA is correct. Or propose a change in HA so that it also accepts PPM as units of measurement for Formaldehyde 🤷🏻‍♂️ 

Any suggestion is welcome.

julianjwong, 2022-05-17:
I'd say that given these two sensors both have different units it's probably best to suggest a change in HA for the units?

dmulcahey, 2022-05-17:
We had the same issue with VOC in HA. we can create a ppm formaldehyde entity similar to how we did for VOC

maxperron, 2022-05-19:
#1406 

I believe it's the same devices and issues

drmulligan, 2022-10-05:
#1707 is also related.

julianjwong, 2022-10-05:
Not sure when this happened but pm2.5 is now showing for me. Has spikes to over 40000 but generally working

vikitor87, 2022-12-17:
Today I have implemented the quirk for this sensor, and it shows everything apparently correct except PM2.5. 
[image]

In the HomeAssistant logs I get this error.
```
Logger: homeassistant
Source: zha_quirks/ts0601_air_quality.py:158
First occurred: 19:22:42 (1529 occurrences)
Last logged: 19:35:22

Error doing job: Task exception was never retrieved
Traceback (most recent call last):
  File "/usr/local/lib/python3.10/site-packages/zigpy_znp/zigbee/application.py", line 588, in on_af_message
    self.packet_received(
  File "/usr/local/lib/python3.10/site-packages/zigpy/application.py", line 855, in packet_received
    self.handle_message(
  File "/usr/local/lib/python3.10/site-packages/zigpy/application.py", line 361, in handle_message
    return sender.handle_message(
  File "/usr/local/lib/python3.10/site-packages/zigpy/device.py", line 370, in handle_message
    return endpoint.handle_message(
  File "/usr/local/lib/python3.10/site-packages/zigpy/endpoint.py", line 224, in handle_message
    handler(hdr, args, dst_addressing=dst_addressing)
  File "/usr/local/lib/python3.10/site-packages/zigpy/zcl/__init__.py", line 373, in handle_message
    self.handle_cluster_request(hdr, args, dst_addressing=dst_addressing)
  File "/usr/local/lib/python3.10/site-packages/zhaquirks/tuya/__init__.py", line 1431, in handle_cluster_request
    status = getattr(self, handler_name)(*args)
  File "/usr/local/lib/python3.10/site-packages/zhaquirks/tuya/__init__.py", line 1449, in handle_get_data
    getattr(self, dp_handler)(record)
  File "/usr/local/lib/python3.10/site-packages/zhaquirks/tuya/__init__.py", line 1482, in _dp_2_attr_update
    value = dp_map.converter(value)
  File "/config/zha_quirks/ts0601_air_quality.py", line 158, in <lambda>
    lambda x: None if (value == 0xaaac or value == 0xaaab) else x,
NameError: name 'value' is not defined
```
Any clue to get the PM2.5 values? I have observed that in Z2M it is possible, and there is a thread about it.
[Added new model Tuya Air quality sensor TS0601 _TZE200_dwcarsat](https://github.com/Koenkk/zigbee-herdsman-converters/pull/3745/files)
Thanks.

javicalle, 2022-12-17:
Replace this:
```python
lambda x: None if (value == 0xaaac or value == 0xaaab) else x
```

With:
```python
lambda x: None if (x == 0xaaac or x == 0xaaab) else x
```

vikitor87, 2022-12-17:
You've got it. Now the value appears, perfect! I think it should be divided by 10, or if the value is true (from 65 to 75 ug/m3), I have a short time to live.

javicalle, 2022-12-17:
Then you can try with;
```python
lambda x: None if (x == 0xaaac or x == 0xaaab) else x/10
```

ylemoigne, 2022-12-18:
@vikitor87 Can you share the quirk file you've implemented ?

vikitor87, 2022-12-18:
Of course! @ylemoigne . It is the same quirk file published at the beginning but modifying code line 158
<details><summary>ts0601_air_quality.py</summary>
<p>

```
"""Tuya Air Quality sensor."""

from typing import Dict

from zigpy.profiles import zha
from zigpy.quirks import CustomDevice
from zigpy.zcl.clusters.general import Basic, GreenPowerProxy, Groups, Ota, Scenes, Time

from zigpy.zcl.clusters.measurement import PM25

from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODELS_INFO,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
)

from zhaquirks.tuya import DPToAttributeMapping, TuyaLocalCluster, TuyaNewManufCluster

from zhaquirks.tuya.air import (
    TuyaAirQualityCO2,
    TuyaAirQualityFormaldehyde,
    TuyaAirQualityHumidity,
    TuyaAirQualityTemperature,
    TuyaAirQualityVOC,
    TuyaCO2ManufCluster,
)

class TuyaCO2Sensor(CustomDevice):
    """Tuya Air quality device."""

    signature = {
        # NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.0: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)]
        # device_version=1
        # SizePrefixedSimpleDescriptor(endpoint=1, profile=260, device_type=81, device_version=1,
        # input_clusters=[0, 4, 5, 61184],
        # output_clusters=[25, 10])
        MODELS_INFO: [
            ("_TZE200_8ygsuhe1", "TS0601"),
            ("_TZE200_yvx5lh6k", "TS0601"),
        ],
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.SMART_PLUG,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaCO2ManufCluster.cluster_id,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            }
        },
    }

    replacement = {
        ENDPOINTS: {
            1: {
                DEVICE_TYPE: zha.DeviceType.ON_OFF_LIGHT,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
[... 166 lines cut ...]
                    TuyaAirQualityPM25,
                    TuyaAirQualityCO2,
                    TuyaAirQualityFormaldehyde,
                    TuyaAirQualityHumidity,
                    TuyaAirQualityTemperature,
                    TuyaAirQualityVOC,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            },
            242: {
                PROFILE_ID: 41440,
                DEVICE_TYPE: 97,
                INPUT_CLUSTERS: [],
                OUTPUT_CLUSTERS: [GreenPowerProxy.cluster_id],
            },
        }
    }
```
</p>
</details>

· Entities

[image]
[image]

gudvinr, 2022-12-18:
Excluding `0xaaac` and `0xaaab` is not enough, over the time you'll notice that it produces some very off 1 values that will be normal on next measurement

ylemoigne, 2022-12-18:
@vikitor87 thanks, it works (well at least it looks better now, i don't know the accuracy of the device)

lonnie776, 2022-12-30:
So after much head scratching and googling, I was able to get my 6 sensors to show up using ts0601_air_quality.py from above. However now I am fairly certain Formaldehyde, PM2.5, and VOC are all incorrect and will require some other scaling factor. From my brief research my values are out of the normal for indoor conditions. I have also noticed the Unit of Measure is incorrect for VOC and Formaldehyde, but this can be fixed with Customize. 

Has anyone been able to get sane readings from Tuya and compared them to the values reported by Home Assistant?

javicalle, 2022-12-30:
From the various reports that I have found of this type of sensor, I think that there are at least 2 different types of sensors with mixed sensors/DPs.
Look for a quirk for your exact `manufacturer` to be sure.

RSDynamics, 2023-04-08:
Not a lot of action here but I have the TS0601 _TZE200_dwcarsat
I copied this quirk as the values seems to be mixed up. 
Values seem to be at the right place now but the accuracy of the sensor is not very high. 
An almost constant humidity of 105% and CO2 level of around 362ppm are just impossible.
PM25 is spiking to 4000 at random times.
Is there a way to add an offset to the values?
CO2 is about 100ppm to low and humidity is 40% to high.

RSDynamics, 2023-04-12:
No, did not had the time yet. 
What I do think it is a different issue than a offset. The humidity stays within about 1% so of the 105% so that is also not very realistic.
I think temperature is in the same component and is reliable so not sure what the issue is at the moment.

codyrocco, 2023-06-23:
still no news? 
anyway, event on z2m, co2 values seems to be...fixed, at around 350 :)))
attached are values from zha... i should be very dead now!

[image]

martin3000, 2023-07-15:
I added this entry to my configuration.yaml:
zha:
..  custom_quirks_path: .homeassistant/custom_zha_quirks/

(with HA core)

I stored the ts0601_air_quality.py in this directory. The formaldehyde values are better now, but with 40ppm outside on the balcony still to high. I think it should not be more than 0.1ppm.

martin3000, 2023-07-16:
<!DOCTYPE html>
<html>
<body>

  | device unit | HA unit | conv. fact. | ok if
-- | -- | -- | -- | --
CO2 | ppm | ppm |   | <1000 ppm (outside air: 400ppm)
Formaldehyde | µg/m³ | ppm | /1240 | <0.1 ppm
Pm2.5 | µg/m³ | µg/m³ |   | <50 µg/m³
VOC | ppm | µg/m³ | *2500 | <3000 µg/m³
Temperature | °C | °C |   |  
Humidity | % | % |   |  
</body>
</html>

martin3000, 2023-07-17:
By comparing the values with the tuya app, I have now found the following:

```
    dp_to_attribute: Dict[int, DPToAttributeMapping] = {
        2: DPToAttributeMapping(
            TuyaAirQualityPM25.ep_attribute, "measured_value",  lambda x: None if (x>5000) else x
        ),
        18: DPToAttributeMapping(
            TuyaAirQualityTemperature.ep_attribute, "measured_value", lambda x: x * 10
        ),
        19: DPToAttributeMapping(
            TuyaAirQualityHumidity.ep_attribute, "measured_value", lambda x: x * 10
        ),
        20: DPToAttributeMapping(
           TuyaAirQualityFormaldehyde.ep_attribute, "measured_value", lambda x: x * 1e-9
        ),
        21: DPToAttributeMapping(
            TuyaAirQualityVOC.ep_attribute, "measured_value", lambda x: x * 1e-6
        ),
        22: DPToAttributeMapping(
            TuyaAirQualityCO2.ep_attribute, "measured_value", lambda x: x * 1e-6,
        ),
    }
```
Maybe it is better to change the formaldehyde unit from ppm to ppb and leave the factor as 1e-6.

martin3000, 2023-07-18:
VOC is displayed in the tuya app with the unit mg/m³, in the HA tuya integration it shows up as ppm. The numbers seem to be the same. If zhaquirks sends a value of 10, the tuya app shows it as 0,010 mg/m³ (would be 10 µg/m³) and the tuya HA integration shows it as 0,010 ppm (would be 25 µg/m³).

Abdull, 2024-01-18:
/unstale

frog32, 2024-04-05:
The combined work of folks in this thread already results in a much better quirk than is currently available from zha-device-handlers. Would any of the contributors mind submitting a pull request so less tech-savvy people can use it as well and are not getting garbage?

Eis-T, 2024-09-07:
Hi, I'm trying to get this device to work and it turns out I don't have the skills yet to get this sensor to work in HA. Can somebody who figured this one out post a config / quirk here? First time working with quirks so I might be making obvious mistakes.

martin3000, 2024-09-07:
Just pair it in ZHA and it will work. You don't need any skills.

darth-hp, 2024-11-25:
> By comparing the values with the tuya app, I have now found the following:
> 
> ```
>     dp_to_attribute: Dict[int, DPToAttributeMapping] = {
>         2: DPToAttributeMapping(
>             TuyaAirQualityPM25.ep_attribute, "measured_value",  lambda x: None if (x>500) else x
>         ),
>         18: DPToAttributeMapping(
>             TuyaAirQualityTemperature.ep_attribute, "measured_value", lambda x: x * 10
>         ),
>         19: DPToAttributeMapping(
>             TuyaAirQualityHumidity.ep_attribute, "measured_value", lambda x: x * 10
>         ),
>         20: DPToAttributeMapping(
>            TuyaAirQualityFormaldehyde.ep_attribute, "measured_value", lambda x: x * 1e-6
>         ),
>         21: DPToAttributeMapping(
>             TuyaAirQualityVOC.ep_attribute, "measured_value", lambda x: x * 1e-6
>         ),
>         22: DPToAttributeMapping(
>             TuyaAirQualityCO2.ep_attribute, "measured_value", lambda x: x * 1e-6,
>         ),
>     }
> ```
> 
> The formaldehyde unit must be changed from ppm to ppb (customize: section).

Hi Martin,

How do I get this to work for ZHA?

Kind regards

closed 2025-02-17 by prairiesnpr

martin3000, 2025-02-25:
Every few minutes, the pm2.5 sensor sends the value 0xAAAC (43692)
For my TZE200_dwcarsat I had a custom quirks with this code:
   lambda x: None if (x == 0xaaac or x == 0xaaab or x == 0xaa00 or x>5000) else x
or alternatively:
   lambda x: None if (x>500) else x
This removed the strange peak values.