zigpy/zha-device-handlers #1564: [Device Support Request] Aqara Smart Light Switch (No Neutral, Double Rocker)

aallen90, 2022-05-14:
This is a wall switch with two buttons. I have a few of these devices wired to control combination ceiling light/fan devices (1 button toggles the light, and the other toggles the fan).

**Is your feature request related to a problem? Please describe.**

Basic device controls work when paired to HA (on/off), but HA reports inaccurate device and diagnostic information.
 Since these are mains-powered, I think they may have the ability to act as a "Router", but I suspect the lack of a matching quirk causes the  `Power Source` to show "Battery or Unknown" and act as an "EndDevice". 

Additionally, when pairing the device to HA, it reports the two switch endpoints as light entities.

**Describe the solution you'd like**

A to quirk match the device and translate cluster attributes correctly.

- Zigbee Device Type: `EndDevice` (Should be `Router`?)
- Power Source: `Battery or Unknown` (Should be `Mains`? It's powered by 120V AC)
- device_temperature: reports `32.4 °F` or `0.2 °C`, which suggests the default unit of measure is incorrect (It's summertime here, so reported temp is far too cold)
-  ZHA device type name: `lumi.light.b2laus01` (should be `lumi.switch.b2laus01`)
   - I corrected this in my instance by setting the `zha.device_config` entry to `switch`, then used "Switch as X" to have the `light` and `fan` entities for my use-case, but `switch` should be the default.

<details>
<summary>Device signature</summary>

```json
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4447, maximum_buffer_size=127, maximum_incoming_transfer_size=100, server_mask=11264, maximum_outgoing_transfer_size=100, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0100",
      "in_clusters": [
        "0x0000",
        "0x0002",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0009"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "2": {
      "profile_id": 260,
      "device_type": "0x0100",
      "in_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006"
      ],
      "out_clusters": []
    },
    "242": {
      "profile_id": 41440,
      "device_type": "0x0061",
      "in_clusters": [],
      "out_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "LUMI",
  "model": "lumi.switch.b2laus01",
  "class": "zigpy.device.Device"
}
```

[... 406 lines cut ...]
[2031:1:0x0002]: 'async_initialize' stage succeeded
[2031:1:0x0000]: 'async_initialize' stage succeeded
[2031:1:0x0003]: 'async_initialize' stage succeeded
[2031:1:0x0019]: 'async_initialize' stage succeeded
[0x07EF:2:0x0006] Received ZCL frame: b'\x18g\x01\x00\x00\x00\x10\x00'
[0x07EF:2:0x0006] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=0, is_reply=1, disable_default_response=1, reserved=0, *is_cluster=False, *is_general=True), tsn=103, command_id=1, *is_reply=True)
[0x07EF:2:0x0006] Decoded ZCL frame: OnOff:Read_Attributes_rsp(status_records=[ReadAttributeRecord(attrid=0x0000, status=<Status.SUCCESS: 0>, value=TypeValue(type=Bool, value=<Bool.false: 0>))])
[2031:2:0x0006]: finished channel initialization
[2031:2:0x0006]: 'async_initialize' stage succeeded
[2031:2:0x0000]: 'async_initialize' stage succeeded
[2031:2:0x0003]: 'async_initialize' stage succeeded
[2031](lumi.switch.b2laus01): power source: Battery or Unknown
[2031](lumi.switch.b2laus01): completed initialization
[0x4DA5](Daylight): Device seen - marking the device available and resetting counter
[0x4DA5](Daylight): Update device availability -  device available: True - new availability: True - changed: False
[0x17A4](lumi.switch.b2laus01): last_seen is 62581.99680876732 seconds ago and ping attempts have been exhausted, marking the device unavailable
[0x17A4](lumi.switch.b2laus01): Update device availability -  device available: False - new availability: False - changed: False

```

</details>

**Additional context**

[image]

aallen90, 2022-05-29:
My initial device setup logs were were re-adding an already initialized device. I've since removed & added the device, and updated the initial post with the logs captured.

I'm trying to write a quirk for this, though I'm struggling to find a good reference handler to base this on. I also don't use Python normally, so I'm ironing out my environment setup.

The `CtrlLn` and `CtrlNeutral` classes seemed like a good starting point, since the inline comments say "Aqara single and double key switch device", but the endpoints are wildly different. Adding "lumi.switch.b2laus01" to the MODELS_INFO array won't match. I believe this device is a newer model than those. `PlugMAEU01` looks fairly similar, just for a single switch.

I also did some digging and found that the UK version of this device (requested in issue #1134) has the same device signature, so if I can get this working well, it's likely both models can leverage the quirk.

cobrapower, 2022-12-03:
Hello ! I have the same issue with the EU version, I would like to know if the quirk you make have some updates since your last comment. I'm not on HA since long enough to know how to help but I would enjoy to get into it if you need to.

closed 2023-06-08 by github-actions[bot]