zigpy/zha-device-handlers #1568: [Device Support Request] _TZE200_nklqjk62

Stejsce, 2022-05-17:
I would like to ask for addition of device PJ-ZGD01 (Garage door controlle).

<details>
<summary>Device signature </summary>

{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0051",
      "in_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0xef00"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "242": {
      "profile_id": 41440,
      "device_type": "0x0061",
      "in_clusters": [],
      "out_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "_TZE200_nklqjk62",
  "model": "TS0601",
  "class": "zigpy.device.Device"
}

</details>

<details>
<summary> Diagnostic information </summary>

 "data": {
    "ieee": "**REDACTED**",
    "nwk": 18163,
    "manufacturer": "_TZE200_nklqjk62",
    "model": "TS0601",
    "name": "_TZE200_nklqjk62 TS0601",
    "quirk_applied": false,
    "quirk_class": "zigpy.device.Device",
    "manufacturer_code": 4417,
    "power_source": "Mains",
    "lqi": 66,
    "rssi": null,
    "last_seen": "2022-05-17T22:54:24",
    "available": true,
    "device_type": "Router",
    "signature": {
      "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
      "endpoints": {
        "1": {
          "profile_id": 260,
          "device_type": "0x0051",
          "in_clusters": [
            "0x0000",
            "0x0004",
            "0x0005",
            "0xef00"
          ],
          "out_clusters": [
            "0x000a",
            "0x0019"
          ]
        },
        "242": {
          "profile_id": 41440,
          "device_type": "0x0061",
          "in_clusters": [],
          "out_clusters": [
            "0x0021"
          ]
        }
      }
    }

</details>

javicalle, 2022-05-17:
Duplicate of #1260

javicalle, 2022-05-28:
@Stejsce I have proposed a quirk for your device:
* https://github.com/zigpy/zha-device-handlers/issues/1260#issuecomment-1140250505

Can we continue the conversation in the other thread and close this one?

Thanks for your understanding.

Stejsce, 2022-05-31:
> @Stejsce I have proposed a quirk for your device:
> 
> * [[Device Support Request] TS0601 GDC311ZBQ1 Tuya Loratap Garage Switcher #1260 (comment)](https://github.com/zigpy/zha-device-handlers/issues/1260#issuecomment-1140250505)
> 
> Can we continue the conversation in the other thread and close this one?
> 
> Thanks for your understanding.

Hi @javicalle , no problem. Thank you for solving this request.

javicalle, 2022-05-31:
It would be nice if you could validate the proposed quirk.

Regards.

closed 2022-05-31 by javicalle