zigpy/zha-device-handlers #1569: [BUG] Tradfri E1524 Remote Automation Triggers Incorrect.

TheDevFreak, 2022-05-18:
**Describe the bug**
The triggers registered in hass automations do not work. checked the quirk it's loading (zhaquirks.ikea.fivebtnremotezha.IkeaTradfriRemote1) and the quirk is expecting, for example: `PARAMS: {"move_mode": 0, "rate": 83},` for a long press of dim up, however when i long press dim up the event I get has `"rate": 84`. Seems to be same for long press of dim down.

**Expected behavior**
The automation triggers are correct

<details>
<summary>Device signature - this can be acquired by clicking on the "Zigbee Device Signature" button in the device settings</summary>

```
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4476, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0820",
      "in_clusters": [
        "0x0000",
        "0x0001",
        "0x0003",
        "0x0020",
        "0x1000",
        "0xfc7c"
      ],
      "out_clusters": [
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0008",
        "0x0019",
        "0x1000"
      ]
    }
  },
  "manufacturer": "IKEA of Sweden",
  "model": "TRADFRI remote control",
  "class": "zhaquirks.ikea.fivebtnremotezha.IkeaTradfriRemote1"
}
```

</details>

<details>
<summary>Diagnostic information - this can be acquired by clicking on the "Download Diagnostics" button in the device settings</summary>

```
{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2022.5.4",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.9.9",
    "docker": true,
    "arch": "x86_64",
    "timezone": "Europe/London",
    "os_name": "Linux",
    "os_version": "5.10.108",
    "supervisor": "2022.05.2",
    "host_os": "Home Assistant OS 7.6",
    "docker_version": "20.10.9",
    "chassis": "vm",
    "run_as_root": true
  },
  "custom_components": {
    "nodered": {
      "version": "1.0.8",
      "requirements": []
    },
    "battery_sim": {
      "version": "1.0",
      "requirements": []
    },
    "dyson_cloud": {
      "version": "0.15.0",
      "requirements": [
        "libdyson==0.8.7"
      ]
    },
    "adaptive_lighting": {
      "version": "1.0.14",
      "requirements": []
    },
    "scheduler": {
      "version": "v0.0.0",
      "requirements": []
    },
    "browser_mod": {
      "version": "1.5.3",
      "requirements": []
    },
    "ocpp": {
      "version": "0.4.31",
      "requirements": [
        "ocpp>=0.14.1",
        "websockets>=10.2"
      ]
    },
    "dyson_local": {
      "version": "0.16.4-4",
      "requirements": [
        "libdyson==0.8.11"
      ]
    },
    "pyscript": {
      "version": "1.3.3",
      "requirements": [
        "croniter==1.3.4",
        "watchdog==2.1.6"
      ]
    },
    "lovelace_gen": {
      "version": "0.1.1",
      "requirements": [
        "jinja2"
      ]
    },
    "blitzortung": {
      "version": "1.0.1",
      "requirements": [
        "paho-mqtt>=1.5.0"
      ]
    },
    "circadian_lighting": {
      "version": "2.0.8",
      "requirements": []
    },
    "hacs": {
      "version": "1.24.5",
      "requirements": [
        "aiogithubapi>=21.11.0"
      ]
    },
    "webrtc": {
      "version": "v2.3.0",
      "requirements": []
    },
    "hass_agent_notifier": {
      "version": "2022.3.15",
      "requirements": []
    },
    "templatesensor": {
      "version": "0.1.3",
      "requirements": []
    }
  },
  "integration_manifest": {
    "domain": "zha",
    "name": "Zigbee Home Automation",
    "config_flow": true,
    "documentation": "https://www.home-assistant.io/integrations/zha",
    "requirements": [
      "bellows==0.29.0",
      "pyserial==3.5",
      "pyserial-asyncio==0.6",
      "zha-quirks==0.0.73",
      "zigpy-deconz==0.16.0",
      "zigpy==0.45.1",
      "zigpy-xbee==0.14.0",
      "zigpy-zigate==0.7.4",
      "zigpy-znp==0.7.0"
    ],
    "usb": [
      {
        "vid": "10C4",
        "pid": "EA60",
        "description": "*2652*",
        "known_devices": [
          "slae.sh cc2652rb stick"
        ]
      },
      {
        "vid": "10C4",
        "pid": "EA60",
        "description": "*sonoff*plus*",
        "known_devices": [
          "sonoff zigbee dongle plus"
        ]
      },
      {
        "vid": "10C4",
        "pid": "EA60",
        "description": "*tubeszb*",
        "known_devices": [
          "TubesZB Coordinator"
        ]
      },
      {
        "vid": "1A86",
        "pid": "7523",
        "description": "*tubeszb*",
        "known_devices": [
          "TubesZB Coordinator"
        ]
      },
      {
        "vid": "1A86",
        "pid": "7523",
        "description": "*zigstar*",
        "known_devices": [
          "ZigStar Coordinators"
        ]
      },
      {
        "vid": "1CF1",
        "pid": "0030",
        "description": "*conbee*",
        "known_devices": [
          "Conbee II"
        ]
      },
      {
        "vid": "10C4",
        "pid": "8A2A",
        "description": "*zigbee*",
        "known_devices": [
          "Nortek HUSBZB-1"
        ]
      },
      {
        "vid": "10C4",
        "pid": "8B34",
        "description": "*bv 2010/10*",
        "known_devices": [
          "Bitron Video AV2010/10"
        ]
      }
    ],
    "codeowners": [
      "@dmulcahey",
      "@adminiuga"
    ],
    "zeroconf": [
      {
        "type": "_esphomelib._tcp.local.",
        "name": "tube*"
      }
    ],
    "after_dependencies": [
      "usb",
      "zeroconf"
    ],
    "iot_class": "local_polling",
    "loggers": [
      "aiosqlite",
      "bellows",
      "crccheck",
      "pure_pcapy3",
      "zhaquirks",
      "zigpy",
      "zigpy_deconz",
      "zigpy_xbee",
      "zigpy_zigate",
      "zigpy_znp"
    ],
    "is_built_in": true
  },
  "data": {
    "ieee": "**REDACTED**",
    "nwk": 46405,
    "manufacturer": "IKEA of Sweden",
    "model": "TRADFRI remote control",
    "name": "IKEA of Sweden TRADFRI remote control",
    "quirk_applied": true,
    "quirk_class": "zhaquirks.ikea.fivebtnremotezha.IkeaTradfriRemote1",
    "manufacturer_code": 4476,
    "power_source": "Battery or Unknown",
    "lqi": 156,
    "rssi": -61,
    "last_seen": "2022-05-18T13:34:46",
    "available": true,
    "device_type": "EndDevice",
    "signature": {
      "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4476, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
      "endpoints": {
        "1": {
          "profile_id": 260,
          "device_type": "0x0820",
          "in_clusters": [
            "0x0000",
            "0x0001",
            "0x0003",
            "0x0020",
            "0x1000",
            "0xfc7c"
          ],
          "out_clusters": [
            "0x0003",
            "0x0004",
            "0x0005",
            "0x0006",
            "0x0008",
            "0x0019",
            "0x1000"
          ]
        }
      }
    },
    "entities": [
      {
        "entity_id": "button.ikea_of_sweden_tradfri_remote_control_3687dafe_identify",
        "name": "IKEA of Sweden TRADFRI remote control"
      },
      {
        "entity_id": "sensor.bens_remote_power",
        "name": "IKEA of Sweden TRADFRI remote control"
      }
    ],
    "neighbors": [],
    "endpoint_names": [
      {
        "name": "NON_COLOR_CONTROLLER"
      }
    ],
    "user_given_name": "Ben's Remote",
    "device_reg_id": "c8f767aa7733a0f305272a4b7687774d",
    "area_id": "6ada5d8f56b641e9961812a3bcc08f56"
  }
}
```

</details>

TheJulianJES, 2022-05-18:
https://github.com/zigpy/zha-device-handlers/pull/318 might be related.

MattWestb, 2022-05-18:
All device i have have being fixed but the maintainer have have not merging it for over 4 weeks and was missing HA 2022.5  https://github.com/zigpy/zha-device-handlers/pull/1511.

closed 2022-05-23 by dmulcahey