zigpy/zha-device-handlers #1570: [BUG] E1812 shortcut button no longer has an action in Homeassistant

bwims, 2022-05-19:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to ZHA integration
2. Click on add device
3. Scroll down to device details
4. See error - there are only 2 diagnotic entities, identify and power

**Expected behavior**
There should be a sensor action for a switch

**Screenshots**
If applicable, add screenshots to help explain your problem.

<details>
<summary>Device signature - this can be acquired by clicking on the "Zigbee Device Signature" button in the device settings</summary>

```
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4476, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0820",
      "in_clusters": [
        "0x0000",
        "0x0001",
        "0x0003",
        "0x0009",
        "0x0020",
        "0x1000"
      ],
      "out_clusters": [
        "0x0003",
        "0x0004",
        "0x0006",
        "0x0008",
        "0x0019",
        "0x0102",
        "0x1000"
      ]
    }
  },
  "manufacturer": "IKEA of Sweden",
  "model": "TRADFRI SHORTCUT Button",
  "class": "zhaquirks.ikea.shortcutbtn.IkeaTradfriShortcutBtn"
}
```

</details>

<details>
<summary>Diagnostic information - this can be acquired by clicking on the "Download Diagnostics" button in the device settings</summary>

```
{
  "home_assistant": {
    "installation_type": "Home Assistant Container",
    "version": "2022.4.6",
    "dev": false,
    "hassio": false,
    "virtualenv": false,
    "python_version": "3.9.9",
    "docker": true,
    "arch": "aarch64",
    "timezone": "Europe/London",
    "os_name": "Linux",
    "os_version": "5.10.103-v8+",
    "run_as_root": true
  },
  "custom_components": {
    "simpleicons": {
      "version": "v2.0.0",
      "requirements": [
        "simpleicons==6.16.0"
      ]
    },
    "hacs": {
      "version": "1.24.2",
      "requirements": [
        "aiogithubapi>=21.11.0"
      ]
    },
    "nodered": {
      "version": "1.0.8",
      "requirements": []
    }
  },
  "integration_manifest": {
    "domain": "zha",
    "name": "Zigbee Home Automation",
    "config_flow": true,
    "documentation": "https://www.home-assistant.io/integrations/zha",
    "requirements": [
      "bellows==0.29.0",
      "pyserial==3.5",
      "pyserial-asyncio==0.6",
      "zha-quirks==0.0.72",
      "zigpy-deconz==0.14.0",
      "zigpy==0.44.2",
      "zigpy-xbee==0.14.0",
      "zigpy-zigate==0.7.4",
      "zigpy-znp==0.7.0"
    ],
    "usb": [
      {
        "vid": "10C4",
        "pid": "EA60",
        "description": "*2652*",
        "known_devices": [
          "slae.sh cc2652rb stick"
        ]
      },
      {
        "vid": "10C4",
        "pid": "EA60",
        "description": "*sonoff*plus*",
        "known_devices": [
          "sonoff zigbee dongle plus"
        ]
      },
      {
        "vid": "10C4",
        "pid": "EA60",
        "description": "*tubeszb*",
        "known_devices": [
          "TubesZB Coordinator"
        ]
      },
      {
        "vid": "1A86",
        "pid": "7523",
        "description": "*tubeszb*",
        "known_devices": [
          "TubesZB Coordinator"
        ]
      },
      {
        "vid": "1A86",
        "pid": "7523",
        "description": "*zigstar*",
        "known_devices": [
          "ZigStar Coordinators"
        ]
      },
      {
        "vid": "1CF1",
        "pid": "0030",
        "description": "*conbee*",
        "known_devices": [
          "Conbee II"
        ]
      },
      {
        "vid": "10C4",
        "pid": "8A2A",
        "description": "*zigbee*",
        "known_devices": [
          "Nortek HUSBZB-1"
        ]
      },
      {
        "vid": "10C4",
        "pid": "8B34",
        "description": "*bv 2010/10*",
        "known_devices": [
          "Bitron Video AV2010/10"
        ]
      }
    ],
    "codeowners": [
      "@dmulcahey",
      "@adminiuga"
    ],
    "zeroconf": [
      {
        "type": "_esphomelib._tcp.local.",
        "name": "tube*"
      }
    ],
    "after_dependencies": [
      "usb",
      "zeroconf"
    ],
    "iot_class": "local_polling",
    "loggers": [
      "aiosqlite",
      "bellows",
      "crccheck",
      "pure_pcapy3",
      "zhaquirks",
      "zigpy",
      "zigpy_deconz",
      "zigpy_xbee",
      "zigpy_zigate",
      "zigpy_znp"
    ],
    "is_built_in": true
  },
  "data": {
    "ieee": "**REDACTED**",
    "nwk": 35209,
    "manufacturer": "IKEA of Sweden",
    "model": "TRADFRI SHORTCUT Button",
    "name": "IKEA of Sweden TRADFRI SHORTCUT Button",
    "quirk_applied": true,
    "quirk_class": "zhaquirks.ikea.shortcutbtn.IkeaTradfriShortcutBtn",
    "manufacturer_code": 4476,
    "power_source": "Battery or Unknown",
    "lqi": 255,
    "rssi": -24,
    "last_seen": "2022-05-19T13:32:32",
    "available": true,
    "device_type": "EndDevice",
    "signature": {
      "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4476, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
      "endpoints": {
        "1": {
          "profile_id": 260,
          "device_type": "0x0820",
          "in_clusters": [
            "0x0000",
            "0x0001",
            "0x0003",
            "0x0009",
            "0x0020",
            "0x1000"
          ],
          "out_clusters": [
            "0x0003",
            "0x0004",
            "0x0006",
            "0x0008",
            "0x0019",
            "0x0102",
            "0x1000"
          ]
        }
      }
    },
    "entities": [
      {
        "entity_id": "sensor.ikea_of_sweden_tradfri_shortcut_button_fb9ab6fe_power",
        "name": "IKEA of Sweden TRADFRI SHORTCUT Button"
      },
      {
        "entity_id": "button.ikea_of_sweden_tradfri_shortcut_button_fb9ab6fe_identify",
        "name": "IKEA of Sweden TRADFRI SHORTCUT Button"
      }
    ],
    "neighbors": [],
    "endpoint_names": [
      {
        "name": "NON_COLOR_CONTROLLER"
      }
    ],
    "user_given_name": null,
    "device_reg_id": "e07208ac2d471dd7ae5c7338f48c8792",
    "area_id": null
  }
}
```

</details>

<details>
<summary>Additional logs</summary>

```
Paste any additional debug logs here.
```

</details>

**Additional context**
Add any other context about the problem here.

I looked for updates to HA to see if there had already been changes to fix this but could not find any.

Many thanks for any advice you can provide!

MattWestb, 2022-05-19:
If like having working device automatons in ZHA use the quirk from https://github.com/zigpy/zha-device-handlers/pull/1511 and putting them in your reconfigured local quirk folder and you is getting all automatons working.

bwims, 2022-05-28:
Any ideas how I can debug this?

MattWestb, 2022-05-28:
In HA go `Developer Tools` -> `EVENTS`  and put `zha_event` in the `Listen to events` feeld and clicking `start listening`.
If single press its sending:
```
{
    "event_type": "zha_event",
    "data": {
        "device_ieee": "ec:1b:bd:ff:fe:d0:9f:82",
        "unique_id": "ec:1b:bd:ff:fe:d0:9f:82:1:0x0006",
        "device_id": "db02546a32aba15feb91376606f27e3c",
        "endpoint_id": 1,
        "cluster_id": 6,
        "command": "on",
        "args": [],
        "params": {}
    },
```
Double:
```
{
    "event_type": "zha_event",
    "data": {
        "device_ieee": "ec:1b:bd:ff:fe:d0:9f:82",
        "unique_id": "ec:1b:bd:ff:fe:d0:9f:82:1:0x0006",
        "device_id": "db02546a32aba15feb91376606f27e3c",
        "endpoint_id": 1,
        "cluster_id": 6,
        "command": "off",
        "args": [],
        "params": {}
    },
```
And long:
```
{
    "event_type": "zha_event",
    "data": {
        "device_ieee": "ec:1b:bd:ff:fe:d0:9f:82",
        "unique_id": "ec:1b:bd:ff:fe:d0:9f:82:1:0x0008",
        "device_id": "db02546a32aba15feb91376606f27e3c",
        "endpoint_id": 1,
        "cluster_id": 8,
        "command": "move_with_on_off",
        "args": [
            0,
            83
        ],
        "params": {
            "move_mode": 0,
            "rate": 83
        }
    },
```
And long release:
```
{
    "event_type": "zha_event",
    "data": {
        "device_ieee": "ec:1b:bd:ff:fe:d0:9f:82",
        "unique_id": "ec:1b:bd:ff:fe:d0:9f:82:1:0x0008",
        "device_id": "db02546a32aba15feb91376606f27e3c",
        "endpoint_id": 1,
        "cluster_id": 8,
        "command": "stop_with_on_off",
        "args": [],
        "params": {}
    },
```
If you is not getting any event try `reconfigure device`  from the device card and pressing the button then sending commands to it so its not sleeping and can reviving the commands.
If not working delete the device from the device card and waiting one minute and try paring it one more time and doing the test.

bwims, 2022-06-10:
Thanks @MattWestb !  The Reconfigure fixed it.    I had to continually press the button while it was going on so all the relevant bindings occurred, then it said it failed, but it was by that point function.

Thanks again!
B.

closed 2022-06-10 by bwims