zigpy/zha-device-handlers #1571: [Device Support Request] _TZE200_vrjkcam9 TS0601

Celer21, 2022-05-19:
Hello, 
i bought this Water/Gas valve
https://zigbee.blakadder.com/IOTPerfect_PF-PM02D-TYZ.html

I would like to see this device to be supported in ZHA.
It shoud have one on/off switch.

Thank for adding support for this device
PS: This is my first device support request, so please be patient with me.
<details>
<summary>Device signature - this can be acquired by clicking on the "Zigbee Device Signature" button in the device settings</summary>

```
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0051",
      "in_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0xef00"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZE200_vrjkcam9",
  "model": "TS0601",
  "class": "zigpy.device.Device"
}
```

</details>

<details>
<summary>Diagnostic information - this can be acquired by clicking on the "Download Diagnostics" button in the device settings</summary>

```
{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2022.5.5",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.9.9",
    "docker": true,
    "arch": "x86_64",
    "timezone": "Europe/Prague",
    "os_name": "Linux",
    "os_version": "5.15.38",
    "supervisor": "2022.05.2",
    "host_os": "Home Assistant OS 8.0",
    "docker_version": "20.10.14",
    "chassis": "vm",
    "run_as_root": true
  },
  "custom_components": {
    "scheduler": {
      "version": "v0.0.0",
      "requirements": []
    },
    "browser_mod": {
      "version": "1.5.2",
      "requirements": []
[... 564 lines cut ...]
[0xB048:1:0xef00] No explicit handler for cluster command 0x01: b'\x00\x00\x1a\x05\x00\x01\x01'
[0xB048:1:0xef00] Received ZCL frame: b'\x09\x1F\x01\x00\x00\x1A\x05\x00\x01\x01'
[0xB048:1:0xef00] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.CLUSTER_COMMAND: 1>, is_manufacturer_specific=0, is_reply=1, disable_default_response=0, reserved=0, *is_cluster=True, *is_general=False), tsn=31, command_id=1, *is_reply=True)
[0xB048:1:0xef00] Unknown cluster command 1 b'\x00\x00\x1a\x05\x00\x01\x01'
[0xB048:1:0xef00] Received command 0x01 (TSN 31): b'\x00\x00\x1a\x05\x00\x01\x01'
[0xB048:1:0xef00] No explicit handler for cluster command 0x01: b'\x00\x00\x1a\x05\x00\x01\x01'
[0xB048:1:0xef00] Received ZCL frame: b'\x09\x1F\x01\x00\x00\x1A\x05\x00\x01\x01'
[0xB048:1:0xef00] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.CLUSTER_COMMAND: 1>, is_manufacturer_specific=0, is_reply=1, disable_default_response=0, reserved=0, *is_cluster=True, *is_general=False), tsn=31, command_id=1, *is_reply=True)
[0xB048:1:0xef00] Unknown cluster command 1 b'\x00\x00\x1a\x05\x00\x01\x01'
[0xB048:1:0xef00] Received command 0x01 (TSN 31): b'\x00\x00\x1a\x05\x00\x01\x01'
[0xB048:1:0xef00] No explicit handler for cluster command 0x01: b'\x00\x00\x1a\x05\x00\x01\x01'
[0xE280](lumi.weather): Device seen - marking the device available and resetting counter
[0xE280](lumi.weather): Update device availability -  device available: True - new availability: True - changed: False
[0x6827](lumi.weather): Device seen - marking the device available and resetting counter
[0x6827](lumi.weather): Update device availability -  device available: True - new availability: True - changed: False
[0xB048:1:0xef00] Received ZCL frame: b'\x09\x1F\x01\x00\x00\x1A\x05\x00\x01\x01'
[0xB048:1:0xef00] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.CLUSTER_COMMAND: 1>, is_manufacturer_specific=0, is_reply=1, disable_default_response=0, reserved=0, *is_cluster=True, *is_general=False), tsn=31, command_id=1, *is_reply=True)
[0xB048:1:0xef00] Unknown cluster command 1 b'\x00\x00\x1a\x05\x00\x01\x01'
[0xB048:1:0xef00] Received command 0x01 (TSN 31): b'\x00\x00\x1a\x05\x00\x01\x01'
[0xB048:1:0xef00] No explicit handler for cluster command 0x01: b'\x00\x00\x1a\x05\x00\x01\x01'
[0xBAB0:1:0x000a] Received ZCL frame: b'\x00\x05\x00\x07\x00'
[0xBAB0:1:0x000a] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl(frame_type=
```

</details>

javicalle, 2022-05-19:
Can you try with that quirk:

<details>
<summary>ts0601_air_quality.py</summary>

```python
"""Tuya Valve."""

from typing import Dict

from zigpy.profiles import zha
from zigpy.quirks import CustomDevice
import zigpy.types as t
from zigpy.zcl.clusters.general import Basic, Groups, Ota, Scenes, Time
from zigpy.zcl.clusters.smartenergy import Metering

from zhaquirks import DoublingPowerConfigurationCluster
from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODELS_INFO,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
)
from zhaquirks.tuya import TuyaLocalCluster
from zhaquirks.tuya.mcu import (
    DPToAttributeMapping,
    TuyaDPType,
    TuyaMCUCluster,
    TuyaOnOff,
)

class TuyaValveManufCluster(TuyaMCUCluster):
    """On/Off Tuya cluster with extra device attributes."""

    attributes = TuyaMCUCluster.attributes.copy()
    attributes.update(
        {
            0xEF01: ("time_left", t.uint32_t, True),
            0xEF02: ("state", t.enum8, True),
            0xEF03: ("last_valve_open_duration", t.uint32_t, True),
            0xEF04: ("dp_6", t.uint32_t, True),
        }
    )

    dp_to_attribute: Dict[int, DPToAttributeMapping] = {
        1: DPToAttributeMapping(
            TuyaOnOff.ep_attribute,
            "on_off",
            dp_type=TuyaDPType.BOOL,
        ),
        6: DPToAttributeMapping(
            TuyaMCUCluster.ep_attribute,
            "dp_6",
            TuyaDPType.VALUE,
        ),
        7: DPToAttributeMapping(
            DoublingPowerConfigurationCluster.ep_attribute,
            "battery_percentage_remaining",
            TuyaDPType.VALUE,
        ),
        11: DPToAttributeMapping(
            TuyaMCUCluster.ep_attribute,
            "time_left",
            TuyaDPType.VALUE,
        ),
        12: DPToAttributeMapping(
            TuyaMCUCluster.ep_attribute,
            "state",
            TuyaDPType.VALUE,
        ),
        15: DPToAttributeMapping(
            TuyaMCUCluster.ep_attribute,
            "last_valve_open_duration",
            TuyaDPType.VALUE,
        ),
    }

    data_point_handlers = {
        1: "_dp_2_attr_update",
        6: "_dp_2_attr_update",
        7: "_dp_2_attr_update",
        11: "_dp_2_attr_update",
        12: "_dp_2_attr_update",
        15: "_dp_2_attr_update",
    }

class TuyaValve(CustomDevice):
    """Tuya valve device."""

    signature = {
        MODELS_INFO: [("_TZE200_81isopgh", "TS0601")],
        # SizePrefixedSimpleDescriptor(endpoint=1, profile=260, device_type=81, device_version=1,
        # input_clusters=[0, 4, 5, 61184], output_clusters=[25, 10])
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.SMART_PLUG,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaValveManufCluster.cluster_id,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            }
        },
    }

    replacement = {
        ENDPOINTS: {
            1: {
                DEVICE_TYPE: zha.DeviceType.SMART_PLUG,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaOnOff,
                    DoublingPowerConfigurationCluster,
                    TuyaValveManufCluster,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            }
        }
    }
```

</details>

There are some guides here to install local quirk:
* https://github.com/zigpy/zha-device-handlers/discussions/693

Celer21, 2022-05-20:
Hi @javicalle,
quirk ts0601_air_quality.py is working!
Thank you!!!
I have to update model number 
`MODELS_INFO: [("_TZE200_vrjkcam9", "TS0601")],
`
Final quirk:
```
"""Tuya Valve."""

from typing import Dict

from zigpy.profiles import zha
from zigpy.quirks import CustomDevice
import zigpy.types as t
from zigpy.zcl.clusters.general import Basic, Groups, Ota, Scenes, Time
from zigpy.zcl.clusters.smartenergy import Metering

from zhaquirks import DoublingPowerConfigurationCluster
from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODELS_INFO,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
)
from zhaquirks.tuya import TuyaLocalCluster
from zhaquirks.tuya.mcu import (
    DPToAttributeMapping,
    TuyaDPType,
    TuyaMCUCluster,
    TuyaOnOff,
)

class TuyaValveManufCluster(TuyaMCUCluster):
    """On/Off Tuya cluster with extra device attributes."""

    attributes = TuyaMCUCluster.attributes.copy()
    attributes.update(
        {
            0xEF01: ("time_left", t.uint32_t, True),
            0xEF02: ("state", t.enum8, True),
            0xEF03: ("last_valve_open_duration", t.uint32_t, True),
            0xEF04: ("dp_6", t.uint32_t, True),
        }
    )

    dp_to_attribute: Dict[int, DPToAttributeMapping] = {
        1: DPToAttributeMapping(
            TuyaOnOff.ep_attribute,
            "on_off",
            dp_type=TuyaDPType.BOOL,
        ),
        6: DPToAttributeMapping(
            TuyaMCUCluster.ep_attribute,
            "dp_6",
            TuyaDPType.VALUE,
        ),
        7: DPToAttributeMapping(
            DoublingPowerConfigurationCluster.ep_attribute,
            "battery_percentage_remaining",
            TuyaDPType.VALUE,
        ),
        11: DPToAttributeMapping(
            TuyaMCUCluster.ep_attribute,
            "time_left",
            TuyaDPType.VALUE,
        ),
        12: DPToAttributeMapping(
            TuyaMCUCluster.ep_attribute,
            "state",
            TuyaDPType.VALUE,
        ),
        15: DPToAttributeMapping(
            TuyaMCUCluster.ep_attribute,
            "last_valve_open_duration",
            TuyaDPType.VALUE,
        ),
    }

    data_point_handlers = {
        1: "_dp_2_attr_update",
        6: "_dp_2_attr_update",
        7: "_dp_2_attr_update",
        11: "_dp_2_attr_update",
        12: "_dp_2_attr_update",
        15: "_dp_2_attr_update",
    }

class TuyaValve(CustomDevice):
    """Tuya valve device."""

    signature = {
        MODELS_INFO: [("_TZE200_vrjkcam9", "TS0601")],
        # SizePrefixedSimpleDescriptor(endpoint=1, profile=260, device_type=81, device_version=1,
        # input_clusters=[0, 4, 5, 61184], output_clusters=[25, 10])
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.SMART_PLUG,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaValveManufCluster.cluster_id,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            }
        },
    }

    replacement = {
        ENDPOINTS: {
            1: {
                DEVICE_TYPE: zha.DeviceType.SMART_PLUG,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaOnOff,
                    DoublingPowerConfigurationCluster,
                    TuyaValveManufCluster,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            }
        }
    }

```

closed 2022-05-20 by Celer21

javicalle, 2022-05-20:
> I have to update model number

Ouuuch... of course. And I have 'renamed' the filename...

The quirk has also generated a battery entity in HA?

Celer21, 2022-05-20:
It only generated on/off switch. 
There is no battery entity and device has no battery inside...
So everything seems wokring ok.

[image]

javicalle, 2022-05-20:
> device has no battery inside...

Noted. Thanks for clarifying.

M4TH76, 2022-07-17:
Hi
I have the same valve switch, it show up in ha(TS0601
by _TZE200_vrjkcam9), but no on/off button!?!
Did I miss something?
Do I still need to add the quirk?

javicalle, 2022-07-25:
No one does a PR for this device, so you will need to copy the local quirk
or make a PR to add the device to the mainstream.