zigpy/zha-device-handlers #1579: [Device Support Request] _TZ3210_e5t9bfdv TS0501B

cuppa-joe, 2022-05-23:
This is a single channel LED controller that erroneously reports color support. I think it would need the same fix as _TZ3000_4whigl8i TS0501B. 

<details>
<summary>Device signature - this can be acquired by clicking on the "Zigbee Device Signature" button in the device settings</summary>

```
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0101",
      "in_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0008",
        "0x0300",
        "0x1000",
        "0xef00"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "242": {
      "profile_id": 41440,
      "device_type": "0x0061",
      "in_clusters": [],
      "out_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "_TZ3210_e5t9bfdv",
  "model": "TS0501B",
  "class": "zigpy.device.Device"
}
```

</details>

<details>
<summary>Diagnostic information - this can be acquired by clicking on the "Download Diagnostics" button in the device settings</summary>

```
{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2022.5.5",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.9.9",
    "docker": true,
    "arch": "x86_64",
    "timezone": "America/Chicago",
    "os_name": "Linux",
    "os_version": "5.15.41",
    "supervisor": "2022.05.2",
    "host_os": "Home Assistant OS 8.1",
    "docker_version": "20.10.14",
    "chassis": "vm",
    "run_as_root": true
  },
  "custom_components": {
    "simpleicons": {
[... 438 lines cut ...]
    "endpoint_names": [
      {
        "name": "DIMMABLE_LIGHT"
      },
      {
        "name": "unknown 97 device_type of 0xa1e0 profile id"
      }
    ],
    "user_given_name": null,
    "device_reg_id": "fe54c254e0d13c7854270086ac05e524",
    "area_id": "381b81e7b08f4e66a3cebbe7cfe6c11e"
  }
}
```

</details>

<details>
<summary>Additional logs</summary>

```
Paste any additional debug logs here.
```

</details>

javicalle, 2022-05-23:
I think it's more like the `_TZ3210_9q49basr` one (although I think both quirks do exactly the same thing).
Could you try to add your device to the `ts0501bs.py` quirk?:
* https://github.com/zigpy/zha-device-handlers/blob/dev/zhaquirks/tuya/ts0501bs.py

If you are not familiar, there are some guides about it. Maybe the most popular can be:
* https://github.com/zigpy/zha-device-handlers/discussions/693#discussioncomment-857274

MattWestb, 2022-05-23:
Per "tuya standard" is all TS0501B Dimmable light lights https://developer.tuya.com/en/docs/iot-device-dev/tuya-endtocloud-logic?id=Kav5tfxsbsncf#title-14-Lights and i think its can being time using `MODEL` instead of `MODEL_INFO` and merger the 2 tuya quirks and also the LIDL ones to one more general quirk that is patching all tuya dimmebal light so user is not getting problems all the time with new versions.

And i think we can using the same Principe for more tuya devices that is in the tuya standard list and all or most is coming with the same bugs.

javicalle, 2022-05-23:
Not sure if can be done because of the diferent `TuyaManufCluster`. But I agree to merge all of them if not possible in the same quirk, at least at the same file.

MattWestb, 2022-05-23:
2 quirks is deleting the color cluster and one is patching the attribute {0x400A: 0} on the color cluster (that shall not being present on one dimmer light). Witch way is right is one good question and the tuya manufacture cluster is not being used so all can being treated the same way.

One thinking of the LIDL ones is the A version i think its being Zigbee certificated versions but looks having the same bugs as the "normal" tuya B ones.

MattWestb, 2022-05-23:
The tuya manufacture cluster can being used for "tuya scenes" switches but  we have not seen how they is using it and the TS004F and the remote LIDL have shipping is using normal Zigbee commands for changing the lights and no magic is being done as IKEA is doing with there remotes and lights with custom scene commands.

javicalle, 2022-05-23:
> 2 quirks is deleting the color cluster and one is patching the attribute {0x400A: 0} on the color cluster (that shall not being present on one dimmer light). Witch way is right is one good question and the tuya manufacture cluster is not being used so all can being treated the same way.

Attribute `0x400A` is the device `ColorCapabilities`:
* https://github.com/zigpy/zigpy/blob/b166e2e1f22c49bad201d7ecdf202f0fc9b7b087/zigpy/zcl/clusters/lighting.py#L118

I bet that setting to 0 will disable any color capabilities for the device, and probably would have the same effect that removing the cluster.

cuppa-joe, 2022-05-23:
> I think it's more like the `_TZ3210_9q49basr` one (although I think both quirks do exactly the same thing). Could you try to add your device to the `ts0501bs.py` quirk?:
> 
>     * https://github.com/zigpy/zha-device-handlers/blob/dev/zhaquirks/tuya/ts0501bs.py
> 
> 
> If you are not familiar, there are some guides about it. Maybe the most popular can be:
> 
>     * [Editing (ZHA) quirks in a (HA) container or local. #693 (comment)](https://github.com/zigpy/zha-device-handlers/discussions/693#discussioncomment-857274)

This works! I originally tried ts0501b.py and completely missed ts0501bs.py.

javicalle, 2022-06-02:
If device is working you can create the PR to add your device to the mainstream?

Just edit from the web the file content adding your device to the MODELS_INFO and follow the instructions:
* https://github.com/zigpy/zha-device-handlers/blob/dev/zhaquirks/tuya/ts0501bs.py

Reference this issue in the first PR comment as follows:
```
Fixes: #1579
```

This way other users will be able to use the device in ZHA thanks to you!

closed 2022-12-06 by github-actions[bot]

VCTGomes, 2024-11-29:
Hi. This device supports Do-Not Disturb feature, the Tuya version from power-on behavior. I think it uses custom clusters. Is it possible to implement it?

kelthor85, 2025-02-12:
> Hi. This device supports Do-Not Disturb feature, the Tuya version from power-on behavior. I think it uses custom clusters. Is it possible to implement it?

Please, I've been searching for days, wondering where on earth this DND feature is! I have a super bright led strip attached to a dimmer on a mirror in my parent's bedroom. Let's just say the way the house came alive at 3 am when power returned was not ideal!

Niek, 2025-10-04:
Not sure if I should open a new issue, but device `_TZ3210_syh4kuef` has the exact same issue. Single color reported as full color.

PR #2136 (2023-01-26): add two manufacturers to TS0501bs
fixes #2135 
fixes #1579